matthias-bs/BresserWeatherSensorReceiver #3: SSL error

meijer3, 2022-08-02:
**Describe the bug**
I get a SSL error with my Heltec Lora esp32. 

**To Reproduce**
Steps to reproduce the behavior:
The only thing in the 'MQTT example' script that I changed is the following.
```
    const char ssid[] = "WiFiSSID";
    const char pass[] = "WiFiPassword";
    #define HOSTNAME "hostname"
    const char MQTT_HOST[] = "xxx.yyy.zzz.com";
    const int  MQTT_PORT = 8883; // I used port 1883
    const char MQTT_USER[] = ""; 
    const char MQTT_PASS[] = "";
```

**Expected behavior**
The other example shows the lora messages (that works), my laptop can connect to MQTT, so that works.
I would assume the scripts works without any certs

**Debug Log**
```
15:01:28.190 -> Checking wifi...
15:01:28.190 -> MQTT connecting... [I][ssl_client.cpp:127] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
15:01:28.238 -> [E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():216]: (-80) UNKNOWN ERROR CODE (0050)
15:01:28.238 -> [E][WiFiClientSecure.cpp:133] connect(): start_ssl_client: -80
15:01:28.238 -> .[I][ssl_client.cpp:127] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
15:01:29.237 -> [E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():216]: (-80) UNKNOWN ERROR CODE (0050)
15:01:29.237 -> [E][WiFiClientSecure.cpp:133] connect(): start_ssl_client: -80
```

**Embedded Device (please complete the following information):**
 - Board:  #https://heltec.org/project/wifi-lora-32/
 - Arduino Version: 1.8.15

matthias-bs, 2022-08-02:
Which Board Support Package are you using? I am using esp32 1.0.6
Besides, I am using Arduino 1.8.19.
I am not sure if this might relevant, but I think it's worth trying.

BTW: Port 8883 is for secure MQTT (SSL), port 1833 for non-secure MQTT. You have to make sure that your client uses a port which is provided by your MQTT broker.

The issue seems to be related rather to the [arduino-mqtt](https://github.com/256dpi/arduino-mqtt) library than to BresserWeatherSensorReceiver.

Maybe it helps to check if the examples provided in [arduino-mqtt](https://github.com/256dpi/arduino-mqtt) are working with your setup.

matthias-bs, 2022-08-02:
Ahh - maybe most important: which version of [arduino-mqtt](https://github.com/256dpi/arduino-mqtt) are you using? Mine is v2.5.0.

meijer3, 2022-08-03:
Thanks for your quick reply, I looked at arduino-mqtt example and I noticed that they used `WiFiClient` instead of `WiFiClientSecure`. Which works for me,
I guess `the net.setInsecure();` is not working somehow. (I disabled it for now)

matthias-bs, 2022-08-16:
Hopefully fixed in https://github.com/matthias-bs/BresserWeatherSensorReceiver/commit/c7c415ab40b0115ec7b6908ec9dfdfd8cab5b324
Could you please check?

matthias-bs, 2022-08-16:
Hopefully fixed now: https://github.com/matthias-bs/BresserWeatherSensorReceiver/commit/ee16ad16bd72915772de8a7dd8ff3c89048ae29b
Could you please check?

closed 2022-11-09 by matthias-bs

meijer3, 2022-11-13:
I checked, I think its actually only happening if I selected the heltec board in Arduino IDE. If I pick esp32dev it is working...

matthias-bs, 2022-11-13:
Thank you for the feedback! For other reasons, I recommend to update to v0.4.0, though.