matthias-bs/BresserWeatherSensorReceiver #4: Mosquitto MQTT as server does not work if no SSL is used

franki29, 2022-08-11:
**Describe the bug**
If I compile your code and try without using ssl to connect to my mqtt server, the server response with a "socket error (unknown) on client"

**To Reproduce**
Steps to reproduce the behavior:
1. Step1 'compile'
2. Step2 'connect to a none secure mosquitto mqtt server.'
4. See error

**Expected behavior**
Successfully connection to MQTT server without using ssl

It is working if I change the code to:
#elif defined(ESP8266)
   // BearSSL::WiFiClientSecure net;
   **WiFiClient net;**
&
 #if (!defined(CHECK_PUB_KEY) and !defined(CHECK_CA_ROOT) and !defined(CHECK_FINGERPRINT))
       **// net.setInsecure();**
    #endif

Seems to be a problem using BearSSL::WiFiClientSecure net;

Br. Frank

matthias-bs, 2022-08-11:
same as https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/3

matthias-bs, 2022-08-16:
Hopefully fixed in https://github.com/matthias-bs/BresserWeatherSensorReceiver/commit/c7c415ab40b0115ec7b6908ec9dfdfd8cab5b324
Could you please check?

matthias-bs, 2022-08-16:
Hopefully fixed now: https://github.com/matthias-bs/BresserWeatherSensorReceiver/commit/ee16ad16bd72915772de8a7dd8ff3c89048ae29b
Could you please check?

franki29, 2022-10-08:
Hi using #define USE_WIFI is working for esp8266. 
Only compiling is failing at my side because in WeatherSensor.h you use #include <WeatherSensorCfg.h> at my side I have to change it to #include "WeatherSensorCfg.h"

closed 2022-10-08 by franki29

matthias-bs, 2022-10-08:
Hi,

thanks for testing and feedback! `#include "WeatherSensorCfg.h"` makes more sense, because it contains the sketch specific settings and not general definitions of the library.

In some (but not all) of the examples, there is
```
#include "WeatherSensorCfg.h"
#include "WeatherSensor.h"
```
I will see if I can remove the inconsistencies/redundancies.

This also relates to issue #1 and issue #11.

Thank you for pointing this out!

Best regards
Matthias