matthias-bs/BresserWeatherSensorReceiver #5: Add mqtt information of Wind direction - D = wind direction 0..F = N..NNE..E..S..W..NNW

franki29, 2022-08-11:
Hi, currently it looks like your code is only using the information of degree for wind direction, but my bresser 6 in 1 can also send this information as direction in letters like NW, NNW ...

Is it possible for you to add this information and send it to the mqtt server?

Br. Frank

matthias-bs, 2022-08-11:
Hi,
the sensor sends the wind direction in degrees, but a mapping function for translating this into letters can be implemented for sure. I have done something similar for translation of wind speed from m/s to Bft, see: https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/44bda263af1caef053e9b23d5d12a88098e79dde/src/WeatherSensor.cpp#L573

Instead of comparing values one by one, if have implemented the comparison in to stages to speed it up.

I think this should be done in a similar fashion:
1. quadrant (N/E/S/W)
2. sector (NNW/NW/WNW) etc.

Cheers
Matthias

matthias-bs, 2022-08-11:
A function prototype could look like this:
`void WeatherSensor::winddir_deg_to_str(float deg, char * buf);`

matthias-bs, 2022-08-12:
Could you please check this?

In WeatherSensor.h:
```
#include <string>
[...]

const std::string COMPASS_POINTS[17] = {
    "N", "NNE", "NE", "ENE", 
    "E", "ESE", "SE", "SSE",
    "S", "SSW", "SW", "WSW",
    "W", "WNW", "NW", "NNW",
    "N"
};
[...]

class WeatherSensor {
    public:
        void winddir_flt_to_str(float dir, char * buf)
        {
            std::string point = COMPASS_POINTS[(int)((dir + 11.25)/22.5)];
            strncpy(buf, point.c_str(), sizeof(buf-1));
            buf[point.length()] = '\0';
        };
[...]
```
And in your sketch e.g.:
```
    for (float d=0; d <= 360; d += 11.25)
    {
      char buf[4];
      weatherSensor.winddir_flt_to_str(d, buf);
      Serial.printf("Wind dir: [%5.1fdeg] [%-3s]\n",
        d, buf);
    }
```

franki29, 2022-08-12:
HI, I tried and the result is: (sure I made something wrong as it should be only one direction?
{"sensor_id": 14200203,"ch:":0,"battery_ok":1,"temp_c":24.5,"humidity":50,"wind_gust":0.0,"wind_avg":0.0,"wind_dir":158.0,"rain":541.6}
Wind dir: [  0.0deg] [N  ]
Wind dir: [ 11.2deg] [NNE]
Wind dir: [ 22.5deg] [NNE]
Wind dir: [ 33.8deg] [NE ]
Wind dir: [ 45.0deg] [NE ]
Wind dir: [ 56.2deg] [ENE]
Wind dir: [ 67.5deg] [ENE]
Wind dir: [ 78.8deg] [E  ]
Wind dir: [ 90.0deg] [E  ]
Wind dir: [101.2deg] [ESE]
Wind dir: [112.5deg] [ESE]
Wind dir: [123.8deg] [SE ]
Wind dir: [135.0deg] [SE ]
Wind dir: [146.2deg] [SSE]
Wind dir: [157.5deg] [SSE]
Wind dir: [168.8deg] [S  ]
Wind dir: [180.0deg] [S  ]
Wind dir: [191.2deg] [SSW]
Wind dir: [202.5deg] [SSW]
Wind dir: [213.8deg] [SW ]
Wind dir: [225.0deg] [SW ]
Wind dir: [236.2deg] [WSW]
Wind dir: [247.5deg] [WSW]
Wind dir: [258.8deg] [W  ]
Wind dir: [270.0deg] [W  ]
Wind dir: [281.2deg] [WNW]
Wind dir: [292.5deg] [WNW]
Wind dir: [303.8deg] [NW ]
Wind dir: [315.0deg] [NW ]
Wind dir: [326.2deg] [NNW]
Wind dir: [337.5deg] [NNW]
Wind dir: [348.8deg] [N  ]
Wind dir: [360.0deg] [N  ]
{"rssi":-85.5}

matthias-bs, 2022-08-12:
The for-loop was just intended to show that the mapping from degrees to text works as intended. The following code snipped should provide the desired result:
```
        [...]
        sprintf(&mqtt_payload[strlen(mqtt_payload)], ",\"wind_dir\":%.1f", weatherSensor.wind_direction_deg);
        char buf[4];
        weatherSensor.winddir_flt_to_str(weatherSensor.wind_direction_deg, buf);
        sprintf(&mqtt_payload[strlen(mqtt_payload)], ",\"wind_dir_txt\":\"%s\"", buf);
        [...]

franki29, 2022-08-12:
Hi, yes, this is working, thanks: :+1: 
{"sensor_id": 14200203,"ch:":0,"battery_ok":1,"temp_c":33.0,"humidity":24,"wind_gust":0.8,"wind_avg":0.8,"wind_dir":158.0,"rain":541.6},"wind_dir":158.0,"wind_dir_txt":"SSE"
{"sensor_id": 14200203,"ch:":0,"battery_ok":1,"temp_c":32.9,"humidity":24,"wind_gust":1.4,"wind_avg":1.2,"wind_dir":202.0,"rain":541.6},"wind_dir":202.0,"wind_dir_txt":"SSW"
{"rssi":-66.5}
{"sensor_id": 14200203,"ch:":0,"battery_ok":1,"temp_c":32.9,"humidity":24,"wind_gust":0.6,"wind_avg":0.6,"wind_dir":202.0,"rain":541.6},"wind_dir":202.0,"wind_dir_txt":"SSW"
{"sensor_id": 14200203,"ch:":0,"battery_ok":1,"temp_c":32.9,"humidity":24,"wind_gust":0.8,"wind_avg":0.8,"wind_dir":202.0,"rain":541.6},"wind_dir":202.0,"wind_dir_txt":"SSW"
{"rssi":-68.5}
{"sensor_id": 14200203,"ch:":0,"battery_ok":1,"temp_c":33.0,"humidity":24,"wind_gust":0.0,"wind_avg":0.0,"wind_dir":158.0,"rain":541.6},"wind_dir":158.0,"wind_dir_txt":"SSE"

closed 2022-08-12 by franki29