matthias-bs/BresserWeatherSensorReceiver #6: Adding Domoticz Support ?

franki29, 2022-08-12:
Hi, is it possible to add mqtt Domoticz Support?

Currently I am using node red  to convert json string into Domoticz message:

msg1.payload = {"idx":1810,"nvalue":0,"svalue":msg.payload.wind_dir+";"+msg.payload.wind_dir_txt+";"+msg.payload.wind_avg*10+";"+msg.payload.wind_gust*10+";"+msg.payload.temp_c+";"+msg.payload.temp_c}

Where : 

IDX = id of your device (This number can be found in the devices tab in the column "IDX")
WB = Wind bearing (0-359)
WD = Wind direction (S, SW, NNW, etc.)
WS = 10 * Wind speed [m/s]
WG = 10 * Gust [m/s]
22 = Temperature
24 = Temperature Windchill

MQTT Topic is domoticz/in

matthias-bs, 2022-08-12:
Yes, this is certainly possible. You just have to format the JSON string with sprintf() accordingly. Just adapt the example - if you encounter any problems, please ask.

matthias-bs, 2022-08-12:
Maybe a domoticz example can be implemented in addition to the existing MQTT example.

franki29, 2022-08-14:
HI, I made a quick and dirty hack:

At // MQTT topics:
const char MQTT_PUB_DOMO[]        = "domoticz/in";

At void publishWeatherdata(bool complete):

 char domo_payload[PAYLOAD_SIZE];
 char domo2_payload[PAYLOAD_SIZE];

sprintf(domo_payload, "{\"idx\":877,\"nvalue\":0,\"svalue\":\"%.1f", weatherSensor.wind_direction_deg);
    sprintf(&domo_payload[strlen(domo_payload)], ";%s", buf);
sprintf(&domo_payload[strlen(domo_payload)], ";%.1f", weatherSensor.wind_avg_meter_sec *10);
      sprintf(&domo_payload[strlen(domo_payload)], ";%.1f", weatherSensor.wind_gust_meter_sec *10); 
      sprintf(&domo_payload[strlen(domo_payload)], ";%.1f", weatherSensor.temp_c);
        sprintf(&domo_payload[strlen(domo_payload)], ";%.1f", weatherSensor.temp_c);
        sprintf(&domo_payload[strlen(domo_payload)], "\"}");

sprintf(domo2_payload, "{\"idx\":878,\"nvalue\":0,\"svalue\":\"%.1f", weatherSensor.rain_mm *100);
      sprintf(&domo2_payload[strlen(domo2_payload)], ";%.1f", weatherSensor.rain_mm);
     sprintf(&domo2_payload[strlen(domo2_payload)], "\"}");

client.publish(MQTT_PUB_DOMO, &domo_payload[0], false,0);
client.publish(MQTT_PUB_DOMO, &domo2_payload[0], false,0);

Br. Frank

matthias-bs, 2022-08-14:
Hi,

That's good! I think this might be useful for others, too!

What do you think of adding your code as another example?  Either by making a fork/pull request or just adding your file as an attachment to this discussion,  so I can add it.

Best regards
Matthias

franki29, 2022-08-14:
Hi, as I am not a programmer, I think it is a too dirty hack.

But for people who want to use domoticz implementation through mqtt it might be a helpful example.
May be you can clean the code a little bit if you find some stupid things that might can be better and add it as a new example.
Here my files and a picture

I also forgot to implement my third Sensor:

char domo3_payload[PAYLOAD_SIZE];

  sprintf(domo3_payload, "{\"idx\":876,\"nvalue\":0,\"svalue\":\"%.1f",weatherSensor.temp_c);
     sprintf(&domo3_payload[strlen(domo3_payload)], ";%d", weatherSensor.humidity);
     sprintf(&domo3_payload[strlen(domo3_payload)], ";0\"}");

client.publish(MQTT_PUB_DOMO, &domo3_payload[0], false,0)
Br. Frank

[BresserWeatherSensorMQTT_domo_example.zip](https://github.com/matthias-bs/BresserWeatherSensorReceiver/files/9333281/BresserWeatherSensorMQTT_domo_example.zip)
[image]
;

Br

matthias-bs, 2022-08-14:
Thank you! I will have a look at it when time permits.

matthias-bs, 2022-08-15:
Hi,

Could you please check where the dew point and the "feels-like ... temperature are expected?
```
    // domoticz virtual wind sensor
    sprintf(domo_payload, "{\"idx\":%d,\"nvalue\":0,\"svalue\":\"%.1f", 
       DOMO_WIND_IDX, weatherSensor.sensor[i].wind_direction_deg);
    char buf[4];
    sprintf(&domo_payload[strlen(domo_payload)], ";%s", winddir_flt_to_str(weatherSensor.sensor[i].wind_direction_deg, buf));
    sprintf(&domo_payload[strlen(domo_payload)], ";%.1f", weatherSensor.sensor[i].wind_avg_meter_sec *10);
    sprintf(&domo_payload[strlen(domo_payload)], ";%.1f", weatherSensor.sensor[i].wind_gust_meter_sec *10);
    sprintf(&domo_payload[strlen(domo_payload)], ";%.1f", weatherSensor.sensor[i].temp_c);
    sprintf(&domo_payload[strlen(domo_payload)], ";%.1f", weatherSensor.sensor[i].temp_c);
    sprintf(&domo_payload[strlen(domo_payload)], "\"}");
        
    // domoticz virtual rain sensor
    sprintf(domo2_payload, "{\"idx\":%d,\"nvalue\":0,\"svalue\":\"%.1f", 
      DOMO_RAIN_IDX, weatherSensor.sensor[i].rain_mm *100);
    sprintf(&domo2_payload[strlen(domo2_payload)], ";%.1f", weatherSensor.sensor[i].rain_mm);
    sprintf(&domo2_payload[strlen(domo2_payload)], "\"}");
     
    // domoticz virtual temp & humidity sensor
    sprintf(domo3_payload, "{\"idx\":%d,\"nvalue\":0,\"svalue\":\"%.1f", DOMO_TH_IDX, weatherSensor.sensor[i].temp_c);
    sprintf(&domo3_payload[strlen(domo3_payload)], ";%d", weatherSensor.sensor[i].humidity);
    sprintf(&domo3_payload[strlen(domo3_payload)], ";0\"}");
```
Best regards
Matthias

franki29, 2022-08-16:
Hi , it  does not compile using ES8266:

In file included from /home/frank/code/BresserWeatherSensorReceiver/examples/BresserWeatherSensorDomoticz/WeatherSensor.cpp:56:
WeatherSensor.h:210:25: error: 'NUM_SENSORS' was not declared in this scope
  210 |         sensor_t sensor[NUM_SENSORS];      // !> sensor data array
      |                         ^~~~~~~~~~~
/home/frank/code/BresserWeatherSensorReceiver/examples/BresserWeatherSensorDomoticz/WeatherSensor.h: In member function 'void WeatherSensor::clearSlots(uint8_t)':
WeatherSensor.h:232:30: error: 'NUM_SENSORS' was not declared in this scope
  232 |             for (int i=0; i< NUM_SENSORS; i++) {
      |                              ^~~~~~~~~~~
WeatherSensor.h:233:40: error: 'sensor' was not declared in this scope; did you mean 'Sensor'?
  233 |                 if ((type == 0xFF) || (sensor[i].s_type == type)) {
      |                                        ^~~~~~
      |                                        Sensor
/home/frank/code/BresserWeatherSensorReceiver/examples/BresserWeatherSensorDomoticz/WeatherSensor.cpp: In member function 'bool WeatherSensor::getData(uint32_t, uint8_t, uint8_t, void (*)())':
WeatherSensor.cpp:153:29: error: 'NUM_SENSORS' was not declared in this scope
  153 |             for (int i=0; i<NUM_SENSORS; i++) {
      |                             ^~~~~~~~~~~
WeatherSensor.cpp:154:22: error: 'sensor' was not declared in this scope; did you mean 'Sensor'?
  154 |                 if (!sensor[i].valid) {
      |                      ^~~~~~
      |                      Sensor
WeatherSensor.cpp:164:52: error: 'sensor' was not declared in this scope; did you mean 'Sensor'?
  164 |                 if (((flags & DATA_TYPE) != 0) && (sensor[i].s_type == type)) {
      |                                                    ^~~~~~
      |                                                    Sensor
/home/frank/code/BresserWeatherSensorReceiver/examples/BresserWeatherSensorDomoticz/WeatherSensor.cpp: In member function 'bool WeatherSensor::genMessage(int, uint32_t, uint8_t, uint8_t)':
WeatherSensor.cpp:266:5: error: 'sensor' was not declared in this scope; did you mean 'Sensor'?
  266 |     sensor[i].sensor_id               = id;
      |     ^~~~~~
      |     Sensor
/home/frank/code/BresserWeatherSensorReceiver/examples/BresserWeatherSensorDomoticz/WeatherSensor.cpp: In member function 'int WeatherSensor::findSlot(uint32_t, DecodeStatus*)':
WeatherSensor.cpp:304:28: error: 'sensor_ids_exc' was not declared in this scope
  304 |     uint8_t n_exc = sizeof(sensor_ids_exc)/4;
      |                            ^~~~~~~~~~~~~~
WeatherSensor.cpp:316:28: error: 'sensor_ids_inc' was not declared in this scope
  316 |     uint8_t n_inc = sizeof(sensor_ids_inc)/4;
      |                            ^~~~~~~~~~~~~~
WeatherSensor.cpp:335:21: error: 'NUM_SENSORS' was not declared in this scope
  335 |     for (int i=0; i<NUM_SENSORS; i++) {
      |                     ^~~~~~~~~~~
WeatherSensor.cpp:337:14: error: 'sensor' was not declared in this scope; did you mean 'Sensor'?
  337 |         if (!sensor[i].valid && (free_slot < 0)) {
      |              ^~~~~~
      |              Sensor
/home/frank/code/BresserWeatherSensorReceiver/examples/BresserWeatherSensorDomoticz/WeatherSensor.cpp: In member function 'DecodeStatus WeatherSensor::decodeBresser6In1Payload(uint8_t*, uint8_t)':
WeatherSensor.cpp:677:5: error: 'sensor' was not declared in this scope; did you mean 'Sensor'?
  677 |     sensor[slot].sensor_id   = id_tmp;
      |     ^~~~~~
      |     Sensor
In file included from /home/frank/code/BresserWeatherSensorReceiver/examples/BresserWeatherSensorDomoticz/BresserWeatherSensorDomoticz.ino:118:
WeatherSensor.h:210:25: error: 'NUM_SENSORS' was not declared in this scope
  210 |         sensor_t sensor[NUM_SENSORS];      // !> sensor data array
      |                         ^~~~~~~~~~~
/home/frank/code/BresserWeatherSensorReceiver/examples/BresserWeatherSensorDomoticz/WeatherSensor.h: In member function 'void WeatherSensor::clearSlots(uint8_t)':
WeatherSensor.h:232:30: error: 'NUM_SENSORS' was not declared in this scope
  232 |             for (int i=0; i< NUM_SENSORS; i++) {
      |                              ^~~~~~~~~~~
WeatherSensor.h:233:40: error: 'sensor' was not declared in this scope; did you mean 'Sensor'?
  233 |                 if ((type == 0xFF) || (sensor[i].s_type == type)) {
      |                                        ^~~~~~
      |                                        Sensor
/home/frank/code/BresserWeatherSensorReceiver/examples/BresserWeatherSensorDomoticz/BresserWeatherSensorDomoticz.ino: In function 'void publishWeatherdata()':
BresserWeatherSensorDomoticz:364:23: error: 'class WeatherSensor' has no member named 'sensor'; did you mean 'Sensor'?
  364 |     if (weatherSensor.sensor[i].wind_ok && weatherSensor.sensor[i].temp_ok) {
      |                       ^~~~~~
      |                       Sensor
BresserWeatherSensorDomoticz:364:58: error: 'class WeatherSensor' has no member named 'sensor'; did you mean 'Sensor'?
  364 |     if (weatherSensor.sensor[i].wind_ok && weatherSensor.sensor[i].temp_ok) {
      |                                                          ^~~~~~
[... 39 lines cut ...]
  381 |         DOMO_RAIN_IDX, weatherSensor.sensor[i].rain_mm *100);
      |                                      ^~~~~~
      |                                      Sensor
[... cut ...]
      |                                                                                                       Sensor
BresserWeatherSensorDomoticz:391:75: error: 'class WeatherSensor' has no member named 'sensor'; did you mean 'Sensor'?
  391 |       sprintf(&domo3_payload[strlen(domo3_payload)], ";%d", weatherSensor.sensor[i].humidity);
      |                                                                           ^~~~~~
      |                                                                           Sensor
exit status 1
'NUM_SENSORS' was not declared in this scope

franki29, 2022-08-16:
And there is a { too much in your domoticz example at:

// Generate WiFi network instance
#if defined(ESP32)
    WiFiClientSecure net;
**#elif defined(ESP8266) {**
   // BearSSL::WiFiClientSecure net;
   WiFiClient net;
#endif

matthias-bs, 2022-08-16:
> Hi , it does not compile using ES8266:
> 
Please use the updated files from https://github.com/matthias-bs/BresserWeatherSensorReceiver/tree/main/src, modify `WeatherSensorCfg.h.template` as required and save it as `WeatherSensor.Cfg`. If you only have on sensor, you can set `NUM_SENSORS` to 1.

matthias-bs, 2022-08-16:
> And there is a { too much in your domoticz example at:
> 
> // Generate WiFi network instance #if defined(ESP32) WiFiClientSecure net; **#elif defined(ESP8266) {** // BearSSL::WiFiClientSecure net; WiFiClient net; #endif

Fixed!

franki29, 2022-08-16:
Still compile problems using port 1883

Arduino: 1.8.19 (Linux), Board: "LOLIN(WEMOS) D1 mini (clone), 80 MHz, Flash, Disabled (new aborts on oom), Disabled, All SSL ciphers (most compatible), 32KB cache + 32KB IRAM (balanced), Use pgm_read macros for IRAM/PROGMEM, DOUT (compatible), 40MHz, 4MB (FS:2MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

/home/frank/code/BresserWeatherSensorReceiver/examples/BresserWeatherSensorDomoticz/BresserWeatherSensorDomoticz.ino: In function 'void mqtt_setup()':
BresserWeatherSensorDomoticz:291:14: error: 'class WiFiClient' has no member named 'setInsecure'
  291 |          net.setInsecure();
      |              ^~~~~~~~~~~
exit status 1
'class WiFiClient' has no member named 'setInsecure'

matthias-bs, 2022-08-16:
This part should not be compiled:
```
       if (MQTT_PORT != 1883) {
         net.setInsecure();
       }
```
Did you create a `secrets.h` and change MQTT_PORT to 1883? (preferred)
Or did you edit the example section in BresserWeatherSensorDomoticz.ino accordingly?

franki29, 2022-08-16:
Yes , i use a secret.h from and change the port to 1883

matthias-bs, 2022-08-16:
And is there a line `#define SECRET` in it which prevents that the code block in *.ino is used?

franki29, 2022-08-16:
Yes, at the first line of secrets.h.

franki29, 2022-08-16:
It compiles if I do:
 #endif
    #if (!defined(CHECK_PUB_KEY) and !defined(CHECK_CA_ROOT) and !defined(CHECK_FINGERPRINT))
       if (MQTT_PORT != 1883) {
//         net.setInsecure();
       }
    #endif

franki29, 2022-08-16:
Only perceived_temperature is missing in your example for domoticz.

matthias-bs, 2022-08-16:
Well, this is strange! If MQTT_PORT  __is__ 1883 and the line gets executed if MQTT_PORT __is not__ 1883, then I am missing something.
Could you please add the following line after `net.setInsecure();` and check the result?
`Serial.printf("MQTT_PORT: %d\n", MQTT_PORT);`

matthias-bs, 2022-08-16:
> Only perceived_temperature is missing in your example for domoticz.
Yes, please check my question above. Same as for dew point.

franki29, 2022-08-17:
HI, there is no serial print out.
    #if (!defined(CHECK_PUB_KEY) and !defined(CHECK_CA_ROOT) and !defined(CHECK_FINGERPRINT))
       if (MQTT_PORT != 1883) {
//        net.setInsecure();
Serial.printf("MQTT_PORT: %d\n", MQTT_PORT);
       }
    #endif

franki29, 2022-08-17:
Remember you have it twice in:
at line  224 where you switch to WifiClient (and this is working)
and in line 291. I think because you switch to WifiClient in line 224, the compile error happens. ?

matthias-bs, 2022-08-17:
Ahhh, o.k. - I got it wrong! On ESP8266, you have to use WiFiClient for non-secure connections and BearSSL::WiFiClientSecure for secure connections (TLS). If you use TLS, but you cannot validate in any way, you have to call `net.setInsecure()` - the comment in [WiFiClientSecureBearSSL.h](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.h) says:

   > // Don't validate the chain, just accept whatever is given.  VERY INSECURE!

Besides, I learned that it is required to set the time (via NTP) first in order to be able to do x.509 validation. ([BearSSL_Validation.ino](https://github.com/esp8266/Arduino/blob/313b3c07ecccbe6fee24aa9fa447c4aed16ca499/libraries/ESP8266WiFi/examples/BearSSL_Validation/BearSSL_Validation.ino#L27))

It's a little bit confusing (for me, at least) that there are subtle differences between the ESP32 and the ESP8266 in that part of the libraries and that the differences between secure and non-secure connections are not entirely hidden by the API...

However, now I should be able to provide a real fix!

matthias-bs, 2022-08-17:
Regarding the calculated values: Could you please check below if I understood this correctly?
```
    // domoticz virtual wind sensor
      sprintf(domo_payload, "{\"idx\":%d,\"nvalue\":0,\"svalue\":\"%.1f", 
         DOMO_WIND_IDX, weatherSensor.sensor[i].wind_direction_deg);
      char buf[4];
      sprintf(&domo_payload[strlen(domo_payload)], ";%s", winddir_flt_to_str(weatherSensor.sensor[i].wind_direction_deg, buf));
      sprintf(&domo_payload[strlen(domo_payload)], ";%.1f", weatherSensor.sensor[i].wind_avg_meter_sec *10);
      sprintf(&domo_payload[strlen(domo_payload)], ";%.1f", weatherSensor.sensor[i].wind_gust_meter_sec *10);
      sprintf(&domo_payload[strlen(domo_payload)], ";%.1f", weatherSensor.sensor[i].temp_c);
      sprintf(&domo_payload[strlen(domo_payload)], ";%.1f", weatherSensor.sensor[i].temp_c); // <--- windchill / heat index here???
      sprintf(&domo_payload[strlen(domo_payload)], "\"}");
      Serial.printf("%s: %s\n", MQTT_PUB_DOMO, domo_payload);
      client.publish(MQTT_PUB_DOMO, domo_payload, false, 0);      
    
   
    // domoticz virtual temp & humidity sensor
      sprintf(domo3_payload, "{\"idx\":%d,\"nvalue\":0,\"svalue\":\"%.1f", DOMO_TH_IDX, weatherSensor.sensor[i].temp_c);
      sprintf(&domo3_payload[strlen(domo3_payload)], ";%d", weatherSensor.sensor[i].humidity);
      sprintf(&domo3_payload[strlen(domo3_payload)], ";0\"}"); // <--- dew point here?
      Serial.printf("%s: %s\n", MQTT_PUB_DOMO, domo3_payload);
      client.publish(MQTT_PUB_DOMO, domo3_payload, false, 0);

```

franki29, 2022-08-18:
Yes, the second one is the "Chill" factor:
If I am right it should work with:
 sprintf(&domo_payload[strlen(domo_payload)], ";%.1f", weatherSensor.sensor[i].temp_c);
      sprintf(&domo_payload[strlen(domo_payload)], ";%.1f",  perceived_temperature(weatherSensor.sensor[i].temp_c,weatherSensor.sensor[i].wind_avg_meter_sec,weatherSensor.sensor[i].humidity));

franki29, 2022-08-18:
Hi, I also add the Humindex:
for the formula I used part of: https://github.com/Dariu5/Humidity-meter-with-TFT-display-and-DHT22-sensor
[BresserWeatherSensorDomoticz_example.zip](https://github.com/matthias-bs/BresserWeatherSensorReceiver/files/9371804/BresserWeatherSensorDomoticz_example.zip)

franki29, 2022-08-18:
[image]
[image]

matthias-bs, 2022-08-18:
What is Humindex? It is different from the Heat Index you provided the link to earlier and which is already part of WeatherUtils or `perceived_temperature()`, respectively?

matthias-bs, 2022-08-18:
How about my previous question regarding the desired place for the dew point value? ;-)

franki29, 2022-08-19:
Hi, Humidex is an index number introduced by Canadian meteorologists to describe the perceived temperature of an average person as a function of air temperature and humidity or dew point. ( I got it from your link in WaetherUtils.cpp)
It is just a other way to show "feels temperature"
https://myscope.net/hitzeindex-gefuehle-temperatur/
[image]

franki29, 2022-08-19:
Regarding dew point: this is calulated by domoticz and it is not needed to send.
[image]

matthias-bs, 2022-08-19:
Ah, o.k. I will add Humidex to WeatherUtils. Thanks for pointing this out.

matthias-bs, 2022-08-19:
Thank you for your response regarding dew point display in Domoticz!

closed 2022-08-20 by franki29

matthias-bs, 2022-08-23:
Added Humidex.