matthias-bs/BresserWeatherSensorReceiver #7: Add windchill information

franki29, 2022-08-12:
Hi, is it possible to add windchill calculation into your code?

matthias-bs, 2022-08-12:
Sure it is possible from a technical point of view. I am confident that someone has done this before. Did you search for a possible implementation (maybe on Github) which can be adopted?
At first glance, there are two possible approaches -  a rather time consuming floating point calculation (which should not be done on each and every data frame) or a table look-up (maybe with linear interpolation).
Besides, it is a little bit out-of-scope with respect to this library. It seems rather to be application specific.
Regards
Matthias

matthias-bs, 2022-08-12:
https://de.m.wikipedia.org/wiki/Windchill

franki29, 2022-08-13:
Hi, I found some interesting side here:
https://www.brunweb.de/wetterstation-berechnungen/

Regards
Frank

matthias-bs, 2022-08-14:
Hi,

I have added [BresserWeatherSensorReceiver/src/WeatherUtils.h](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/src/WeatherUtils.h) and [BresserWeatherSensorReceiver/src/WeatherUtils.cpp](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/src/WeatherUtils.cpp).
You can try it by copying both files in the same directory as you *.ino file and adding `#include "WeatherUtils.h"`to your sketch.

Best regards
Matthias

franki29, 2022-08-14:
Hi, I tried to implement, but I get result 0.0:

  sprintf(&domo_payload[strlen(domo_payload)], ";%.1f", calcwindchill);

What did I made wrong?

franki29, 2022-08-14:
Or is it because there i no wind currently?
Than windchill should be set to actual temperature, what do you think?

matthias-bs, 2022-08-14:
You have to pass parameters in the function call: `calcwindchill(weatherSensor.temp_c, weatherSensor.wind_avg_meter_sec)`
And: The windchill formular is only valid if temperature <= 10°C and windspeed >4.8 km/h!
Maybe you should try calheatindex() today if you live in Germany or anywhere else in Central Europe! ;-)

matthias-bs, 2022-08-14:
> Or is it because there i no wind currently? Than windchill should be set to actual temperature, what do you think?

Yes, if the conditions are not met, the actual temperature should be returned. I will all this later.

franki29, 2022-08-14:
Ah, I am learning.
I add now:
 if (weatherSensor.temp_c >=11) {        
        sprintf(&domo_payload[strlen(domo_payload)], ";%.1f", calcheatindex(weatherSensor.temp_c,weatherSensor.humidity));
        }
        if (weatherSensor.temp_c <=10) {
          sprintf(&domo_payload[strlen(domo_payload)], ";%.1f", calcwindchill(weatherSensor.temp_c, weatherSensor.wind_avg_meter_sec));     
        }
And get now a heatindex that seems to be OK

franki29, 2022-08-14:
[image]

franki29, 2022-08-14:
[image]

matthias-bs, 2022-08-14:
Almost correct! ;-)

Windchill: Results are valid for temperatures <= 10°C and windspeeds >4.8 km/h only!
Heat index: Results are valid for temperatures >= 16,7°C and humidity >40% only!

This can be combined as "Gefühlte Temperatur" or "Feels like xx°C"

franki29, 2022-08-14:
Hmmm,
maybe this will work?

if ((weatherSensor.temp_c >=11) && (weatherSensor.humidity > 40)) {        
        sprintf(&domo_payload[strlen(domo_payload)], ";%.1f", calcheatindex(weatherSensor.temp_c,weatherSensor.humidity));
        }
        else  if ((weatherSensor.temp_c <=10) && ( weatherSensor.wind_avg_meter_sec *3.6 > 4.8)) {
          sprintf(&domo_payload[strlen(domo_payload)], ";%.1f", calcwindchill(weatherSensor.temp_c, weatherSensor.wind_avg_meter_sec));     
        }
       else {
        sprintf(&domo_payload[strlen(domo_payload)], ";%.1f", weatherSensor.temp_c);
       }

matthias-bs, 2022-08-15:
Added perceptive_temperature() in [BresserWeatherSensorReceiver/src/WeatherUtils.h](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/src/WeatherUtils.h) and [BresserWeatherSensorReceiver/src/WeatherUtils.cpp](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/src/WeatherUtils.cpp). (https://github.com/matthias-bs/BresserWeatherSensorReceiver/commit/1f1f97a02fc07bcf028ae8d01fc59a0c84bacd70)

closed 2022-08-19 by franki29