matthias-bs/BresserWeatherSensorReceiver #8: Support of BME280

franki29, 2022-08-19:
Hi, me again, sorry.
It would be also nice to add BME280 support to have also in house data.
Please find attached a very dirty hack again for supporting BME280.
[BresserWeatherSensorDomoticz_example.zip](https://github.com/matthias-bs/BresserWeatherSensorReceiver/files/9381596/BresserWeatherSensorDomoticz_example.zip)

[image]

It is tricky using a ESP8266 because of limited GPIO's, but at my side it is working with a wemos nodemcu clone device
Br Frank

matthias-bs, 2022-08-19:
Hi,
sorry to say this, but this is clearly out of scope. I think it's time for your own fork. You seem to have a solution (at least partly) already - please let me know if you need help.
Best regards
Matthias

P.S.: I'm using the BME280 in my flora2 project. Nice little devices, but the rise in prices over the last months makes you dizzy...

franki29, 2022-08-20:
OK, thanks for information.
Br. Frank

closed 2022-08-20 by franki29