matthias-bs/BresserWeatherSensorReceiver #10: Bresser Lightning Sensor

franki29, 2022-09-17:
Hi, as you wrote in the Wiki, you also like to add the lightning sensor.
As there seems to be no reaction at https://github.com/merbanan/rtl_433/issues/2140, i like to share my first tests decoder with you.
If it possible could you please test it with your device?

Thanks

[bresser_lightning.zip](https://github.com/matthias-bs/BresserWeatherSensorReceiver/files/9591297/bresser_lightning.zip)

matthias-bs, 2022-09-17:
Hi,
I think the Lightning Sensors are interesting from a technical point of view, but I don't have one yet and lightnings are quite rare in my area, so currently I'm hesitating to buy one. Sorry, I'm afraid currently I won't be able to test the decoder.
Cheers, 
M.

franki29, 2022-09-18:
OK, same here regarding the lightning.
Is it possible to implement the test decoder for you as I am not able to do?
Changing values is not the problem for me if there will be changes in the decoder, but implementation in your current code is not possible for me.

Thanks and best regards

matthias-bs, 2022-09-18:
Hi,
I will keep it in mind but I cannot make any promise if and when I will come back to it.
BR,
M.

matthias-bs, 2023-02-08:
As an alternative, for integration into the base station:
https://www.dfrobot.com/product-1828.html

AndreyMihaylov85, 2023-05-26:
Hello,
i was able to run it  with attached decoder, 
[bresser_lightning2.zip](https://github.com/matthias-bs/BresserWeatherSensorReceiver/files/11575858/bresser_lightning2.zip)

here is the  result, 

{"time" : "2023-05-26 13:47:47", "model" : "Bresser_lightning", "id" : 42806, "Frage1" : 50, "Kilometer" : 0, "CTR" : 0, "Frage2" : 0, "mic" : "CRC", "battery_low" : 0}#012'

now Im trying to figured out why Weewx Sdr driver can not identify the model even thought all it seems fine into sdr.py 

class Bresser_lightning(Packet):
    #{"time" : "2023-05-26 13:32:11", "model" : "Bresser_lightning", "id" : 42806, "Frage1" : 50, "Kilometer" : 0, "CTR" : 0, "Frage2" : 0, "mic" : "CRC", "battery_low" : 0}

    IDENTIFIER = "Bresser_lightning"

    @staticmethod
    def parse_json(obj):
        pkt = dict()
        pkt['dateTime'] = Packet.parse_time(obj.get('time'))
        pkt['usUnits'] = weewx.METRICWX
        pkt['station_id'] = obj.get('id')
        pkt['Frage1'] = Packet.get_float(obj, 'Frage1')
        pkt['Kilometer'] = Packet.get_float(obj, 'Kilometer')
        pkt['CTR'] = Packet.get_float(obj, 'CTR')
        pkt['Frage2'] = Packet.get_float(obj, 'Frage2')
        pkt['battery_low'] = 0 if obj.get('battery') == 'OK' else 1

        # deal with different labels from rtl_433
        for dst, src in [('Frage1', 'Frage1'),
                         ('Kilometer', 'Kilometer'),
                         ('CTR', 'CTR'),
                         ('Frage2', 'Frage2')]:
            if src in obj:
                pkt[dst] = Packet.get_float(obj, src)
        return  Bresser_lightningPacket.insert_ids(pkt)

    @staticmethod
    def insert_ids(pkt):
        station_id = pkt.pop('station_id', '0000')
        pkt = Packet.add_identifiers(pkt, station_id,  Bresser_lightning.__name__)
        return pkt

matthias-bs, 2023-05-26:
@AndreyMihaylov85 Excellent! Thank you for sharing this! As already mentioned in https://github.com/merbanan/rtl_433/issues/2140, the device is based on Austria Micro System's AS3935 Franklin Lightning Sensor IC. So the answer to Frage1 or Frage2 could be "Energy of the Single Lightning".  The IC provides this as a 20-bit value, but it might be that only the 16 most significant bits are transmitted here (in this case as "Frage2").

According to the "4Cast PRO SF WIFI Weather Center with 7in1 Outdoor sensor" manual, also "Period of time since last lightning" and "Number of lightnings per hour" are shown on the display. This means, also transmitting of number of lightning strikes between the transmission intervals would be reasonable. From this, the values shown could be calculated by the base station.

Anyway, I'll try to integrate the decoder as soon as my spare time permits.

matthias-bs, 2023-06-24:
Hi,

Finally I bought a Bresser Lightning Sensor and started on integrating/modifying the decoder.
Here is a first draft: https://github.com/matthias-bs/BresserWeatherSensorReceiver/tree/10-bresser-lightning-sensor
Please check with the "Basic" example.

Cheers,
Matthias

franki29, 2023-07-02:
Hi, does not work for my sensor. 
no data, and the diggest check is failing.
Also it does not recognize my Bresser weather station (6in1)
Br. Frank

franki29, 2023-07-02:
SX1276] Data:		 D4 73 69 B5 8 AA A2 90 AA AA AA 0 0 0 0 0 0 0 0 0 7F FF FF FF FF FF FF
Raw Data: 73 69 B5 08 AA A2 90 AA AA AA 00 00 00 00 00 00 00 00 00 7F FF FF FF FF FF FF 15 

This are the information when  I use the older software and press the button at my sensor to send the data.
(under the reset button there is a hole with a hidden button)

matthias-bs, 2023-07-06:
Oh, I forgot to mention that you have to modify your `WeatherSensorCfg.h`! In 
https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/9b5bc1b60424c305e201b6a53dde98a2a96f1d14/src/WeatherSensorCfg.h#L173 I added the ID of my lightning sensor to make debugging easier. You should remove the ID or replace it with your own (0x1FA2)!

If I inject the raw data you posted above, I get the following:
```
[D][WeatherSensor.cpp:1028] decodeBresserLightningPayload(): ID: 0x1FA2  TYPE: 3  CTR: 0  batt_low: 0  distance_km: 0  unknown1: 0x83a  unknown2: 0x0000
Id: [    1FA2] Typ: [3] Battery: [OK ] Lightning Counter: [  0] Distance: [ 0km] unknown1: [0x03A] unknown2: [0x0000] RSSI: [-103.0dBm]
```
Let's see if we can get some test data on Sunday... ;-)

matthias-bs, 2023-07-06:
I have added a few hints about emulating lightning strikes for this kind of sensor:
https://github.com/matthias-bs/BresserWeatherSensorReceiver/wiki#lightning-sensor

closed 2023-07-09 by matthias-bs