matthias-bs/BresserWeatherSensorReceiver #11: Radio module not recognized (radio.beginFSK failed)

MariaGardener, 2022-09-24:
Hi,

I have a problem with radio module initialization and fixed it as follows:

I changed in WeatherSensor.cpp the line 
`static SX1276 radio = new Module(PIN_RECEIVER_CS, PIN_RECEIVER_IRQ, PIN_RECEIVER_RST, PIN_RECEIVER_GPIO);`

with hard coded pin settings for my case:

`static SX1276 radio = new Module(27, 21,32, 33);`

Then it works. There might be something wrong with IN_RECEIVER_CS, PIN_RECEIVER_IRQ, PIN_RECEIVER_RST, PIN_RECEIVER_GPIO
since they are not known in this context though I have declared them in WeatherSensorCfg.h

I would appreciate any advice, may I have done something wrong.

Cheers Maria

matthias-bs, 2022-10-24:
I hope I can provide a clean solution now.

1. I have modified the library and the example [BresserWeatherSensorMQTT.ino](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/examples/BresserWeatherSensorMQTT/BresserWeatherSensorMQTT.ino)

    This example now does not have a configuration file of its own any longer, but uses the one in BresserWeatherSensorReceiver/src. You can change it in your local Arduino/libraries/BresserWeatherSensorReceiver/src directory if you are always using the same setup (receiver type and wiring).

    I will change the other examples in the same way.

2. I have added a customized version in [BresserWeatherSensorMQTTCustom](https://github.com/matthias-bs/BresserWeatherSensorReceiver/tree/main/examples/BresserWeatherSensorMQTTCustom). Please check the [Readme.md](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/examples/BresserWeatherSensorMQTTCustom/Readme.md).

    Following this structure, you are able to use a customized configuration per sketch.

matthias-bs, 2022-11-09:
I assume this has been fixed in release [v0.4.0](https://github.com/matthias-bs/BresserWeatherSensorReceiver/releases/tag/v0.4.0).

closed 2022-11-09 by matthias-bs