matthias-bs/BresserWeatherSensorReceiver #12: [HELP] How to compile this correctly

gitpower2017, 2022-10-19:
Hello.

I could not find any instructions in the wiki on how to compile the software.

In the file "WeatherSensorCFG.h" I have defined my ESP8266, cc1101 and the 5in1 sensor. 

Only how does it go on then?

Greetings Lothar

gitpower2017, 2022-10-19:
Hello.

I have make lot of test.

But was is this Error?

undefined reference to `_ZN9RainGauge6updateE2tmff'
undefined reference to `Z19windspeed_ms_to_bftf'

and a lot more.

--- LONG ---

/home/ls/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /tmp/arduino-sketch-F6DF9EA8EA0AD8A141217D84F3C1ACEB/sketch/BresserWeatherSensorMQTT.ino.cpp.o: in function `_Z10mqtt_setupv':
/home/ls/Dokumente/Wetter5in1_BresserWeatherSensorReceiver/BresserWeatherSensorReceiver/examples/BresserWeatherSensorMQTT/BresserWeatherSensorMQTT.ino:326: undefined reference to `_ZN9RainGauge6updateE2tmff'
/home/ls/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /tmp/arduino-sketch-F6DF9EA8EA0AD8A141217D84F3C1ACEB/sketch/BresserWeatherSensorMQTT.ino.cpp.o: in function `_ZNKSt12__shared_ptrIN7BearSSL19WiFiClientSecureCtxELN9__gnu_cxx12_Lock_policyE0EE3getEv':
/home/ls/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/xtensa-lx106-elf/include/c++/10.3.0/bits/shared_ptr_base.h:1325: undefined reference to `_Z18winddir_flt_to_strfPc'
/home/ls/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /home/ls/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/xtensa-lx106-elf/include/c++/10.3.0/bits/shared_ptr_base.h:1325: undefined reference to `_Z19windspeed_ms_to_bftf'
/home/ls/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /tmp/arduino-sketch-F6DF9EA8EA0AD8A141217D84F3C1ACEB/sketch/BresserWeatherSensorMQTT.ino.cpp.o: in function `_ZN7BearSSL19WiFiClientSecureCtx11setInsecureEv':
/home/ls/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.h:68: undefined reference to `_Z12calcdewpointff'
/home/ls/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /home/ls/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.h:69: undefined reference to `_Z21perceived_temperaturefff'
/home/ls/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /tmp/arduino-sketch-F6DF9EA8EA0AD8A141217D84F3C1ACEB/sketch/BresserWeatherSensorMQTT.ino.cpp.o: in function `_ZN10MQTTClient5beginEPKciR6Client':
/home/ls/bin/Arduino/libraries/MQTT/src/MQTTClient.h:102: undefined reference to `_ZN9RainGauge10currentDayEv'
/home/ls/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /home/ls/bin/Arduino/libraries/MQTT/src/MQTTClient.h:102: undefined reference to `_ZN9RainGauge11currentWeekEv'
/home/ls/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /home/ls/bin/Arduino/libraries/MQTT/src/MQTTClient.h:102: undefined reference to `_ZN9RainGauge12currentMonthEv'
/home/ls/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /tmp/arduino-sketch-F6DF9EA8EA0AD8A141217D84F3C1ACEB/sketch/BresserWeatherSensorMQTT.ino.cpp.o:(.text._Z18publishWeatherdatab+0x108): undefined reference to `_ZN9RainGauge6updateE2tmff'
/home/ls/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /tmp/arduino-sketch-F6DF9EA8EA0AD8A141217D84F3C1ACEB/sketch/BresserWeatherSensorMQTT.ino.cpp.o: in function `_Z18publishWeatherdatab':
/home/ls/Dokumente/Wetter5in1_BresserWeatherSensorReceiver/BresserWeatherSensorReceiver/examples/BresserWeatherSensorMQTT/BresserWeatherSensorMQTT.ino:422: undefined reference to `_Z18winddir_flt_to_strfPc'
/home/ls/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /home/ls/Dokumente/Wetter5in1_BresserWeatherSensorReceiver/BresserWeatherSensorReceiver/examples/BresserWeatherSensorMQTT/BresserWeatherSensorMQTT.ino:424: undefined reference to `_Z19windspeed_ms_to_bftf'
/home/ls/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /home/ls/Dokumente/Wetter5in1_BresserWeatherSensorReceiver/BresserWeatherSensorReceiver/examples/BresserWeatherSensorMQTT/BresserWeatherSensorMQTT.ino:428: undefined reference to `_Z19windspeed_ms_to_bftf'
/home/ls/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /home/ls/Dokumente/Wetter5in1_BresserWeatherSensorReceiver/BresserWeatherSensorReceiver/examples/BresserWeatherSensorMQTT/BresserWeatherSensorMQTT.ino:430: undefined reference to `_Z12calcdewpointff'
[... cut ...]
/home/ls/Dokumente/Wetter5in1_BresserWeatherSensorReceiver/BresserWeatherSensorReceiver/examples/BresserWeatherSensorMQTT/BresserWeatherSensorMQTT.ino:573: undefined reference to `_ZN13WeatherSensor10genMessageEijhh'
/home/ls/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /home/ls/Dokumente/Wetter5in1_BresserWeatherSensorReceiver/BresserWeatherSensorReceiver/examples/BresserWeatherSensorMQTT/BresserWeatherSensorMQTT.ino:579: undefined reference to `_ZN13WeatherSensor7getDataEjhhPFvvE'
collect2: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1 #

matthias-bs, 2022-10-19:
Hi,
it seems you have a mix of sketch and library from different releases.

I just tried it with an empty Arduino library directory and it worked as follows:

1. Check that the required Board Manager URL is in the Preferences (e.g. the one for ESP8266):

[image]

2. Select  **BresserWeatherSensorReceiver** in the Library Manager - the current version [as of now] is v0.3.3:
 
[image]

3. The Library Manager will ask if to install the **library dependencies** - select **"Install all"**:

[image]

4. From the **File** Menu, select **Examples->BresserWeatherSensorReceiver->BresserWeatherSensorMQTT**:

[image]

5. Select your board:

[image]

6. Modify `BresserWeatherSensorCfg.h` as required, typically:
```
// Number of sensors to be received
#define NUM_SENSORS 1
...
// Use pinning for LoRaWAN Node 
//#define LORAWAN_NODE
...
// Select type of receiver module
#define USE_CC1101
//#define USE_SX1276
```
Please check that your pinning matches the default:
```
#elif defined(ESP32)
    #define PIN_RECEIVER_CS   27
    
    // CC1101: GDO0 / RFM95W/SX127x: G0
    #define PIN_RECEIVER_IRQ  21 
    
    // CC1101: GDO2 / RFM95W/SX127x: G1
    #define PIN_RECEIVER_GPIO 33
    
    // RFM95W/SX127x - GPIOxx / CC1101 - RADIOLIB_NC
    #define PIN_RECEIVER_RST  32

#elif defined(ESP8266)
    #define PIN_RECEIVER_CS   15
    
    // CC1101: GDO0 / RFM95W/SX127x: G0
    #define PIN_RECEIVER_IRQ  4 
    
    // CC1101: GDO2 / RFM95W/SX127x: 
    #define PIN_RECEIVER_GPIO 5 
    
    // RFM95W/SX127x - GPIOxx / CC1101 - RADIOLIB_NC
    #define PIN_RECEIVER_RST  2
#endif
``` 
7. Edit the mandatory settings in `BresserWeatherSensorMQTT.ino`:
```
#define USE_SECUREWIFI          // use secure WIFI
//#define USE_WIFI              // use non-secure WIFI
...
// Map sensor IDs to Names - the number of entries must match NUM_SENSORS
SensorMap sensor_map[NUM_SENSORS] = {
    {0x39582376, "WeatherSensor"}
};
...
#ifndef SECRET
    const char ssid[] = "WiFiSSID";
    const char pass[] = "WiFiPassword";
    ...

    #define    MQTT_PORT     8883
    const char MQTT_HOST[] = "xxx.yyy.zzz.com";
    const char MQTT_USER[] = ""; // leave blank if no credentials used
    const char MQTT_PASS[] = ""; // leave blank if no credentials used
...
```

- Set the WiFi settings (ssid/pass) according to your Access Point
- Set `USE_SECUREWIFI`/`USE_WIFI`, `MQTT_PORT`, `MQTT_HOST`, `MQTT_USER` and `MQTT_PASS` according to your MQTT broker's IP address and configuration
   - for a non-secure setup: `USE_WIFI` and typically `MQTT_PORT` 1883
   - for a secure-setup: `USE_SECUREWIFI` and typically `MQTT_PORT` 8883 (and other security measures as needed)
   - set `MQTT_USER` and `MQTT_PASS` as required by the MQTT broker
 - SensorMap is just for your convenience - if you do not know your sensor's ID yet, leave it as it is. Just change the number of entries to match `NUM_SENSORS`.

You can of course copy the **secrets** to `secrets.h` and make your changes there instead of modifying the template in the sketch. In this case, do not forget to add the following in `secrets.h`:
```
#define SECRET
```

8. Now you want to save your changes. You will be asked to select a new directory, because the example resides in the Arduino/libraries folder which is treated as read-only:

[image]

9. Finally you should be able to compile your sketch without any errors or warnings:

[image]

Hope this helps!

Best regards
Matthias

gitpower2017, 2022-10-20:
Thank you for the detailed instructions.

I noticed a small error:

`#define NUM_SENSORS 1`

```
Causes an error:
BresserWeatherSensorMQTT:163:1: error: too many initializers for 'SensorMap [1]'
  163 | };
      | ^
exit status 1
too many initializers for 'SensorMap [1]''.

```

The original entry works
`#define NUM_SENSORS 2 `

Maybe there is a possibility to add this instruction to the wiki.
A version with LoRaWan and TTN might be interesting for some people.

Best regards
Lothar

gitpower2017, 2022-10-20:
Hello.

I have made file test today. It is not working properly yet.
It works:
- WLAN
- MQTT Connect
- MQTT Send "Status Onlnie".

After that reboot:

```
BresserWeatherSensorMQTT 20221006

Attempting to connect to SSID: schrottie....connected!
Checking wifi...
MQTT connecting... connected!
ESPWeather-39B225/status: online

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffffdc0 end: 3fffffc0 offset: 01a0
3fffff60:  3ffeed60 3ffe8a91 3ffeee24 40204a75  
3fffff70:  00000020 00000000 3ffeeb60 3ffef270  
3fffff80:  3ffe8c49 3ffeee08 3ffeeb60 402020b6  
3fffff90:  3fffdad0 3ffe8a91 3ffeec4c 402026e8  
3fffffa0:  feefeffe 00000000 3ffef25c 40209c70  
3fffffb0:  feefeffe feefeffe 3ffe86a0 40101355  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v00050cc0
~ld

BresserWeatherSensorMQTT 20221006`
...
```

I have a D1 mini from AZ Delivery.
As a board I use "D1 mini (clone)" (Wemo)
I have deactivated the LED in "BresserWeatherSensor MQTT"

My pinout. (Picture)
[image]

Which ESP8266 + CC1101 do you use and how does the pinout look like? 

Gruß
Lothar

matthias-bs, 2022-10-20:
> Thank you for the detailed instructions.
>
> I noticed a small error:
>
> `#define NUM_SENSORS 1`
>
> ```
> Causes an error:
> BresserWeatherSensorMQTT:163:1: error: too many initializers for 'SensorMap [1]' [...]

Therefore I have added the comment
```
// Map sensor IDs to Names - the number of entries must match NUM_SENSORS
```
in the tutorial and changed NUM_SENSORS and the number of entries to 1. Presumably most users might want to start with one sensor.

> Maybe there is a possibility to add this instruction to the wiki. A version with LoRaWan and TTN might be interesting for some people.

At least I have added a link to the tutorial to the README.md.

Regarding LoRaWAN and TTN (and Helium Network): see https://github.com/matthias-bs/BresserWeatherSensorTTN :wink:

matthias-bs, 2022-10-20:
> Hello.
>
> I have made file test today. It is not working properly yet. It works:
>
>     * WLAN
>
>     * MQTT Connect
>
>     * MQTT Send "Status Onlnie".
>
>
> After that reboot:
>
> ```
> BresserWeatherSensorMQTT [...]

Here you can find my pinout.

For debugging, I suggest you switch to [BresserWeatherSensorBasic](https://github.com/matthias-bs/BresserWeatherSensorReceiver/tree/main/examples/BresserWeatherSensorBasic) and enable `#define _DEBUG_MODE_`.
Then you should see what is going on in the serial console.

Maybe something is still wrong with the SW, maybe your CC1101 or ESP8266 is broken. (I spent a lot of time with a CC1101 which was dead-on-delivery. And just recently I pulled my hair out with a partially defect ESP32 module.)

gitpower2017, 2022-10-21:
Hello.

I tested three different ESP8266 boards. 
All three have the same error. After MQTT has transmitted the status online, there is a reboot.
I have the same error with the BresserWeatherSensorBasis.ino.
Another firmware e.g. Tasmota or ESPeasy works.

On "[uPesy ESP32 Wroom DevKi](https://www.upesy.com/products/upesy-esp32-wroom-devkit-board)t" the software runs without reboot.
However, I do not see any communication between ESP & CC1101.

I used the BresserWeatherSensorBasic & BresserWeatherSensorMQTT for testing.

gitpower2017, 2022-10-21:
Hello.

Hello.

I have successfully tested with an [older](https://github.com/matthias-bs/Bresser_Weather_Sensor_CC1101_MQTT) version the ESP8266 + cc1101.

Here the reception works and data is transmitted via MQTT.

The current version has a bug?

----- DEBUG Mode -----
MQTT connecting... connected!
[CC1101] Initializing ... 
success!
[CC1101] Setup complete - awaiting incoming messages...
[CC1101] Data:		 D4 E5 40 7F F7 1F F7 EF AB FE 67 BF E8 FF 1A BF 80 08 E0 08 10 54 01 98 40 17 00
[CC1101] R [0xD4] RSSI: -73.000000 LQI: 111
Id: [      BF] Battery: [OK ] Temp: [15.4C] Hum: [ 98%] Wind max: [0.8m/s] Wind avg: [0.8m/s] Wind dir: [315.0deg] Rain: [  74.0mm] 
{"sensor_id":       BF,"battery_ok":1,"temp_c":15.4,"humidity":98,"wind_gust":0.8,"wind_avg":0.8,"wind_gust_bft":0,"wind_avg_bft":0,"wind_dir":315.0,"rain":74.0}

matthias-bs, 2022-10-21:
Did you receive any debug messages with the current version?
I have an idea about the reason. I hope I can provide a solution next week.

gitpower2017, 2022-10-22:
With the current version I did not get a DEBUG message.

```
BresserWeatherSensorMQTT 20221006

Attempting to connect to SSID: schrottie....connected!
Checking wifi...
MQTT connecting... connected!
ESPWeather-39B225/status: online

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffffdc0 end: 3fffffc0 offset: 01a0
3fffff60:  3ffeed60 3ffe8a91 3ffeee24 40204a75  
3fffff70:  00000020 00000000 3ffeeb60 3ffef270  
3fffff80:  3ffe8c49 3ffeee08 3ffeeb60 402020b6  
3fffff90:  3fffdad0 3ffe8a91 3ffeec4c 402026e8  
3fffffa0:  feefeffe 00000000 3ffef25c 40209c70  
3fffffb0:  feefeffe feefeffe 3ffe86a0 40101355  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v00050cc0
~ld

BresserWeatherSensorMQTT 20221006`
...

```

franki29, 2022-10-22:
Hi @gitpower2017 , do you use a USB cable for the power? Sometimes cheap USB Cable does not give enough power. Just try a different one. I had same issues with other software, and I am using this software with same devices like you without any problem since weeks.

Br. Frank

gitpower2017, 2022-10-22:
> Hi @gitpower2017 , do you use a USB cable for the power? Sometimes cheap USB Cable does not give enough power. Just try a different one. I had same issues with other software, and I am using this software with same devices like you without any problem since weeks.

An older software version works without any problems.
And even the current BASIS version brings the same error. Therefore I suspect a bug in the software.
Other projects like ESPeasy work without problems.
Werd this but with better power supply test.

franki29, 2022-10-24:
Hi @gitpower2017 , I tested the new version again and got also a WDT Error.
At my side it was a wrong setting in WeatherSensorCfg.h file at line 8
// Use pinning for LoRaWAN Node 
Please change #define LORAWAN_NODE to
//#define LORAWAN_NODE and make the correct settings at the section 
// Select type of receiver module
#define USE_CC1101
and that the right settings under
 #elif defined(ESP8266)  depending on your wiring

What is happening is that the state for the radio Lib returns -2 if the CC1101 is not correctly working.
@matthias-bs maybe you can add something like "CC1101 not detected" in the code?

Hope this helps
Frank

matthias-bs, 2022-10-24:
I am currently tracking this down... I am quite sure that the way I try to use `WeatherSensorCfg.h` is wrong, which results in the library using a different version than the sketch.

matthias-bs, 2022-10-24:
Yes, I can confirm the expected root cause.

matthias-bs, 2022-10-24:
I hope I can provide a clean solution now.
1. I have modified the library and the example [BresserWeatherSensorMQTT.ino](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/examples/BresserWeatherSensorMQTT/BresserWeatherSensorMQTT.ino)

   This example now does not have a configuration file of its own any longer, but uses the one in `BresserWeatherSensorReceiver/src`. You can change it in your local `Arduino/libraries/BresserWeatherSensorReceiver/src` directory if you are always using the same setup (receiver type and wiring).

   I will change the other examples in the same way.

2. I have added a customized version in [BresserWeatherSensorMQTTCustom](https://github.com/matthias-bs/BresserWeatherSensorReceiver/tree/main/examples/BresserWeatherSensorMQTTCustom). Please check the [Readme.md](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/examples/BresserWeatherSensorMQTTCustom/Readme.md).

   Following this structure, you are able to use a customized configuration per sketch.

gitpower2017, 2022-10-25:
Hello.

The bugfix seems to work.
I was able to compile the version " BresserWeatherSensorMQTTCustom" and it runs.
However, I only see {"rssi:" -90.5} no other values.
Can I enable a DEBUG for the CC1101 output?

The version "BresserWeatherSensorBasic" & "BresserWeatherSensorMQTT" still brings the error "Soft WDT reset".

PS: I use "USE_WIFI" however the name is (I think) inappropriate. Is perhaps "MQTT_NOSECURE" better? 

Is there an example for the secure.h?

gitpower2017, 2022-10-25:
[WeatherSensorCfg.h.zip](https://github.com/matthias-bs/BresserWeatherSensorReceiver/files/9858610/WeatherSensorCfg.h.zip)

Enclosed is my file.

I am checking my pinout.

franki29, 2022-10-25:
Hi @gitpower2017 , you should choose the right device at:
// Select appropriate sensor message format(s)
#define BRESSER_5_IN_1
#define BRESSER_6_IN_1

If you have a older device use 
 #define BRESSER_5_IN_1
//#define BRESSER_6_IN_1

for a newer one use:
//#define BRESSER_5_IN_1
#define BRESSER_6_IN_1

to figure out what device you have look at " Readme"

Your pinout should be like https://github.com/Man-fred/culfw-esp8266/wiki/Hardware

gitpower2017, 2022-10-25:
'll build myself another CC1101 + D1mini.
Maybe something is broken in the old one.
After all, I already had values.
Currently I get no data with any software

matthias-bs, 2022-10-25:
Did you receive any debug messages with the current version?
I have

> Hello.
> 
> The bugfix seems to work. I was able to compile the version " BresserWeatherSensorMQTTCustom" and it runs. However, I only see {"rssi:" -90.5} no other values. Can I enable a DEBUG for the CC1101 output?
Yes, you can enable debug output. 

Presumably you have an RF signal quality issue. If you get the RSSI value, the communication between ESP and RF transceiver seems to be o.k.
> 
> The version "BresserWeatherSensorBasic" & "BresserWeatherSensorMQTT" still brings the error "Soft WDT reset".

These examples use the default WeatherSensorCfg.h which will not match your setup. You get a WDT reset, because RF transceiver initialization gets stuck in an endless loop, because the wrong transceiver response is expected. 
> 
> PS: I use "USE_WIFI" however the name is (I think) inappropriate. Is perhaps "MQTT_NOSECURE" better?
> 
> Is there an example for the secure.h?

Please see https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/12#issuecomment-1284508006 for the contents of `secure.h`

matthias-bs, 2022-10-25:
> Hi @gitpower2017 , you should choose the right device at: // Select appropriate sensor message format(s) #define BRESSER_5_IN_1 #define BRESSER_6_IN_1
>
> If you have a older device use #define [...]

You do not have to select the decoder any more. The SW tries the first one; if decoding fails it tries the other one.
The defines are only provided to disable a decoder if a user knows it won't be used (to save code space and run time).

gitpower2017, 2022-10-25:
I have now done some test.

With the version "Bresser_Weather_Sensor_CC1101_MQTT" I get value displayed. I have to reboot the ESP most 2-3x.
[Bresser_Weather_Sensor_CC1101_MQTT.zip](https://github.com/matthias-bs/BresserWeatherSensorReceiver/files/9860429/Bresser_Weather_Sensor_CC1101_MQTT.zip)

LOG:
```
.connected!
Checking wifi...
MQTT connecting... connected!
[CC1101] Initializing ...
success!
[CC1101] Setup complete - awaiting incoming messages...
[CC1101] Receive failed - failed, code 0
[CC1101] Data:           D4 E4 40 7F F7 2F FB EF EB FD A6 A9 DD FF 1B BF 80 08 D0 04 10 14 02 59 56 22 00
[CC1101] R [0xD4] RSSI: -85.000000 LQI: 115
Id: [      BF] Battery: [OK ] Temp: [21.4C] Hum: [ 59%] Wind max: [0.8m/s] Wind avg: [0.4m/s] Wind dir: [292.5deg] Rain: [  25.6mm]
{"sensor_id":       BF,"battery_ok":1,"temp_c":21.4,"humidity":59,"wind_gust":0.8,"wind_avg":0.4,"wind_gust_bft":0,"wind_avg_bft":0,"wind_dir":292.5,"rain":25.6}
[CC1101] Receive failed - failed, code 0
[CC1101] Data:           D4 E4 40 7F F7 2F 5B EF EB FD A6 A9 DD FF 9B BF 80 0C D0 04 10 14 02 51 52 22 00
[CC1101] R [0xD4] RSSI: -85.500000 LQI: 127
decodeBresser5In1Payload: Parity wrong at 0
[CC1101] R [0xD4] RSSI: -85.500000 LQI: 127
[CC1101] Receive failed - failed, code 0
[CC1101] Data:           D4 F3 40 7F F7 2F F9 EF EB FD A6 A9 DD FF 1C BD 80 08 90 06 10 14 02 59 56 22 20
[CC1101] R [0xD4] RSSI: -83.500000 LQI: 127
decodeBresser5In1Payload: Parity wrong at 0
[CC1101] R [0xD4] RSSI: -83.500000 LQI: 127
[CC1101] Receive failed - failed, code 0
[CC1101] Data:           D4 E4 40 7F F7 2F F7 EF EB FF A6 A9 DD FF 1B BF 80 08 D0 08 10 14 02 59 56 22 00
[CC1101] R [0xD4] RSSI: -86.500000 LQI: 107
decodeBresser5In1Payload: Parity wrong at 8
[CC1101] R [0xD4] RSSI: -86.500000 LQI: 107
[CC1101] Receive failed - failed, code 0
[CC1101] Receive failed - failed, code 0
[CC1101] Data:           D4 E3 40 7E F7 2F E7 AF E9 F5 E6 A9 DD FF 5C BF 80 08 D0 08 10 16 02 59 54 22 00
[CC1101] R [0xD4] RSSI: -85.000000 LQI: 127
decodeBresser5In1Payload: Parity wrong at 0
[CC1101] R [0xD4] RSSI: -85.000000 LQI: 127
{"sensor_id":       BF,"battery_ok":1,"temp_c":21.4,"humidity":59,"wind_gust":0.8,"wind_avg":0.4,"wind_gust_bft":0,"wind_avg_bft":0,"wind_dir":292.5,"rain":25.6}
{"rssi":-85,"lqi":127}
[CC1101] Receive failed - failed, code 0
[CC1101] Receive failed - failed, code 0
[CC1101] Receive failed - failed, code 0
[CC1101] Receive failed - failed, code 0
[CC1101] Receive failed - failed, code 0
[CC1101] Data:           D4 E4 40 7F F7 2F F7 EF E9 FD A7 A9 DD FF 1B BF 80 08 D0 08 10 16 02 58 56 22 00
[CC1101] R [0xD4] RSSI: -82.500000 LQI: 127
Id: [      BF] Battery: [OK ] Temp: [21.6C] Hum: [ 58%] Wind max: [0.8m/s] Wind avg: [0.8m/s] Wind dir: [292.5deg] Rain: [  25.6mm]
{"sensor_id":       BF,"battery_ok":1,"temp_c":21.6,"humidity":58,"wind_gust":0.8,"wind_avg":0.8,"wind_gust_bft":0,"wind_avg_bft":0,"wind_dir":292.5,"rain":25.6}
[CC1101] Receive failed - failed, code 0
[CC1101] Receive failed - failed, code 0

```

I have attached my sketch including the console output.

With the same hardware and the "BresserWeatherSensorMQTTCustom" version I get only the rssi values displayed.
[BresserWeatherSensorMQTTCustom.zip](https://github.com/matthias-bs/BresserWeatherSensorReceiver/files/9860430
/BresserWeatherSensorMQTTCustom.zip)

LOG:
```
connected!
Checking wifi...
MQTT connecting... connected!
ESPWeatherLSD-0878CA/status: online
ESPWeatherLSD-0878CA/status: online
ESPWeatherLSD-0878CA/radio: {"rssi":-103}
ESPWeatherLSD-0878CA/status: online
ESPWeatherLSD-0878CA/radio: {"rssi":-99}
ESPWeatherLSD-0878CA/status: online
ESPWeatherLSD-0878CA/radio: {"rssi":-101.5}
ESPWeatherLSD-0878CA/status: online
ESPWeatherLSD-0878CA/radio: {"rssi":-102}
ESPWeatherLSD-0878CA/status: online
ESPWeatherLSD-0878CA/radio: {"rssi":-97}
ESPWeatherLSD-0878CA/status: online
ESPWeatherLSD-0878CA/radio: {"rssi":-102.5}
ESPWeatherLSD-0878CA/status: online
ESPWeatherLSD-0878CA/radio: {"rssi":-101.5}
ESPWeatherLSD-0878CA/status: online
ESPWeatherLSD-0878CA/radio: {"rssi":-102}
ESPWeatherLSD-0878CA/status: online
ESPWeatherLSD-0878CA/radio: {"rssi":-97.5}
```

I have attached my sketch including the console output.

matthias-bs, 2022-10-25:
Oh, I put the `#define _DEBUG_MODE_` in the wrong place in WeatherSensorCfg.h! Can you please move it up before the line `#define DEBUG_PORT Serial`?

Can you please increase `RX_TIMEOUT` (default is 90s)? In the logfile above (https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/12#issuecomment-1290532723), there are many receive/parity errors. It seems your reception is not good. With the rule that getData expects to receive a message before a timeout of 90s occurs, you probably never receive a message in time. (Bresser_Weather_Sensor_CC1101_MQTT works in a different way - its getWeatherdata() function is non-blocking and there are 100 retries until the main loop is resumed.)

From your sensor ID (length: 1 byte), I assume that you have a sensor using the 5-in-1 protocol. That's good, because all data is contained in one RF message as opposed to the 6-in-1 protocol where the data is contained in two messages. Just to be sure, you could change the line no. 624 in WeatherDataMQTTCustom.ino
from
`decode_ok = weatherSensor.getData(RX_TIMEOUT, DATA_COMPLETE, 0, &clientLoopWrapper);`
to
`decode_ok = weatherSensor.getData(RX_TIMEOUT, 0, 0, &clientLoopWrapper);`?

The default line tries to receive all values even if contained in two separate messages until a timeout occurs (`RX_TIMEOUT`). The alternative proposal is satisfied if only one message can be decoded successfully.

matthias-bs, 2022-10-25:
Here are some hints about improving the radio signal range:
https://github.com/matthias-bs/Bresser_Weather_Sensor_CC1101_MQTT/issues/3

franki29, 2022-10-25:
And now come the strange thing : I also have a bresser 5in1 Sensor (was a cheap offer at Bresser so I could not resits) and this sensor **is also not recognize.!**

franki29, 2022-10-25:
It look like that the switch between the 6in1 and 5in1 does not work.
I change in WeatherSensor.ccp the line 240 to 245 to:
   #ifdef BRESSER_5_IN_1
               // if (decode_res == DECODE_PAR_ERR ||
                  //  decode_res == DECODE_CHK_ERR || 
                  //  decode_res == DECODE_DIG_ERR) {
                    decode_res = decodeBresser5In1Payload(&recvData[1], sizeof(recvData) - 1);
               // }
           #endif

And than I receive the 5in1 data and 6in1 
ESPWeatherLSD-BABA07/data/14200203: {"id":14200203,"ch":0,"battery_ok":1,"wind_gust":0.0,"wind_avg":0.0,"wind_dir":112.0,"rain":709.6,"rain_d":0.0,"rain_w":0.0,"rain_m":0.0}
ESPWeatherLSD-BABA07/extra: {"wind_dir_txt":"ESE","wind_gust_bft":0,"wind_avg_bft":0}
ESPWeatherLSD-BABA07/data/      63: {"id":00000063,"ch":0,"battery_ok":1,"temp_c":21.8,"humidity":62,"wind_gust":0.0,"wind_avg":0.0,"wind_dir":202.5,"rain":0.0,"rain_d":0.0,"rain_w":0.0,"rain_m":0.0}
ESPWeatherLSD-BABA07/extra: {"wind_dir_txt":"SSW","wind_gust_bft":0,"wind_avg_bft":0,"dewpoint_c":14.2,"perceived_temp_c":24.3}

matthias-bs, 2022-10-25:
I would prefer to move some items to separate issues...

I have successfully used the same code with two different kinds of sensors - one is using the 5-in-1 protocol, the other one the 6-in-1 protocol. But - after reviewing it, I found the following issue:

We have the following initialization:
https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/33dda04392f0978cf06207ef004bf53f0c6858d5/src/WeatherSensor.cpp#L201

Then we have
```
            #ifdef BRESSER_6_IN_1
                decode_res = decodeBresser6In1Payload(&recvData[1], sizeof(recvData) - 1);
            #endif
            #ifdef BRESSER_5_IN_1
                if (decode_res == DECODE_PAR_ERR ||
                    decode_res == DECODE_CHK_ERR || 
                    decode_res == DECODE_DIG_ERR) {
                    decode_res = decodeBresser5In1Payload(&recvData[1], sizeof(recvData) - 1);
                }
            #endif
```
If we did not define BRESSER_6_IN_1 (i.e. we changed the default config), then `decode_res` will still equal `DECODE_INVALID` and `decodeBresser5In1Payload()` will not be called, even if `BRESSER_5_IN_1` is defined. This is a bug.

In the default configuration, both `BRESSER_6_IN_1` and `BRESSER_5_IN_1` are defined and we do not have any problem.
 
decodeBresser6In1Payload() will return one of the following:
   * DECODE_DIG_ERR - digest check error
   * DECODE_CHK_ERR - checksum error
   * DECODE_SKIP - decoding was o.k., but sensor has been skipped in `findSlot()` due to the ID filter rules (`sensor_ids_inc` / `sensor_ids_exc`)
   * DECODE_FULL - decoding was o.k., but sensor data could not be stored because the array `sensor[]` was already full (in `findSlot()`)
   * DECODE_OK - decoding o.k.
 
Actually DECODE_PAR_ERR is not used in `decodeBresser6In1Payload()`, but I'd rather like to leave it as it is.

matthias-bs, 2022-10-25:
> > Here are some hints about improving the radio signal range: [matthias-bs/Bresser_Weather_Sensor_CC1101_MQTT#3](https://github.com/matthias-bs/Bresser_Weather_Sensor_CC1101_MQTT/issues/3)
> 
> In another cc1101 bresser project, you can set a FrequencyOffset. The dispersion from the receiver is high.

Which project are you referring to? How is the offset determined?

May I ask you to open another issue as a request for a new feature?

matthias-bs, 2022-10-25:
> It look like that the switch between the 6in1 and 5in1 does not work. I change in WeatherSensor.ccp the line 240 to 245 to: #ifdef BRESSER_5_IN_1 // if (decode_res == DECODE_PAR_ERR || // decode_res [...]

This leads to another situation - did you change NUM_SENSORS to 2?
If not, the following will happen (assuming that no decoding error occurs):
   * If you receive a 6-in-1 message first, it will be stored.
      * if you receive a 5-in-1 message next, you will get DECODE_FULL
      * if you receive a 6-in-1 message next with the values missing in the first one, you will get DECODE_OK (sensor[slot].complete = true)
      * if you receive a 6-in-1 message next with the same values as in the first one, you will get DECODE_OK (sensor[slot] is updated, sensor[slot].complete = FALSE)
      * in all cases above, the 5-in-1 message will not be handled
   * if you receive the 5-in-1 message first, it will be stored and you are done (DECODE_OK)

gitpower2017, 2022-10-26:
I am now flooded with a bit much info.
What do I need to do for the next test?

matthias-bs, 2022-10-26:
Sorry about this. Some comments have been more directed at @franki29. :-) 
Please make another try with BresserWeatherSensorMQTTCustom. 
If you just enable debug messages and increase the RX_TIMEOUT as described in https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/12#issuecomment-1290777297 you should be able to get valid MQTT messages or at least we should get debug info. Please leave both BRESSER_5_IN_1 and BRESSER_6_IN_1 defined in your own config. 
Thanks for your patience!

franki29, 2022-10-26:
> > It look like that the switch between the 6in1 and 5in1 does not work. I change in WeatherSensor.ccp the line 240 to 245 to: #ifdef BRESSER_5_IN_1 // if (decode_res == DECODE_PAR_ERR || // decode_res == DECODE_CHK_ERR || // decode_res == DECODE_DIG_ERR) { decode_res = decodeBresser5In1Payload(&recvData[1], sizeof(recvData) - 1); // } #endif
> > And than I receive the 5in1 data and 6in1 ESPWeatherLSD-BABA07/data/14200203: {"id":14200203,"ch":0,"battery_ok":1,"wind_gust":0.0,"wind_avg":0.0,"wind_dir":112.0,"rain":709.6,"rain_d":0.0,"rain_w":0.0,"rain_m":0.0} ESPWeatherLSD-BABA07/extra: {"wind_dir_txt":"ESE","wind_gust_bft":0,"wind_avg_bft":0} ESPWeatherLSD-BABA07/data/ 63: {"id":00000063,"ch":0,"battery_ok":1,"temp_c":21.8,"humidity":62,"wind_gust":0.0,"wind_avg":0.0,"wind_dir":202.5,"rain":0.0,"rain_d":0.0,"rain_w":0.0,"rain_m":0.0} ESPWeatherLSD-BABA07/extra: {"wind_dir_txt":"SSW","wind_gust_bft":0,"wind_avg_bft":0,"dewpoint_c":14.2,"perceived_temp_c":24.3}
> 
> This leads to another situation - did you change NUM_SENSORS to 2? If not, the following will happen (assuming that no decoding error occurs):
> 
>     * If you receive a 6-in-1 message first, it will be stored (incomplete data set stored, waiting for the remaining values)
>       
>       * if you receive a 5-in-1 message next, you will get DECODE_FULL
>       * if you receive a 6-in-1 message next with the values missing in the first one, you will get DECODE_OK (sensor[slot].complete = true)
>       * if you receive a 6-in-1 message next with the same values as in the first one, you will get DECODE_OK (sensor[slot] is updated, sensor[slot].complete = FALSE)
>       * in all cases above, the 5-in-1 message will not be handled
> 
>     * if you receive the 5-in-1 message first, it will be stored and you are done (DECODE_OK)
> 
> 
> Ah - and in case of the soil moisture sensor, also the 6-in-1 protocol is used, but all data is contained in a single message.

Hi @matthias-bs , yes I changed it to NUM_SENSORS to 2. But the code that gitpower2017 was using did not work without changing the line 240 to 245.

gitpower2017, 2022-10-26:
I fetched the current GIT, increased the value for RX timeout to 100 & 120 sec. Nothing has changed, I only see the RSSI value.

If I enable #define _DEBUG_MODE_ in the src/WeatherSensorCfg.h it does not compile.

```
In file included from /home/ls/Dokumente/BresserWeatherSensorReceiver/examples/BresserWeatherSensorMQTTCustom/src/WeatherSensor.cpp:59:
/home/ls/Dokumente/BresserWeatherSensorReceiver/examples/BresserWeatherSensorMQTTCustom/src/WeatherSensor.cpp: In member function 'DecodeStatus WeatherSensor::decodeBresser6In1Payload(uint8_t*, uint8_t)':
/home/ls/Dokumente/BresserWeatherSensorReceiver/examples/BresserWeatherSensorMQTTCustom/src/WeatherSensor.cpp:691:21: error: 'chksum' was not declared in this scope
  691 |         DEBUG_PRINT(chksum, HEX);
      |                     ^~~~~~
/home/ls/Dokumente/BresserWeatherSensorReceiver/examples/BresserWeatherSensorMQTTCustom/src/WeatherSensorCfg.h:95:49: note: in definition of macro 'DEBUG_PRINT'
   95 |     #define DEBUG_PRINT(...) { DEBUG_PORT.print(__VA_ARGS__); }
      | ^~~~~~~~~~~
exit status 1
Fehler beim Kompilieren für das Board NodeMCU 1.0 (ESP-12E Module).

```

So that one can use the secret.h, still a #define SECRET is missing

I guess the code for the secret.h should look like this:

```
#define SECRET

#if defined(SECRET)
 #include "secret.h
#endif
```

gitpower2017, 2022-10-26:
> Hi @gitpower2017 for testing just do following:
> 
>     1. Do not compile with Debug as it will fail (I already open a ticket for that)
> 
>     2. Change in WeatherSensor.ccp the line 240 to 245 to:
>        #ifdef BRESSER_5_IN_1
>        // if (decode_res == DECODE_PAR_ERR ||
>        // decode_res == DECODE_CHK_ERR ||
>        // decode_res == DECODE_DIG_ERR) {
>        decode_res = decodeBresser5In1Payload(&recvData[1], sizeof(recvData) - 1);
>        // }
>        #endif
> 
> 
> This is of course not a final solution but it should work for you

The same.

I will rebuild the CC1101+esp8266 this weekend.

matthias-bs, 2022-10-26:
> I fetched the current GIT, increased the value for RX timeout to 100 & 120 sec. Nothing has changed, I only see the RSSI value.
>
> If I enable #define _DEBUG_MODE_ in the src/WeatherSensorCfg.h it [...]

I have added comments about secrets.h in the sketches and renamed the define to be consistent with the file name, i.e. `SECRETS` and `secrets.h`.

In the sketch:
```
#include "secrets.h"

#ifndef SECRETS
    // Optionally copy everything between BEGIN secrets / END secrets to secrets.h
    // Otherwise, leave secrets.h as an empty file and edit the contents below.

    // BEGIN secrets
    #define SECRETS
    [...]
    // END secrets
#endif
```

gitpower2017, 2022-11-02:
I have performed many tests.
For calibration I used SignalDuino.
Result:
1) Correct frequency is 868.4
2) SignalDuino receives a correct record every 12 seconds (always)

I have entered the frequency 868.4 in the WeatherSensor.cpp (line 82).

Your implementation receives mostly one correct packet, but also many wrong ones.
Occasionally it does not find a sync at all and I have to reboot the ESP.
The protocol switch between 5in1 and 6in1 does not seem to work properly.

In conclusion, YES it does work occasionally.
Most of the time it does not.

Maybe you can create a 5in1 and a 6in1 version. Very few probably have both.

matthias-bs, 2022-11-02:
Regarding your Postscript: Does it fix your reception problem and/or the protocol switching issue?
Maybe you have an USB power bank as a supply for testing?

I have used the library with both protocols without changes. I'm also planning to use it with a 5-in-1-protocol weather station and a 6-in-1-protocol soil moisture sensor, therefore two versions are not an option for me.

If you still think the protocol switching does not work, just disable the unwanted protocol in the header file.

gitpower2017, 2022-11-03:
My problem is the USB power supply.
Changed that and the whole thing works very well.

Thanks for the great project.
I am already looking forward to a regular operation.
Unfortunately, I understand the source code only insufficiently and can therefore little optimize or adjust the code.

I think the ticket can be closed.

Greetings
Lothar

closed 2022-11-03 by matthias-bs