matthias-bs/BresserWeatherSensorReceiver #14: Calibrate from CC1101?

gitpower2017, 2022-10-29:
I have found two pages on how to calibrate a CC1101 module.
It here on an approximation method to determine the optimal frequency setting, in relation to the transmitter.
Perhaps this could be adapted to the ESP?

[FreqTest.ino](https://github.com/pa-pa/AskSinPP/blob/master/examples/FreqTest/FreqTest.ino)
[ermittlung-der-cc1101-frequenz](https://asksinpp.de/Grundlagen/FAQ/Fehlerhafte_CC1101.html#ermittlung-der-cc1101-frequenz)

matthias-bs, 2022-10-29:
For sure this can be ported to ESP32/Esp8266. I would rather use an RF transceiver with better reliability on frequency/frequency stability. Like the RFM95W/SX1276...

gitpower2017, 2022-10-30:
It is important that the CC1101 can be calibrated by offset.
The CC1101 is very inexpensive and is therefore very often used.
I was about a suitable function or special firmware
be pleased

Piratejim13, 2024-04-26:
Good morning
I am looking for the wiring to use BresserWeatherSensorReceiver it with an ESP32 and a CC1101.
Here are the pins of my CC1101
1 GND
2 3.3V
3 GDO0
4 CKN
5 SCK
6 MOSI
7 MISO/GDO1
8 GDO2

can you help me connect my ESP32 and my CC1101
Thank you

matthias-bs, 2024-04-26:
@Piratejim13 Please open a new issue if the topic is not related to an existing one.

matthias-bs, 2024-12-09:
Optional frequency offset added in https://github.com/matthias-bs/BresserWeatherSensorReceiver/pull/207.