matthias-bs/BresserWeatherSensorReceiver #15: The JSON string is not formed correctly (ID)

gitpower2017, 2022-11-02:
**Describe the bug**
there is an error in the generation of the JSON. 
The ID can also contain letters (z.B. BF) and must be in "".

`Client [ESPWeatherLSD-39B225] Cannot parse "ESPWeatherLSD-39B225/data/LSDWetterSensor": {"id":000000BF,"ch":0,"battery_ok":1,"temp_c":16.9,"humidity":59,"wind_gust":0.0,"wind_avg":0.4,"wind_dir":202.5,"rain":28.8,"rain_d":0.0,"rain_w":0.0,"rain_m":0.0}
`

matthias-bs, 2022-11-02:
Fixed in https://github.com/matthias-bs/BresserWeatherSensorReceiver/commit/b880b30ded0a883a4586234dc6080ff23caf6dca and https://github.com/matthias-bs/BresserWeatherSensorReceiver/commit/6191c8ec74d3b4c2c7d80f2e61b48953bf70de9c.

matthias-bs, 2022-11-02:
N.B. All three sensors I used for testing have only the digits 0...9 in there IDs - that's why I did not notice this problem.

gitpower2017, 2022-11-08:
Yes.
Kann geschlossen werden.

Danke

closed 2022-11-09 by matthias-bs