matthias-bs/BresserWeatherSensorReceiver #16: LILYGO® TTGO LORA32 868/915Mhz ESP32 LoRa OLED - pinouts

pmknowles, 2022-11-02:
The pinout for this device is different as the SX1278 is hardwired.
GPIO5  -- SX1278's SCK
GPIO19 -- SX1278's MISO
GPIO27 -- SX1278's MOSI
GPIO18 -- SX1278's CS
GPIO14 -- SX1278's RESET
GPIO26 -- SX1278's IRQ(Interrupt Request)

There doesn't seem to be a place to edit the default pins. The board is available in the Boards Manager but I'm not sure if that filters through to your code.

The MQTT output goes online then 25 seconds later goes to dead. I can't get the test versions to compile so not sure whether it an issue from the weather station to the radio or radio to ESP32. Any ideas?

matthias-bs, 2022-11-02:
You might want to check if https://github.com/matthias-bs/BresserWeatherSensorReceiver/tree/main/examples/BresserWeatherSensorMQTTCustom/src works for you. In this example, the library part is placed in the sub-directory /src of the sketch. In there is the file [WeatherSensorCfg.h](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/examples/BresserWeatherSensorMQTTCustom/src/WeatherSensorCfg.h) where you can define the pins as needed.

I also recommend to enable debug output by uncommenting
```
#define _DEBUG_MODE_ 
```
at the beginning.

Please try the latest version of the repository instead of any of the releases.

Let me now if the compilation still fails and provide a log file. I just tried to compile for the targets TTGO LoRa32-OLED V1 and TTGO LoRa32-OLED V2.1.6 - both  worked.

matthias-bs, 2022-11-02:
What is the difference between the SX1276 and the SX1278? 
My library uses SX1276 from https://github.com/jgromes/RadioLib:
https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/6191c8ec74d3b4c2c7d80f2e61b48953bf70de9c/src/WeatherSensor.cpp#L66

But RadioLib also provides SX1278 support.

matthias-bs, 2022-11-02:
As far as I can see from https://www.semtech.com/products/wireless-rf/lora-connect, SX1278 is for 137MHz to 525MHz, while SX1276 for 137MHz to 1020MHz.
Which frequency does your weather sensor use?

pmknowles, 2022-11-02:
It’s 868MHz there are quite a few typos on Lilygo’s website

Regards,

Phil

pmknowles, 2022-11-02:
> Relation between SX1278 and SX1276 module classes in RadioLib:
> https://jgromes.github.io/RadioLib/class_physical_layer.html<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fjgromes.github.io%2FRadioLib%2Fclass_physical_layer.html&data=05%7C01%7C%7C2b88a31e99024b517a5d08dabcff6f11%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638030102499402460%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=uWLXMkm8E0w5e0%2FLgVM1ZtCcvXYu2E0CcF8q5iJlYqM%3D&reserved=0>
>
> So it might work without any modifications...

The Arduino Board Manager for V2 arduino-esp32/pins_arduino.h at master · espressif/arduino-esp32 (github.com)<https://github.com/espressif/arduino-esp32/blob/master/variants/ttgo-lora32-v2/pins_arduino.h>
Has this which doesn’t help
// SPI LoRa Radio
#define LORA_SCK    5   // GPIO5 - SX1276 SCK
#define LORA_MISO   19  // GPIO19 - SX1276 MISO
#define LORA_MOSI   27  // GPIO27 - SX1276 MOSI
#define LORA_CS     18  // GPIO18 - SX1276 CS
#define LORA_RST    12  // GPIO14 - SX1276 RST
#define LORA_IRQ    26  // GPIO26 - SX1276 IRQ (interrupt request)
And V2.1
arduino-esp32/pins_arduino.h at master · espressif/arduino-esp32 (github.com)<https://github.com/espressif/arduino-esp32/blob/master/variants/ttgo-lora32-v21new/pins_arduino.h>
// SPI LoRa Radio
#define LORA_SCK    5   // GPIO5 - SX1276 SCK
#define LORA_MISO   19  // GPIO19 - SX1276 MISO
#define LORA_MOSI   27  // GPIO27 - SX1276 MOSI
#define LORA_CS     18  // GPIO18 - SX1276 CS
#define LORA_RST    23  // GPIO23 - SX1276 RST
#define LORA_IRQ    26  // GPIO26 - SX1276 IRQ (interrupt request)
#define LORA_D1     33  // GPIO33 - SX1276 IO1 (for LMIC Arduino library)
#define LORA_D2     32 // GPIO32 - SX1276 IO2
V1
// SPI LoRa Radio
#define LORA_SCK 5        // GPIO5 - SX1276 SCK
#define LORA_MISO 19     // GPIO19 - SX1276 MISO
#define LORA_MOSI 27    // GPIO27 - SX1276 MOSI
#define LORA_CS 18     // GPIO18 - SX1276 CS
#define LORA_RST 14   // GPIO14 - SX1276 RST
#define LORA_IRQ 26  // GPIO26 - SX1276 IRQ (interrupt request)
Mine appears to be a V1 and the G1 isn’t connected to the ESP32

Regards,

Phil

matthias-bs, 2022-11-02:
What is G1? And what is actually the problem?
I think you could create an SPI object with your SPI pins and pass this together with the other pins to the Module constructor.

Best regards,
Matthias

pmknowles, 2022-11-02:
I don’t know but it’s in WeatherSensorCfg.h

    // CC1101: GDO2 / RFM95W/SX127x: G1
    #define PIN_RECEIVER_GPIO 33

I can’t get the custom version to compile loads of compile errors like this
undefined reference to `RainGauge::update(tm, float, float)'

With the MQTT version with debug turned on this is all I get

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:1

load:0x3fff0030,len:1184

load:0x40078000,len:13160

load:0x40080400,len:3036

entry 0x400805e4

BresserWeatherSensorMQTT 20221006

Attempting to connect to SSID: pkm-24m.connected!

Checking wifi...

MQTT connecting... connected!

ESPWeather/status: online

And MQTT explorer shows
ESPWeather/status “online”
And about 20 seconds later
ESPWeather/status “dead”

Giving up for today...

Regards,

Phil

franki29, 2022-11-03:
Hi @pmknowles ,
try in WeatherSensorCfg.h with:

// Use pinning for LoRaWAN Node 
**//**#define LORAWAN_NODE

#elif defined(ESP32)
    #define PIN_RECEIVER_CS   18
    
    // CC1101: GDO0 / RFM95W/SX127x: G0
    #define PIN_RECEIVER_IRQ  26 
    
    // CC1101: GDO2 / RFM95W/SX127x: G1
    #define PIN_RECEIVER_GPIO 33
    
    // RFM95W/SX127x - GPIOxx / CC1101 - RADIOLIB_NC
    #define PIN_RECEIVER_RST  23

At my side it is working with that settings

pmknowles, 2022-11-03:
Afraid that didn’t work. I’m pretty sure the ESP32 isn’t talking to the SX1276. The debug output stops after the first MQTT message saying online. No more output. The MQTT message changes to dead 18 seconds later.

I’ve deleted the Library and re-installed. I now can compile the Basic version but if I try to change anything (like turn on debugging or set fixedpoint rather than having both fixed and floating in refuses to compile.

I’m going to try to implement a Lora transmit and receive to an RFM95W that I have to confirm I can communicate to the SX1276. That will need to wait until the weekend.

franki29, 2022-11-03:
Hi @pmknowles , can you share your WeatherSensorCfg.h that you are using?

pmknowles, 2022-11-03:
This is the version for my ESP dev board and the RFM95W

#if !defined(WEATHER_SENSOR_CFG_H)
#define WEATHER_SENSOR_CFG_H

// Number of sensors to be received
#define NUM_SENSORS 2

// Use pinning for LoRaWAN Node
//#define LORAWAN_NODE
#define ESP32

// List of sensor IDs to be excluded - can be empty
uint32_t const sensor_ids_exc[] = {};
//uint32_t const sensor_ids_exc[] = { 0x39582376 };

// List of sensor IDs to be included - if empty, handle all available sensors
uint32_t const sensor_ids_inc[] = {};
//uint32_t const sensor_ids_inc[] = { 0x83750871 };

// Select appropriate sensor message format(s)
#define BRESSER_5_IN_1
#define BRESSER_6_IN_1

// Select type of receiver module
//#define USE_CC1101
#define USE_SX1276

// Disable data type which will not be used to save RAM
#define WIND_DATA_FLOATINGPOINT
#define WIND_DATA_FIXEDPOINT

//#define _DEBUG_MODE_

// Arduino default SPI pins
//
// Board   SCK   MOSI  MISO
// ESP8266 D5    D7    D6
// ESP32   D18   D23   D19
#if defined(LORAWAN_NODE)
    #define PIN_RECEIVER_CS   14

    // CC1101: GDO0 / RFM95W/SX127x: G0
    #define PIN_RECEIVER_IRQ  4

    // CC1101: GDO2 / RFM95W/SX127x: G1
    #define PIN_RECEIVER_GPIO 16

    // RFM95W/SX127x - GPIOxx / CC1101 - RADIOLIB_NC
    #define PIN_RECEIVER_RST  12
#elif defined(ESP32)
    #define PIN_RECEIVER_CS   27

    // CC1101: GDO0 / RFM95W/SX127x: G0
    #define PIN_RECEIVER_IRQ  21

    // CC1101: GDO2 / RFM95W/SX127x: G1
    #define PIN_RECEIVER_GPIO 33

    // RFM95W/SX127x - GPIOxx / CC1101 - RADIOLIB_NC
    #define PIN_RECEIVER_RST  32

#elif defined(ESP8266)
    #define PIN_RECEIVER_CS   15

    // CC1101: GDO0 / RFM95W/SX127x: G0
    #define PIN_RECEIVER_IRQ  4

    // CC1101: GDO2 / RFM95W/SX127x:
    #define PIN_RECEIVER_GPIO 5

    // RFM95W/SX127x - GPIOxx / CC1101 - RADIOLIB_NC
    #define PIN_RECEIVER_RST  2
#endif

#endif

If I enable debug mode I still only get

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:1

load:0x3fff0030,len:1344

load:0x40078000,len:13864

load:0x40080400,len:3608

entry 0x400805f0

BresserWeatherSensorMQTT 20221006

Attempting to connect to SSID: pkm-24m.connected!

Checking wifi...

MQTT connecting... connected!

ESPWeather-A1B1FC/status: online

Regards,

Phil

matthias-bs, 2022-11-03:
Looks like you are using one of the old releases (instead of the HEAD of the repository).
I've created release v0.4.0 this morning. With this release it should be possible to compile all variants, enabling debug output should really give you debug output and the code should use the pins you actually define in your header file. 
Please give it another try!

franki29, 2022-11-03:
Hi @pmknowles , so we are now not talking about the TTGO anymore, right?
And what example did you use now for testing?

pmknowles, 2022-11-03:
I used 3.3 version in the Arduino Library as at least with the RFM95W I can use the pins stated in the code

pmknowles, 2022-11-04:
Success!
{"id":"000000AF","ch":0,"battery_ok":1,"temp_c":4.1,"humidity":94,"wind_gust":3.0,"wind_avg":1.3,"wind_dir":292.5,"rain":88.8,"rain_d":0.0,"rain_w":0.0,"rain_m":0.0}

When I ran with serial monitor I got the message that the SX1276 was initialised which I wasn’t before.

This is the pinning for the TTGO LoRa OLED v1.0

#define SCK 5

#define MISO 19

#define MOSI 27

#define SS 18

#define RST 14

#define DIO0 26

Thank you for your help

Regards,

Phil

matthias-bs, 2022-11-13:
Hi Phil,

I just added a pin definition for TTGO_LORA32 in [WeatherSensorCfg.h](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/src/WeatherSensorCfg.h) which uses the defines from the board specific header files. Can we close this issue then?

Best regards
Matthias

pmknowles, 2023-01-03:
I have had another go at this and I think that the issue is in WeatherSensor.cpp with these lines

    #ifdef USE_CC1101
        int state = radio.begin(868.3, 8.21, 57.136417, 270, 10, 32);
    #else
        int state = radio.beginFSK(868.3, 8.21, 57.136417, 250, 10, 32);
    #endif

I think there should be brackets round USE_CC1101

pmknowles, 2023-01-03:
It would also explain why I couldn't get any output from turning Debug on because all the #ifdef statements for _DEBUG_MODE_ are missing brackets too.

matthias-bs, 2023-01-03:
I don't think so.
First I had a look at _Bjarne Stroustrup: The C++ Programming Language_. According to it (chapter R16 in my German version)

`#if defined IDENTIFIER`
and
`#if defined (IDENTIFIER)`
and
`#ifdef IDENTIFIER`
are equivalent.

Next I added the following in `WeatherSensor.cpp`:
```
    #ifdef USE_CC1101
        #warning "Building for CC1101"
        int state = radio.begin(868.3, 8.21, 57.136417, 270, 10, 32);
    #else
        #warning "Building for SX1276"
        int state = radio.beginFSK(868.3, 8.21, 57.136417, 250, 10, 32);
    #endif
```
The warning is printed according to the selected define in `WeatherSensorCfg.h`.

I recommend to add some kind of warning in your file to see if the sketch is compiled with the copy of the library files you are expecting.

matthias-bs, 2023-01-03:
> It would also explain why I couldn't get any output from turning Debug on because all the #ifdef statements for _DEBUG_MODE_ are missing brackets too.

Here the order of the define and the macros using it was wrong in a previous version.

pmknowles, 2023-01-03:
I installed the library as a zip to a fresh installation of Arduino IDE. It compiled but the serial monitor produced no output after the MQTT connection and didn’t start the radio chip. I inserted serial.print statements and it didn’t reach the radio.begin statement for the SX1276. I only got it to work by deleting the ifdef statement completely. I’m not sure why it is ignoring the else so not starting the SX1276.

matthias-bs, 2023-01-04:
You are building the sources in https://github.com/matthias-bs/BresserWeatherSensorReceiver/tree/main/examples/BresserWeatherSensorMQTTCustom with all changes made in https://github.com/matthias-bs/BresserWeatherSensorReceiver/tree/main/examples/BresserWeatherSensorMQTTCustom/src, right?

Lordcybertracker, 2023-01-22:
I've now got a V1 to run.
Since this was without a display, it was just lying around anyway.

But I took it as Lorawan_Node:

// Arduino default SPI pins
//
// Board SCK MOSI MISO
// ESP8266 D5 D7 D6
// ESP32 D18 D23 D19
#if defined(LORAWAN_NODE)
    #define PIN_RECEIVER_CS 18
    
    // CC1101: GDO0 / RFM95W/SX127x: G0
    #define PIN_RECEIVER_IRQ 26
    
    // CC1101: GDO2 / RFM95W/SX127x: G1
    #define PIN_RECEIVER_GPIO 33
    
    // RFM95W/SX127x - GPIOxx / CC1101 - RADIOLIB_NC
    #define PIN_RECEIVER_RST 23

matthias-bs, 2023-01-22:
LORAWAN_NODE was just intended as a define for my own board.

closed 2023-03-09 by matthias-bs