matthias-bs/BresserWeatherSensorReceiver #18: Question: Does bresser 7-channel work?

kroon040, 2022-11-29:
Hi,

I bought this one
https://www.bresser.de/en/Weather-Time/WLAN-Weather-Stations-Centers/BRESSER-Smart-Home-7-Channel-Tuya-Thermometer-Hygrometer.html

The outdoor sensor  is the 7006300, is this working?

matthias-bs, 2022-11-29:
Hi,

I don't know. I did not have the chance to test this sensor yet. Chances are not bad - the Soil Temperature and Moisture sensor (https://www.bresser.de/en/Weather-Time/BRESSER-soil-temperature-and-moisture-meter.html) which is working is listed in the accessories. But you never know...

Cheers
Matthias

matthias-bs, 2022-11-29:
BTW: 7006300 is probably the set, while the outdoor sensor is either 7009999 or 7009974.

kroon040, 2022-11-30:
Hi,

I go to try.

I have an cc1101 a schema that I used a time ago and the print also with a wemos d1 mini

[image]

I'm looking for the right config. I think the most are right, only RST, what pin is this on the cc1101?

```
#elif defined(ESP8266)
    #define PIN_RECEIVER_CS   15
    
    // CC1101: GDO0 / RFM95W/SX127x: G0
    #define PIN_RECEIVER_IRQ  4 
    
    // CC1101: GDO2 / RFM95W/SX127x: 
    #define PIN_RECEIVER_GPIO 5 
    
    // RFM95W/SX127x - GPIOxx / CC1101 - RADIOLIB_NC
    #define PIN_RECEIVER_RST  2
#endif
```

matthias-bs, 2022-11-30:
The CC1101 does not have a reset pin - you can assign RADIOLIB_NC.

kroon040, 2022-12-01:
Hi,

I can confirm that it works. I only needed to use the mqttcustom.ino, because the other crashed.

I there a default interval of those sensors, I think 1 minute, but I noticed sometimes 1 minute, 2 and also some time more that 10minutes.

Thnx for the nice project

matthias-bs, 2022-12-01:
Hi,

Thank you very much for the feedback? Could you please provide the sensor's part number?

Regarding the default interval, it is most likely in the order of 1 minute. For the Soil Moisture/Temperature sensor (7009972) the transmit interval is provided in the manual (1 min). If the interval seems to be longer, some messages could not be decoded due reception errors. You can have a look at this by enabling debug output (just uncomment `#define _DEBUG_MODE_` in `WeatherSensorCfg.h`).

Please also see the [Wiki section 'Radio Transceiver Range](https://github.com/matthias-bs/BresserWeatherSensorReceiver/wiki#radio-transceiver-range) for possible improvements.

Best regards
Matthias

kroon040, 2022-12-01:
Hi,

I have compiled it on an esp8266, and run some tests.

Laptop and tamsota device in the **same** vlan:

- SLAAC works it get an ipv6 address. (global address)
- ping works
-  webui listen to the address, use the http://[2001:xxxx:xxxx:xxxx]
- got and ipv6 local (FE80).
-  can't ping local address
- als webui can't ping.

I have default tasmota on **other** vlan than the laptop

- SLAAC works, get an ipv6 in that vlan.
- ping doesn't work, looks like the implementation doesn't use the default gateway from this address?
- webui not working ofcourse (ping also not working no routing)
- got and ipv6 local (FE80).
-  can't ping local address
- als webui can't ping.

kroon040, 2022-12-02:
Hi, 

Looking on the sensor it is 7006300 see picture. Looking on de bresser site, it should be the 7009999 (both range 150m)

[image]

I also going to flash the bresser station with tasmota.

https://templates.blakadder.com/bresser_7006300.html

matthias-bs, 2022-12-02:
Thank you very much for this information! May I ask you another favor? Could you please help to find out which protocol the sensor uses?
In order to do this, just comment out the line `#define BRESSER_5_IN_1` in `WeatherSensorCfg. h`.  If decoding still works afterwards, we know that the 6-in-1 protocol is used.

kroon040, 2022-12-02:
Hi,

I commented uit de Bresser_5_IN_1, and still working.

I noticed something else. The hostname to MQTT is correct, but in my wifi router the name is ESP_CHIPID

I noticed also in another project it sometimes breaks with arduino version of the esp. I'm using 3.0.2 with esp8266

matthias-bs, 2022-12-04:
Hi,

thanks again for testing! 

I just googled for "ESP8266 WiFi hostname" and got this hit:
https://github.com/esp8266/Arduino/issues/2826

Yes, there seem to be a few issues related to setting the hostname...

Cheers,
Matthias

matthias-bs, 2022-12-04:
I have one more question: Could you please provide the sensor_id? Thanks in advance again!

kroon040, 2022-12-06:
IN debug mode, I got errors compiling

```
In file included from /home/richard/Downloads/BresserWeatherSensorReceiver-0.4.0/examples/BresserWeatherSensorMQTTCustom/src/WeatherSensor.cpp:59:
/home/richard/Downloads/BresserWeatherSensorReceiver-0.4.0/examples/BresserWeatherSensorMQTTCustom/src/WeatherSensor.cpp: In member function 'DecodeStatus WeatherSensor::decodeBresser6In1Payload(uint8_t*, uint8_t)':
/home/richard/Downloads/BresserWeatherSensorReceiver-0.4.0/examples/BresserWeatherSensorMQTTCustom/src/WeatherSensor.cpp:692:21: error: 'chksum' was not declared in this scope
  692 |         DEBUG_PRINT(chksum, HEX);
      |                     ^~~~~~
/home/richard/Downloads/BresserWeatherSensorReceiver-0.4.0/examples/BresserWeatherSensorMQTTCustom/src/WeatherSensorCfg.h:95:49: note: in definition of macro 'DEBUG_PRINT'
   95 |     #define DEBUG_PRINT(...) { DEBUG_PORT.print(__VA_ARGS__); }
      |                                                 ^~~~~~~~~~~
exit status 1
Error compiling for board LOLIN(WEMOS) D1 R2 & mini
```.

Also a question. I want to disable the blue led complete, is this also possible? Not for power and not for receiving data

kroon040, 2022-12-06:
IN debug mode, I got errors compiling

```
In file included from /home/richard/Downloads/BresserWeatherSensorReceiver-0.4.0/examples/BresserWeatherSensorMQTTCustom/src/WeatherSensor.cpp:59:
/home/richard/Downloads/BresserWeatherSensorReceiver-0.4.0/examples/BresserWeatherSensorMQTTCustom/src/WeatherSensor.cpp: In member function 'DecodeStatus WeatherSensor::decodeBresser6In1Payload(uint8_t*, uint8_t)':
/home/richard/Downloads/BresserWeatherSensorReceiver-0.4.0/examples/BresserWeatherSensorMQTTCustom/src/WeatherSensor.cpp:692:21: error: 'chksum' was not declared in this scope
  692 |         DEBUG_PRINT(chksum, HEX);
      |                     ^~~~~~
/home/richard/Downloads/BresserWeatherSensorReceiver-0.4.0/examples/BresserWeatherSensorMQTTCustom/src/WeatherSensorCfg.h:95:49: note: in definition of macro 'DEBUG_PRINT'
   95 |     #define DEBUG_PRINT(...) { DEBUG_PORT.print(__VA_ARGS__); }
      |                                                 ^~~~~~~~~~~
exit status 1
Error compiling for board LOLIN(WEMOS) D1 R2 & mini
```.

matthias-bs, 2022-12-06:
Can can you please try to comment out line 692 in WeatherSensor.cpp?
I'll try to fix that in the next release.

kroon040, 2022-12-06:
Hi,

Is it possible to disable the led on the wemos complete?

Here the debug log

```
kroonen-weer-F44ED3/status: online
[CC1101] Initializing ... 
success!
[CC1101] Setup complete - awaiting incoming messages...
[CC1101] Receive failed - code: 0
[CC1101] Receive failed - code: 0
[CC1101] Data:		 D4 31 2C 55 1 84 20 29 BB BB BB BB B0 4 96 96 BB B0 A5 0 0 0 0 80 10 60 0
[CC1101] R [D4] RSSI: -89.5
Raw Data: 31 2C 55 01 84 20 29 BB BB BB BB B0 04 96 96 BB B0 A5 00 00 00 00 80 10 60 00 40 
kroonen-weer-F44ED3/data/55018420: {"id":"55018420","ch":1,"battery_ok":1,"temp_c":4.9,"humidity":96,"wind_gust":44.4,"wind_avg":44.4,"wind_dir":1221.0,"uv":44.4}
kroonen-weer-F44ED3/extra: {"wind_dir_txt":"","wind_gust_bft":12,"wind_avg_bft":12,"dewpoint_c":4.3,"perceived_temp_c":-5.1}
kroonen-weer-F44ED3/status: online
kroonen-weer-F44ED3/radio: {"rssi":-89.5}
```

matthias-bs, 2022-12-07:
Hi,

thanks a lot again! So the sensor type is the upper four bits of byte[6] (the one following the ID) => 2.
```
Raw Data:  31 2C 55 01 84 20 ->2<-9 BB BB BB BB B0 04 96 96 BB B0 A5 00 00 00 00 80 10 60 00
```
You can disable the LED by commenting out this line:
https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/baef2e4126bd922c4c2dcdc92688c72967c676b9/examples/BresserWeatherSensorMQTTCustom/BresserWeatherSensorMQTTCustom.ino#L108

Cheers
Matthias

kroon040, 2022-12-08:
thanks commenting out worked.

regards Richard

closed 2022-12-08 by kroon040