matthias-bs/BresserWeatherSensorReceiver #20: Add documentation for the Adafruit LoRa Radio FeatherWing - RFM95W (ADA3231)

sgtJohnny, 2022-12-09:
I have successfully tested and used the RFM95W "FeatherWing", Adafruit ADA3232, with a ESP32-S2 feather. 

Maybe it would be worth adding this to the documentation. The easiest way to implement is:

_Solder-Bridges on the Module/Wing:_
**E** to **IRQ**
**D** to **CS**
**C** to **RST**
**A** to **DI01**

_definitions in WeatherSensorCfg.h:_

 #define PIN_RECEIVER_CS   6
 #define PIN_RECEIVER_IRQ  5
 #define PIN_RECEIVER_GPIO 11
 #define PIN_RECEIVER_RST  9

matthias-bs, 2022-12-10:
Hi, 
Excellent! Thank you for sharing! I will add it.

Cheers
Matthias

matthias-bs, 2022-12-10:
Added in 0ba58faa231d566fb301ebdd34f4bf06b8903cdc and d7401db0e2f1c9de61890552b730eea3ef86926d

matthias-bs, 2022-12-10:
Which version of the ESP32S2 Feather are you using?
Is the pinout for those identical?
https://www.adafruit.com/product/4769
https://www.adafruit.com/product/5303

The latter features a BME280 Barometer/Temperature/Humidity sensor - very handy to have an integrated indoor sensor in addition to the remote radio sensors.

sgtJohnny, 2022-12-10:
I'm using the "ESP32-S2 Feather" https://www.adafruit.com/product/5303

There are quite a few models with the same pinout:

Product ID: 5000  // ESP32-S2, 4MB Flash, 2MB PSRAM
Product ID: 5303  // ESP32-S2, 4MB Flash, 2MB PSRAM, WITH BME280
Product ID: 5300  // ESP32-S2 4MB Flash, 2MB PSRAM,  Integrated Display
Product ID: 5477  // ESP32-S3, 4MB Flash, 2MB PSRAM
Product ID: 5323  // ESP32-S3, 8MB Flash, 0MB PSRAM
Product ID: 5483  // ESP32-S3 4MB Flash, 2MB PSRAM,  Integrated Display

This one: https://www.adafruit.com/product/4769 is actually a board from unexpected maker and has different pinout.

Yes BME280 could be handy, but it might be not the best idea to use the on-board one. My Board (no BME) gets quite warm, due to the 5V to 3.3V LDO on board, that has to supply power to the RF and CPU, so temperature readings would be off a few degrees. I've decided to use a BME680 breakout-board and attach with the STEMMA QT cable.

matthias-bs, 2022-12-10:
Maybe the heating from the LDO can be limited by intermittent operation: wake up from deep sleep, take a measurement, turn on WiFi and radio transceiver and do everything else - then go to deep sleep again.

matthias-bs, 2023-01-05:
Hi,

can this issue be closed?

Thanks and regards
Matthias

closed 2023-01-05 by sgtJohnny