matthias-bs/BresserWeatherSensorReceiver #21: Issue with negative temperatures on Bresser 7002510

sgtJohnny, 2022-12-10:
**Describe the bug**
In the last few days, temperatures got below zero. This caused abnormal rain-meter readings on my Bresser 5-in-1 (7002510), which uses the 6-in-1 decoder of the library. 

The problem seems to be, that a received message is interpreted as "temp, hum, wind, rain" OK, but this is not possible, since the to messages are either "wind & rain" or "temp, hum & rain". 

To investigate, I logged the raw messages and found the following behavior:

Message 1 (Correct Reading)  (Type: Wind, Rain) :
**10.12.2022  21:41:07**  Data: 55 49 14 40 07 ed 18 ff ff ff 27 08 **ff fb 77** ff 01 02 00 00 00 00 00 00 00 00
Rain = 48.8mm

Message 2 (False Reading) (Type: Wind, Rain, Temp, Hum):
**10.12.2022  21:41:19**  Data: 2c c3 14 40 07 ed 18 ff ff ff 27 08 **99 7a 94** ff f0 dd 00 00 00 00 00 00 00 00
Rain = 66857mm

I marked bytes 12,13,14 in bold. 

For checking if the temperature is OK, there is the following function:
`temp_ok  = humidity_ok = msg[12] <= 0x99 && (msg[13] & 0xf0) <= 0x90;`

In Message 1,  0xFF, 0xFB, 0x77 is clearly read false for "temp_ok"
In Message 2,  0x99, 0x7a, 0x94 are read true for "temp_ok"  (which would be correct since this message comes after the wind/rain one)

For checking if the rain is OK, there is the following function:
` rain_ok = msg[12] <= 0x99 && msg[13] <= 0x99 && msg[14] <= 0x99;`

In Message 1, 0xFF, 0xFB, 0x77 is read true for "rain_ok"
In Message 2, 0x99, 0x7a, 0x94 is read true for "rain_ok"

The temperature (temp_raw) in this data example would be calculated as follows
900+90+7 = 997
997-1000 = -3  * 0.1f = -0.3°C

This is correct and was displayed on the indoor station, this poses a problem with the "rain_ok" check since it also calculates a rain level of 66857mm because the validity-check returns true. 

Maybe an implementation of cross-check would help, but I'm not fully sure.
Example:
`rain_ok = msg[12] <= 0x99 && msg[13] <= 0x99 && msg[14] <= 0x99 && !temp_ok;`

matthias-bs, 2022-12-10:
Yes, this is a known bug inherited with the 6-in-1 decoder from the rtl_433 project. There already seems to be a fix:
https://github.com/merbanan/rtl_433/issues/2184

I did not find the time to catch up with it yet. I fear it will have to wait until after christmas...

franki29, 2022-12-13:
Hi, yes that is right as we do not know if it works for other version of the Bresser Sensor, but for me and some others that have the same outdoor sensor it is working. There is also a second solution, but than you get a compile error.

MacH-21, 2022-12-13:
@franki29 @matthias-bs @sgtJohnny @merbanan,

I have the 6in1 outdoor sensor and have been testing the flags please see issue 1214 last posted 16hrs ago.
we know msg16 (lsb 0 - 1) indicates which bytes etc are valid. see issue 1214
there are other flags such as Battery status and Temperature sign flag.

For the 6in1 I use-
flags = (msg[16] & 0x0f) = 0 
(Valid Wspeed,Wgust,Wdirection,Battery flag,Temperature Sign flag,Temperature,Humidity,UV)
Battery status flag   = (msg[13] >> 2) & 1; (0=Lo, 1=Ok)
Temperature sign flag = (msg[13] >> 3) & 1; (0=Positive, 1=Negative

flags = (msg[16] & 0x0f) = 1
(Valid Wgust,Wspeed,Wdirection,Rain)to 0.000mm

The 5in1 does not have a UV sensor -- Just ignore this value.
also Battery and Temperature may use different bits for flags

specifications say -40 to +80 
we need the Temperature sign flag
what would happen if +40 to +80 we need the Temperature sign flag
outdoor and remote sensors are capable of reading these values, they use the Sensitron SHT31 which is capable of -40 to +125
Benjamin/Christian is there a way to list/merge these different posts?

matthias-bs, 2022-12-13:
That's fine - we are watching your efforts in the rtl_433 project with great interest! ;-)

sgtJohnny, 2022-12-13:
I have fixed locally, for me both solutions work
`rain_ok = msg[12] <= 0x65 && msg[13] <= 0x99 && msg[14] <= 0x99;`
of like in the post on rt433:
`rain_ok = (msg[16] & 0x01);`

matthias-bs, 2023-01-18:
Still testing, but looks good:
https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/test-6in1-protocol/src/WeatherSensor.cpp

closed 2023-02-08 by matthias-bs