matthias-bs/BresserWeatherSensorReceiver #22: decodeBresser5In1Payload() extracts rain-value from three digits instead of four digits

aanban, 2022-12-23:
I use this setup: 
* Bresser 5-in-1 weather station, 
* CC1101 868MHz receiver 
* Wemos D1 mini µC. 

The received readings from the 5-in-1 sensor are confirmed by the base station. The SW worked right away, Respect! 

This year white Christmas could be forgotten anyway, it's raining instead. But at least the rain counter is incrementing. Coincidentally, at my station there was just a change from 8980 (898.0mm) to 9112 (911.2mm) .

The 5-in-1 manual states  rain-value range as [0 ... 999.9mm]. 

I wonder if there is a special reason why only **three digits** are used for the extraction of the rain value and the existing **fourth digit** of msg[24] is ignored. Or it this a bug?

I made a change in **[decodeBresser5In1Payload() line 542](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/35c14b463db9c56aa210604a2a81c58e64b599f2/src/WeatherSensor.cpp#L542)**  from: 
```c
int rain_raw = (msg[23]&0x0f) + ((msg[23]&0xf0)>>4) * 10 + (msg[24]&0x0f) * 100;
```

to this:
```c
int rain_raw = (msg[23]&0x0f) + ((msg[23]&0xf0)>>4) * 10 + (msg[24]&0x0f) * 100 + ((msg[24]&0xf0)>>4) * 1000 ; 
```

Example of an received frame, the rain-value is found in msg[23] and msg[24]
```
16:01:42.005 > raw= E4 1C 7F EB 5F EA EF E7 FE 6A CD 6E FF 1B E3 80 14 A0 15 10 18 01 95 32 91 00
16:01:42.011 >      |____________________XOR_redundancy__| CS |___________________________Data__|
16:01:42.019 > 
16:01:42.019 > msg[14]       [e3]       ID   = 227 
16:01:42.022 > msg[15]       [80]       Type = 0 
16:01:42.025 > msg[20,21,25] [18,01,00] T = 118 =  11.8 °C
16:01:42.028 > msg[22]       [95]       H = 95 %
16:01:42.030 > msg[17]       [a0]       Wind-Dir-Raw = 2250 °
16:01:42.036 > msg[16,17]    [14,a0]    Wind-Gust-Raw = 20 
16:01:42.038 > msg[18,19]    [15,10]    Wind-Raw = 15 
16:01:42.044 > msg[23,24]    [32,91]    Rain = 9132 total =913.2mm delta =  0.0mm
16:01:42.050 > msg[25]       [00]       Battery=Ok
16:01:42.052 > 24   12s E3-00 BAT=OK  T= 11.8°C H=95% wMax= 2.0m/s wAve= 1.5m/s wDir=225.0° R=913.2mm RSSI=-73.5dBm
```

Maybe someone with an 5-in-1 station can confirm?

matthias-bs, 2022-12-23:
Hi,

thank you for reporting. The decoder function was taken from the [rtl_433](https://github.com/merbanan/rtl_433) project, i.e. [rtl_433/blob/master/src/devices/bresser_5in1.c](https://github.com/merbanan/rtl_433/blob/master/src/devices/bresser_5in1.c).
I always wondered why the rain counter has such a small range, but I failed to compare it to the manual.
Considering this and the fact that the upper nibble of msg[24] is currently considered as unused, I'd say you are right!
The following example also gives a hint in the same direction:
https://github.com/merbanan/rtl_433/blob/7d5129786b35287f503a0af1d6243fd59e0db0da/src/devices/bresser_5in1.c#L33

BTW: Which device/manual are you referring to? 2022-04-Manual_7002510_Weather-Center-5in1_de-en_BRESSER_v042022a.pdf states "Rainfall measuring range: 0...9999 mm". So, the rain counter value could in fact extend into msg[25], which also contains the battery state (bit 7) and the temperature sign (bit 3). That would mean that the rain counter MSB would be somewhere in bits 6:4 and 2:0. That seems a little strange...
See https://github.com/merbanan/rtl_433/blob/7d5129786b35287f503a0af1d6243fd59e0db0da/src/devices/bresser_5in1.c#L42

Regards
Matthias

matthias-bs, 2022-12-23:
Let's see what happens if the rain continues and your sensor crosses the 999.9 mm mark.

aanban, 2022-12-23:
> Hi,
>
> thank you for reporting. The decoder function was taken from the [rtl_433](https://github.com/merbanan/rtl_433) project, i.e. [...]

OOps, your are right the manual states 9999mm (page 9 in German folder) as max value. 

Wert                            | Bereich
--------------------------------|----------------------------------------------
Temperatur - Einheit            | °C / °F
Temperatur-Messbereich          | -40 °C ... 60 °C
Luftfeuchtigkeits_Messbereich   | 1% ... 99 %
Niederschlags-Einheit           | mm, inch
Niederschlags-Messbreich        | 0 ... 9999 mm ( 0 ... 397.7 inch)
Windgeschwindigkeit Einheit     | mph, ms/s , km/h , knot
Windgeschwindigkeit Messbereich | 0...112 mph, 0...50 m/s , 0...180 km/h, 0...97knots
Windrichtungs-Anzeigen          | 16

aanban, 2022-12-23:
> Let's see what happens if the rain continues and your sensor crosses the 999.9 mm mark.

I don't need a flood. Well I think it will take years for the counter to jump to 9999.9mm. Before that surly the battery is low, and then I will see if the counter then starts again at 0. 

Cheers Andreas 

p.s.
But I am wondering about something else. It is unbelievable how many bytes are wasted for this simple XOR check. Wouldn't it have made more sense to use an error correction like Reed-Solomon instead? As far as I have observed, mostly just single bytes within the frame are broken, which could be easily rescued with an error correction. 

OK, actually we have to be glad that the frame was defined super-simple. Only then it was possible to guess the unknown format.

closed 2023-01-11 by matthias-bs

matthias-bs, 2023-01-13:
@aanban I created a pull request in the rtl_433 project according to your finding. Hope this is o.k. for you...

aanban, 2023-01-14:
> @aanban I created a pull request in the rtl_433 project according to your finding. Hope this is o.k. for you...

Everything is perfect, thanks for doing this for me. 

I also have a small addendum. Meanwhile, the battery of my 5in1 outdoor station was empty, so I had to replace it some days ago. Doing so, the rainfall counter jumps to 0. Probably we will not get out so easily what happens when the counter jumps from 999.9 mm to 1000.0mm, i.e. where to find it within the frame.

matthias-bs, 2023-01-14:
O.k., thank you for your message. My rain gauge is currently at 410mm - let's see if the batteries last until an overflow occurs...