matthias-bs/BresserWeatherSensorReceiver #26: Home Assistant: Client <unknown> disconnected due to protocol error.

ga3ns3blu3mchen, 2022-12-31:
Hi,
i have issues to connect the weatherstation to my Home Assistant by MQTT. Testing with the BresserWaetherSensorWaiting gives me perfect values. But when i try a MQTT connection, the Serial Monitor shows me a endless try to MQTT connect and in Home Assistant i get a "Client <unknown> disconnected due to protocol error." message. Anyone else got it Working with Home Assistant?

BR

matthias-bs, 2023-01-03:
Hi,

I have no experience with Home Assistant. Is there any other debug output available?
Does HA provide an MQTT broker?
Did you cross-check the MQTT port, MQTT login/password and the type of connection (secure vs. non-secure)?

Best regards
Matthias

ga3ns3blu3mchen, 2023-01-03:
Hi Matthias,

Thanks for the response.
unfortunately not. When i uncomment the _DEBUG_MQTT_ line, i get a compile error:

`
BresserWeatherSensorMQTT:602:21: error: 'genWeatherdata' was not declared in this scope

  602 |         decode_ok = genWeatherdata(0 /* slot */, 0x01234567 /* ID */, 1 /* type */, 0 /* channel */);

      |                     ^~~~~~~~~~~~~~

exit status 1

'genWeatherdata' was not declared in this scope
`
In HA you can setup a Mosquitto Broker. I checked the MQTT settings, and they seem to be correct. In the Broker log i see the esp trying to connect but stopping with the "disconnected due to protocol error". 
I tried everything with an online MQTT broker and it works like a charm, so it has something to do with the HA provided broker.
Maybe an incompatible MQTT library or issues with the MQTT topics.
I was hoping that someone ran into the same issue and knows a solution.

matthias-bs, 2023-01-03:
O.k., thanks for your feedback. I will check why genWeatherdata() cannot be found.

matthias-bs, 2023-01-03:
Fixed _DEBUG_MQTT_ feature:
https://github.com/matthias-bs/BresserWeatherSensorReceiver/commit/a696ff70f6c984b9f4201ea70f607e8d3a91afbc

lschans, 2023-01-13:
I have burned a whole evening trying to figure this out. I have a HA myself with the same broker. Same endless........... In the terminal.

A pointer in the right direction to solve this would be nice.

ga3ns3blu3mchen, 2023-01-13:
Hi Lschans,

in my case the #define USE_SECUREWIFI was the problem. Use #define USE_WIFI and it will work. And there is a change in the format of the json string from Release 0.4 to 0.5
HA can only work with the 0.4 Release json string format. The change was also only for the data string, extra ist working in 0.5. I wanted to inform @matthias-bs about it as soon as i have time to compare the code and give him the exact change which causes the issue for HA.
@matthias-bs thank you for this amazing project. You should add a "buy me a coffee" button to your git :)

matthias-bs, 2023-01-14:
@ga3ns3blu3mchen Thanks for the kind feedback! I'd be happy to make some changes to it (or to add a specific example sketch) compatible to HA.

Good idea to add a "buy me a coffee" button, I will think about this! ;-)

lschans, 2023-01-14:
@ga3ns3blu3mchen Sadly this didn't work for me. I tried all different configuration options and it just keeps loading with a bunch of ....... and then nothing.

If I have some time this weekend I will give this a try https://esphome.io/custom/custom_component.html

Mainly because to my opinion ESPHome is the better integration when you have Home Assistant. If I can hack something together I will either push this as a fork to this project, or contact @matthias-bs  and ask if he want's to host the code in his repo. I am not that much of a 'maintaining a repo' guy. And this is most of his code anyway.

Let me know how you people would think about this as an ESPHome integration?

matthias-bs, 2023-01-16:
Are using an ESP32 or an ESP8266? With ESP32 and setting "Core Debug Level: Verbose" in the IDE, you might get some helpful information from the WiFi protocol stack.

Due to some network problems today, I noticed a few things:
1. according to some sources, it is recommended to add a `delay(10);` between `WiFi.begin()` and polling `WiFi.status()`
2. the dots are printed at several places in the code (not really helpful for localization of a problem...)
3. in some cases, the program does never recover from a networking problem, but a reset would help

To address the third item, add a retry counter and performed a timed restart, e.g.:
```
    WiFi.begin(ssid, pass);
    delay(10);
    int cnt = 0;
    while (WiFi.status() != WL_CONNECTED)
    {
        Serial.print(".");
        delay(1000);
        if (++cnt == 10)
            ESP.deepSleep(SLEEP_INTERVAL * 1000);
    }
    Serial.println(F("connected!"));
```

I don't have any experience with ESPHome, but I'm open to make some modifications if I can help or to add another example (or a real integration).

lschans, 2023-01-20:
So far nothing of the tips and tricks helped, my network connection is fine. Either using secure or normal WiFi.
Using secure wifi it fetches the time and this all works fine. It's somehow related to the MQTT (mosquitto server) Home Assistant ships with. I confirmed the server works with other MQTT connectors. 

I didn't have much time for coding last week because of some crazyness at the job. I wil keep testing and see if I can find out what exactly is keeping it from connecting to home assistant. 

I'll keep you guys updated.

matthias-bs, 2023-03-16:
Any update on this?

closed 2023-03-26 by matthias-bs