matthias-bs/BresserWeatherSensorReceiver #30: error 'NUM_SENSORS' was not declared in this scope

mczakk, 2023-01-14:
Hi, i'm trying to compile this code but keep getting the same error, 
WeatherSensor.h:204:25: error: 'NUM_SENSORS' was not declared in this scope
         sensor_t Sensor[NUM_SENSORS];      //!< sensor data array

as you can see from the screenshots 'NUM_SENSORS' is defined!
[image]
[image]

can you offer any assistance?

matthias-bs, 2023-01-14:
Hi,

Are you using the Arduino IDE for compiling? Your screenshots do not look familiar to me.
Arduino treats some things regarding header files in a special way...

Have you tried to include WeatherSensorCfg.h in WeatherSensor.h?

If you are using Arduino, it also makes a difference where the files are located - according to your screenshot, the header files and the WeatherSensor.cpp are not located in the same directory.

In WeatherSensor.cpp, WeatherSensorCfg.h is included before WeatherSensor.h, though.

Maybe you can try to use the directory structure of example/BresserWeatherSensorMQTTCustom for your project?

Best regards
Matthias

mczakk, 2023-01-18:
Hi Matthias, 
I tried using the MQTT code with my esp32, and it works :) Now I have another issue, I can publish a message to Hivemq, but I also need to publish it to AWS, is there a way to publish a second message to a different endpoint?

Cheers

Matt

matthias-bs, 2023-01-18:
Hi,
that's nice to read!

If you want to publish your data to another MQTT broker, you need a second MQTTClient object for setting up and maintaining a separate connection.

I would try to set up the AWS connection first (if I remember correctly from reading about it, you really have to set up the correct security measures for this). In the second step, I would integrate both connections.
Maybe you cannot keep both connections in parallel (due to RAM limitations), but have to use them one after another.

Best regards
Matthias

mczakk, 2023-01-18:
I was thinking that it would be easier to connect to one broker first, i.e HiveMQ, then disconnect and connect to AWS. I found instrctions to connect to AWS here: [https://how2electronics.com/connecting-esp32-to-amazon-aws-iot-core-using-mqtt/](url) , but am struggling to incorporate that code into your .ino file :( (c++ coding is not my forte!)
I'm inclined to go down the ESP32 TTN route using your code from here: [https://github.com/matthias-bs/BresserWeatherSensorTTN](url) then integrate AWS and MQTT from there.
I'll let you know how it goes!!

Cheers

Matt

matthias-bs, 2023-01-18:
Can we close this issue then? Feel free to open another one with a different topic if required.

mczakk, 2023-01-18:
I dont see why not!, I'm sure I'll be quick to shout if i can't get your TTN code to work!!

closed 2023-01-18 by mczakk