matthias-bs/BresserWeatherSensorReceiver #33: MQTT Topics ändern

Lordcybertracker, 2023-01-22:
Hallo,
ich hab es doch ans laufen bekommen.

Jedoch hab ich 2 Fragen?

1. Gibt es eine Möglichkeit die Topics zu ändern?
Da diese Daten auf meinem Server zusammenlaufen und da diese von meinen Bezeichnungen abweichen.
So müßte ich extra alles anders Definieren.
2. Windrichtung. Da die Anlagen ja nicht immer richtig ausgerichtet sind gib es eien weg dieses anzupassen?
Danke

matthias-bs, 2023-01-22:
1. Die Topics sind einfach Strings. Man kann sie natürlich anpassen, muss nur aufpassen dass die reservierte Länge der Variablen eingehalten wird.
2. Dafür kann man sicher eine Funktion implementieren, die einen Offset (positiv oder negativ) addiert. Man muss dann ggf. bei Ergebnissen <0° oder >360° wieder 360° addieren bzw. subtrahieren. Der Offset kann per #define im Code gesetzt werden.

Oder man berücksichtigt den Offset auf der Client-Seite.

Lordcybertracker, 2023-01-22:
Danke.
Hab es einfach in meinem Json Decoder angepasst. 
Daten laufen jetzt per MQTT über Nodered in meine Grafana.

closed 2023-01-22 by Lordcybertracker