matthias-bs/BresserWeatherSensorReceiver #35: JSON Parsen schlägt schief.

Lordcybertracker, 2023-01-24:
HAllo,m
Beim  Decodieren der JSON bekom ich immer Fehler:
25.1.2023, 00:29:18[node: 4b3131542aa8ddaf](http://nodered.hsn-tech.de:1880/#)function : (error)
"SyntaxError: Unexpected number in JSON at position 7"

Die DAten die roh ankommen sind:
25.1.2023, 00:29:18[node: e320d83476070742](http://nodered.hsn-tech.de:1880/#)
ESP-Bresser -0CDFBC/data/ A : msg.payload : string[163]
"{"id":0000000A,"ch":0,"battery_ok":1,"temp_c":-0.6,"humidity":99,"wind_gust":0.8,"wind_avg":0.8,"wind_dir":292.5,"rain":0.0,"rain_d":0.0,"rain_w":0.0,"rain_m":0.0}"

Woran kann das liegen?

matthias-bs, 2023-01-25:
Das liegt mit Sicherheit an der id. Versuche bitte einmal den Wert statt als Hexadezimalzahl als Dezimalzahl auszugeben, also den Formatstring im snprintf()-Funktionsaufruf von %08X (hex with leading zeroes and capital letters) in %lu (unsigned int or unsigned long) zu ändern.

Lordcybertracker, 2023-01-25:
Danke.
Ich bekomme den Empfänger heute Abend wieder. Dann teste ich das mal.
Kannst du noch Aufschlüsseln. Welcher Wert was ist in Deutsch. Das hab ich noch nicht kapiert also avg und gust.

matthias-bs, 2023-01-25:
avg - average -> durchschnittliche Windgeschwindigkeit
gust -> Windgeschwindigkeit in Böen

Lordcybertracker, 2023-01-25:
Danke das war es.

closed 2023-01-25 by matthias-bs