matthias-bs/BresserWeatherSensorReceiver #37: LoRa32 V 2,1 1.6.1 [SX1276] Error initialising

Jo59Devs, 2023-02-07:
TTGO-Lora32-V2.1 Board
Labeled with T3 v1.6.1 20210104

Settings from: [here](https://github.com/espressif/arduino-esp32/blob/master/variants/ttgo-lora32-v21new/pins_arduino.h) doesn't work.
Setting SX1276:
    #define PIN_RECEIVER_CS      18
    #define PIN_RECEIVER_IRQ    26
    #define PIN_RECEIVER_GPIO  33
    #define PIN_RECEIVER_RST    23

Please help!

matthias-bs, 2023-02-07:
Which board did you select in the Arduino IDE?
In which way can "doesn't work" be described in more detail? Could you please provide some output from the compiler (or the running software, if compiling works fine)?

Jo59Devs, 2023-02-07:
Hi, I use VisualStudioCode and the compiling works fine.
Selected Board is:
`[env:esp32dev]`
`platform = espressif32@5.0.0`
`board = esp32dev`
`framework = arduino`

Result:
Attempting to connect to SSID: MyXXXX.connected!
Checking wifi...
MQTT connecting... connected!
Bresser-5in1-AA79C0/status: online
[  1357][E][WeatherSensor.cpp:125] begin(): [SX1276] Error initialising: [-2]

I have tested many Pin-Settings.....
Best Regards

matthias-bs, 2023-02-07:
Currently I am only using Arduino IDE, for VisualStudioCode you have to go a little bit more into details.

First we have to ensure that your desired pin config is actually used. Could you please provide the compiler output? There should be some output from `WeatherSensorCfg.h`
- "ARDUINO_TTGO_LoRa32_V1 defined; using on-board transceiver"
 
    or

- "Generic ESP32; assuming this is the LoRa_Node board (DFRobot Firebeetle32 + Adafruit RFM95W LoRa Radio)"

`WeatherSensorCfg.h` tries to detect a define set by the Arduino IDE (or in your case some config file) and choose the matching pin defines.

In case no specific board definition is found, the section starting with
```
#elif defined(ESP32)
    // Generic pinning for ESP32 development boards
```
should be used.

In any case, you have to find the pinning four you board in some documentation (maybe here: https://www.thethingsnetwork.org/forum/t/big-esp32-sx127x-topic-part-2/11973) .

matthias-bs, 2023-02-07:
More pinning info to choose from:
http://www.lilygo.cn/prod_view.aspx?TypeId=50003&Id=1130&FId=t3:50003:3
https://github.com/Xinyuan-LilyGo/TTGO-LoRa-Series

Jo59Devs, 2023-02-07:
I cannot find log files, but the following Code-Section ist highlighted:

#elif defined(TTGO_LORA32)
    #define PIN_RECEIVER_CS   18
    // CC1101: GDO0 / RFM95W/SX127x: G0
    #define PIN_RECEIVER_IRQ  26
    // CC1101: GDO2 / RFM95W/SX127x: G1
    // n.c. on v1/v2?, LORA_D1 on v21
    #define PIN_RECEIVER_GPIO 33
    // RFM95W/SX127x - GPIOxx / CC1101 - RADIOLIB_NC
    #define PIN_RECEIVER_RST  23
#elif defined(ADAFRUIT_FEATHER_ESP32S2)

Ok, I know the Info above.

matthias-bs, 2023-02-07:
Is this your board?
https://www.aliexpress.com/item/32915894264.html

Jo59Devs, 2023-02-07:
Yes, exactly:
[this](https://de.aliexpress.com/item/32872517875.html?spm=a2g0o.order_list.order_list_main.5.6d015c5fHCoC3A&gatewayAdapt=glo2deu)
I try to compile the Project in Arduino Studio, and it seems the compiling works.

matthias-bs, 2023-02-07:
So which pinout is required?
Is there any define (from the config or IDE) to let the compiler detect the board without having to change a source file manually?

Jo59Devs, 2023-02-07:
"Is there any define (from the config or IDE...."
Not really, i have changed to  #define USE_SX1276 and #define TTGO_LORA32
The required Pins are:
    #define PIN_RECEIVER_CS   18
    #define PIN_RECEIVER_IRQ  26
    #define PIN_RECEIVER_GPIO 33
    #define PIN_RECEIVER_RST  23
Why it doesn't work with VSCode is beyond me :)

Jo59Devs, 2023-02-07:
It now also works with VSCode:
`[env:ttgo-lora32-v21]`
`platform = espressif32`
`board = ttgo-lora32-v21`
framework = arduino`
`monitor_speed = 115200`
`lib_deps = `
	`jgromes/RadioLib@^5.6.0`
	`bblanchon/ArduinoJson@^6.20.0`
	`256dpi/MQTT@^2.5.0`

Further changes:
#define LED_GPIO        25      // LED pin

matthias-bs, 2023-02-07:
O.k., thank you!

I also added it in [WeatherSensorCfg.h](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/src/WeatherSensorCfg.h): https://github.com/matthias-bs/BresserWeatherSensorReceiver/commit/81a588970bebe018a40c71b813b3f17c165196c8

I didn't notice the menu items for Board Revisions in the Arduino IDE before:
[image]

The selection above sets the define `ARDUINO_TTGO_LoRa32_v21new`.

Jo59Devs, 2023-02-08:
My problem was, that I didn't choose the right board in VSCode. 
This caused a lot of misunderstandings. 
The problem is solved and can be closed. 

Thank you for your effort and your work here, Matthias.

Best Regards Jo

matthias-bs, 2023-02-08:
You are welcome!

Cheers
Matthias

closed 2023-02-08 by matthias-bs