matthias-bs/BresserWeatherSensorReceiver #38: Range Inquiry

garudaonekh, 2023-02-17:
Hi,
Very interested with this one.
My cOncern is range. what is its range in compare with traditonal RF module like Superheterodyne type?

I use to use CC1101 to decode weather station using another library but the range is disappointed.

Thanks

matthias-bs, 2023-02-17:
Hi,

please read my comments here: https://github.com/matthias-bs/BresserWeatherSensorReceiver/wiki#radio-transceiver-range.

With the CC1101 and a distance of ~15m from the sender, it works (with a noticeable rate of invalid frames) if the receiver is in a good position next to the window.

If you are fine with spending a little more money, use one of the SX1276 based solutions with a proper antenna.

Cheers
Matthias

garudaonekh, 2023-02-19:
Yes, I will use SX1278 if required. Do you have any detail about its range?  I need longer range because I use for agriculture project

matthias-bs, 2023-02-19:
No, I don't think it is possible to make a general estimation.

garudaonekh, 2023-02-22:
How about sending? Does it perform better than the normal RF433?

matthias-bs, 2023-02-22:
I do not quite understand that question. The sensors are sending, `BresserWeatherSensorReceiver` is, well, receiving.

And what do you mean by "normal RF433"?

The sensors supported by this project are using FSK modulation at 868 MHz. There are some common sensors (mostly using OOK modulation) at 433 MHz. To get a rough idea, you could compare the indicated range of some devices in the 433 MHz and in the 868 MHz band (e.g. on Bresser's website).

The actual range depends on many factors -

Transceiver:
* Modulation
* Symbol rate
* Oscillator frequency accuracy and stability
* Output power / Input sensitivity
* Antenna connection losses
* Antenna gain
* ...

Transmission channel (over-the-air):
* Distance
* Interference
* Reflections
* Damping
* ...

garudaonekh, 2023-02-23:
Yes, yours is only for receiving. But my project also need to send to remote switch as well using RC-Switch. My intent is to use CC1101 for both

matthias-bs, 2023-02-23:
Which range do you need?

garudaonekh, 2023-02-23:
I work on farm project so range does matters. I am looking to achieve several hundred meters on open field.

garudaonekh, 2023-02-23:
There are 10dbm and 20dbm CC1101 on the market. Do you think this can improve transmission of CC1101 using ASk/OOK?
They however have same receiving sensitivity

matthias-bs, 2023-02-23:
Probably yes, but I can't tell how much. And I don't know if the higher power version is allowed in your country.

For several hundred meters, I'd go for a LoRa module like EBYTE E22/E32. There seem to be Arduino libraries for both (however I did't use any of them).

With LoRa modulation much better range can be achieved than with FSK/ASK/OOK at the same output power.

garudaonekh, 2023-02-23:
The reason I try to use it instead of LORA because the RF433 remote switches are very cheap

closed 2023-03-06 by matthias-bs