matthias-bs/BresserWeatherSensorReceiver #39: Fail to compile

singinglarvae, 2023-02-20:
Hi There,

I am not pretty sure if it is a bug or a missconfiguration of my own, but anyhow, here's my issue.

I use Ubuntu 22.04LTS, Arduino IDE 1.8.19 and ESP 8266 (I tried both [dev and stable version](url=https://arduino-esp8266.readthedocs.io/en/latest/installing.html))

When trying to compile on a ESP866 generic module, I have got the following error :+1: 

`In file included from /home/marc/Arduino/BresserWeatherSensorBasic/BresserWeatherSensorBasic.ino:54:
sketch/WeatherSensorCfg.h:315:48: note: '#pragma message: Receiver chip: [SX1276]'
  315 | #pragma message("Receiver chip: " RECEIVER_CHIP)
      |                                                ^
sketch/WeatherSensorCfg.h:316:170: note: '#pragma message: Pin config: RST->2, CS->15, GD0/G0/IRQ->4, GDO2/G1/GPIO->5'
  316 | #pragma message("Pin config: RST->" STR(PIN_RECEIVER_RST) ", CS->" STR(PIN_RECEIVER_CS) ", GD0/G0/IRQ->" STR(PIN_RECEIVER_IRQ) ", GDO2/G1/GPIO->" STR(PIN_RECEIVER_GPIO) )
      |                                                                                                                                                                          ^
Archiving built core (caching) in: /tmp/arduino_cache_703725/core/core_esp8266com_esp8266_generic_xtal_80,vt_flash,exception_disabled,stacksmash_disabled,ssl_all,mmu_3232,non32xfer_fast,ResetMethod_nodemcu,CrystalFreq_26,FlashFreq_40,FlashMode_dout,eesz_1M64,led_2,sdk_nonosdk_190703,ip_lm2f,dbg_Disabled,lvl_None____,wipe_none,baud_115200_551d4108386915ba00dc00bafc339e8d.a
/usr/share/arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: sketch/BresserWeatherSensorBasic.ino.cpp.o:(.text.setup+0x8): undefined reference to `_ZN13WeatherSensor5beginEv\'
/usr/share/arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: sketch/BresserWeatherSensorBasic.ino.cpp.o:(.text.loop+0x3c): undefined reference to `_ZN13WeatherSensor10getMessageEv\'
/usr/share/arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: sketch/BresserWeatherSensorBasic.ino.cpp.o:(.text.loop+0x67): undefined reference to `_ZN13WeatherSensor10getMessageEv\'
collect2: error: ld returned 1 exit status
exit status 1
Erreur de compilation pour la carte Generic ESP8266 Module`

I guess "pragma messages" are just alarms, but I could not get why I have got undefined references ? Wrong liking ?
 
Any idea ?

matthias-bs, 2023-02-20:
Hi,

the pragma message are just for your information, to allow you to compare the setup used by the sketch with the actual hardware setup.

Yes, you got a linker error. The sketch is compiled, but the library seems to be missing. Did you install "BresserWeatherSensorReceiver" with the Arduino IDE's Library Manager?

Regards
Matthias

singinglarvae, 2023-02-21:
Hi,

Shame on me... I did not get the link for the build instructions :-S And therefoer, I did not install the librairy using the library manager...

Followed the instructions and I solved my issue. Just need to tweak a few things but it is already looking much better!

Thanks for your quick reply,

regards,

Marc

closed 2023-02-21 by singinglarvae