matthias-bs/BresserWeatherSensorReceiver #41: Add support for Sainlogic 7-in-1 WS3500

fgnm, 2023-03-03:
Hi,
I have a Sainlogic WS3500 Weather Station which communicate with its display at 868Mhz (inside the display there's a ESP8266 and a CC1101 receiver).
I'd like to receive date inside my own ESP8266 with CC1101 receiver, I guess that your library for Bresser 7-in-1 could be adapted also to my Sainlogic, but I think that the initial parameter configurations aren't right, I always get `[CC1101] Receive failed: [0]`, do you think can be added? If so, can you teach me how to identify correct parameters for my WS?

Thanks

matthias-bs, 2023-03-03:
Hi,
hard to tell... As the first step, could you please enable debug output? (see https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/DEBUG_OUTPUT.md) If it is really a variant of the Bresser 7-in-1 protocol, there should be a chance to figure out the details.

To identify unknown RF parameters, code rate, encoding etc. you could have a look at the RTL_433 project. There are many people with great experience.

Regards
Matthias

matthias-bs, 2023-03-04:
Maybe you find a solution via any of the other model/brand names for this device:
https://www.reddit.com/r/myweatherstation/comments/xn0kak/sainlogic_ws3500_data/
"I live in Portugal and in October 2021 I acquired a Sainlogic WS3500, also known as Froggit WH3000SE."

This seems to be about integrating it via WiFi instead of the proprietary radio protocol:
https://github.com/phifogg/ioBroker.sainlogic

fgnm, 2023-03-06:
Hi @matthias-bs thank you so much for the info. Very helpful to know the real manufacturer is Froggit and not Sainlogic :)
However I cannot try RTL_433 because I don't have the required hardware. I've just a CC1101 which is not supported by the lib.
I'll try in the following day if I can get one, I hope that one of the already made profiles may recognize my WS.

For the WiFi thing, yes is the protocol I already use, but I'm forced to use their own Display Console.. I'd like to use my own receiver because it's not really convenient to have the display near the WS + I will probably needs to add a Mobile Network so having my own hardware would be much better anyway

fgnm, 2023-03-07:
Hey I think I got something very interesting.. RTL_433 was able to successfully capture and decode Weather Station data! The real model name is `Fineoffset-WH65B`
[image]

So, sorry if I bother you, but it's my first time working on this kind of stuff, could you help me to understand and setup correct parameters?

This is the cu8 file from RTL_433: [g004_868.3M_1000k.cu8.txt](https://github.com/matthias-bs/BresserWeatherSensorReceiver/files/10907662/g004_868.3M_1000k.cu8.txt)

And this is the protocol to decode data: https://github.com/merbanan/rtl_433/blob/88ae8f84737dc2fbb3897012313dae7f58d9d4e8/src/devices/fineoffset.c#L158

If I understand correctly Preamble and Sync word are the same of the bresser, however I always get `[CC1101] Receive failed: [0]`, looks like that this condition is not true https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/src/WeatherSensor.cpp#L212 , indeed `recvData[0]` is never `0xD4`.

I guess that the reason is because wrong parameters here: https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/src/WeatherSensor.cpp#L92 how can I identify them correctly?

Many thanks

fgnm, 2023-03-07:
Sorry for the dumb question but where I should get those?
This is de debug output while running RTL_433, just run with `rtl_433 -f 868.3M -vvv -R 78`
```
Fine Offset Electronics, WH25, WH32B, WH24, WH65B, HP1000 Temperature/Humidity/Pressure Sensor: Exact bit width (in us) is 57.67 vs 58.00, 79 bit measured
fineoffset_WH0290_callback: short package. Row length: 178. Header index: 202
bitbuffer:: Number of rows: 1 
[00] {178} 55 6a aa aa aa 96 ea 12 08 00 01 3b 13 80 00 00 06 00 21 ad 61 90 00 
pulse_demod_pcm(): Fine Offset Electronics, WH25, WH32B, WH24, WH65B, HP1000 Temperature/Humidity/Pressure Sensor
bitbuffer:: Number of rows: 1 
[00] {178} 55 6a aa aa aa 96 ea 12 08 00 01 3b 13 80 00 00 06 00 21 ad 61 90 00 
Pulse data: 262 pulses
[  0] Pulse:    0, Gap:    3, Period:    3
[  1] Pulse:    3, Gap:    5, Period:    8
[  2] Pulse:    2, Gap:    5, Period:    7
[  3] Pulse:    2, Gap:    4, Period:    6
[  4] Pulse:    2, Gap:    4, Period:    6
[  5] Pulse:    2, Gap:    3, Period:    5
[  6] Pulse:    3, Gap:    6, Period:    9
[  7] Pulse:    2, Gap:    3, Period:    5
[  8] Pulse:    3, Gap:    3, Period:    6
[  9] Pulse:    1, Gap:   59, Period:   60
[ 10] Pulse:   59, Gap:   56, Period:  115
[ 11] Pulse:   59, Gap:   58, Period:  117
[ 12] Pulse:   58, Gap:   58, Period:  116
[ 13] Pulse:   58, Gap:   12, Period:   70
[ 14] Pulse:    2, Gap:   44, Period:   46
[ 15] Pulse:   63, Gap:   13, Period:   76
[ 16] Pulse:    4, Gap:    7, Period:   11
[ 17] Pulse:    1, Gap:   13, Period:   14
[ 18] Pulse:    2, Gap:    7, Period:    9
[ 19] Pulse:    1, Gap:    5, Period:    6
[ 20] Pulse:   58, Gap:   59, Period:  117
[ 21] Pulse:   58, Gap:   59, Period:  117
[ 22] Pulse:   57, Gap:   59, Period:  116
[ 23] Pulse:   60, Gap:   40, Period:  100
[ 24] Pulse:    1, Gap:    7, Period:    8
[ 25] Pulse:    1, Gap:    5, Period:    6
[ 26] Pulse:   59, Gap:   59, Period:  118
[ 27] Pulse:   30, Gap:    2, Period:   32
[ 28] Pulse:   25, Gap:   58, Period:   83
[ 29] Pulse:   61, Gap:   56, Period:  117
[ 30] Pulse:   59, Gap:   24, Period:   83
[ 31] Pulse:    2, Gap:   31, Period:   33
[ 32] Pulse:   57, Gap:    4, Period:   61
[ 33] Pulse:    1, Gap:   55, Period:   56
[ 34] Pulse:   58, Gap:   57, Period:  115
[ 35] Pulse:   60, Gap:   57, Period:  117
[ 36] Pulse:   59, Gap:   57, Period:  116
[ 37] Pulse:   60, Gap:   56, Period:  116
[ 38] Pulse:   45, Gap:    1, Period:   46
[ 39] Pulse:   12, Gap:   58, Period:   70
[ 40] Pulse:   59, Gap:   24, Period:   83
[ 41] Pulse:    1, Gap:   32, Period:   33
[ 42] Pulse:   59, Gap:   72, Period:  131
[ 43] Pulse:    1, Gap:   31, Period:   32
[ 44] Pulse:    1, Gap:    7, Period:    8
[ 45] Pulse:    2, Gap:   14, Period:   16
[ 46] Pulse:    2, Gap:    7, Period:    9
[ 47] Pulse:    1, Gap:   15, Period:   16
[ 48] Pulse:    2, Gap:    7, Period:    9
[ 49] Pulse:    1, Gap:    9, Period:   10
[ 50] Pulse:   59, Gap:   57, Period:  116
[ 51] Pulse:  106, Gap:    1, Period:  107
[ 52] Pulse:    8, Gap:   58, Period:   66
[ 53] Pulse:  177, Gap:   58, Period:  235
[ 54] Pulse:   59, Gap:   58, Period:  117
[ 55] Pulse:   57, Gap:  231, Period:  288
[ 56] Pulse:   58, Gap:  116, Period:  174
[ 57] Pulse:   58, Gap:  292, Period:  350
[ 58] Pulse:   60, Gap:  335, Period:  395
[... 179 lines cut ...]
[238] Pulse:    4, Gap:    1, Period:    5
[239] Pulse:    3, Gap:   11, Period:   14
[240] Pulse:    2, Gap:    5, Period:    7
[241] Pulse:    2, Gap:    1, Period:    3
[242] Pulse:    2, Gap:    1, Period:    3
[243] Pulse:    8, Gap:    2, Period:   10
[244] Pulse:    4, Gap:   10, Period:   14
[245] Pulse:    5, Gap:   34, Period:   39
[246] Pulse:    2, Gap:    5, Period:    7
[247] Pulse:    2, Gap:    5, Period:    7
[248] Pulse:    3, Gap:   29, Period:   32
[249] Pulse:    2, Gap:    3, Period:    5
[250] Pulse:    4, Gap:    4, Period:    8
[251] Pulse:    1, Gap:    2, Period:    3
[252] Pulse:    1, Gap:    1, Period:    2
[253] Pulse:    7, Gap:    3, Period:   10
[254] Pulse:    5, Gap:    3, Period:    8
[255] Pulse:    5, Gap:    1, Period:    6
[256] Pulse:    5, Gap:    5, Period:   10
[257] Pulse:    2, Gap:    5, Period:    7
[258] Pulse:    3, Gap:   42, Period:   45
[259] Pulse:    2, Gap:    5, Period:    7
[260] Pulse:    2, Gap:    5, Period:    7
[261] Pulse:    1, Gap:    2, Period:    3
```

matthias-bs, 2023-03-07:
Never mind, this is not a dumb question!

We need the following information (data for Bresser in braces):
1. carrier frequency:                   (868.3 MHz)
2. bit rate:                                     (8.22 kbps)
3. frequency deviation:              (57.136417 kHz)
4. Rx bandwidth:                        (270.0 kHz (CC1101) / 250 kHz (SX1276))
5. output power:                        (10 dBm)
6. preamble length:                   (40 bits)

Let's start some guesswork:
1. You already set 868.3 MHz and got correct results, so this is the required frequency
2. According to the rtl_433 output, the bit width is 58.00 µs (57.67 µs)
    with 1 second = 1,000,000 µs there 1,000,000 / 58.0 (or 57.67) = 17,241.4 ( or 17,340.0) bits per second
   You have to find the nearest matching data rate in the CC1101 data sheet (Section 12. "Data Rate Programming")
3. not sure - how about https://triq.org/rtl_433/OPERATION.html#analysis?
4. not sure - how about https://triq.org/rtl_433/OPERATION.html#analysis?
5. don't care, we only want to receive  
6. same as above, same sync word (as you already found)

I was lucky to be able to build upon other people's work! ;-)

Here is an excel table with some CC1101 register settings as a reference: https://github.com/RFD-FHEM/RFFHEM/issues/607#issuecomment-889429946

matthias-bs, 2023-03-07:
Maybe it helps to check out https://github.com/NorthernMan54/rtl_433_ESP. Currently only 433 MHz devices with OOK modulation seem to be supported, but maybe someone has done a fork for something which could help.

matthias-bs, 2023-03-07:
Looking at the spectrogram, I'd say the frequency deviation is something like 40 kHz, but I do not have any experience with this. Maybe it's best to ask in the rtl_433 forum.

matthias-bs, 2023-03-07:
Please give `rtl_433 -A` (maybe constrained to 868.3 MHz if possible) a try.

fgnm, 2023-03-07:
Many thanks for all your explaination.. This is the output of `rtl_433 -f 868.3M -A`, it arrives each 16sec so it's the WS for sure
```
Detected FSK package	2023-03-07 13:30:39
Analyzing pulses...
Total count:  137,  width: 11.99 ms		(11986 S)
Pulse width distribution:
 [ 0] count:    1,  width:    0 us [0;0]	(   0 S)
 [ 1] count:   28,  width:    2 us [2;2]	(   2 S)
 [ 2] count:    4,  width:    3 us [3;3]	(   3 S)
 [ 3] count:    5,  width:   19 us [18;20]	(  19 S)
 [ 4] count:    5,  width:   40 us [37;46]	(  40 S)
 [ 5] count:   30,  width:   58 us [50;63]	(  58 S)
 [ 6] count:   10,  width:   11 us [9;12]	(  11 S)
 [ 7] count:    5,  width:   28 us [25;31]	(  28 S)
 [ 8] count:   29,  width:    1 us [1;1]	(   1 S)
 [ 9] count:    8,  width:    5 us [5;6]	(   5 S)
 [10] count:    4,  width:  118 us [117;120]	( 118 S)
 [11] count:    1,  width:  172 us [172;172]	( 172 S)
 [12] count:    1,  width:  229 us [229;229]	( 229 S)
 [13] count:    2,  width:    7 us [7;7]	(   7 S)
 [14] count:    1,  width:  291 us [291;291]	( 291 S)
 [15] count:    3,  width:    4 us [4;4]	(   4 S)
Gap width distribution:
 [ 0] count:    5,  width:    2 us [2;2]	(   2 S)
 [ 1] count:   11,  width:    4 us [4;4]	(   4 S)
 [ 2] count:    8,  width:    5 us [5;6]	(   5 S)
 [ 3] count:    6,  width:    3 us [3;3]	(   3 S)
 [ 4] count:   27,  width:   57 us [49;62]	(  57 S)
 [ 5] count:   22,  width:    1 us [1;1]	(   1 S)
 [ 6] count:    7,  width:   38 us [33;44]	(  38 S)
 [ 7] count:    8,  width:   24 us [21;28]	(  24 S)
 [ 8] count:    8,  width:   14 us [13;17]	(  14 S)
 [ 9] count:    7,  width:  160 us [131;175]	( 160 S)
 [10] count:    7,  width:    8 us [7;9]	(   8 S)
 [11] count:    3,  width:   74 us [74;74]	(  74 S)
 [12] count:    8,  width:  111 us [98;117]	( 111 S)
 [13] count:    5,  width:  292 us [273;323]	( 292 S)
 [14] count:    2,  width: 1039 us [960;1118]	(1039 S)
Pulse period distribution:
 [ 0] count:    1,  width:    2 us [2;2]	(   2 S)
 [ 1] count:   15,  width:    5 us [5;6]	(   5 S)
 [ 2] count:   12,  width:    7 us [7;9]	(   7 S)
 [ 3] count:    8,  width:   66 us [55;75]	(  66 S)
 [ 4] count:   13,  width:   18 us [16;22]	(  18 S)
 [ 5] count:   26,  width:  110 us [85;132]	( 110 S)
 [ 6] count:   14,  width:   11 us [10;14]	(  11 S)
 [ 7] count:    8,  width:   43 us [37;51]	(  43 S)
 [ 8] count:    9,  width:   27 us [25;30]	(  27 S)
 [ 9] count:    8,  width:  168 us [153;176]	( 168 S)
 [10] count:    7,  width:  242 us [216;291]	( 242 S)
 [11] count:    4,  width:  333 us [287;349]	( 333 S)
 [12] count:    3,  width:    3 us [3;3]	(   3 S)
 [13] count:    2,  width: 1040 us [961;1119]	(1040 S)
 [14] count:    2,  width:  453 us [441;465]	( 453 S)
Pulse timing distribution:
 [ 0] count:    1,  width:    0 us [0;0]	(   0 S)
 [ 1] count:   34,  width:    2 us [2;2]	(   2 S)
 [ 2] count:   10,  width:    3 us [3;3]	(   3 S)
 [ 3] count:   10,  width:   19 us [17;23]	(  19 S)
 [ 4] count:   13,  width:   40 us [33;49]	(  40 S)
 [ 5] count:   56,  width:   57 us [49;63]	(  57 S)
 [ 6] count:   13,  width:   11 us [9;13]	(  11 S)
 [ 7] count:   10,  width:   27 us [24;31]	(  27 S)
 [ 8] count:   51,  width:    1 us [1;1]	(   1 S)
 [ 9] count:   16,  width:    5 us [5;6]	(   5 S)
 [10] count:   13,  width:  115 us [98;131]	( 115 S)
 [11] count:    7,  width:  166 us [152;175]	( 166 S)
 [12] count:    8,  width:  276 us [229;323]	( 276 S)
 [13] count:    7,  width:    7 us [7;8]	(   7 S)
 [14] count:   14,  width:    4 us [4;4]	(   4 S)
[... 5 lines cut ...]
Use a flex decoder with -X 'n=name,m=FSK_PCM,s=1,l=1,r=1024'
pulse_demod_pcm(): Analyzer Device
bitbuffer:: Number of rows: 1 
[00] {8569} 30 e0 c6 06 18 60 63 00 00 00 00 00 00 00 03 ff ff bf ff ff ff fe 00 00 00 00 00 00 00 ff ff ff ff ff ff ff e0 00 00 00 00 00 00 07 ff 7f ff ff ff ff fe 00 00 00 00 00 00 00 7f ff ff ff ff ff ff f8 00 00 00 00 00 00 00 ff ef ff 3f ff ff ff 00 00 00 00 00 00 00 1f ff ff ff ff fd ff e0 00 00 00 00 00 00 03 ff ff ff ef ff ff fe 00 00 00 00 00 00 00 7f ff ff ff ff ff ff f0 00 00 00 00 00 00 00 7f ff ff ff ff ff ff 80 00 00 00 00 00 00 7f ff ff ff ff ff ff ff 00 00 00 00 00 00 01 ff ff df ff ff ff ff c0 00 00 00 00 00 00 3f ff ff ff ff ff ff f0 00 00 00 04 00 00 03 ff ff ff ff ff ff ff 00 00 00 00 00 00 00 3f ff ff ff ff ff ff f8 00 00 00 00 00 00 00 ff ff ff ff ff ff ff 00 06 00 00 00 00 00 1f ff ff f7 ff ff ff f0 00 00 00 00 00 00 03 ff ff ff ff ff ff ff 00 00 00 00 00 00 00 3f ff ff ff ff ff ff 7c 00 00 00 00 00 00 03 ff ff ff ff ff ff ff c0 00 00 00 00 00 00 0f ff ff ff ff ff ff fc 00 00 00 00 00 00 01 ff ff ff ff f7 ff ff 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff ff ff ff ff ff 80 00 00 00 00 00 00 1f ff ff ff ff ff ff ff ff ff ff ff ff ff ff 00 00 00 00 00 00 00 1f ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff fe 00 00 00 00 00 00 80 7f ff ff ff ff ff ff f8 00 00 00 00 00 00 07 ff ff ff ff ff ff ff 80 00 00 00 00 20 00 00 00 00 00 00 00 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00 80 7f ff ff ff ff ff ff f8 00 00 00 00 00 02 00 03 00 00 00 00 00 0f ff ff ff ff ff ff fe 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7f ff ff ff ff ff ff f0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 07 ff ff ff ff ff ff ff ff ff ff ff ff ff ff f8 00 00 00 00 00 00 00 00 00 00 00 00 00 07 ff ff ff ff ff ff ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 1f ff ff  [... cut ...]
 [ 0] count:    1,  width:   76 us [76;76]	(  76 S)
Gap width distribution:
Pulse period distribution:
Pulse timing distribution:
 [ 0] count:    1,  width:   76 us [76;76]	(  76 S)
 [ 1] count:    1,  width: 10001 us [10001;10001]	(10001 S)
Level estimates [high, low]:   1487,     12
RSSI: -10.4 dB SNR: 20.9 dB Noise: -31.4 dB
Frequency offsets [F1, F2]:     -48,      0	(-0.7 kHz, +0.0 kHz)
Guessing modulation: Single pulse detected. Probably Frequency Shift Keying or just noise...
view at https://triq.org/pdv/#AAB102004C27118155

```

fgnm, 2023-03-07:
Tried just a simple test using `int state = radio.begin(868.3, 17.24, 40, 270, 10, 32);`, now it seems recognize correctly the sync word, I can see in the debug output:
```
[CC1101] Data: D4 24 10 64 82 7D 22 00 00 00 00 00 03 00 10 A4 E4 54 62 D5 8F 3F FD F6 BF CB FD 
[CC1101] R [D4] RSSI: -68.5
Digest check failed - [8EBA] Vs [91E0] (1F5A)
```

So of course the decoder is failing because isn't the same of the bresser, but I could try to port the RTL_433 for the WH68B in the same format you did for the bresser

matthias-bs, 2023-03-07:
Woohoo! Excellent!

fgnm, 2023-03-07:
Well more or less I got it working (copied the code from RTL_433), in the following days I'll try to make a clean implementation of it. Meanwhile, I've also tried to get the same result with an Heltec ESP32 WiFi Lora V2, which should have an SX1276 built in module, with the following parameters `radio.beginFSK(868.3, 17.24, 40, 250, 14, 32);` it doesn't receive data but I'm still unsure if the issue is HW or SW, I'll update.

Anyway, thank you so much for all your support and explanation, I didn't expect this was even possible :)

matthias-bs, 2023-03-09:
Which board configuration are you using? Did you check that the pin configuration is correct for the Heltec ESP32 WiFi Lora V2?

matthias-bs, 2023-03-15:
Any update on this?

fgnm, 2023-03-16:
I tried with Heltec ESP32 WiFi Lora V2 in many ways but wasn't able to succeed, so I tried with another ESP32 with an SX1276 sensor but still unable to receive any data.. I mean to code itself works and I can read some signals but are not the one from the WS... Finally I ends up using the CC1101 and ESP32 which works pretty well with parameters posted previosly.
The SX1276 remains a mistery for me.. I'm really noob in this field, but I can't see any clear reason why if the frequencies are the same one sensor works and the other no :/

closed 2023-03-26 by matthias-bs