matthias-bs/BresserWeatherSensorReceiver #42: Weather station 7002531

Jerome67, 2023-03-03:
Hi, 
i'm trying to use your code with a **WeMos (non-official) D1 mini with CC110**1.
My weather station is 7002531 and i try to get external sensor data ( _temperature, humidity and wind_ ).
But impossible to launch, an error appear in the serial monitor. After some search on your Github page, another guy have the same problem ( _solution was change the power supply 5V USB : [https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/12](url) ). 
But in my case i want to monitor data and send it on my website with GET method.

Have you some solution to test CC1101 receptor with my bresser weather station external sensor ?
Thanks

**This is my monitor message :** 

`Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffffdd0 end: 3fffffc0 offset: 01a0
3fffff70:  3ffee6f8 00000000 3ffee734 4020124d  
3fffff80:  00000020 00000000 3ffeea38 40203d95  
3fffff90:  3fffdad0 00000000 3ffeea38 40201051  
3fffffa0:  feefeffe feefeffe 3ffeea8c 40204590  
3fffffb0:  feefeffe feefeffe 3ffe85d8 40101281  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v00045ac0
~ld
`

matthias-bs, 2023-03-03:
Hi,

this is most likely a wiring problem between the D1 mini and the CC1101 module. Could you please enable debug output (see https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/DEBUG_OUTPUT.md)?

Regards
Matthias

Jerome67, 2023-03-04:
Thanks for your fast reply.
This is the message i got when debug is ON

`--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffffdb0 end: 3fffffc0 offset: 01a0
3fffff50:  3ffee9e4 3ffe8972 3ffeece8 402012f9  
3fffff60:  00000020 feefeffe feefeffe feefeffe  
3fffff70:  00000000 402061d8 00000000 00000000  
3fffff80:  0001c200 00000000 3ffeece8 3ffeed50  
3fffff90:  3fffdad0 00000000 3ffeece8 40201051  
3fffffa0:  feefeffe feefeffe 3ffeed3c 40204aa8  
3fffffb0:  feefeffe feefeffe 3ffe85d8 401015c9  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v00046800
~ld

SDK:2.2.2-dev(38a443e)/Core:3.0.1=30001000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:c0b69df
[CC1101] Initializing ... 
[CC1101] Error initialising: [-2]`

matthias-bs, 2023-03-04:
No that helpful... Could you please paste the compiler output here?

I think it's a wiring problem or a hardware defect. Did the CC1101 module already work in another setup? I already had one module which was dead on arrival...

Jerome67, 2023-03-04:
I just try with another mini D1 and it's the same.
I'm trying with another ESP8266 NodeMCU.
How i can test if the CC1101 module working correctly ?

This is the compiler output : 
`In file included from C:\Users\Jérôme\Desktop\DOMOTIQUE\BresserWeatherSensorReceiver-main\examples\BresserWeatherSensorBasic\BresserWeatherSensorBasic.ino:54:
C:\Users\J�r�me\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src/WeatherSensorCfg.h:312:48: note: '#pragma message: Receiver chip: [CC1101]'
  312 | #pragma message("Receiver chip: " RECEIVER_CHIP)
      |                                                ^
In file included from C:\Users\Jérôme\Desktop\DOMOTIQUE\BresserWeatherSensorReceiver-main\examples\BresserWeatherSensorBasic\BresserWeatherSensorBasic.ino:54:
C:\Users\J�r�me\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src/WeatherSensorCfg.h:313:170: note: '#pragma message: Pin config: RST->2, CS->15, GD0/G0/IRQ->4, GDO2/G1/GPIO->5'
  313 | #pragma message("Pin config: RST->" STR(PIN_RECEIVER_RST) ", CS->" STR(PIN_RECEIVER_CS) ", GD0/G0/IRQ->" STR(PIN_RECEIVER_IRQ) ", GDO2/G1/GPIO->" STR(PIN_RECEIVER_GPIO) )
      |                                                                                                                                                                          ^
In file included from C:\Users\J�r�me\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src\WeatherSensor.cpp:62:
C:\Users\J�r�me\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src\WeatherSensorCfg.h:312:48: note: '#pragma message: Receiver chip: [CC1101]'
  312 | #pragma message("Receiver chip: " RECEIVER_CHIP)
      |                                                ^
In file included from C:\Users\J�r�me\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src\WeatherSensor.cpp:62:
C:\Users\J�r�me\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src\WeatherSensorCfg.h:313:170: note: '#pragma message: Pin config: RST->2, CS->15, GD0/G0/IRQ->4, GDO2/G1/GPIO->5'
  313 | #pragma message("Pin config: RST->" STR(PIN_RECEIVER_RST) ", CS->" STR(PIN_RECEIVER_CS) ", GD0/G0/IRQ->" STR(PIN_RECEIVER_IRQ) ", GDO2/G1/GPIO->" STR(PIN_RECEIVER_GPIO) )
      |                                                                                                                                                                          ^
Executable segment sizes:
ICACHE : 32768           - flash instruction cache 
IROM   : 251384          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 29581   / 32768 - code in IRAM          (IRAM_ATTR, ISRs...) 
DATA   : 1496  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 2152  ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 26744 )         - zeroed variables      (global, static) in RAM/HEAP 
Le croquis utilise 284613 octets (27%) de l'espace de stockage de programmes. Le maximum est de 1044464 octets.
Les variables globales utilisent 30392 octets (37%) de mémoire dynamique, ce qui laisse 51528 octets pour les variables locales. Le maximum est de 81920 octets.
esptool.py v3.0
Serial port COM3
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 50:02:91:e7:6f:da
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 288768 bytes to 211253...
Writing at 0x00000000... (7 %)
Writing at 0x00004000... (15 %)
Writing at 0x00008000... (23 %)
Writing at 0x0000c000... (30 %)
Writing at 0x00010000... (38 %)
Writing at 0x00014000... (46 %)
Writing at 0x00018000... (53 %)
Writing at 0x0001c000... (61 %)
Writing at 0x00020000... (69 %)
Writing at 0x00024000... (76 %)
Writing at 0x00028000... (84 %)
Writing at 0x0002c000... (92 %)
Writing at 0x00030000... (100 %)
Wrote 288768 bytes (211253 compressed) at 0x00000000 in 7.0 seconds (effective 331.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin..`

matthias-bs, 2023-03-04:
Did you cross-check that the connection assumed by the compiler matches your wiring?
RST->2
CS->15
GD0/G0/IRQ->4
GDO2/G1/GPIO->5

matthias-bs, 2023-03-04:
Apart from that, nothing suspicious in the compiler output.

Jerome67, 2023-03-04:
I just test with a real ESP8266 NodeMcu, it's exactly the same. I think the cc1101 module is down. 
I double (triple) check pin :/

Jerome67, 2023-03-04:
I just test with multimetre the power vcc, it's ok. 
The vcc power is 3.3V, the D2 pin 1.66V, D8 pin 3.3V, all the others 0.00V

matthias-bs, 2023-03-04:
Is D2 used for CC1101 reset? 1.66V is neither high nor low... Maybe it toggles between both states.
You could try to disconnect it, because there is also a software reset via SPI (I think).

Jerome67, 2023-03-05:
I haven't reset pin with my CC1101. D2 is for SDA pin.
The numbers defined in configuration file is about GPIO number(e.g GPIO3) or digital pin number (e.g D1...)?

matthias-bs, 2023-03-05:
The numbers are the GPIO numbers. See https://github.com/esp8266/Arduino/blob/master/variants/ for board specific mappings.

Jerome67, 2023-03-06:
Thanks for the link, very useful information.
It was correctly configured with the correct GPIO numbers.
I ordered a new CC1101 module, I will wait to receive it to be sure

matthias-bs, 2023-03-06:
I hope it works then.

matthias-bs, 2023-03-15:
Thanks for your feedback!

Jerome67, 2023-03-15:
I open the ticket but for another question. 
I try to insert the received data into my variables in order to send them every 5 minutes via GET method but nothing is present in the variables after receiving data. 
How can i do this ?

matthias-bs, 2023-03-15:
Sorry, I don't have a clear picture of what you are doing. Would you mind sharing your code?

Jerome67, 2023-03-15:
i want to use "my_var_humidity" in the loopCallback in order to sent it in GET method ( code for sent is written and work - but the my_var_humidity is empty )

`if (weatherSensor.sensor[i].humidity_ok) {
                Serial.printf("Hum: [%3d%%] ",
                    weatherSensor.sensor[i].humidity);
my_var_humidity = weatherSensor.sensor[i].humidity;
            }`

matthias-bs, 2023-03-15:
It depends on the scope of your variable, i.e. where your variable is declared. If it is inside the callback function, is an 'automatic' variable (contents only valid during execution of the function). If it is a global variable (declared outside any function), its value is retained at any time.
If this doesn't help, I'd need some more context of the code.

Jerome67, 2023-03-16:
Ok this is the solution, it wasn't a global variable.
It seems good, working all the night, all the data was receive and add on my database. 
This morning, temp. is negative outside, the data contain -99.6°C. 
I see another issue with this problem with a fix. Have you fix on the stable version or i must modify the file as specified on the issue solution?

Jerome67, 2023-03-16:
[image]

[image]

 This is my last data in database. 
The type is float for my variable i use to store in database

matthias-bs, 2023-03-16:
There are some things between the library and your database which I don't know about. I have both a 5-in-1 and a 6-in-1 sensor running all winter long without an issue like this (but not with the example sketch, though). And no one else has reported an issue like this.

Jerome67, 2023-03-16:
I just check, it's not a database error. The -99.90, -99.70, -99.50... etc value coming from arduino code. How i can check raw data ? 
(-99.50 => -0.5°c )

Jerome67, 2023-03-16:
I use the callback file, i check the real values tonight.
The wind data from the sensor is m/s or km/h in raw data ?

Jerome67, 2023-03-16:
> If you shared your code on Github, maybe I would be able to give you an advice. Or if you could provide some output from the serial console with debug level set...

I share it soon, thanks

matthias-bs, 2023-03-16:
> I use the callback file, i check the real values tonight. The wind data from the sensor is m/s or km/h in raw data ?

m/s

Jerome67, 2023-03-16:
This is my code : [https://github.com/Jerome67/Arduino_test/blob/main/Bresser_GET](url)
Actually it's working fine ( outside temperature is positive ).
Perhaps a problem with float variable ? I'm most confortable with PHP/MySQL, who don't need to declare specific type of every variable
Every 5 minutes, data from bresser sensor (outside) and DHT22 (inside) are sent on my website with GET method ( and save into MySQL database )

matthias-bs, 2023-03-17:
O.k., I'd suggest to split the problem in two parts:
1. Do you get the correct values from the external sensor?
2. Can you pass negative values with the HTTP Get command correctly?

For 1), you might want to check the printf() output from the main loop.

For 2), I'd test it by setting the variables to some test value instead of waiting for a negative sensor value.
    I'm not familiar with the HTTP client, but I would test the following:
    1. Do an explicit typecast of the float variable, i.e. `String serverPath = serverName + "?db=exterieur&temp_ext="+String(temp_ext)+...`
    2. Add double quotes around the parameter, i.e. `temp_ext=\""+String(temp_ext)+"\""+...`

matthias-bs, 2023-03-17:
By the way, you are using the callback in a way that was not intended and might lead to problems.

`weatherSensor.getData()` tries to receive data from the sensor for a certain time (60s in your example). This is required, because the sensors only transmit at certain intervals (like e.g. 12 to 60 seconds) and maybe some retries are needed (due to transmission errors). For this time, execution blocks, which might lead to problems like an MQTT protocol timeout or no response to a user interaction (like a key press). In those cases, the callback can be used to take some action (like servicing the MQTT protocol) or aborting the loop to take immediate action. In any case, the execution time of the callback should be as short as possible.

I would rather move your entire code block commented `// PARTIE GET` to the main loop. And I'd create both the WiFiClient and HTTPClient objects at global scope (i.e. outside of all function functions). This saves execution time and energy (In your case, the objects are created at the beginning and deleted at the end of your block [block means everything between `{}`]). Please see [BresserWeatherSensorMQTT.ino](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/examples/BresserWeatherSensorMQTT/BresserWeatherSensorMQTT.ino) for a similar implementation.

matthias-bs, 2023-03-22:
Maybe the data from this sensor is different from the ones which have used up to now. Could you send some raw data, please?

Jerome67, 2023-03-22:
You need raw data when temperature is under 0°c ? Because now the winter is finish. 
What type of raw data you need ? The variable or raw decode data from cc1101 ?

matthias-bs, 2023-03-22:
Yes, raw data as provided by the receiver (before decoding) of temperatures below zero would be helpful.

matthias-bs, 2023-03-26:
Compile with "Debug Level: Verbose", run the sketch and copy output from the serial console

marxu, 2023-03-30:
Hello jerome67. 
Could you tell me the purchase link of the CC1101 that works for you?
I would like to test with one that I have. Mod. 7003100CM3000. And share the experience here.

matthias-bs, 2023-03-30:
This one works for me: https://www.amazon.de/gp/product/B075PFQ57G/
Please notice that the header pins have a 1.27mm pitch.

Jerome67, 2023-03-30:
> 

I order this one [AliExpress cc1101 868mhz](https://fr.aliexpress.com/item/1005002603656188.html?spm=a2g0o.productlist.main.1.1fba4668iElAY9&algo_pvid=6cf1c6a9-56a8-4cc7-82d5-38f70c14511d&algo_exp_id=6cf1c6a9-56a8-4cc7-82d5-38f70c14511d-0&pdp_ext_f=%7B%22sku_id%22%3A%2212000021352867203%22%7D&pdp_npi=3%40dis%21EUR%211.98%211.74%21%21%21%21%21%402102176616801616985127009d0767%2112000021352867203%21sea%21FR%214067058457&curPageLogUid=MFvB7JrykNeX)
It's a 2.00mm pitch, don't forget to order 2.00mm header or solder directly on the board

Jerome67, 2023-04-05:
> Compile with "Debug Level: Verbose", run the sketch and copy output from the serial console.

Hi, i try today Debug Level, this is what you need ?

`[CC1101] Data: D4 7C 5A 2C 10 05 12 18 FE 67 FE 00 08 03 56 70 FF F0 71 00 00 00 00 00 00 00 00 
08:58:48.024 -> [CC1101] R [D4] RSSI: -74.5
08:58:48.024 -> find_slot(): ID=2C100512
08:58:48.024 -> find_slot(): Storing into slot #0
08:58:48.024 -> Temp: 1  Hum: 1  UV: 1  Wind: 1  Rain: 0  Moist: 0
08:58:48.024 -> Id: [2C100512] Typ: [1] Battery: [OK ] Ch: [0] Temp: [  3.5C] Hum: [ 70%] Wind max: [ 1.9m/s] Wind avg: [ 1.8m/s] Wind dir: [  0.0deg] Rain: [-----.-mm] Moisture: [--%] RSSI: [-74.5dBm]`

If you need this kind of data, i can store it into a file when temperature is negative ( tonight )

matthias-bs, 2023-04-05:
Yes, exactly! That's good!

Jerome67, 2023-04-05:
Is it possible to store raw data into a variable ? 

(Debug/Raw informations are provided by RadioLib library ? )

matthias-bs, 2023-04-05:
Thank you! I will look into this as soon as my time permits.

matthias-bs, 2023-04-10:
Hi,
I checked the raw data you provided - yes, the decoding function decodeBresser6In1Payload() returns wrong values for your type of sensor. Do you still have the correct values (i.e. as displayed by the original weather station) matching the raw data you placed on github?

It seems, that the changes I made on January 14 (https://github.com/matthias-bs/BresserWeatherSensorReceiver/commit/f757b6c38995f1e67670149b35bd15b9b3ecff87) work for some types of sensors, but break things for yours.

I'll create a branch with reverting those modifications for testing. If it works for you, a common solution for all types of sensors handled by this decoder has to be found.
 
Regards
Matthias

matthias-bs, 2023-04-10:
The affected device is [BRESSER 3-in-1 Professional Wind Gauge / Anemometer](https://www.bresser.de/en/Brand/Bresser/BRESSER-3-in-1-Professional-Wind-Gauge-Anemometer.html).

matthias-bs, 2023-04-10:
I changed the decoder function back to the formula used in https://github.com/merbanan/rtl_433/blob/master/src/devices/bresser_6in1.c, which - according to the comments - should work for the 7002531.
Could you please check if this works for you?
Please see https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/workaround_7002531_3in1_wind_gauge/src/WeatherSensor.cpp

matthias-bs, 2023-04-11:
Please also use the other source files from the branch "workaround_7002531_3in1_wind_gauge".

Jerome67, 2023-04-11:
Ok indeed it's my fault, there was a duplicate library

This is now what i have with negative temperature
The temperature is correct but the minus sign (-) don't appear

[Raw data 11/04/2023](https://github.com/Jerome67/home/blob/main/raw_data_11042023)

matthias-bs, 2023-04-11:
Well, according to https://github.com/merbanan/rtl_433/issues/1587, the decoder should have worked for your sensor, but it seems it has never been tested with negative temperatures.
I've updated https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/workaround_7002531_3in1_wind_gauge/src/WeatherSensor.cpp, please give it another try.

Jerome67, 2023-04-12:
Ok it seems good now, here the [raw data](https://github.com/Jerome67/home/blob/main/raw_data_12042023) for the last update

matthias-bs, 2023-04-12:
Thanks for the feedback and providing data again. I'd like to open an issue in the rtl_433 project. May I use the raw data you recorded for this purpose?

I didn't find a way to distinguish between your sensor and other sensors using the same decoder automatically. As a workaround, I would add a define `SENSOR_IDS_DECODE3IN1` to [src/WeatherSensorCfg.h](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/src/WeatherSensorCfg.h). The user would have to provide the IDs of all sensors of this type, otherwise the standard decoding algorithm will be used.

matthias-bs, 2023-04-12:
Could you please check this again?
https://github.com/matthias-bs/BresserWeatherSensorReceiver/tree/workaround_7002531_3in1_wind_gauge/src
Please note that 3 files have been changed.

Jerome67, 2023-04-12:
Yes, of course you can use my raw data on an issue in the rtl_433 project

This is some simple [raw data](https://github.com/Jerome67/home/blob/main/raw_data_12042023_2) with the 3 new files.
Do you need data with negative values ?

matthias-bs, 2023-04-13:
Yes, negative values would be great. Please note that I just made another minor modification.

Jerome67, 2023-04-13:
ok, I'll do after holiday, next week

Jerome67, 2023-04-25:
Hi, this is the new [raw data](https://github.com/Jerome67/home/blob/main/raw_data_25042023) with the last file updated

matthias-bs, 2023-04-26:
Thanks again!

Jerome67, 2023-04-28:
In addition, I think this document may be useful to others. This is the wiring instruction on a nodemcu esp8266 that i made : 
[ESP8266 NodeMCU + CC1101](https://github.com/Jerome67/home/blob/main/ESP8266_NodeMCU_cc1101.pdf)

matthias-bs, 2023-05-08:
Hi,

Is there anything else I can do or can this issue be closed?
Unfortunately I did not find the time to post an issue regarding the protocol differences in the rtl_433 project, yet....

Cheers
Matthias

Jerome67, 2023-05-09:
Hi,
All seem's ok now, this issue can be closed. Thanks a lot !

closed 2023-05-09 by Jerome67