matthias-bs/BresserWeatherSensorReceiver #43: Help: Bresser 9602510, D1 mini, CC1101

stundenblume, 2023-03-06:
Hi,
I would like to use your software to integrate a Bresser 9602510 (5in1) sensor via MQTT into my system.
I can't get it to work. I already stopped after "Starting execution..."

I followed your [instructions](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/DEBUG_OUTPUT.md). Modified WeatherSensorCfg.h (#define USE_CC1101) and compiled  BresserWeatherSensorBasic without issues.
Wiring is according this [diagram](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/ESP8266_WeMos_D1-Mini_pinout_cc1101_v2.pdf).
The original receiver gets data.
I assuming the sensor is broadcasting and don't have do be paired.
I assuming I do not have to provide the sensor type.

Can I check somehow if the CD1101 is working/wired correctly?

Thanks for your help,
Sebastian

**Compile Output:**
```
In file included from /home/setz/projekte/sketchbook/BresserWeatherSensorBasic/BresserWeatherSensorBasic.ino:54:
/home/setz/projekte/sketchbook/libraries/BresserWeatherSensorReceiver/src/WeatherSensorCfg.h:336:48: note: '#pragma message: Receiver chip: [CC1101]'
  336 | #pragma message("Receiver chip: " RECEIVER_CHIP)
      |                                                ^
/home/setz/projekte/sketchbook/libraries/BresserWeatherSensorReceiver/src/WeatherSensorCfg.h:337:170: note: '#pragma message: Pin config: RST->2, CS->15, GD0/G0/IRQ->4, GDO2/G1/GPIO->5'
  337 | #pragma message("Pin config: RST->" STR(PIN_RECEIVER_RST) ", CS->" STR(PIN_RECEIVER_CS) ", GD0/G0/IRQ->" STR(PIN_RECEIVER_IRQ) ", GDO2/G1/GPIO->" STR(PIN_RECEIVER_GPIO) )
      |                                                                                                                                                                          ^
. Variables and constants in RAM (global, static), used 29896 / 80192 bytes (37%)
║   SEGMENT  BYTES    DESCRIPTION
╠══ DATA     1496     initialized variables
╠══ RODATA   1584     constants       
╚══ BSS      26816    zeroed variables
. Instruction RAM (IRAM_ATTR, ICACHE_RAM_ATTR), used 61719 / 65536 bytes (94%)
║   SEGMENT  BYTES    DESCRIPTION
╠══ ICACHE   32768    reserved space for flash instruction cache
╚══ IRAM     28951    code in IRAM    
. Code in flash (default, ICACHE_FLASH_ATTR), used 249508 / 1048576 bytes (23%)
║   SEGMENT  BYTES    DESCRIPTION
╚══ IROM     249508   code in flash   
esptool.py v3.0
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 84:cc:a8:b1:e1:87
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 285696 bytes to 209752...
Writing at 0x00000000... (7 %)
Writing at 0x00004000... (15 %)
Writing at 0x00008000... (23 %)
Writing at 0x0000c000... (30 %)
Writing at 0x00010000... (38 %)
Writing at 0x00014000... (46 %)
Writing at 0x00018000... (53 %)
Writing at 0x0001c000... (61 %)
Writing at 0x00020000... (69 %)
Writing at 0x00024000... (76 %)
Writing at 0x00028000... (84 %)
Writing at 0x0002c000... (92 %)
Writing at 0x00030000... (100 %)
Wrote 285696 bytes (209752 compressed) at 0x00000000 in 4.8 seconds (effective 480.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

**Serial Output:**
```
sdl��|�d�|�$�c|����s�b�#��og�lng���c8��dsdsdp�g��d��b'�|���c��'o�$��l �ngl o;Ǜ�'#l�$p�'�{������cn�<�#��og�l �gnd`g{���gc��`{��gc��`��{l`��o�dStarting execution...
```
[image]
[image]

stundenblume, 2023-03-06:
Yes

[image]

matthias-bs, 2023-03-06:
Hi,

> I assuming the sensor is broadcasting and don't have do be paired.

That's correct!

> I assuming I do not have to provide the sensor type.

That's correct. Are you sure about the part number 9602510? I could not find it on Bresser's website.

According to your photos, the wiring is correct - it's exactly the same as mine (except for the Wake/RST connection which is missing [that is not the cause of the problem])

Maybe the CC1101 module is broken. I had the same issue - only 2 out of 3 modules I ordered are working; this does not seem to be uncommon.

stundenblume, 2023-03-06:
Thank you, I will order a new module. Will have a look into SPI.

9602510 is correct.

[image]
[image]

matthias-bs, 2023-03-06:
Thank you for the confirmation of the part number!

I hope a new module will solve the problem!

stundenblume, 2023-03-11:
Hi,
I got the new modules and build a complete new item. With the same result. :-/
I assume I did something wrong.
Will delete and reinstall your library. If that doesn't help, I will have a look with a bus-pirate.
Can you maybe provide me a compiled version?
Thanks,
Sebastian

matthias-bs, 2023-03-11:
Sorry for that! Yes, I'll provide it tomorrow.

stundenblume, 2023-03-11:
No Problem! I'm happy than I can use your code!

stundenblume, 2023-03-11:
Got a little further:
After reinstalling your library and updating radiolib I get this error:
```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffffdd0 end: 3fffffd0 offset: 01a0
3fffff70:  3fffde60 00000000 3ffeea68 3ffeeae8  
3fffff80:  3ffee718 00000000 3ffee75c 4020127d  
3fffff90:  00000020 00000000 00000000 3ffeeae8  
3fffffa0:  3fffdad0 00000000 3ffeea68 40201058  
3fffffb0:  3fffdad0 00000000 3ffeeabc 402053fc  
3fffffc0:  feefeffe feefeffe 3fffdab0 40101409  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v00046b90
~ld
Starting execution...
```
It seems that that output is not generated by your code? Because it's before `Starting execution...`

matthias-bs, 2023-03-12:
No, the output is coming from the Arduino bootloader/operating system which actually runs your sketch. `Soft WDT reset` means there is a reset from the "Soft Watchdog". This means a task in the sketch is running too long before returning control to the OS which would service the watchdog. A possible cause could be an endless loop, which is sometimes used in libraries such as RadioLib if a fatal error occurred (i.e. HW configuration failed).

You can use an exception decoder to pinpoint the part of the code which caused the exception.

matthias-bs, 2023-03-12:
I am starting a new build with updated libraries now:
BresserWeatherSensorReceiver: v0.6.1
MQTT v2.5.1
RadioLib v5.7.0
ArduinoJson v6.20.1

- using Arduino IDE 2.0.4 with ESP8266 board package v3.1.1
- changing the following in ~/Arduino/libraries/BresserWeatherSensorReceiver/src/WeatherSensorReceiverCfg.h as follows:
```
// ------------------------------------------------------------------------------------------------
// --- Radio Transceiver ---
// ------------------------------------------------------------------------------------------------
// Select type of receiver module (if not yet defined based on the assumptions above)
#if ( !defined(USE_CC1101) && !defined(USE_SX1276) )
    #define USE_CC1101
    //#define USE_SX1276
#endif
    ```
    (uncommented USE_CC1101, commented USE_SX1276)
    - loading the sketch source code via File->Examples->BresserWeatherSensorReceiver->BresserWeatherSensorBasic
    - selecting the board "LOLIN(WEMOS) D1 R2 & mini"
    - selecting "Debug port: Serial"
    ```
- Build output:
    ```
Using board 'd1_mini' from platform in folder: /home/mp/.arduino15/packages/esp8266/hardware/esp8266/3.1.1
Using core 'esp8266' from platform in folder: /home/mp/.arduino15/packages/esp8266/hardware/esp8266/3.1.1
/home/mp/.arduino15/packages/esp8266/tools/python3/3.7.2-post1/python3 -I /home/mp/.arduino15/packages/esp8266/hardware/esp8266/3.1.1/tools/mkbuildoptglobals.py /tmp/.mount_arduindagvyJ/resources/app/node_modules/arduino-ide-extension/build 10607 /tmp/arduino/sketches/6B9DE325620D63245258CB1972B47822 /tmp/arduino/sketches/6B9DE325620D63245258CB1972B47822/core/build.opt /tmp/.arduinoIDE-unsaved2023212-30795-2ypj2f.95nnh/BresserWeatherSensorBasic/BresserWeatherSensorBasic.ino.globals.h /home/mp/.arduino15/packages/esp8266/hardware/esp8266/3.1.1/cores/esp8266/CommonHFile.h
Clean build, created dir /tmp/arduino/sketches/6B9DE325620D63245258CB1972B47822/core
Note: optional global include file '/tmp/.arduinoIDE-unsaved2023212-30795-2ypj2f.95nnh/BresserWeatherSensorBasic/BresserWeatherSensorBasic.ino.globals.h' does not exist.
  Read more at https://arduino-esp8266.readthedocs.io/en/latest/faq/a06-global-build-options.html
Detecting libraries used...
/home/mp/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -D_GNU_SOURCE -DESP8266 @/tmp/arduino/sketches/6B9DE325620D63245258CB1972B47822/core/build.opt -I/home/mp/.arduino15/packages/esp8266/hardware/esp8266/3.1.1/tools/sdk/include -I/home/mp/.arduino15/packages/esp8266/hardware/esp8266/3.1.1/tools/sdk/lwip2/include -I/home/mp/.arduino15/packages/esp8266/hardware/esp8266/3.1.1/tools/sdk/libc/xtensa-lx106-elf/include -I/tmp/arduino/sketches/6B9DE325620D63245258CB1972B47822/core -c @/home/mp/.arduino15/packages/esp8266/hardware/esp8266/3.1.1/tools/warnings/none-g++ -Os -g -free -fipa-pta -Werror=return-type -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++17 -ffunction-sections -fdata-sections -fno-exceptions -DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x8000 -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DDEBUG_ESP_PORT=Serial -DARDUINO=10607 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_WEMOS_D1MINI\"" "-DARDUINO_BOARD_ID=\"d1_mini\"" -DFLASHMODE_DIO -I/home/mp/.arduino15/packages/esp8266/hardware/esp8266/3.1.1/cores/esp8266 -I/home/mp/.arduino15/packages/esp8266/hardware/esp8266/3.1.1/variants/d1_mini /tmp/arduino/sketches/6B9DE325620D63245258CB1972B47822/sketch/BresserWeatherSensorBasic.ino.cpp -o /dev/null
Alternatives for WeatherSensorCfg.h: [BresserWeatherSensorReceiver@0.6.1]
ResolveLibrary(WeatherSensorCfg.h)
  -> candidates: [BresserWeatherSensorReceiver@0.6.1]
/home/mp/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -D_GNU_SOURCE -DESP8266 @/tmp/arduino/sketches/6B9DE325620D63245258CB1972B47822/core/build.opt -I/home/mp/.arduino15/packages/esp8266/hardware/esp8266/3.1.1/tools/sdk/include -I/home/mp/.arduino15/packages/esp8266/hardware/esp8266/3.1.1/tools/sdk/lwip2/include -I/home/mp/.arduino15/packages/esp8266/hardware/esp8266/3.1.1/tools/sdk/libc/xtensa-lx106-elf/include -I/tmp/arduino/sketches/6B9DE325620D63245258CB1972B47822/core -c @/home/mp/.arduino15/packages/esp8266/hardware/esp8266/3.1.1/tools/warnings/none-g++ -Os -g -free -fipa-pta -Werror=return-type -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++17 -ffunction-sections -fdata-sections -fno-exceptions -DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x8000 -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DDEBUG_ESP_PORT=Serial -DARDUINO=10607 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_WEMOS_D1MINI\"" "-DARDUINO_BOARD_ID=\"d1_mini\"" -DFLASHMODE_DIO -I/home/mp/.arduino15/packages/esp8266/hardware/esp8266/3.1.1/cores/esp8266 -I/home/mp/.arduino15/packages/esp8266/hardware/esp8266/3.1.1/variants/d1_mini -I/home/mp/Arduino/libraries/BresserWeatherSensorReceiver/src /tmp/arduino/sketches/6B9DE325620D63245258CB1972B47822/sketch/BresserWeatherSensorBasic.ino.cpp -o /dev/null
Alternatives for RadioLib.h: [RadioLib@5.7.0]
ResolveLibrary(RadioLib.h)
  -> candidates: [RadioLib@5.7.0]
[... cut ...]
╚══ IRAM     29799    code in IRAM    
. Code in flash (default, ICACHE_FLASH_ATTR), used 251356 / 1048576 bytes (23%)
║   SEGMENT  BYTES    DESCRIPTION
╚══ IROM     251356   code in flash   
Using library BresserWeatherSensorReceiver at version 0.6.1 in folder: /home/mp/Arduino/libraries/BresserWeatherSensorReceiver 
Using library RadioLib at version 5.7.0 in folder: /home/mp/Arduino/libraries/RadioLib 
Using library SPI at version 1.0 in folder: /home/mp/.arduino15/packages/esp8266/hardware/esp8266/3.1.1/libraries/SPI 
```

matthias-bs, 2023-03-12:
Binaries: https://github.com/matthias-bs/BresserWeatherSensorReceiver/tree/main/bin

matthias-bs, 2023-03-12:
Serial console output (115200 baud):
```
SDK:2.2.2-dev(38a443e)/Core:3.1.1=30101000/lwIP:STABLE-2_1_3_RELEASE/glue:1.2-65-g06164fb/BearSSL:b024386
Starting execution...
[CC1101] Initializing ... 
success!
[CC1101] Setup complete - awaiting incoming messages...
[CC1101] Receive failed: [0]
[CC1101] Receive failed: [0]
[CC1101] Receive failed: [0]
[CC1101] Receive failed: [0]
[CC1101] Receive failed: [0]
[CC1101] Receive failed: [0]
[CC1101] Receive failed: [0]
[CC1101] Data: D4 5B 71 39 58 23 76 18 FF 66 FF 23 48 0C 46 82 FF F0 73 AA 00 00 00 00 08 00 00 
[CC1101] R [D4] RSSI: -87.5
Digest check failed - [5B71] != [37C0]
[CC1101] Data: D4 AD C0 39 58 23 76 18 FF 76 FF 23 48 7F 8A 89 FF 01 CB AA 42 00 20 00 00 00 00 
[CC1101] R [D4] RSSI: -92.0
Digest check failed - [ADC0] != [6433]
[CC1101] Receive failed: [0]
[CC1101] Data: D4 94 F4 74 50 85 86 49 BB BB BB BB B0 21 96 07 BB B0 22 00 00 00 00 00 00 00 00 
[CC1101] R [D4] RSSI: -80.0
find_slot(): ID=74508586
find_slot(): Storing into slot #0
Temp: 1  Hum: 0  UV: 0  Wind: 0  Rain: 0  Moist: 1
Id: [74508586] Typ: [4] Battery: [OK ] Ch: [1] Temp: [ 21.9C] Hum: [---%] Wind max: [--.-m/s] Wind avg: [--.-m/s] Wind dir: [---.-deg] Rain: [-----.-mm] Moisture: [40%] UV index: [-.-%] RSSI: [-80.0dBm]
[CC1101] Receive failed: [0]
[CC1101] Data: D4 3B 8D 39 58 23 76 18 FF FF FF 23 48 FF 8A 89 FF 01 43 AA 00 00 80 00 00 00 40 
[CC1101] R [D4] RSSI: -92.5
find_slot(): ID=39582376
find_slot(): Storing into slot #0
Temp: 0  Hum: 0  UV: 0  Wind: 1  Rain: 1  Moist: 0
Id: [39582376] Typ: [1] Battery: [OK ] Ch: [0] Temp: [---.-C] Hum: [---%] Wind max: [ 0.0m/s] Wind avg: [ 0.0m/s] Wind dir: [234.0deg] Rain: [  757.6mm] Moisture: [--%] UV index: [-.-%] RSSI: [-92.5dBm]
```

stundenblume, 2023-03-12:
Thank you for your binary. It works as expected!
Now I have to figure out why my compiled code does not work.

```{$lܟ<�l�|�$�c<����s�#�c��gn�lno���#p�${$sd8�g��d��co�|���b��og�$��l`�'od`orǛ�gc$�lx�n�sĜ����bg�|�#��no�d`�gnl`o;���gc��`r��gc��`�����l`��g�d
SDK:2.2.2-dev(38a443e)/Core:3.1.1=30101000/lwIP:STABLE-2_1_3_RELEASE/glue:1.2-65-g06164fb/BearSSL:b024386
Starting execution...
[CC1101] Initializing ... 
success!
[CC1101] Setup complete - awaiting incoming messages...
[CC1101] Receive failed: [0]
[CC1101] Data: D4 33 99 36 14 52 11 18 FF FF FF 13 28 20 46 53 FF F0 5A AA 00 00 00 00 00 00 00 
[CC1101] R [D4] RSSI: -55.5
find_slot(): ID=36145211
find_slot(): Storing into slot #0
Temp: 1  Hum: 1  UV: 1  Wind: 1  Rain: 0  Moist: 0
Id: [36145211] Typ: [1] Battery: [OK ] Ch: [0] Temp: [ 20.4C] Hum: [ 53%] Wind max: [ 0.0m/s] Wind avg: [ 0.0m/s] Wind dir: [132.0deg] Rain: [-----.-mm] Moisture: [--%] UV index: [0.0] RSSI: [-55.5dBm]
[CC1101] Data: D4 13 FB 36 14 52 11 18 FF FF FF 13 28 FF FC FB FF 01 0C AA 00 00 00 00 00 00 00 
[CC1101] R [D4] RSSI: -69.5
find_slot(): ID=36145211
find_slot(): Storing into slot #0
Temp: 0  Hum: 0  UV: 0  Wind: 1  Rain: 1  Moist: 0
Id: [36145211] Typ: [1] Battery: [OK ] Ch: [0] Temp: [---.-C] Hum: [---%] Wind max: [ 0.0m/s] Wind avg: [ 0.0m/s] Wind dir: [132.0deg] Rain: [   30.4mm] Moisture: [--%] UV index: [-.-%] RSSI: [-69.5dBm]
[CC1101] Receive failed: [0]```

matthias-bs, 2023-03-12:
What was the reason? Maybe configured for SX1276 as per default?

stundenblume, 2023-03-12:
Right now I just used your .bin to find inner rest. ;-)
After the kids are sleeping I will try to compile by myself again.

My environment:
BresserWeatherSensorReceiver: v0.6.1
MQTT v2.5.1 
RadioLib v5.7.0
ArduinoJson v6.20.1 
Arduino IDE 2.0.4

stundenblume, 2023-03-12:
[...]
ESP8266 by ESP8266 Community v3.1.1

.../libraries/BresserWeatherSensorReceiver/src/WeatherSensorReceiverCfg.h:
```
#if ( !defined(USE_CC1101) && !defined(USE_SX1276) )
    #define USE_CC1101
    //#define USE_SX1276
#endif
```

**differences:**
Board: LOLIN(WEMOS) D1 mini (clone) (not the cause, that board works also)
Debug Port: Disabled (not the cause, it works to)

I don't know why, but now the self compiled code works as expected. I will have messed up something...
Sorry for taking your time and not able to reproduce the behavior. :-/

matthias-bs, 2023-03-12:
No worries! I'm happy that it works now!

closed 2023-03-12 by stundenblume