matthias-bs/BresserWeatherSensorReceiver #44: WADLBECK

dmolner, 2023-03-08:
Hello,
I am trying work with a Waldbeck station.
I have an OLED TTGO LORA32 board.
I have this mistake.
Start run ...
[42] [E] [Weathersensor.cpp: 130] Begin (): [sx1276] Initializing error: [-104]

I read https://github.com/mathias-bs/bresserweathersensorreceiver/issues/41, but I'm not sure if it can work.

Best regards,
David.

matthias-bs, 2023-03-08:
Hi,
Did you look up the error code in RadioLib?
Did you do an internet search for "Waldbeck Weather Station Protocol"? Most likely this is a badge of some product already supported by rtl_433.
Regards,
Matthias

matthias-bs, 2023-03-08:
Which model do you have?
Did you already modify the code?

dmolner, 2023-03-08:
> [sx1276] Initializing error: [-104]
I found this in RadioLib
/*!
  \brief The supplied receiver bandwidth value is invalid.
*/
#define RADIOLIB_ERR_INVALID_RX_BANDWIDTH                      (-104)

> Did you do an internet search for "Waldbeck Weather Station Protocol"? Most likely this is a badge of some product already supported by rtl_433.
I think its possible work
https://github.com/merbanan/rtl_433/issues/1750

dmolner, 2023-03-08:
> Which model do you have? Did you already modify the code?

Model Nº 10032850
Which part of code i need modify.
Sorry.

fgnm, 2023-03-09:
To add on this, I'm still not able to receive data with the Heltec board (which is basically the same of @dmolner).. Currently I can only receive data with the CC1101 sensor attached to an ESP8266.. I haven't time to look in depth yet, but at least in theory I don't see any reason why the SX1276 shouldn't work

dmolner, 2023-03-09:
After doing some tests.
I'm definitely lost, now I only get the message from

Starting execution...

I will wait for you to @fgnm post what you have achieved

matthias-bs, 2023-03-09:
Did you select the correct variant of the TTGO LORA32 board?  The internal connections differ and for the V2 an additional wire is needed. (see https://github.com/matthias-bs/BresserWeatherSensorReceiver#configuration-by-selecting-a-supported-board-in-the-arduino-ide)

dmolner, 2023-03-09:
I can't find DIO1 in the board

> Additionally you might uncomment `RADIOLIB_DEBUG`:
Starting execution...
M	SX127x

matthias-bs, 2023-03-09:
If the board shown in the picture above is the one you are using, it's the V2 which needs the extra wire between LoRa_DIO1 and GPIO33 (did you draw the red wire to show me this?).
Did you select Board: "TTGO LoRa32-OLED"/Board Revision: "TTGO LoRa32 V2" in the Arduino IDE?

I do not understand your comment regarding the Heltec board. If you are using a Heltec board (which one?) you need to set the pin configuration by yourself or select Board: "Heltec Wireless Stick" if it is this specific one.

In general, to track down these kind of issues, I need as much information as possible:

- board/board revision (your HW)
- board/board revision (selected by you in the Arduino IDE)
- LoRa module frequency (some boards are available as 433 MHz and as 868 MHz versions)
- external wiring (if applicable)
- compiler output
- run-time serial console output (with debug level set to "Debug" before compiling)
- any code changes you have made (preferably in a github fork of the project)
- in case of a weather sensor related issue: brand, model, radio frequency)

matthias-bs, 2023-03-09:
> [image]
> 
> > Additionally you might uncomment `RADIOLIB_DEBUG`:
> > Starting execution...
> > M	SX127x
> 
> After some test more With an Heltec board in serial console I see 16:46:20.610 -> Starting execution... 16:46:20.654 -> M SX127x 16:46:20.654 -> 16:46:20.654 -> address: 0x1 16:46:20.654 -> bits: 2 0 16:46:20.654 -> value: 0b1 16:46:20.654 -> current: 0b1100 16:46:20.654 -> mask: 0b111 16:46:20.654 -> new: 0b1001 16:46:20.654 -> read: 0b1100 16:46:20.654 ->
> 
> But nothing more.

RadioLib tries to change the bits [2:0] the OP_MODE register (read-modify-write access):
#define RADIOLIB_SX127X_REG_OP_MODE                            0x01

The value of bits [2:1] shall be 0b1, the current value (of the entire register) is 0b1100. So the new value shall be 0b1001, but it is read back as 0b1100 - the same value as before. So the changing register failed.

This means the SPI write access failed. Maybe the pin configuration is still wrong (SPI, reset) or there is a HW defect.

dmolner, 2023-03-09:
> If the board shown in the picture above is the one you are using, it's the V2 which needs the extra wire between LoRa_DIO1 and GPIO33 (did you draw the red wire to show me this?). Did you select Board: "TTGO LoRa32-OLED"/Board Revision: "TTGO LoRa32 V2" in the Arduino IDE?
Yes this is my board.
[image]

> 
> I do not understand your comment regarding the Heltec board. If you are using a Heltec board (which one?) you need to set the pin configuration by yourself or select Board: "Heltec Wireless Stick" if it is this specific one.

I test with this other board, but i prefer TTGO board.
> 
> In general, to track down these kind of issues, I need as much information as possible:
> 
> * board/board revision (your HW)
TTGO ESP32 LoRa 868MHz
> * board/board revision (selected by you in the Arduino IDE)
TTGO LoRa32-OLED
> * LoRa module frequency (some boards are available as 433 MHz and as 868 MHz versions)
868MHz
> * external wiring (if applicable)
Wire between LoRa_DIO1 and GPIO33 (same the picture)
> * compiler output
`In file included from C:\Users\dmoln\Baixades\Arduino\BresserWeatherSensorReceiver-main\examples\WALDBEKC\BresserWeatherSensorBasic\BresserWeatherSensorBasic.ino:54:
C:\Users\dmoln\Baixades\Arduino\BresserWeatherSensorReceiver-main\examples\WALDBEKC\BresserWeatherSensorBasic\WeatherSensorCfg.h:106:88: note: #pragma message: ARDUINO_heltec_wireless_stick defined; using on-board transceiver
     #pragma message("ARDUINO_heltec_wireless_stick defined; using on-board transceiver")
                                                                                        ^
In file included from C:\Users\dmoln\Baixades\Arduino\BresserWeatherSensorReceiver-main\examples\WALDBEKC\BresserWeatherSensorBasic\BresserWeatherSensorBasic.ino:54:
C:\Users\dmoln\Baixades\Arduino\BresserWeatherSensorReceiver-main\examples\WALDBEKC\BresserWeatherSensorBasic\WeatherSensorCfg.h:336:48: note: #pragma message: Receiver chip: [SX1276]
 #pragma message("Receiver chip: " RECEIVER_CHIP)
                                                ^
C:\Users\dmoln\Baixades\Arduino\BresserWeatherSensorReceiver-main\examples\WALDBEKC\BresserWeatherSensorBasic\WeatherSensorCfg.h:337:170: note: #pragma message: Pin config: RST->RST_LoRa, CS->SS, GD0/G0/IRQ->DIO0, GDO2/G1/GPIO->DIO1
 #pragma message("Pin config: RST->" STR(PIN_RECEIVER_RST) ", CS->" STR(PIN_RECEIVER_CS) ", GD0/G0/IRQ->" STR(PIN_RECEIVER_IRQ) ", GDO2/G1/GPIO->" STR(PIN_RECEIVER_GPIO) )
                                                                                                                                                                          ^
In file included from C:\Users\dmoln\Baixades\Arduino\BresserWeatherSensorReceiver-main\examples\WALDBEKC\BresserWeatherSensorBasic\WeatherSensor.h:66,
                 from C:\Users\dmoln\Baixades\Arduino\BresserWeatherSensorReceiver-main\examples\WALDBEKC\BresserWeatherSensorBasic\BresserWeatherSensorBasic.ino:55:
c:\Users\dmoln\OneDrive\Documentos\Arduino\libraries\RadioLib\src/RadioLib.h:53:61: note: #pragma message: RADIOLIB_PLATFORM="ESP32"
   #pragma message(RADIOLIB_VAR_NAME_VALUE(RADIOLIB_PLATFORM))
                                                             ^
c:\Users\dmoln\OneDrive\Documentos\Arduino\libraries\RadioLib\src/RadioLib.h:54:66: note: #pragma message: RADIOLIB_VERSION_MAJOR=(0x05)
   #pragma message(RADIOLIB_VAR_NAME_VALUE(RADIOLIB_VERSION_MAJOR))
                                                                  ^
c:\Users\dmoln\OneDrive\Documentos\Arduino\libraries\RadioLib\src/RadioLib.h:55:66: note: #pragma message: RADIOLIB_VERSION_MINOR=(0x07)
   #pragma message(RADIOLIB_VAR_NAME_VALUE(RADIOLIB_VERSION_MINOR))
                                                                  ^
c:\Users\dmoln\OneDrive\Documentos\Arduino\libraries\RadioLib\src/RadioLib.h:56:66: note: #pragma message: RADIOLIB_VERSION_PATCH=(0x00)
   #pragma message(RADIOLIB_VAR_NAME_VALUE(RADIOLIB_VERSION_PATCH))
                                                                  ^
c:\Users\dmoln\OneDrive\Documentos\Arduino\libraries\RadioLib\src/RadioLib.h:57:66: note: #pragma message: RADIOLIB_VERSION_EXTRA=(0x00)
   #pragma message(RADIOLIB_VAR_NAME_VALUE(RADIOLIB_VERSION_EXTRA))
                                                                  ^`
> * run-time serial console output (with debug level set to "Debug" before compiling)
`18:48:35.851 -> ets Jun  8 2016 00:22:57
18:48:35.851 -> 
18:48:35.851 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
18:48:35.851 -> configsip: 188777542, SPIWP:0xee
18:48:35.851 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
18:48:35.851 -> mode:DIO, clock div:1
18:48:35.851 -> load:0x3fff0030,len:1184
18:48:35.851 -> load:0x40078000,len:13220
18:48:35.851 -> ho 0 tail 12 room 4
18:48:35.851 -> load:0x40080400,len:3028
18:48:35.851 -> entry 0x400805e4
18:48:36.006 -> Starting execution...
18:48:36.006 -> M	SX127x`

> * any code changes you have made (preferably in a github fork of the project)
Not change nothing.
Only uncomment line 70
#define ARDUINO_TTGO_LoRa32_V2
> * in case of a weather sensor related issue: brand, model, radio frequency)
WALDBECK Mod Nº 10032850 868.3MHz

Do not you worry.
I'm sorry to bother you so much.

matthias-bs, 2023-03-09:
Please note that you can set the "Board" and in some cases also the "Board Revision" in the Arduino IDE:
[image]

If you configure the board in this way, you do not have to uncomment the define in `WeatherSensorCfg.h`, because it is passed to the compiler automatically.

matthias-bs, 2023-03-09:
Please make sure that you don't mix configurations: in your description above, the first items are related to the TTGO LoRa32 board, but then the compiler output shows `Arduino_heltec_wireless_stick` is defined.

matthias-bs, 2023-03-09:
Searching for the Waldbeck model number 10032850 revealed the model name "Huygens" which seems to be mentioned more often when it comes to re-branding and protocol compatibility.

dmolner, 2023-03-10:
> Please make sure that you don't mix configurations: in your description above, the first items are related to the TTGO LoRa32 board, but then the compiler output shows `Arduino_heltec_wireless_stick` is defined.

Yes, it is because I make one test with a Heltec board, but i'm working with TTGO

closed 2023-03-26 by matthias-bs