matthias-bs/BresserWeatherSensorReceiver #45: 6-in-1 Decoder fails PR #40

matthias-bs, 2023-03-09:
@jorgenavarroortiz: The 6-in-1 decoder fails for some reason after integrating https://github.com/matthias-bs/BresserWeatherSensorReceiver/pull/40.

matthias-bs, 2023-03-09:
In decodeBresser7In1Payload(), the "data whitening" modifies the payload buffer in-place, rendering it useless for subsequent decoding attempts.

@jorgenavarroortiz Please make a local copy of the payload buffer (strongly preferred) or move this decoder to the end of the chain.

matthias-bs, 2023-03-09:
Workaround: Comment out `#define BRESSER_7_IN_1` in https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/e69774b7375aecbde922bf0d078055341a8e7fec/src/WeatherSensorCfg.h#L216.

jorgenavarroortiz, 2023-03-09:
Hi! I have submit a pull request. I have made a local copy of the payload buffer. It works with my weather station (7-in-1). Please test it with the other weather stations (5-in-1 and 6-in-1), it should work.

matthias-bs, 2023-03-09:
Looks good! I'll check it tomorrow!

closed 2023-03-28 by matthias-bs