matthias-bs/BresserWeatherSensorReceiver #48: Issues after update

mczakk, 2023-03-23:
Hi Matthias, me again :0
I've just tried to update my esp32 board after noticing an error on the display, and it has me slightly confused after your updates to the WeatherSensorCcg.h file
My board is a T3_V1.6.1, with the LoRa DI01 wired manually to socket 12 (as labelled on the board. I cant find any reference to GPIO 33 for my board, and have had to use 
`
#define PIN_LMICDI01 12 
`
in my TTGO_ESP32_Bresser_with _display.ino code, and change line 187 to 
`if defined (ARDUINO_TTGO_LoRa32_V2)
To get the new version of WeatherSensor.cfg.h to work I have to select "Board: TTGO LoRa32-OLED" / "Board Revision: TTGO LoRa32 V2" in the Arduino IDE.
It does throw this error when compiling :
`c:\Users\mtc20tfq\Arduino\libraries\MCCI_LoRaWAN_LMIC_library\src\hal\getpinmap_thisboard.cpp: In function 'const Arduino_LMIC::HalPinmap_t* Arduino_LMIC::GetPinmap_ThisBoard()':
c:\Users\mtc20tfq\Arduino\libraries\MCCI_LoRaWAN_LMIC_library\src\hal\getpinmap_thisboard.cpp:71:72: note: #pragma message: Board not supported -- use an explicit pinmap
         #pragma message("Board not supported -- use an explicit pinmap")`
but does appear to work ok. 
Its just a little confusing!!
:)

matthias-bs, 2023-03-26:
Hi,
What do you mean by "I cant find any reference to GPIO 33 for my board"? Is GPIO33 not available at the pin headers?
Regarding the pragma messages: those are not compiler errors, just notes. I've left them in from the original LMIC examples, but I admit this makes no sense if a board which is supported has been selected. I will change this in the next release.

matthias-bs, 2023-06-27:
I tried to suppress pragma messages from the LMIC library code, but it didn't work. You would have to modify the code itself.

closed 2023-06-27 by matthias-bs