matthias-bs/BresserWeatherSensorReceiver #49: Code modification needed to get 5-in-1 WS working

Noamby42, 2023-03-26:
Hey! I had a problem where I had to replace "else if" by "if" in WeatherSensor.cpp L235 to make it work, otherwise it gives me a checksum error. Any ideas if it's just me or the bug is knew?

matthias-bs, 2023-03-26:
Hi,
No there is no known issue regarding this.
Are you sure the checksum error just comes from the 6-in-1 decoder which is tried before the 5-in-1 decoder? If this is the case, the data should be decoded correctly and you can ignore those checksum errors.
Otherwise: could you please provide some debug output?
Regards
Matthias

Noamby42, 2023-03-26:
The error is coming from the 5in1 decoder, because the information is not decoded for either the 6in1 (obviously) or the 5in1.

Here what I got when 6in1 decoder is defined and the "else if" is present :

[ 11334][V][WeatherSensor.cpp:219] getMessage(): [SX1276] Data: D4 ED 0F FF FF 0F FF EF 68 FE AD FF FF FF 12 F0 00 00 F0 00 10 97 01 52 00 00 00 
[ 11336][D][WeatherSensor.cpp:221] getMessage(): [SX1276] R [D4] RSSI: -37.5
[ 11342][D][WeatherSensor.cpp:658] decodeBresser6In1Payload(): Digest check failed - [ED0F] != [5666]

The debug log never show me any kind of error for the 5in1 decoder, but no decoded data either.

If I desactivate the 6in1 decoder without removing the "else", I get some errors beacause there is no "if" before the "else if" : 

C:\Users\Noam-\Documents\Arduino\libraries\BresserWeatherSensorReceiver-0.6.1\src\WeatherSensor.cpp: In member function 'DecodeStatus WeatherSensor::getMessage()':
C:\Users\Noam-\Documents\Arduino\libraries\BresserWeatherSensorReceiver-0.6.1\src\WeatherSensor.cpp:235:17: error: expected '}' before 'else'
                 else if (decode_res == DECODE_INVALID ||
                 ^~~~
C:\Users\Noam-\Documents\Arduino\libraries\BresserWeatherSensorReceiver-0.6.1\src\WeatherSensor.cpp:212:34: note: to match this '{'
         if (recvData[0] == 0xD4) {
                                  ^
C:\Users\Noam-\Documents\Arduino\libraries\BresserWeatherSensorReceiver-0.6.1\src\WeatherSensor.cpp:250:5: warning: no return statement in function returning non-void [-Wreturn-type]
     } // if (state == RADIOLIB_ERR_NONE)
     ^
C:\Users\Noam-\Documents\Arduino\libraries\BresserWeatherSensorReceiver-0.6.1\src\WeatherSensor.cpp: At global scope:
C:\Users\Noam-\Documents\Arduino\libraries\BresserWeatherSensorReceiver-0.6.1\src\WeatherSensor.cpp:252:5: error: expected unqualified-id before 'return'
     return decode_res;
     ^~~~~~
C:\Users\Noam-\Documents\Arduino\libraries\BresserWeatherSensorReceiver-0.6.1\src\WeatherSensor.cpp:253:1: error: expected declaration before '}' token
 }
 ^

After removing the "else", I get the decoded data of my Weather station.
I've updated the library manually few hours ago and I didn't succeed to reproduce the checksum error for the 5in1 decoder, the gap between the bitsSet and the expectedBitsSet was of 1 or 2 ; maybe I've changed a value somewhere in the library by mistake.

PS : I use a Lilygo V2.1 1.6 for the ‎7002510 5-in-1 Bresser WS

matthias-bs, 2023-03-26:
Thanks for the compiler and console output!

Yes, 
https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/49da1553d47d159c6fc0b247d8bd129cb1519fa5/src/WeatherSensor.cpp#L235
has to be changed to `if (decode_res == DECODE_INVALID || ` in order to allow disabling of the 6-in-1 decoder!

But I don't understand why you do not get decoded data with the 6-in-1 decoder enabled. I have a setup with two sensors; one of them needs the 6-in-1 decoder the other one the 5-in-1 decoder. So the 5-in-1 decoder is known to work also if the 6-in-1 decoder is run before.

The only pitfall I observed is that I sometimes received another (6-in-1 type) sensor from somewhere in my neighbourhood. In this case - depending on the setting of `NUM_SENSORS` - no attempt is made to try to receive your own sensor in the current cycle. In this case, you have to filter the sensor IDs as desired (`SENSOR_IDS_EXC`/`SENSOR_IDS_INC`).

matthias-bs, 2023-03-27:
Fixed in https://github.com/matthias-bs/BresserWeatherSensorReceiver/commit/e982909c912606a64fd2ce583df80b84144b293f

matthias-bs, 2023-03-29:
Could you please re-check with the updated version and the information above?

Noamby42, 2023-04-08:
Hi Matthias sorry for the late response, working well without compilation error, thanks for the fix

closed 2023-04-08 by Noamby42