matthias-bs/BresserWeatherSensorReceiver #50: Bresser model 7 in 1 7003100CM3000

marxu, 2023-03-29:
Hello.
I just bought this weather station and I would like to try the CC1101, but I don't know which one to buy. Can you tell me where to buy it? So I would test if it works, and I would indicate it here. Thank you.

matthias-bs, 2023-03-29:
Hi! Well, this is a little bit difficult. These modules are sold on Amazon, Ebay and by various electronics suppliers. I don't know any module where the actual manufacturer is know or which I would prefer over others.

Unfortunately, there have been several reports about modules which had a defect on arrival.

Maybe you can send a link our two and I can confirm if it should work according to its product description from a technical point of view. The manufacturing quality may be another issue...

matthias-bs, 2023-03-29:
Could you please provide the part number on the sensor (if available)?

marxu, 2023-03-29:
Where do I have to look for the part number of the sensor?

matthias-bs, 2023-03-29:
Maybe on the lid of the battery compartment?

marxu, 2023-03-29:
Thank you very much for answering Mattias-bs.
I don't know how to locate the sensor number. I've looked and can't find it. 
But it works at 868 mhz, I thought this was enough.
All C1101 are not the same?

matthias-bs, 2023-03-29:
O.k., thanks for looking for the part number.

The CC1101 modules are available as 433 MHz and as 868 MHz versions (same IC, but different PCB assembly) - sometimes the offers are not as obvious as they should be.

The next topic is the header connector - full vs. half through-hole pads, 1.27 vs 2.254 mm pitch.

And finally there is the antenna connector - single pad for wire/coil antenna vs. pads for SMA connector (seems to be rare for 868 MHz versions, though).

And you should check if the antenna is included or not...

On the quality side, there are issues with  defect modules (e.g. no SPI communication, abnormal power consumption), frequency offset and bad range (maybe related to the antenna).

matthias-bs, 2023-05-29:
Hi, do you have any news on that? Can this issue be closed?

matthias-bs, 2023-06-27:
Closed due to several weeks of inactivity.

closed 2023-06-27 by matthias-bs