matthias-bs/BresserWeatherSensorReceiver #53: BresserWeatherSensorReceiverBasic ESP32-WROVER-E with c1101

windmillBart, 2023-05-04:
Hi,
I,m trying to use the BresserWeatherSensorReceiverBasic example with an ESP32-WROVER-E board in combination with a c1101 868 Mhz module. I'm using Arduino IDE 1.8 to compile the sketches.

I made the required changes in the WeathersensorCfg.h file to define the ESP32, c1101 and to define the pins as follows:
// Arduino default SPI pins
//
// Board   SCK   MOSI  MISO
// ESP8266 D5    D7    D6
// ESP32   D18   D23   D19

// Generic pinning for ESP32 development boards
#define PIN_RECEIVER_CS   5
// CC1101: GDO0 / RFM95W/SX127x: G0
#define PIN_RECEIVER_IRQ  21
// CC1101: GDO2 / RFM95W/SX127x: G1
#define PIN_RECEIVER_GPIO 4
// RFM95W/SX127x - GPIOxx / CC1101 - RADIOLIB_NC
#define PIN_RECEIVER_RST  32

First question I only have the following pins on the c1101 module: vcc, gnd, miso, mosi, gdo0, gdo2, csn, slck.
the above PIN_RECEIVER_RST 32 is not connected as this would be the 9th pin to connect, I only have 8 available?

when I compile and upload the following message appaers on terminal

    [     4][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   451][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
Starting execution...
[   458][D][WeatherSensor.cpp:85] begin(): [CC1101] Initializing ... 
[   558][E][WeatherSensor.cpp:129] begin(): [CC1101] Error initialising: [-2]

any idea what could be the issue or problem?
thanks in advance,
Bart

matthias-bs, 2023-05-05:
Hi,

you do not need a reset pin for the CC1101 - you can use
`#define PIN_RECEIVER_RST RADIOLIB_NC`

Could you please provide a little more information about the ESP32 board you are using? A link to the supplier/manufacturer page with datasheet/pinout would be helpful. Which board did you select in the Arduino IDE?

The error above means the communication between the ESP32 and the CC1101 did not work properly. The reason for this can be a wrong connection or a defective CC1101 module (this is not uncommon).

Regards
Matthias

matthias-bs, 2023-06-27:
Closed due to several weeks of inactivity.

closed 2023-06-27 by matthias-bs