matthias-bs/BresserWeatherSensorReceiver #54: CC1101 & Esp8266

grotma, 2023-05-21:
Hi,
first thanks a lot for the project.
Unfortunately I have troubles to get it working. I'm using a CC1101 together with a CH340 Node MCU V3 ESP8266. I believe I connected it correctly, as the sketch confirms successful connection. But I doesn't manage to process the data, and I'm currently not sure why.

18:13:05.690 -> success!
18:13:05.690 -> [CC1101] Setup complete - awaiting incoming messages...
18:13:06.214 -> ..............................[CC1101] Receive failed: [0]
18:13:20.982 -> ...................................[CC1101] Receive failed: [0]
18:13:38.162 -> ......................................[CC1101] Receive failed: [0]
18:13:57.106 -> ...................
18:14:06.094 -> Sensor timeout
18:14:06.715 -> ......[CC1101] Data: D4 CF 59 73 9B 64 83 58 FD BF FD 09 08 21 06 62 BB B0 5E 00 00 00 00 00 00 00 00 
18:14:09.615 -> [CC1101] R [D4] RSSI: -87.5
18:14:09.615 -> Parity wrong at column 0
18:14:09.615 -> ........................[CC1101] Receive failed: [0]
18:14:21.631 -> ........................[CC1101] Receive failed: [0]
18:14:33.636 -> ...................................................[CC1101] Receive failed: [0]
18:14:59.071 -> .......[CC1101] Receive failed: [0]
18:15:02.499 -> .........
18:15:06.496 -> Sensor timeout
18:15:07.067 -> ......[CC1101] Receive failed: [0]
18:15:09.750 -> ....................................[CC1101] Receive failed: [0]
18:15:27.324 -> ..........................................[CC1101] Receive failed: [0]
18:15:48.310 -> ......................................
18:16:06.838 -> Sensor timeout
18:16:07.458 -> .....[CC1101] Data: D4 D6 A7 73 93 44 83 18 FC BD FC 13 58 20 86 64 BB B0 85 00 00 00 00 00 00 00 00 
18:16:09.611 -> [CC1101] R [D4] RSSI: -88.5
18:16:09.611 -> Parity wrong at column 0
18:16:09.611 -> .......[CC1101] Receive failed: [0]
18:16:12.670 -> .[CC1101] Receive failed: [0]
18:16:12.956 -> ..................[CC1101] Data: D4 3F 82 73 93 44 83 18 FC BD FC 09 0C FF ED 8D FB 01 DB 00 00 00 00 40 00 00 00 
18:16:21.640 -> [CC1101] R [D4] RSSI: -90.5
18:16:21.640 -> Parity wrong at column 0

Anything you can see I've done wrong? Thanks for your help!

matthias-bs, 2023-05-22:
Hi,

It looks like you just have bad reception. To prove this, just run the sketch a little longer; eventually you should be able to receive a valid message. The lines starting with `[CC1101] Data: D4 3F ...` show some valid data, but then there is still a parity error.

Please have a look at https://github.com/matthias-bs/BresserWeatherSensorReceiver/wiki#radio-transceiver-range for some hints about improving signal quality.

BTW: Which weather sensor are you using?

Regards
Matthias

matthias-bs, 2023-05-29:
Hi, do you have any news on that?

grotma, 2023-05-30:
Hi Matthias,
thanks for asking. So far I had no luck with just getting it closer to the weatherstation using the MQTT sketch. Quality of signal was at best between -70 and -80 (but sometimes as well -100), which I believe should be sufficient? I hope it's not just a zigbee device it's detecting.
Anyhow, once I got the time to try a different sized antenna, I'll give you an update!

matthias-bs, 2023-05-31:
Yes, the signal level should be fine. You can find log files in the examples' directories with similar values. I think you've been receiving the correct signal, because the preamble was detected successfully and some parts of the raw data look familiar and different transmissions are similar.

grotma, 2023-06-19:
Hi,
sorry for the late reply. Yes, I tried straight wire as well as the (very cheap looking) coil antenna from the c1101. Still no luck. I'll try with a different setup, I'll keep you updated

grotma, 2023-07-10:
Quick update,
I got a TTGO Lora Oled now and tried, but at least connected to my PC still no joy.
21:11:05.739 -> [178737][D][WeatherSensor.cpp:463] decodeBresser5In1Payload(): Parity wrong at column 0
21:11:17.734 -> [190729][V][WeatherSensor.cpp:220] getMessage(): [SX1276] Data: D4 A5 DB 73 93 44 83 18 FE AD FE 15 C8 27 66 85 BB B0 57 20 00 00 00 00 00 00 00 
21:11:17.734 -> [190731][D][WeatherSensor.cpp:222] getMessage(): [SX1276] R [D4] RSSI: -102.5
21:11:17.782 -> [190738][D][WeatherSensor.cpp:463] decodeBresser5In1Payload(): Parity wrong at column 0
21:11:41.742 -> [214730][V][WeatherSensor.cpp:220] getMessage(): [SX1276] Data: D4 DB CE 73 93 44 83 18 FD 7A FD 13 D8 27 66 85 BB B2 BE 00 00 00 00 00 00 00 00 
21:11:41.742 -> [214732][D][WeatherSensor.cpp:222] getMessage(): [SX1276] R [D4] RSSI: -101.5
21:11:41.742 -> [214739][D][WeatherSensor.cpp:463] decodeBresser5In1Payload(): Parity wrong at column 0
21:11:53.768 -> [226731][V][WeatherSensor.cpp:220] getMessage(): [SX1276] Data: D4 DE 48 73 93 C4 83 18 FD BB FD 13 58 FF D9 C9 FF 01 59 00 00 00 00 84 00 00 00 
21:11:53.768 -> [226732][D][WeatherSensor.cpp:222] getMessage(): [SX1276] R [D4] RSSI: -101.5
21:11:53.768 -> [226739][D][WeatherSensor.cpp:463] decodeBresser5In1Payload(): Parity wrong at column 0

I'll try again getting closer later this week

matthias-bs, 2023-07-10:
Thanks for the information. I hope it is still a matter of range.

grotma, 2023-07-24:
The RSSI is definately a matter of range. I finally got my lenovo laptop out with arduino just below the weatherstation:
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4
[     5][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[    43][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=4 scl=15 freq=100000

BresserWeatherSensorMQTT 20221024

xxxxxxx
connected!
Checking wifi...
MQTT connecting... connected!
BresserDomo-E0A35C/status: online
[  2249][D][WeatherSensor.cpp:85] begin(): [SX1276] Initializing ... 
[  2261][D][WeatherSensor.cpp:98] begin(): success!
[  2261][D][WeatherSensor.cpp:133] begin(): [SX1276] Setup complete - awaiting incoming messages...
[ 13859][V][WeatherSensor.cpp:220] getMessage(): [SX1276] Data: D4 44 29 73 93 44 83 18 FD E6 FE 13 58 FF D9 C9 FF 01 2D 00 00 00 00 00 00 00 00 
[ 13860][D][WeatherSensor.cpp:222] getMessage(): [SX1276] R [D4] RSSI: -76.5
[ 13867][D][WeatherSensor.cpp:463] decodeBresser5In1Payload(): Parity wrong at column 0
[ 25859][V][WeatherSensor.cpp:220] getMessage(): [SX1276] Data: D4 7A B7 73 93 44 83 18 FD E6 FE 13 58 31 96 72 BB B0 2A 00 00 00 00 00 00 00 00 
[ 25861][D][WeatherSensor.cpp:222] getMessage(): [SX1276] R [D4] RSSI: -82.5
[ 25867][D][WeatherSensor.cpp:463] decodeBresser5In1Payload(): Parity wrong at column 0
[ 37859][V][WeatherSensor.cpp:220] getMessage(): [SX1276] Data: D4 DC 19 73 93 44 83 18 FE EE FE 13 58 FF D9 C9 FF 01 24 00 00 00 00 00 00 00 00 
[ 37861][D][WeatherSensor.cpp:222] getMessage(): [SX1276] R [D4] RSSI: -83.0
[ 37868][D][WeatherSensor.cpp:463] decodeBresser5In1Payload(): Parity wrong at column 0
[ 49860][V][WeatherSensor.cpp:220] getMessage(): [SX1276] Data: D4 BF 60 73 93 44 83 18 FE 79 FE 13 58 31 96 72 BB B0 96 00 00 00 00 00 00 00 00 
[ 49862][D][WeatherSensor.cpp:222] getMessage(): [SX1276] R [D4] RSSI: -78.5
[ 49868][D][WeatherSensor.cpp:463] decodeBresser5In1Payload(): Parity wrong at column 0

Unfortunately still the Parity problem. Any idea what else I could try?

matthias-bs, 2023-07-24:
Hi,

I fed your testdata to [BresserWeatherSensorTest.ino](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/examples/BresserWeatherSensorTest/BresserWeatherSensorTest.ino) and disabled the parity/checksum error aborts in `decodeBresser5In1Payload()` (`WeatherSensor.cpp` lines 507 & 525). I got the following results:
```
20:10:27.259 -> testData[5]
20:10:27.259 -> Id: [      8D] Typ: [B] Battery: [OK ] Ch: [0] Temp: [  0.0C] Hum: [  0%] Wind max: [171.2m/s] Wind avg: [ 0.0m/s] Wind dir: [202.5deg] Rain: [    0.0mm] Moisture: [--%] UV index: [-.-%] Light (lux): [--.-Klux] RSSI: [-56.0dBm]
20:10:28.318 -> testData[6]
20:10:28.318 -> Id: [      36] Typ: [F] Battery: [OK ] Ch: [0] Temp: [  0.0C] Hum: [  0%] Wind max: [332.9m/s] Wind avg: [ 0.0m/s] Wind dir: [ 45.0deg] Rain: [    0.0mm] Moisture: [--%] UV index: [-.-%] Light (lux): [--.-Klux] RSSI: [-56.0dBm]
20:10:29.346 -> testData[7]
20:10:29.346 -> Id: [      8D] Typ: [B] Battery: [OK ] Ch: [0] Temp: [  0.0C] Hum: [  0%] Wind max: [273.6m/s] Wind avg: [ 0.0m/s] Wind dir: [ 45.0deg] Rain: [    0.0mm] Moisture: [--%] UV index: [-.-%] Light (lux): [--.-Klux] RSSI: [-56.0dBm]
20:10:30.373 -> testData[8]
20:10:30.373 -> Id: [      36] Typ: [F] Battery: [OK ] Ch: [0] Temp: [  0.0C] Hum: [  0%] Wind max: [102.5m/s] Wind avg: [ 0.0m/s] Wind dir: [ 45.0deg] Rain: [    0.0mm] Moisture: [--%] UV index: [-.-%] Light (lux): [--.-Klux] RSSI: [-56.0dBm]
```
As you can see, the input data is still wrong, you do not even get the same ID on every cycle.

Did you try with different PCs to sort out a potential problem with a noisy power supply? As far as I remember, you already tried different ESP board/receiver/antenna combinations, right?

Another try would be attaching the debug output to an external serial-to-usb converter to get the data to the PC and supplying the MCU/receiver from a USB power bank.

grotma, 2023-07-24:
Yes, that's right, tried initally with esp8266/cc1101 and now with TTGO Lora Oled.
The try 2 weeks ago was on a Dell PC, and now on a lenovo laptop.
Just pluged a different cable (no, doesn't sound too stupid ;-)) but still parity error.
I don't have a serial to usb converter handy, unfortunately.

I will let it run tonight close to the sensor using a USB powerbank and MQTT for logging, but I doubt it'll be different (but who knows?)

matthias-bs, 2023-10-05:
Closed due to inactivity.

closed 2023-10-05 by matthias-bs