matthias-bs/BresserWeatherSensorReceiver #56: Need help: No sensors for BRESSER_5_IN_1

meijer3, 2023-06-05:
With my Heltec lora v2 esp32, I dont have any sensors.
I am running the MQTTcustom example

I added:
```

#if ( !defined(USE_CC1101) && !defined(USE_SX1276) )
    // #define USE_CC1101
    #define USE_SX1276
#endif

... 
#define PIN_RECEIVER_CS 18   // CS
#define PIN_RECEIVER_IRQ 26  // DIO0
#define PIN_RECEIVER_RST 14  // RESET
#define PIN_RECEIVER_GPIO 35 // DIO1
```

`SENSOR_IDS_EXC` &  `SENSOR_IDS_INC` are both empty

I dont know how I can de bug any further. The serial montior outputs only the loop. No sensor.loop
```
Devices/Bresser/status: online
Devices/Bresser/radio: {"rssi":-104}
Devices/Bresser/status: online
Devices/Bresser/radio: {"rssi":-105}
```

matthias-bs, 2023-06-06:
Hi,

I'm not sure if I understand the situation correctly.
Did you use a different hardware, software version or configuration before?

> Thanks for your time, Matthias
> 
> Well I am not that into hardware. I coupled the GND and Vext, but doesn't help.
You got me wrong here - please use the following code in setup():

```
#define VEXT_EN 21
pinMode(VEXT_EN, OUTPUT);
digitalWrite(VEXT_EN, 0);
```

This sets GPIO21 (aka VEXT_EN) low, which should in turn switch on VEXT (to 3.3V). 

> Somethimes it logs `[D][WeatherSensor.cpp:249] getMessage(): [SX1276] Receive failed: [0]`
> 
> I used your project since the beginning and it worked fine before. But there is much refectoring.

What do you mean by this? You mean the changes I made?

> If I start debug mode, I get some sensor data (the 1234567 sensor). So it can read some of the sensors?
> 
Ah, I think I understand the problem! If you enable `_DEBUG_MQTT_` (in `BresserWeatherSensorMQTTCustom.ino`, you replace the sensor data reception by generation of synthetic data. This is for debugging the MQTT part independently of the sensor reception part.

For all other debug information, please see here:
https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/DEBUG_OUTPUT.md
Please enable debug messages and provide some logging data.

> So it worked before. Can I see which sensors are present, so I can white list them or something?

If both `SENSOR_IDS_EXC` & `SENSOR_IDS_INC` are empty, all available sensors would be received (i.e. whitelisted). Each reception cycle is stopped as soon as `NUM_SENSORS` (defined in `WeatherSensorCfg.h`; default: 1) have been received or a timeout occurred.

To test which sensors are available, you could also switch to BresserWeatherSensorBasic.ino temporarily. This is much simpler and prints sensor data as soon as they are received without trying to collect different parts. If there are unknown sensors within range, this might help to find and eventually blacklist them.

Hope this helps a bit!

Cheers
Matthias

matthias-bs, 2023-06-06:
BTW: Which board did you select in the Arduino IDE for the Heltec lora v2 esp32?

meijer3, 2023-06-06:
Well the board is called Heltec Wifi Lora 32(v2), so that seems right. The logging is on verbose. 

```
[D][WeatherSensor.cpp:135] begin(): [SX1276] Setup complete - awaiting incoming messages...
Devices/Bresser/status: online
Devices/Bresser/radio: {"rssi":-99}
```

I did try your version of this repo around  Aug 2022. I think I used the debug mode (I cannot recall so maybe I didn't in the end). However it did show the right sensordata. The temperature was right. 

You refactored a bit and that is fine. Its more dynamic and cleaner now.

I cannot believe it. But within the hours that I changed the code. The battery died.... 
Now with fresh batteries in the Bresser, it worked again. I noticed that the indoor station lost connection.

It is working now, even without the gio21 pin.

Thank you so much for your time anyways, I am very happy with the code

closed 2023-06-06 by meijer3

matthias-bs, 2023-06-07:
I've made a new release v0.9.0 with pin definitions for Heltec WiFi LoRa 32 V2. I'd be happy if you could test this. Only `WeatherSensorCfg.h` has been modified.

Regarding the Vext topic: I just had a brief look at the [PE4295](https://www.psemi.com/products/rf-switches/low-il-rf-switches/pe4259) RF switch datasheet. I'm pretty sure that power must be provided at Vext. Probably the Arduino board support package already takes care of it. It would be great if you could measure the voltages at GPIO21 and Vext to confirm this. (Be careful not to make a short circuit with your meter's probes! Don't don't try to do this if you're not confident.)

Thank you very much in advance,
Matthias

meijer3, 2023-06-07:
Ofcourse and it is working smooth. Thanks so much!
I do measure a steady 0.62v between Vext and ground. Not sure if that passes the low threshold. 
There is no headerpin gio21

matthias-bs, 2023-06-07:
Thank you!

GPIO21 is connected to JP3 pin 18.
Is there any chance to measure the voltage of the RF switch control signal (U7 - SX1276 pin 20 or U8 PE4259 pin 4)?

There are two ways to control the RF switch:
_**Single-pin control mode** enables the switch to
operate with a single control pin (pin 4) supporting
a +3-volt CMOS logic input, and requires a
dedicated +3-volt power supply connection on
pin 6 (VDD). [...]_

_**Complementary-pin control mode** allows the
switch to operate using complementary control
pins CTRL and /CTRL (pins 4 and 6)._"

So it would make sense if Vext were low and CTRL high.