matthias-bs/BresserWeatherSensorReceiver #58: 7 in 1 not receiving data

13, 2023-06-08:
**Context**
- Which example sketch are you using? 
examples/BresserWeatherSensorMQTT
- Which changes to the code did you make (if any)?
USE_CC1101
- Which sensor are you using?
7in1 7003600GYE000

**Describe the bug**
CC1101 is not receiving data

**To Reproduce**
Steps to reproduce the behavior:
Upload sketch

**Expected behavior**
Receive data

**Debug Log**
If applicable, add log file to help explain your problem.
 - Build log file (enable verbose output in the Arduino IDE: File->Preferences - "Show verbose output during: compile")
```
/home/ben/.arduino15/packages/esp8266/tools/python3/3.7.2-post1/python3 -X utf8 -I /home/ben/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/tools/sizes.py --elf /tmp/arduino_build_487795/BresserWeatherSensorMQTT.ino.elf --path /home/ben/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin --mmu "-DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x8000"
. Variables and constants in RAM (global, static), used 32176 / 80192 bytes (40%)
║   SEGMENT  BYTES    DESCRIPTION
╠══ DATA     1696     initialized variables
╠══ RODATA   2984     constants       
╚══ BSS      27496    zeroed variables
. Instruction RAM (IRAM_ATTR, ICACHE_RAM_ATTR), used 62951 / 65536 bytes (96%)
║   SEGMENT  BYTES    DESCRIPTION
╠══ ICACHE   32768    reserved space for flash instruction cache
╚══ IRAM     30183    code in IRAM    
. Code in flash (default, ICACHE_FLASH_ATTR), used 290992 / 1048576 bytes (27%)
║   SEGMENT  BYTES    DESCRIPTION
╚══ IROM     290992   code in flash   
Bibliothek ESP8266WiFi in Version 1.0 im Ordner: /home/ben/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/libraries/ESP8266WiFi  wird verwendet
Bibliothek MQTT in Version 2.5.1 im Ordner: /home/ben/Arduino/libraries/MQTT  wird verwendet
Bibliothek ArduinoJson in Version 6.21.2 im Ordner: /home/ben/Arduino/libraries/ArduinoJson  wird verwendet
Bibliothek BresserWeatherSensorReceiver in Version 0.9.0 im Ordner: /home/ben/Arduino/libraries/BresserWeatherSensorReceiver  wird verwendet
Bibliothek RadioLib in Version 6.0.0 im Ordner: /home/ben/Arduino/libraries/RadioLib  wird verwendet
Bibliothek SPI in Version 1.0 im Ordner: /home/ben/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/libraries/SPI  wird verwendet
```
 - Serial console log file (see [Debug Output Configuration in Arduino IDE](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/DEBUG_OUTPUT.md))
```
22:22:39.684 -> SDK:2.2.2-dev(38a443e)/Core:3.1.1=30101000/lwIP:STABLE-2_1_3_RELEASE/glue:1.2-65-g06164fb/BearSSL:b024386
22:22:39.684 -> 
22:22:39.684 -> 
22:22:39.684 -> BresserWeatherSensorMQTT 20221024
22:22:39.684 -> 
22:22:39.684 -> Attempting to connect to SSID: muhxnetworkfpm close 1 
22:22:39.684 -> mode : sta(8c:aa:b5:cb:65:bd)
22:22:39.684 -> add if0
22:22:39.784 -> ...scandone
22:22:43.500 -> state: 0 -> 2 (b0)
22:22:43.500 -> .state: 2 -> 3 (0)
22:22:43.533 -> state: 3 -> 5 (10)
22:22:43.533 -> add 0
22:22:43.533 -> aid 10
22:22:43.533 -> cnt 
22:22:43.632 -> 
22:22:43.632 -> connected with muhxnetwork, channel 11
22:22:43.632 -> dhcp client start...
22:22:43.665 -> ip:192.168.22.154,mask:255.255.255.0,gw:192.168.22.6
22:22:44.528 -> connected!
22:22:44.528 -> Checking wifi...
22:22:44.528 -> MQTT connecting... connected!
22:22:44.528 -> ESPWeather-CB65BD/status: online
22:22:44.528 -> [CC1101] Initializing ... 
22:22:44.694 -> success!
22:22:44.694 -> [CC1101] Setup complete - awaiting incoming messages...
22:22:53.521 -> pm open,type:2 0

```

**Embedded Device (please complete the following information):**
 - Arduino IDE Version: 1.8.19
 - ESP32/ESP8266 Board Support Package Version: [e.g. ESP32 2.0.7]
 - Library Version 0.9.0
 - Your Board: Wemos D1 mini
 - Selected Board in the Arduino IDE: [brand/model/version]
 - Your Radio Transceiver Module (if external): cc1101 868
 - **Pinning/Wiring** as in the example

matthias-bs, 2023-06-08:
How long did you wait for data?
Which kind of antenna are you using?
What's the distance between the CC1101 and the weather sensor?
What is the power supply to the ESP8266 D1 mini?

13, 2023-06-08:
I had wiring problems (GDO pins swapped).

Now BresserWheaterSensorBasic receives data, but BresserWheaterSensorMQTT is not publishing.
The topics radio and status are published. 

```
00:51:23.197 -> Checking wifi...
00:51:23.197 -> MQTT connecting... connected!
00:51:23.197 -> ESPWeather-CB65BD/status: online
00:51:23.197 -> [CC1101] Initializing ... 
00:51:23.363 -> success!
00:51:23.363 -> [CC1101] Setup complete - awaiting incoming messages...
00:51:28.142 -> [CC1101] Data: D4 BE FD 19 8D AA 8A 18 AA AA AA AA AA 82 AA 88 AA F8 AA AA AA AA AA AA AA AA 00 
00:51:28.175 -> [CC1101] R [D4] RSSI: -27.0
00:51:28.175 -> find_slot(): ID=0000B327
00:51:28.175 -> find_slot(): Storing into slot #0
00:51:32.191 -> pm open,type:2 0
00:51:40.654 -> [CC1101] Data: D4 BE FD 19 8D AA 8A 18 AA AA AA AA AA 82 AA 88 AA F8 AA AA AA AA AA AA AA AA 00 
00:51:40.654 -> [CC1101] R [D4] RSSI: -26.5
00:51:40.654 -> find_slot(): ID=0000B327
00:51:40.654 -> find_slot(): Storing into slot #0
00:51:53.164 -> [CC1101] Data: D4 8E A0 19 8D AA 9A 18 AA AA AA AA AA 82 AA 88 AA F8 AA AA AA AA AA AA AA AA 00 
00:51:53.164 -> [CC1101] R [D4] RSSI: -25.5
00:51:53.164 -> find_slot(): ID=0000B327
00:51:53.164 -> find_slot(): Storing into slot #0
00:52:05.643 -> [CC1101] Data: D4 BE FD 19 8D AA 8A 18 AA AA AA AA AA 82 AA 88 AA F8 AA AA AA AA AA AA AA AA 00 
00:52:05.643 -> [CC1101] R [D4] RSSI: -27.0
00:52:05.643 -> find_slot(): ID=0000B327
00:52:05.677 -> find_slot(): Storing into slot #0
00:52:18.157 -> [CC1101] Data: D4 8E A0 19 8D AA 9A 18 AA AA AA AA AA 82 AA 88 AA F8 AA AA AA AA AA AA AA AA 00 
00:52:18.157 -> [CC1101] R [D4] RSSI: -27.0
00:52:18.157 -> find_slot(): ID=0000B327
00:52:18.157 -> find_slot(): Storing into slot #0
00:52:30.635 -> [CC1101] Data: D4 46 D5 19 8D AA 9A 18 AA AA AA AA AA 82 AA 88 AA FB AA AA AA AA AA AA AA AA 00 
00:52:30.669 -> [CC1101] R [D4] RSSI: -26.0
00:52:30.669 -> find_slot(): ID=0000B327
00:52:30.669 -> find_slot(): Storing into slot #0
00:52:43.148 -> [CC1101] Data: D4 8E A0 19 8D AA 9A 18 AA AA AA AA AA 82 AA 88 AA F8 AA AA AA AA AA AA AA AA 00 
00:52:43.148 -> [CC1101] R [D4] RSSI: -26.0
00:52:43.148 -> find_slot(): ID=0000B327
00:52:43.148 -> find_slot(): Storing into slot #0
00:52:53.668 -> ESPWeather-CB65BD/status: online
00:52:53.668 -> ESPWeather-CB65BD/radio: {"rssi":-26}
00:52:55.660 -> [CC1101] Data: D4 8E A0 19 8D AA 9A 18 AA AA AA AA AA 82 AA 88 AA F8 AA AA AA AA AA AA AA AA 00 
00:52:55.660 -> [CC1101] R [D4] RSSI: -26.5
00:52:55.660 -> find_slot(): ID=0000B327
00:52:55.660 -> find_slot(): Storing into slot #0
00:53:08.137 -> [CC1101] Data: D4 46 D5 19 8D AA 9A 18 AA AA AA AA AA 82 AA 88 AA FB AA AA AA AA AA AA AA AA 00 
00:53:08.170 -> [CC1101] R [D4] RSSI: -27.5
00:53:08.170 -> find_slot(): ID=0000B327
00:53:08.170 -> find_slot(): Storing into slot #0

```

matthias-bs, 2023-06-09:
Could you please add the following code to the end of `decodeBresser7In1Payload()` before line 935 (`return DECODE_OK;`)  in `WeatherSensor.cpp`?

```
 sensor[slot].valid    = true;
 sensor[slot].complete = true;
```

13, 2023-06-09:
Solved, thank you

```
14:21:27.064 -> ESPWeather/status: online
14:21:27.064 -> ESPWeather/radio: {"rssi":-57.5}
14:21:39.542 -> [CC1101] Data: D4 EA AF 19 8D A3 AA 18 AA AA AA AA AA 82 AA 88 9A FC AA AB A2 AA AA AA AA AA 00 
14:21:39.575 -> [CC1101] R [D4] RSSI: -58.5
14:21:39.575 -> find_slot(): ID=0000B327
14:21:39.575 -> find_slot(): Storing into slot #0
14:21:39.575 -> ESPWeather/data/    B327: {"id":45863,"battery_ok":1,"temp_c":22.3,"humidity":56,"wind_gust":0.0,"wind_avg":0.0,"wind_dir":90.0,"uv":0.0,"light_klx":0.1,"rain":0.0,"rain_d":0.0,"rain_w":0.0,"rain_m":0.0}
14:21:39.575 -> ESPWeather/extra: {"wind_dir_txt":"E","wind_gust_bft":0,"wind_avg_bft":0,"dewpoint_c":13.1,"perceived_temp_c":24.7}
14:21:52.048 -> [CC1101] Data: D4 E0 36 19 8D A3 AA 18 AA AA AA AA AA 82 AA 88 9A FC AA AB 8A AA AA AA AA AA 00 
14:21:52.048 -> [CC1101] R [D4] RSSI: -53.5
14:21:52.048 -> find_slot(): ID=0000B327
14:21:52.048 -> find_slot(): Storing into slot #0
14:22:04.554 -> [CC1101] Data: D4 E0 36 19 8D A3 AA 18 AA AA AA AA AA 82 AA 88 9A FC AA AB 8A AA AA AA AA AA 00 
14:22:04.554 -> [CC1101] R [D4] RSSI: -54.0
14:22:04.554 -> find_slot(): ID=0000B327
14:22:04.554 -> find_slot(): Storing into slot #0
14:22:04.554 -> ESPWeather/data/    B327: {"id":45863,"battery_ok":1,"temp_c":22.3,"humidity":56,"wind_gust":0.0,"wind_avg":0.0,"wind_dir":90.0,"uv":0.0,"light_klx":0.1,"rain":0.0,"rain_d":0.0,"rain_w":0.0,"rain_m":0.0}
14:22:04.587 -> ESPWeather/extra: {"wind_dir_txt":"E","wind_gust_bft":0,"wind_avg_bft":0,"dewpoint_c":13.1,"perceived_temp_c":24.7}
14:22:04.587 -> ESPWeather/status: online
14:22:04.587 -> ESPWeather/radio: {"rssi":-54}
```

closed 2023-06-09 by matthias-bs

magicflofla, 2025-01-04:
I got the same problem with my esp32_dev_module with CC1101 and Bresser Weather Station 7in1.
With the `BresserWeatherSensorBasic` i´m receiving data, with` BresserWeahterSensorMQTT` (and MQTTCustom) i get no data.

```
19:30:38.581 -> [ 42174][V][WeatherSensor.cpp:392] getMessage(): [CC1101] Data: D4 8D 1A 1A 58 C4 AD 2A 18 AA AA AA AA AA BA BC 89 FA ED AA AA AA AA AA AA AA AA 
19:30:38.581 -> [ 42186][D][WeatherSensor.cpp:394] getMessage(): [CC1101] R [D4] RSSI: -74.0
19:30:38.581 -> [ 42193][D][WeatherSensorDecoders.cpp:723] decodeBresser7In1Payload(): Digest check failed - [27B0] vs [0A28] (2D98)
19:31:03.557 -> [ 67175][V][WeatherSensor.cpp:392] getMessage(): [CC1101] Data: D4 83 D9 58 C4 C4 AD 2A 18 AA AA AA AA AA BA BC 89 FA ED AA AA AA AA AA AA AA AA 
19:31:03.601 -> [ 67188][D][WeatherSensor.cpp:394] getMessage(): [CC1101] R [D4] RSSI: -74.0
19:31:03.601 -> [ 67195][D][WeatherSensorDecoders.cpp:723] decodeBresser7In1Payload(): Digest check failed - [2973] vs [142E] (3D5D)
19:31:30.481 -> [ 94093][I][BresserWeatherSensorMQTTCustom.ino:880] loop(): Wetterstation-984364/status: online
19:31:30.481 -> 
19:31:30.481 -> [ 94104][I][BresserWeatherSensorMQTTCustom.ino:787] publishRadio(): Wetterstation-984364/radio: {"rssi":-61.5}
19:31:30.526 -> 
19:31:41.053 -> [104673][V][WeatherSensor.cpp:392] getMessage(): [CC1101] Data: D4 8D 1A 1A 58 C4 AD 2A 18 AA AA AA AA AA BA BC 89 FA ED AA AA AA AA AA AA AA AA 
19:31:41.097 -> [104686][D][WeatherSensor.cpp:394] getMessage(): [CC1101] R [D4] RSSI: -74.0
19:31:41.097 -> [104693][D][WeatherSensorDecoders.cpp:723] decodeBresser7In1Payload(): Digest check failed - [27B0] vs [0A28] (2D98)
19:33:00.509 -> [184115][I][BresserWeatherSensorMQTTCustom.ino:880] loop(): Wetterstation-984364/status: online
19:33:00.509 -> 
19:33:00.509 -> [184126][I][BresserWeatherSensorMQTTCustom.ino:787] publishRadio(): Wetterstation-984364/radio: {"rssi":-61.5}
19:33:00.555 -> 
19:33:08.554 -> [192173][V][WeatherSensor.cpp:392] getMessage(): [CC1101] Data: D4 54 D8 D8 58 C4 AD 2A 18 AA AA AA AA AA BA BC 89 FA ED AA AA AA AA AA AA AA AA 
```

I commentet the not used Sensors and configured the CC1101 Sensor. The rest ist not changed. In the  `WeatherSensor.cpp` i find no `decodeBresser7In1Payload()` but i found it in the `WeatherSensorDecoders.cpp` and think .valid and .complete is already set there. Maybe someone got a tip for me.

matthias-bs, 2025-01-04:
Which version are you using?
Did you try with all sensors enabled (i.e. as in the default code)? I suspect that your sensor is actually using the 6-in-1 protocol. Please have a look at the `BresserWeatherSensorBasic` debug output where you saw the expected results.

BTW, `decodeBresser7In1Payload()` was moved to `WeatherSensorDecoders.cpp` a while ago.

magicflofla, 2025-01-04:
I use 0.30.0 and Arduino IDE 2.3.4
I changed the Basic Version to use only the 7in1 also... (i think it worked)

```
20:32:18.580 -> Id: [    F26E] Typ: [1] Ch: [0] St: [0] Bat: [OK ] RSSI: [ -74.0dBm] Temp: [ 23.3C] Hum: [ 41%] Wmax: [ 0.0m/s] Wavg: [ 0.0m/s] Wdir: [ 78.0deg] Rain: [    1.0mm] UVidx: [0.0] Light: [0.0klx] 
20:32:30.963 -> [ 60478][V][WeatherSensor.cpp:392] getMessage(): [CC1101] Data: D4 0A 4E 58 58 C4 AD 2A 18 AA AA AA AA AA BA BC 89 9A EB AA AA E9 AA AA AA AA AA 
20:32:30.963 -> [ 60491][D][WeatherSensor.cpp:394] getMessage(): [CC1101] R [D4] RSSI: -74.0
20:32:31.011 -> [ 60497][D][WeatherSensorDecoders.cpp:723] decodeBresser7In1Payload(): Digest check failed - [A0E4] vs [DD1F] (7DFB)
20:32:43.478 -> [ 73008][V][WeatherSensor.cpp:392] getMessage(): [CC1101] Data: D4 F5 F4 58 C4 AD 2A 18 AA AA AA AA AA BA BC 89 9A EB AA AA AA AA AA AA AA AA 00 
20:32:43.524 -> [ 73021][D][WeatherSensor.cpp:394] getMessage(): [CC1101] R [D4] RSSI: -62.0
20:32:43.524 -> [ 73028][D][WeatherSensor.h:647] log_message():           Byte #: 00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 
20:32:43.542 -> [ 73040][D][WeatherSensor.h:657] log_message(): De-whitened Data: 5F 5E F2 6E 07 80 B2 00 00 00 00 00 10 16 23 30 41 00 00 00 00 00 00 00 00 AA 
20:32:43.542 -> [ 73053][V][WeatherSensorDecoders.cpp:67] findSlot(): find_slot(): ID=0000F26E
20:32:43.542 -> [ 73060][D][WeatherSensorDecoders.cpp:105] findSlot(): sensor[0]: v=0 id=0x0000F26E t=1 c=0
20:32:43.542 -> [ 73068][V][WeatherSensorDecoders.cpp:130] findSlot(): find_slot(): Storing into slot #0
20:32:43.575 -> Id: [    F26E] Typ: [1] Ch: [0] St: [0] Bat: [OK ] RSSI: [ -62.0dBm] Temp: [ 23.3C] Hum: [ 41%] Wmax: [ 0.0m/s] Wavg: [ 0.0m/s] Wdir: [ 78.0deg] Rain: [    1.0mm] UVidx: [0.0] Light: [0.0klx] 
20:32:55.983 -> [ 85487][V][WeatherSensor.cpp:392] getMessage(): [CC1101] Data: D4 34 9F 58 C4 AD 2A 18 AA AA AA AA AA AA BA BC 89 9A EB AA AA AA AA AA AA AA AA 
20:32:55.983 -> [ 85500][D][WeatherSensor.cpp:394] getMessage(): [CC1101] R [D4] RSSI: -74.0
20:32:55.983 -> [ 85506][D][WeatherSensorDecoders.cpp:723] decodeBresser7In1Payload(): Digest check failed - [9E35] vs [7708] (E93D)
```

matthias-bs, 2025-01-04:
In your MQTT logfile, the digest check fails each time. Did your try to run the sketch a little longer and - if enabled - to increase the timeout?
It seems you are encountering many transmission errors. How does the radio transmission path between the sensor and the receiver look like, i.e. what is the distance and are there any walls or windows in between? What kins of antenna are you using?

magicflofla, 2025-01-04:
The Sensor and Receiver are just cm apart, i will install the sensor if it works.  
What Timeouts should i change? I tried to change them already, but did not really know which one.

matthias-bs, 2025-01-04:
Such a short distance could overload the receiver. Please place transmitter and receiver in different rooms with a wall in between.

magicflofla, 2025-01-05:
Thank you for your help, i got a new problem now. I think i destroyed my cc1101, it looks like an hardware problem. I ordered a new one and try it again if i got it.