matthias-bs/BresserWeatherSensorReceiver #60: 7in1 rain always 0.0

13, 2023-06-12:
**Context**
BresserSensorMQTT

**Describe the bug**
Even if it rains, I receive 0.0.
The original display show the actual value like 4.0

matthias-bs, 2023-06-12:
Hmmm, this part of the code was ported from the rtl_433 by @jorgenavarroortiz.

I can't find anything suspicious, maybe the decoding from raw data is not correct. There is a line commented out, maybe this is what is actually needed:
https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/13f4e46eb7a559f8e87481d777e66957f2f71231/src/WeatherSensor.cpp#LL868C1-L868C101

I do not have this kind sensor sensor for testing.

Maybe you can provide some raw data for investigating?

13, 2023-06-12:
With this modification it works.

```
20:32:11.204 -> find_slot(): Storing into slot #0
20:32:11.204 -> muh/WStation/data/B327: {"id":45863,"ch":0,"battery_ok":1,"temp_c":23.6,"humidity":48,"wind_gust":0.0,"wind_avg":0.0,"wind_dir":66.0,"uv":0.0,"light_klx":0.0,"rain":17.2,"rain_d":1.2,"rain_w":1.2,"rain_m":1.2}
20:32:11.204 -> muh/WStation/extra: {"wind_dir_txt":"ENE","wind_gust_bft":0,"wind_avg_bft":0,"dewpoint_c":12.0,"perceived_temp_c":25.3}
20:32:11.238 -> muh/WStation/status: online
20:32:11.238 -> muh/WStation/radio: {"rssi":-45.5}
20:32:23.678 -> [CC1101] Data: D4 4B BA 19 8D AC CA 18 AA AA AA AA AB D8 AA 89 CA E2 AA AA AA AA AA AA AA AA 00 
20:32:23.711 -> [CC1101] R [D4] RSSI: -47.0
20:32:23.711 -> find_slot(): ID=0000B327
20:32:23.711 -> find_slot(): Storing into slot #0
20:32:36.190 -> [CC1101] Data: D4 4B BA 19 8D AC CA 18 AA AA AA AA AB D8 AA 89 CA E2 AA AA AA AA AA AA AA AA 00 
20:32:36.190 -> [CC1101] R [D4] RSSI: -48.0
20:32:36.190 -> find_slot(): ID=0000B327
20:32:36.190 -> find_slot(): Storing into slot #0
20:32:36.190 -> muh/WStation/data/B327: {"id":45863,"ch":0,"battery_ok":1,"temp_c":23.6,"humidity":48,"wind_gust":0.0,"wind_avg":0.0,"wind_dir":66.0,"uv":0.0,"light_klx":0.0,"rain":17.2,"rain_d":1.2,"rain_w":1.2,"rain_m":1.2}
20:32:36.223 -> muh/WStation/extra: {"wind_dir_txt":"ENE","wind_gust_bft":0,"wind_avg_bft":0,"dewpoint_c":12.0,"perceived_temp_c":25.3}
20:32:48.696 -> [CC1101] Data: D4 4B BA 19 8D AC CA 18 AA AA AA AA AB D8 AA 89 CA E2 AA AA AA AA AA AA AA AA 00 
20:32:48.696 -> [CC1101] R [D4] RSSI: -47.0
20:32:48.696 -> find_slot(): ID=0000B327
20:32:48.696 -> find_slot(): Storing into slot #0
20:32:48.696 -> muh/WStation/status: online
20:32:48.696 -> muh/WStation/radio: {"rssi":-47}
20:33:01.200 -> [CC1101] Data: D4 5F 2F 19 8D AC CA 18 AA AA AA AA AB 3C AA 89 CA E2 AA AA AA AA AA AA AA AA 00 
20:33:01.200 -> [CC1101] R [D4] RSSI: -45.5
20:33:01.200 -> find_slot(): ID=0000B327
20:33:01.200 -> find_slot(): Storing into slot #0
20:33:01.200 -> muh/WStation/data/B327: {"id":45863,"ch":0,"battery_ok":1,"temp_c":23.6,"humidity":48,"wind_gust":0.0,"wind_avg":0.0,"wind_dir":66.0,"uv":0.0,"light_klx":0.0,"rain":19.6,"rain_d":3.6,"rain_w":3.6,"rain_m":3.6}
20:33:01.233 -> muh/WStation/extra: {"wind_dir_txt":"ENE","wind_gust_bft":0,"wind_avg_bft":0,"dewpoint_c":12.0,"perceived_temp_c":25.3}
20:33:13.708 -> [CC1101] Data: D4 5F 2F 19 8D AC CA 18 AA AA AA AA AB 3C AA 89 CA E2 AA AA AA AA AA AA AA AA 00 
20:33:13.708 -> [CC1101] R [D4] RSSI: -47.5
20:33:13.708 -> find_slot(): ID=0000B327
20:33:13.708 -> find_slot(): Storing into slot #0

```

jorgenavarroortiz, 2023-06-12:
Hi,

I think Matthias is right. I remove that line because the decoding was not correct initially, and I was waiting for rain in my city. But later I forgot to check it and correct it. Could you please confirm that it works fine uncommenting that line? Thanks.

Best regards,
Jorge

closed 2023-06-13 by matthias-bs