matthias-bs/BresserWeatherSensorReceiver #64: rain value is still the same and seems not to be updated

trueslator, 2023-07-01:
First of all, greate work! It works just fine but there is a little issue with my rain value, it stays all the time at 15,6.
I bought the Bresser station 5 in 1 (Lot No. 7002510-1274880) from ALDI last week for a very good price. And it works just fine with your WeatherSensorReceiver but the Rain value is not changing, doesn't matter if it's raining or not. The original Display shows correct values for rain.

Checking wifi...
MQTT connecting... connected!
ESPWeather-E8AE68/status: online
ESPWeather-E8AE68/data/37040087: {"id":923009159,"ch":0,"battery_ok":1,"temp_c":19.3,"humidity":72,"wind_gust":0.8,"wind_avg":0.8,"wind_dir":68.0,"rain":15.6,"rain_d":0.0,"rain_w":0.0,"rain_m":0.0}
ESPWeather-E8AE68/extra: {"wind_dir_txt":"ENE","wind_gust_bft":0,"wind_avg_bft":0,"dewpoint_c":14.1,"perceived_temp_c":22.9}
Sleeping for 150000 ms

Any idea where to look at?

matthias-bs, 2023-07-02:
Hi Oliver,

Are you using an ESP8266 or an ESP8266?

Best regards
Matthias

trueslator, 2023-07-02:
Hi Matthias,
I'm using a TTGO LoRa32-OLED. The value changed this morning during rain from 15.6 to 17.6.  Rain_D, Rain_W and Rain_M were on 2 but now back to 0. The Rain value is still on 17.6. I'm reading this json with the IOBROKER and manipulate this value by subtracting by 15.6. I will check this again tomorrow . I thought that there was something still in the memory of the ESP therfore I formated it (with zeros). It seems that this value will be send like this by the wether station itself.
Have a nice evening, regards Oliver

matthias-bs, 2023-07-02:
Hi Oliver,

'rain' is the original value sent by the sensor, which is accumulated until a reset, battery change or (unlikely) overflow occurs.
'rain_d' , 'rain_w' and 'rain_m' are the accumulated values of the current (calendar) day, week and month, respectively. Those values calculated by the ESP and are stored in RTC RAM, i.e. kept during deep sleep mode, but not during power off.
In order to work properly, the time and date has to be set correctly, e.g. using Network Time Protocol.

Regards,
Matthias

trueslator, 2023-07-03:
Hi Matthias,
many thanks for help. I will grab a ladder and try to reset the sensor. I want to use the "rain_d" value to get an idea how much water my garden already got from rain. It seems that all reported issues here are coming from misshandling the ESP (unplugin and plug in again). I had some connectivity issues after deep sleep and to solve it I placed the ESP much closer to the router. I think by resetting the rain_d/w/m values I started to be confused. Thanks again - you did a very good job here.

matthias-bs, 2023-07-17:
Hi, any news on this? Can the issue be closed?

Multiplexer2017, 2023-07-18:
Hi Matthias,

Ich habe zur Zeit das Problem mithilfe von Nodered gelöst.
Ein Test mit einem ESP32 und auch mit Firebeetle zeigte das gleiche Problem.
Vermutlich habe ich irgendeine falsche Library benutzt.
Mit Nodered lese ich um 00:00 Uhr den Gesamtregen und subtrahiere den Wert.
Cool währe, wenn man dem Controller das via MQTT sagen könnte.
Also der Controller müsste mehrere Kanäle senden und man kann via Nodered sagen das „Kanal x“ auf Null gesetzt wird.
So könnte man z.B. Regen pro Stunde, pro Tag, pro Woche, Monat usw. selber bestimmen.

Heute kommt der Gewitter Sensor.  Freue mich schon…

Sorry das ich lieber in Deutsch schreibe.

Danke für das coole Projekt.

Issue kann geschlossen werden.

Dirk..

matthias-bs, 2023-07-18:
Hmmm, ich verstehe nicht wo das Problem liegt... Wenn es mit einem ESP32 auch nicht funktioniert, scheint es ja eher an der Uhrzeit als am (nicht vorhandenen)  RTC RAM zu liegen.
Das Rücksetzen des Regenmessers via MQTT ist in https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/66 geplant. Das kann ich so gestalten, dass die verschiedenen Zähler getrennt voneinander rückgesetzt werden können.

Multiplexer2017, 2023-07-18:
Hi Matthias,

Ich habe heute den Blitz Sensor bekommen.
Auf anhieb hat der auch nicht funktioniert.
Ich bin immer davon ausgegangen wenn ich die Arduino IDE starte und dann Updates vorhanden sind, die ich dann auch installiere auch deine Bresser Libraries upgedatet werden.
Als ich dann meine auf dem Rechner befindlichen Dateien mit denen auf GitHub verglich, habe ich dann Unterschiede gefunden.
Ich habe dann die Dateien von GitHub in meinen Arbeitsordner kopiert und der Blitzsensor wurde sofort erkannt.
Ich glaube, ich habe ständig eine alte Version von dir benutzt.
Ob der Regensensor sich nun selbst zurückstellt, werde ich noch testen.
Tja, ich denke es war mein Unwissen welches dir Kopfzerbrechen bereitet hat.
Eventuell stellst du irgendwann mal für „Nichtprofis“ fertig kompilierte Binärdatein zu verfügung. Im WifiManager lässt sich ja das Wichtigste einstellen.

Sorry nochmal… und danke für deine Arbeit…

Dirk.

matthias-bs, 2023-07-18:
Hallo Dirk,

also in der Arduino IDE sollten schon die Bresser Libraries zum Update angeboten werden. Man muss nur aufpassen, dass das Update eben im Library-Pfad landet. Wen man dann ein Archiv herunterläd und beareitet, kommt man eventuell durcheinander...

So oder so - ich hoffe es funktioniert nun auch bei Dir.
Falls nicht, sende mir gerne den Compiler Output und ein Logfile.

Viele Grüße
Matthias

closed 2023-10-05 by matthias-bs