matthias-bs/BresserWeatherSensorReceiver #65: The rain value does not reset

Multiplexer2017, 2023-07-02:
The rain value does not reset

I have the Bresser 7003600.
My problem is:
Rainfall is not reset.
it seems that the internal clock in the weatherSensorReceiver is not running.

{"id":43567,"ch":0,"battery_ok":1,"temp_c":20.7,"humidity":43,"wind_gust":2.3,"wind_avg":2.1,"wind_dir":258.0,"uv":2.1,"light_klx":33.7,"rain":99.6,"rain_d":2.0,"rain_w":2.0,"rain_m":2.0}

the rain values: "rain_d", "rain_w" , "rain_m" are all the same and get bigger when it rains.

I use: 
BresserWeatherSensorMQTTWifiMgr.ino
Arduino Version: 2.1.0
ESP8266 Wemos D1 mini
CC1101

this project is the reason why i bought the Bresser

thank you very much for the great project

Dirk

matthias-bs, 2023-07-02:
Hi Dirk,

If I remember correctly, the ESP8266 does not have a built-in RTC (as the ESP32 has). Thus, the time and date has to be provided via Network Time Protocol (NTP). Could you please check the debug output to see if the time is set correctly?

Best regards
Matthias

matthias-bs, 2023-07-10:
The reason why an ESP8266 cannot provide the rain statistics if sleep mode is used, is that in the current implementation, previous rain values are stored in RTC RAM, which is only available in the ESP32.

This could be fixed by storing the values in the flash file system.

matthias-bs, 2023-07-18:
Not enough information to debug/resolve.

closed 2023-07-18 by matthias-bs