matthias-bs/BresserWeatherSensorReceiver #67: Pressure

chrkan, 2023-07-07:
hello, thanks for this great project. 
I use a COMFORT WEATHER CENTER WITH 7-IN-1 PROFESSIONAL SENSOR, which can also be read out wonderfully. 
But what I'm missing is the air pressure.
or did I overlook this?

matthias-bs, 2023-07-07:
Hi,

The Bresser weather stations have the pressure sensor integrated in the base station unit, not the outdoor sensor unit. Thus, the pressure signal is not contained in the radio signal.

However, it is relatively easy to add an additional pressure sensor, e.g. a BME280 as in [ESP32-e-Paper-Weather-Display](https://github.com/matthias-bs/ESP32-e-Paper-Weather-Display).

Cheers,
Matthias

matthias-bs, 2023-07-07:
Could you please provide the part number of your Weather Center? I'd like to add it to my list of compatible devices.
Thanks in advance!

chrkan, 2023-07-07:
ok thanks i din't no that.

the sensor is the 7803300
https://www.bresser.de/en/Weather-Time/WLAN-Weather-Stations-Centers/BRESSER-WLAN-Comfort-Weather-Station-with-7-in-1-professional-sensor-and-modern-colour-display.html

closed 2023-07-07 by chrkan

matthias-bs, 2023-07-07:
There are still some remaining questions regarding the 7-in-1 protocol:
1. Is there a startup-bit in the flags, i.e. a bit which indicates a reset or a battery change? This bit would be set after such an event and be cleared after some time (e.g. an hour)
2. Is there a sensor type field? In some aspects, the 7-in-1 sensor and the lightning sensor protocols are similar. Currently they are distinguished by the digest check, but there could be a type field.

It would be great if you could share some raw data - one message shortly after a reset/battery change, the other one after maybe 2 hours.

Thanks again in advance!

chrkan, 2023-07-08:
what is the best way to record this? or is there a sketch for it and I also log the seriall monitor

matthias-bs, 2023-07-08:
Just set the debug level to verbose as described in [DEBUG_OUTPUT.md](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/DEBUG_OUTPUT.md) and copy the contents from the serial monitor.

chrkan, 2023-07-08:
`16:06:34.251 -> [CC1101] Data: D4 C4 D6 3A C5 BD FA 18 AA AA AA AA AB FC AA 98 DA 89 A3 2F EC AF 9A AA AA AA 00 
16:06:34.251 -> [CC1101] R [D4] RSSI: -70.5
16:06:34.251 -> find_slot(): ID=0000906F
16:06:34.251 -> find_slot(): Storing into slot #0
16:06:34.251 -> Attempting MQTT connection...connected
16:06:34.371 -> Id: [    906F] Typ: [0] Battery: [OK ] Ch: [0] Temp: [ 32.7C] Hum: [ 23%] Wind max: [ 0.0m/s] Wind avg: [ 0.0m/s] Wind dir: [175.0deg] Rain: [   15.6mm] Moisture: [--%] UV index: [5.3] Light (Klux): [98.5Klux] RSSI: [-70.5dBm]
16:06:46.761 -> [CC1101] Data: D4 D8 81 3A C5 BD FA 18 AA CA AC AA AB FC AA 98 DA 89 A3 2F EC AF 9A AA AA AA 00 
16:06:46.761 -> [CC1101] R [D4] RSSI: -71.5
16:06:46.761 -> find_slot(): ID=0000906F
16:06:46.761 -> find_slot(): Storing into slot #0
16:06:46.761 -> Id: [    906F] Typ: [0] Battery: [OK ] Ch: [0] Temp: [ 32.7C] Hum: [ 23%] Wind max: [ 0.6m/s] Wind avg: [ 0.6m/s] Wind dir: [175.0deg] Rain: [   15.6mm] Moisture: [--%] UV index: [5.3] Light (Klux): [98.5Klux] RSSI: [-71.5dBm]
16:06:59.266 -> [CC1101] Data: D4 C4 D6 3A C5 BD FA 18 AA AA AA AA AB FC AA 98 DA 89 A3 2F EC AF 9A AA AA AA 00 
16:06:59.266 -> [CC1101] R [D4] RSSI: -71.0
16:06:59.266 -> find_slot(): ID=0000906F
16:06:59.266 -> find_slot(): Storing into slot #0
16:06:59.266 -> Id: [    906F] Typ: [0] Battery: [OK ] Ch: [0] Temp: [ 32.7C] Hum: [ 23%] Wind max: [ 0.0m/s] Wind avg: [ 0.0m/s] Wind dir: [175.0deg] Rain: [   15.6mm] Moisture: [--%] UV index: [5.3] Light (Klux): [98.5Klux] RSSI: [-71.0dBm]
16:07:11.765 -> [CC1101] Data: D4 33 A1 3A C5 BD FA 18 AA AA AA AA AB FC AA 98 DA 89 A3 23 9B AF BA AA AA AA 00 

16:07:11.765 -> [CC1101] R [D4] RSSI: -71.5
16:07:11.765 -> find_slot(): ID=0000906F
16:07:11.765 -> find_slot(): Storing into slot #0
16:07:23.695 -> Id: [    906F] Typ: [0] Battery: [OK ] Ch: [0] Temp: [ 32.7C] Hum: [ 23%] Wind max: [ 0.0m/s] Wind avg: [ 0.0m/s] Wind dir: [175.0deg] Rain: [   15.6mm] Moisture: [--%] UV index: [5.1] Light (Klux): [98.9Klux] RSSI: [-71.5dBm]
16:07:24.284 -> [CC1101] Data: D4 E0 D0 3A C5 BC FA 18 AA AA AA AA AB FC AA 98 2A 8E A3 23 9B AF BA AA AA AA 00 
16:07:24.284 -> [CC1101] R [D4] RSSI: -72.0
16:07:24.284 -> Digest check failed - [4A7A] Vs [1238] (5842)
16:07:24.284 -> Digest check failed - [E0D0] != [C44F]
16:07:24.284 -> Parity wrong at column 0
16:07:36.774 -> [CC1101] Data: D4 12 5A 3A C5 B2 8A 08 EA DA AD AA AB FC AA 98 2A 89 A3 23 9B AF BA AA AA AA 00 
16:07:36.774 -> [CC1101] R [D4] RSSI: -71.0
16:07:36.774 -> Digest check failed - [B8F0] Vs [5FFA] (E70A)
16:07:36.899 -> Digest check failed - [125A] != [FFCD]
16:07:36.899 -> Parity wrong at column 0`

Battery out for 30 sec

`16:09:11.574 -> [CC1101] Data: D4 66 07 30 C5 BD EA 10 AB 2A FA AA AA AA AA 9C 3A 89 AA EE FC 2A AB AA AA AA 00 
16:09:11.574 -> [CC1101] R [D4] RSSI: -78.5
16:09:11.574 -> Digest check failed - [CCAD] Vs [32BD] (FE10)
16:09:11.574 -> Digest check failed - [6607] != [57BE]
16:09:11.574 -> Parity wrong at column 0
16:09:24.065 -> [CC1101] Data: D4 44 AD 3A C5 BD EA 10 AE 2A A2 AA AA AA AA 9F 2A 89 AA EE FC AA AA AA AA AA 00 
16:09:24.065 -> [CC1101] R [D4] RSSI: -74.0
16:09:24.065 -> Digest check failed - [EE07] Vs [AF3B] (413C)
16:09:24.065 -> Digest check failed - [44AD] != [70CF]
16:09:24.065 -> Parity wrong at column 0
16:09:36.558 -> [CC1101] Data: D4 C1 D2 3A C5 BD EA 10 AA 3A A2 AA AA AA AA 9E 3A 89 A3 F2 9E AF FA AA AA AA 00 
16:09:36.558 -> [CC1101] R [D4] RSSI: -75.0
16:09:36.558 -> Digest check failed - [6B78] Vs [767D] (1D05)
16:09:36.661 -> Digest check failed - [C1D2] != [BA47]
16:09:36.661 -> Parity wrong at column 0
16:09:49.060 -> [CC1101] Data: D4 7F 47 3A C5 BD CA 10 AA DA AD AA AA AA AA 9E 2A 89 A3 D2 9E AF FA AA AA AA 40 
16:09:49.060 -> [CC1101] R [D4] RSSI: -72.5
16:09:49.060 -> Digest check failed - [D5ED] Vs [EDAE] (3843)
16:09:49.152 -> Digest check failed - [7F47] != [49C1]
16:09:49.152 -> Parity wrong at column 0
16:10:01.552 -> [CC1101] Data: D4 6E 2D 3A C5 3D EA 10 A2 AE AA AA AA AA AB 9E 2A 8F B3 F2 9E AF FA AA 8A AA 00 
16:10:01.614 -> [CC1101] R [D4] RSSI: -76.0
16:10:01.614 -> Digest check failed - [C487] Vs [F1F] (CB98)
16:10:01.614 -> Digest check failed - [6E2D] != [3ED4]
16:10:01.614 -> Parity wrong at column 0
16:10:14.068 -> [CC1101] Data: D4 29 FE 3A C5 BD EB 14 AA AB AA AA AA AA AA 9E 2A 8E A3 F2 9E AF FA AA A3 AA 00 
16:10:14.068 -> [CC1101] R [D4] RSSI: -74.5
16:10:14.068 -> Digest check failed - [8354] Vs [F910] (7A44)
16:10:14.068 -> Digest check failed - [29FE] != [8254]
16:10:14.068 -> Parity wrong at column 0
16:10:26.568 -> [CC1101] Data: D4 2D FE 3A C5 BD EA 10 AA AA AA AE AA AA AA 9E 2A 8E A3 F2 9E AF FB AA AA AA 00 
16:10:26.568 -> [CC1101] R [D4] RSSI: -73.5
16:10:26.568 -> Digest check failed - [8754] Vs [2430] (A364)
16:10:26.568 -> Digest check failed - [2DFE] != [FA83]
16:10:26.568 -> Parity wrong at column 0
[... 83 lines cut ...]
16:17:56.558 -> find_slot(): Storing into slot #0
16:17:56.659 -> Id: [    906F] Typ: [0] Battery: [OK ] Ch: [0] Temp: [ 33.4C] Hum: [ 23%] Wind max: [ 1.6m/s] Wind avg: [ 1.4m/s] Wind dir: [162.0deg] Rain: [    0.0mm] Moisture: [--%] UV index: [4.2] Light (Klux): [89.2Klux] RSSI: [-75.5dBm]
16:18:09.063 -> [CC1101] Data: D4 55 15 3A CD A2 AA 10 A8 FB A8 AA AA AB AA 99 EA 8E A2 38 8E AE 8A 8A AA AA 00 
16:18:09.063 -> [CC1101] R [D4] RSSI: -76.0
16:18:09.063 -> Digest check failed - [FFBF] Vs [6F8] (F947)
16:18:09.063 -> Digest check failed - [5515] != [5535]
[... cut ...]
16:18:34.052 -> Digest check failed - [21C4] Vs [51A1] (7065)
16:18:34.139 -> Digest check failed - [8B6E] != [2359]
16:18:34.139 -> Parity wrong at column 0
16:18:46.584 -> [CC1101] Data: D4 59 6D 3A C5 BC 8A 10 AB DA BC AA AA AA AA 99 FA 8E A2 38 8E AE 8A AA AA 2A 00 
16:18:46.584 -> [CC1101] R [D4] RSSI: -76.5
16:18:46.584 -> Digest check failed - [F3C7] Vs [1E37] (EDF0)
16:18:46.584 -> Digest check failed - [596D] != [2490]
16:18:46.584 -> Parity wrong at column 0
`

I'll check the logs again in a few hours

chrkan, 2023-07-08:
`
18:35:02.952 -> Id: [    906F] Typ: [0] Battery: [OK ] Ch: [0] Temp: [ 33.4C] Hum: [ 24%] Wind max: [ 0.7m/s] Wind avg: [ 0.7m/s] Wind dir: [  9.0deg] Rain: [    0.0mm] Moisture: [--%] UV index: [2.1] Light (Klux): [39.7Klux] RSSI: [-69.5dBm]
18:35:15.425 -> [CC1101] Data: D4 94 03 3A C5 AD 2A 18 AA 3A A3 AA AA AA AA 99 EA 8E A9 3D 9D A8 BA AA AA AA 00 
18:35:15.425 -> [CC1101] R [D4] RSSI: -71.5
18:35:15.425 -> find_slot(): ID=0000906F
18:35:15.888 -> find_slot(): Storing into slot #0
18:35:15.888 -> Id: [    906F] Typ: [0] Battery: [OK ] Ch: [0] Temp: [ 33.4C] Hum: [ 24%] Wind max: [ 0.9m/s] Wind avg: [ 0.9m/s] Wind dir: [ 78.0deg] Rain: [    0.0mm] Moisture: [--%] UV index: [2.1] Light (Klux): [39.7Klux] RSSI: [-71.5dBm]
18:35:27.926 -> [CC1101] Data: D4 9F 69 3A C5 A9 CA 18 AB CA BF AA AA AA AA 99 EA 8E A9 2B C3 AB 7A AA AA AA 00 
18:35:27.926 -> [CC1101] R [D4] RSSI: -74.0
18:35:27.926 -> Digest check failed - [35C3] Vs [459F] (705C)
18:35:28.303 -> Digest check failed - [9F69] != [95DC]
18:35:28.303 -> Parity wrong at column 0
18:35:40.439 -> [CC1101] Data: D4 5F 46 3A C5 A9 CA 18 AB DA BC AA AA AA AA 99 EA 8E A9 2B C3 AB 3A AA AA AA 00 
18:35:40.439 -> [CC1101] R [D4] RSSI: -70.5
18:35:40.439 -> find_slot(): ID=0000906F
18:35:40.439 -> find_slot(): Storing into slot #0
18:35:40.439 -> Id: [    906F] Typ: [0] Battery: [OK ] Ch: [0] Temp: [ 33.4C] Hum: [ 24%] Wind max: [ 1.7m/s] Wind avg: [ 1.6m/s] Wind dir: [ 36.0deg] Rain: [    0.0mm] Moisture: [--%] UV index: [1.9] Light (Klux): [38.2Klux] RSSI: [-70.5dBm]
18:35:52.955 -> [CC1101] Data: D4 18 95 3A C5 A9 CA 18 AB DA BC AA AA AA AA 99 EA 8F A9 2B C3 AB 3A AA AA AA 00 
18:35:52.955 -> [CC1101] R [D4] RSSI: -71.5
18:35:52.955 -> find_slot(): ID=0000906F
18:35:52.955 -> find_slot(): Storing into slot #0
18:35:52.955 -> Id: [    906F] Typ: [0] Battery: [OK ] Ch: [0] Temp: [ 33.4C] Hum: [ 25%] Wind max: [ 1.7m/s] Wind avg: [ 1.6m/s] Wind dir: [ 36.0deg] Rain: [    0.0mm] Moisture: [--%] UV index: [1.9] Light (Klux): [38.2Klux] RSSI: [-71.5dBm]
18:36:05.441 -> [CC1101] Data: D4 A1 F3 3A C5 A9 CA 18 AB 9A B9 AA AA AA AA 99 EA 8F A9 2B C3 AB 3A AA AA AA 00 
18:36:05.441 -> [CC1101] R [D4] RSSI: -68.5
18:36:05.441 -> find_slot(): ID=0000906F
18:36:05.441 -> find_slot(): Storing into slot #0
18:36:05.441 -> Id: [    906F] Typ: [0] Battery: [OK ] Ch: [0] Temp: [ 33.4C] Hum: [ 25%] Wind max: [ 1.3m/s] Wind avg: [ 1.3m/s] Wind dir: [ 36.0deg] Rain: [    0.0mm] Moisture: [--%] UV index: [1.9] Light (Klux): [38.2Klux] RSSI: [-68.5dBm]
18:36:17.922 -> [CC1101] Data: D4 2C 05 3A C5 A9 CA 18 AB 8A B8 AA AA AA AA 99 EA 8F A9 2B C3 AB 3A AA AA AA 00 
18:36:18.132 -> [CC1101] R [D4] RSSI: -69.5
18:36:18.132 -> find_slot(): ID=0000906F
18:36:18.132 -> find_slot(): Storing into slot #0
18:36:18.132 -> Id: [    906F] Typ: [0] Battery: [OK ] Ch: [0] Temp: [ 33.4C] Hum: [ 25%] Wind max: [ 1.2m/s] Wind avg: [ 1.2m/s] Wind dir: [ 36.0deg] Rain: [    0.0mm] Moisture: [--%] UV index: [1.9] Light (Klux): [38.2Klux] RSSI: [-69.5dBm]
18:36:30.421 -> [CC1101] Data: D4 3D F7 3A C5 AE 8A 18 AB BA BB AA AA AA AA 99 FA 8F A9 E3 9F AB DA AA AA AA 00 
18:36:30.548 -> [CC1101] R [D4] RSSI: -69.5
18:36:30.548 -> find_slot(): ID=0000906F
18:36:30.548 -> find_slot(): Storing into slot #0
18:36:30.548 -> Id: [    906F] Typ: [0] Battery: [OK ] Ch: [0] Temp: [ 33.5C] Hum: [ 25%] Wind max: [ 1.1m/s] Wind avg: [ 1.1m/s] Wind dir: [ 42.0deg] Rain: [    0.0mm] Moisture: [--%] UV index: [1.7] Light (Klux): [34.9Klux] RSSI: [-69.5dBm]
18:36:42.939 -> [CC1101] Data: D4 3D F7 3A C5 AE 8A 18 AB BA BB AA AA AA AA 99 FA 8F A9 E3 9F AB DA AA AA AA 00 
18:36:42.939 -> [CC1101] R [D4] RSSI: -69.0
18:36:42.939 -> find_slot(): ID=0000906F
18:36:42.939 -> find_slot(): Storing into slot #0
18:36:42.939 -> Id: [    906F] Typ: [0] Battery: [OK ] Ch: [0] Temp: [ 33.5C] Hum: [ 25%] Wind max: [ 1.1m/s] Wind avg: [ 1.1m/s] Wind dir: [ 42.0deg] Rain: [    0.0mm] Moisture: [--%] UV index: [1.7] Light (Klux): [34.9Klux] RSSI: [-69.0dBm]
18:36:55.418 -> [CC1101] Data: D4 65 C4 3A C5 AE 8A 18 AB 9A B9 AA AA AA AA 99 EA 8F A9 E3 9F AB DA AA AA AA 00 
18:36:55.548 -> [CC1101] R [D4] RSSI: -68.5
18:36:55.548 -> find_slot(): ID=0000906F
18:36:55.548 -> find_slot(): Storing into slot #0
18:36:55.548 -> Id: [    906F] Typ: [0] Battery: [OK ] Ch: [0] Temp: [ 33.4C] Hum: [ 25%] Wind max: [ 1.3m/s] Wind avg: [ 1.3m/s] Wind dir: [ 42.0deg] Rain: [    0.0mm] Moisture: [--%] UV index: [1.7] Light (Klux): [34.9Klux] RSSI: [-68.5dBm]
18:37:07.934 -> [CC1101] Data: D4 6E 09 3A C5 AE 8A 18 AB BA BB AA AA AA AA 99 EA 8F A9 E3 9F AB DA AA AA AA 00 
18:37:07.934 -> [CC1101] R [D4] RSSI: -71.0
18:37:07.934 -> find_slot(): ID=0000906F
18:37:07.934 -> find_slot(): Storing into slot #0
18:37:08.447 -> Id: [    906F] Typ: [0] Battery: [OK ] Ch: [0] Temp: [ 33.4C] Hum: [ 25%] Wind max: [ 1.1m/s] Wind avg: [ 1.1m/s] Wind dir: [ 42.0deg] Rain: [    0.0mm] Moisture: [--%] UV index: [1.7] Light (Klux): [34.9Klux] RSSI: [-71.0dBm]
18:37:20.438 -> [CC1101] Data: D4 43 E9 3A C5 AE 8A 18 AB 8A B8 AA AA AA AA 99 9A 8F A9 E3 9F AB DA AA AA AA 00 
18:37:20.438 -> [CC1101] R [D4] RSSI: -75.0
18:37:20.438 -> find_slot(): ID=0000906F
18:37:20.438 -> find_slot(): Storing into slot #0
18:37:20.438 -> Id: [    906F] Typ: [0] Battery: [OK ] Ch: [0] Temp: [ 33.3C] Hum: [ 25%] Wind max: [ 1.2m/s] Wind avg: [ 1.2m/s] Wind dir: [ 42.0deg] Rain: [    0.0mm] Moisture: [--%] UV index: [1.7] Light (Klux): [34.9Klux] RSSI: [-75.0dBm]
[... cut ...]
18:39:25.553 -> Id: [    906F] Typ: [0] Battery: [OK ] Ch: [0] Temp: [ 33.3C] Hum: [ 24%] Wind max: [ 0.0m/s] Wind avg: [ 0.0m/s] Wind dir: [ 46.0deg] Rain: [    0.0mm] Moisture: [--%] UV index: [1.5] Light (Klux): [32.4Klux] RSSI: [-76.0dBm]
18:39:37.948 -> [CC1101] Data: D4 5F A1 3A E5 AE CA 18 AA AA AA AA AA AA AA 99 EA 8E AD F8 EF AB 3A AA AA AE 00 
18:39:37.948 -> [CC1101] R [D4] RSSI: -73.0
18:39:37.948 -> Digest check failed - [F50B] Vs [5087] (A58C)
18:39:37.948 -> Digest check failed - [5FA1] != [2071]
18:39:37.948 -> Parity wrong at column 0
`

matthias-bs, 2023-07-08:
Thanks a lot! I think I've found what I was looking for!

```
DIGEST:8h8h ID?8h8h WDIR:8h4h 4h 8h WGUST:8h.4h WAVG:8h.4h RAIN:8h8h4h.4h RAIN?:8h TEMP:8h.4hC FLAGS?:4h HUM:8h% LIGHT:8h4h,8h4hKL UV:8h.4h TRAILER:8h8h8h4h

         00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25
Data: D4 C4 D6 3A C5 BD FA 18 AA AA AA AA AB FC AA 98 DA 89 A3 2F EC AF 9A AA AA AA 00

Data: D4 50 04 3A C5 B2 CA 10 A8 8A 8A AA AA AA AA 99 EA 8E A2 38 8E AE 8A AA AA AA 00

Data: D4 2C 95 3A C5 AE CA 18 AA AA AA AA AA AA AA 99 9A 8E A9 8E BA AB FA AA AA AA 00
         -- -- DIGEST
               -- -- ID 
                     -- - DIR 
                         - ?
                       >>> -- bits 7:4 -> type, bit 3 -> startup (0x10 xor 0xAA = 0xBA; 0x18 xor 0xAA = B2) <<<
                              -- - WGUST
                                  - -- WAVG
                                       -- -- -- RAIN
                                                -- RAIN ?
                                                   -- - TEMP
                                                       - FLAGS
                                                         -- HUM
                                                            -- -- -- LIGHT
                                                                     -- - UV
                                                                         - -- -- -- TRAILER
```

The raw data is printed before whitening, so each byte still has to be xor'ed with 0xAA.

chrkan, 2023-07-09:
Nice, Another thing I noticed is the rain value resets after removing the battery

matthias-bs, 2023-07-09:
The sensor battery? Which rain value do you mean? The sensor's or some value calculated by the sketch?

chrkan, 2023-07-09:
Yes, after I took the batteries out of the sensor, the rain_mm value is back to 0 before it was 15.5mm.

Ja es ist wie du in dem post schreibst einfach ein zähler