matthias-bs/BresserWeatherSensorReceiver #79: Find checksum algorithm of Bresser Water Leakage Sensor 7009975

matthias-bs, 2023-08-05:
A basic decoder for the [Bresser Water Leakage Sensor](https://www.bresser.de/en/Weather-Time/Accessories-bresser-2/Sensors/BRESSER-Water-detector.html) has been implemented, see the following [Bitbench](https://triq.net/bitbench#c=%5BBresser%20Water%20Leak%20Sensor%2C%20PN%207009975%5D&c=C7%2070%2035%2097%2004%2008%2057%2070%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20%5BCH7%5D&c=DF%207D%2036%2049%2027%2009%2056%2070%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20%5BCH6%5D&c=9E%2030%2079%2084%2033%2006%2055%2070%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FF%20FD%20DF%20FF%20BF%20FF%20DF%20FF%20FF%20%5BCH5%5D&c=37%20D8%2057%2019%2073%2002%2051%2070%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FF%20FF%20FF%20FF%20FF%20BF%20FF%20EF%20FB%20%5Bset%20CH4%2C%20received%20CH1%20-%3E%20switch%20not%20positioned%20correctly%5D&c=E2%20C8%2068%2027%2091%2024%2054%2070%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20%5BCH4%5D&c=B3%20DA%2055%2057%2017%2040%2053%2070%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20FB%20%5BCH3%5D&c=37%20FA%2084%2073%2003%2002%2052%2070%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FF%20FF%20FF%20DF%20FF%20FF%20FF%20FF%20FF%20%5BCH2%5D&c=27%20F3%2080%2002%2052%2088%2051%2070%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FF%20FF%20FF%20FF%20FF%20DF%20FF%20FF%20FF%20%5BCH1%5D&c=A6%20FB%2080%2002%2052%2088%2059%2070%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FD%20F7%20FF%20FF%20BF%20FF%20FF%20FF%20FF%20%5BCH1%2BNSTARTUP%5D&c=A6%20FB%2080%2002%2052%2088%2059%20B0%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FF%20FF%20FF%20FD%20FF%20F7%20FF%20FF%20FF%20%5BCH1%2BNSTARTUP%2BALARM%5D&c=A6%20FB%2080%2002%2052%2088%2059%2070%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FF%20FF%20BF%20F7%20F7%20FD%207F%20FF%20FF%20%5BCH1%2BNSTARTUP%5D&c=A6%20FB%2080%2002%2052%2088%2059%2070%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FF%20FF%20FF%20FF%20FD%20EF%20FF%20FF%20FF&c=A6%20FB%2080%2002%2052%2088%2059%2070%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20EF&c=A6%20FB%2080%2002%2052%2088%2059%2070%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20FF&c=A6%20FB%2080%2002%2052%2088%2059%2070%2000%2000%2000%2000%2000%2000%2000%2000%2002%20FF%205F%20BF%20FF%20FF%20FF%20FF%20FB%2075&c=A6%20FB%2080%2002%2052%2088%2059%2070%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FF%20FF%20F7%20FF%20FF%20FB%20FF%20FF%20FF&c=A6%20FB%2080%2002%2052%2088%2059%2070%2000%2000%2000%2000%2000%2000%2000%2000%2001%20FF%20F7%20FE%20FB%20FF%20FF%20BF%20FE%20DF&c=A6%20FB%2080%2002%2052%2088%2059%2070%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20FF&c=%5BReset%5D&c=C0%2010%2036%2079%2037%2009%2051%2070%2000%2000%2000%2000%2000%2000%2000%2000%2001%201E%20FD%20FD%20FF%20FF%20FF%20DF%20FF%20FF%20%5BCH1%5D&c=C0%2010%2036%2079%2037%2009%2051%20B0%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FE%20FD%20FF%20AF%20FF%20FF%20FF%20FF%20FD%20%5BCH1%2BALARM%5D&c=%5BReset%5D&c=71%209C%2054%2081%2072%2009%2051%2040%2000%2000%2000%2000%2000%2000%2000%2000%200F%20FF%20FF%20FF%20FF%20FF%20FF%20DF%20FF%20FE%20%5BCH1%2BBATT_LO%5D&c=71%209C%2054%2081%2072%2009%2051%2040%2000%2000%2000%2000%2000%2000%2000%2000%200F%20FE%20FF%20FF%20FF%20FF%20FB%20FF%20FF%20FF&c=71%209C%2054%2081%2072%2009%2051%2040%2000%2000%2000%2000%2000%2000%2000%2000%2007%20FD%20F7%20FF%20DF%20FF%20FF%20DF%20FF%20FF&c=71%209C%2054%2081%2072%2009%2051%2080%2000%2000%2000%2000%2000%2000%2000%2000%201F%20FF%20FF%20F7%20FF%20FF%20FF%20FF%20FF%20FF%20%5BCH1%2BBATT_LO%2BALARM%5D&c=F0%2094%2054%2081%2072%2009%2059%2040%2000%2000%2000%2000%2000%2000%2000%2000%200F%20FF%20DF%20FF%20FF%20FF%20FF%20BF%20FD%20F7%20%5BCH1%2BBATT_LO%2BNSTARTUP%5D&c=F0%2094%2054%2081%2072%2009%2059%2080%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FF%20B7%20FF%20ED%20FF%20FF%20FF%20DF%20FF%20%5BCH1%2BBATT_LO%2BNSTARTUP%2BALARM%5D&f=hhhh%20ID%3Ahhhhhhhh%20TYPE%3A4d%20NSTARTUP%3Ab%20CH%3A3d%20ALARM%3Ab%20NALARM%3Ab%20BATT%3Abb%20FLAGS%3Abbbb%20hhhhhhhhhhhhhhhhhhhhhhhhhhhhhhhhhhhh&cw=4) and
https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/77.

The decoder is based on [bresser_6in1.c](https://github.com/merbanan/rtl_433/blob/master/src/devices/bresser_6in1.c). The ID seems to change on every reset/battery change. The first two bytes are presumably a checksum/crc/digest, but the algorithm is currently unknown. Several attempts have been made, but to no avail.

Any help is appreciated.

amotl, 2023-08-26:
Dear Matthias,

it would be very much coincidental, but maybe source code for decoding data from the Govee Water Leak Detector H5054 helps in one way or another?

It says:
> - Last byte contains the parity bits in index 2-6 (101PPPP1).
  The parity checksum using CRC8 against the first 5 bytes
>
> -- https://github.com/NorthernMan54/rtl_433_ESP/blob/v0.3.0/contrib/govee.c

It is most likely nonsense, as it will be a different device, using a completely different protocol, I did not investigate about it yet, and you most probably know about this implementation already. However, there is always a chance for the right rubberducking at the right time, so apologies for the noise.

With kind regards,
Andreas.

matthias-bs, 2023-08-27:
Dear Andreas,

thanks for the hint, I'll check this. It is quite common that these kind of sensors are rebranded. I already checked for possible similarities with other leakage sensors in the [merbanan/rtl_433](https://github.com/merbanan/rtl_433) project, but without any luck.

Best regards
Matthias

matthias-bs, 2023-10-07:
As discovered by @MaCH, a CRC16/XMODEM  calculation is applied to the 5 bytes of the messages following the 2 bytes with the original CRC.

See
https://github.com/merbanan/rtl_433/issues/2576#issuecomment-1751134478

@MaCH: Thanks a lot for figuring this out!

closed 2023-10-07 by matthias-bs