matthias-bs/BresserWeatherSensorReceiver #80: Support for Heltec Wireless Stick V3 with SX1262

mothra78, 2023-08-13:
Hi, I'd been looking to port your library to use on the Heltec Wireless Stick V3 I bought but recognised it has a sx1262 radio.

I have started to add code to init the SPI and sx1262 but cannot recieve any messages.
Any help would be welcome. 

I have bought a CC1101 to try with an other ESP32

matthias-bs, 2023-08-14:
Hi, did you check [RadioLib](https://github.com/jgromes/RadioLib/tree/master/src/modules) and enable RadioLib's debug output?

matthias-bs, 2023-10-30:
Closed due to inactivity.

phixom, 2023-11-01:
I've also bought this stick. It is possible to add the support?

matthias-bs, 2023-11-01:
@phixom If this is feasible and how much effort it is depends only on RadioLib and how much the API differs between SX1276 and SX1262. (see https://github.com/jgromes/RadioLib/tree/master/src/modules/SX126x). From the first glance I'd say it will be easy, but you never know...

I am not inclined to add this Heltec device to my ever growing zoo of ESP32 boards.

Are you willing to do the testing if I provide the code changes? I think a few iterations will be required. You should also have enough experience to provide some log files.

matthias-bs, 2023-11-01:
Could you please give this a try?
https://github.com/matthias-bs/BresserWeatherSensorReceiver/tree/feature-sx1262

matthias-bs, 2023-11-07:
I would be helpful if anyone with this board could try this:
* Provide compile output of [BresserWeatherSensorBasic.ino](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/examples/BresserWeatherSensorBasic/BresserWeatherSensorBasic.ino) with board "Heltec Wireless Stick" and Core Debug Level: "Debug"
* Log file with one our more supported sensors within range

Thanks!

mhofner21, 2023-11-12:
Hello Matthias,

I'm a complete beginner but have been following your posts with great interest for several months. I have a Heltec Wireless Stick V3 and would be willing to test it. Currently, I'm still working my way through your guides, trying to get everything up and running. I would love to set up a LoRa weather station with your code soon. Oh, and I'm also from Germany.

Best regards,
Matthias

matthias-bs, 2023-11-18:
Hi Matthias,

welcome to the project. Please let me know if you have any questions.

Best regards,
Matthias

mhofner21, 2023-11-18:
Hi Matthias!

Thank you for your response.

Currently, I'm facing some issues with the Heltec Wireless Stick V3. However, these problems are not related to your instructions but rather to Heltec. It's currently impossible for me to load the correct LoRa libraries under MacOS using the Arduino IDE. I keep encountering compilation errors.

While the Heltec Wireless Stick V3 is currently affordable in Germany, it seems unsuitable for me as a beginner.

I'll get in touch as soon as I manage to get it up and running.

Thanks!

evon800c, 2023-11-21:
Hi,

I have a WiFi LoRa 32(V3) board (https://heltec.org/project/wifi-lora-32-v3/) and a Bresser 5-in-1 7002511 available for my experiments.
At the moment I have been able to see the station data with an RTL-SDR USB stick but not with the Heltec indicated above.
if you need a tester, go ahead and "use me".

Davide

matthias-bs, 2023-11-21:
@evon800c 
Hi Davide,

Please try https://github.com/matthias-bs/BresserWeatherSensorReceiver/tree/feature-heltec-wifi-lora-32-v3 with board "Heltec WiFi LoRa 32(V3)" and Core Debug Level: "Debug".

Please provide your compiler and serial console output.

I used the following information:
* https://github.com/espressif/arduino-esp32/blob/master/variants/heltec_wifi_lora_32_V3/pins_arduino.h
* https://github.com/jgromes/RadioLib/blob/master/examples/SX126x/SX126x_Receive_Blocking/SX126x_Receive_Blocking.ino
* https://resource.heltec.cn/download/WiFi_LoRa32_V3/HTIT-WB32LA(F)_V3_Schematic_Diagram.pdf

There is an ambiguity between the schematic (DIO1) and `pins_arduino.h` (D0) - let's see...

Good luck!

Cheers
Matthias

evon800c, 2023-11-22:
Hi,

I'm not very expert on these topics, so I hope not to say too many wrong things, but I agree with you on the FQBN problem and no less on the pin indications.
I personally have a lot of difficulty finding correct information, but it seems to me that the information provided by Heltec is not.

While browsing I found these indications

// SX1262 has the following connections:
// NSS pin: 8
// DIO1 pin: 14
// NRST pin: 12
// BUSY pin: 13

and

static const uint8_t SS_LoRa = 8;
static const uint8_t RST_LoRa = 12;
static const uint8_t BUSY_LoRa = 13;
static const uint8_t DIO1_LoRa = 14;

I recommend these comparison links:
- [output.txt](https://github.com/matthias-bs/BresserWeatherSensorReceiver/files/13444961/output.txt)
- https://github.com/jgromes/RadioLib/issues/706

From what I have been able to observe, the pins with which the sx1262 radio module on this board initializes are 8,14,12,13 (cs, irq, rst, gpio).

With these pins your example "BresserWeatherSensorBasic.ino" initializes the radio module and starts listening but stops there, no other output on serial.

I also had to make some changes to the library code, adding:
- ARDUINO_heltec_wifi_lora_32_V3 -> in case it is interpreted differently
- WIFI_LoRa_32_V3 -> in arduino-ide I see this FQBN otherwise the SX1262 module is not set

I also inserted the same entries in the pin definition part, inserting (8,14,12,13)

Let me know if it's ok and how I can continue to support you.

Davide

files:
[build.zip](https://github.com/matthias-bs/BresserWeatherSensorReceiver/files/13444963/build.zip)
[output.txt](https://github.com/matthias-bs/BresserWeatherSensorReceiver/files/13444964/output.txt)
[serial_monitor.txt](https://github.com/matthias-bs/BresserWeatherSensorReceiver/files/13444965/serial_monitor.txt)

matthias-bs, 2023-11-24:
Hi,

after reading the forum article you mentioned above, I decided not to buy a Heltec board anytime soon. The documentation / libraries are well... too sloppy in my eyes.

It seems (according to your compiler output) you were trying the board definitions by Heltec (from their JSON file) while I am using the plain espressif/arduino-esp32 stuff, i.e. "esp32 by Espressif Systems" V2.0.14 in the Boards Manager. Please let's stick with my approach in the first step.

With this, the compiler will use **-DARDUINO_heltec_wifi_32_lora_V3** and apply the correct (?) pin definitions from [https://github.com/espressif/arduino-esp32/blob/master/variants/heltec_wifi_lora_32_V3/pins_arduino.h](https://github.com/espressif/arduino-esp32/blob/master/variants/heltec_wifi_lora_32_V3/pins_arduino.h).

My code in [src/WeatherSensorCfg.h](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/src/WeatherSensorCfg.h) uses the define mentioned above to set the appropriate pin definitions and to set `#define USE_SX1262`, which finally leads to creation of the radio object in [WeatherSensor.cpp](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/feature-heltec-wifi-lora-32-v3/src/WeatherSensor.cpp):
```
static SX1262 radio = new Module(PIN_RECEIVER_CS, PIN_RECEIVER_IRQ, PIN_RECEIVER_RST, PIN_RECEIVER_GPIO);
```

The `#pragma messages` in the compiler output give some indication if this works correctly.

Of course, you have to use the **[feature-heltec-wifi-lora-32-v3](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/feature-heltec-wifi-lora-32-v3/) branch** of the project!

If the code still hangs while waiting for messages, there are at least two possible error sources:
* Wrong pin definitions
* RadioLib problem
* Bad radio signal

As discussed in https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/91, the blocking variant of RadioLib receive() seems to be flaky, at least with the CC1101. Maybe this is also the case with the SX1262.

I'll try the interrupt variant as soon as I have time.

Cheers, 
Matthias

matthias-bs, 2023-11-25:
I just changed the code to interrupt receive mode. Please try again.

evon800c, 2023-11-25:
Hi,

I reset the Arduino-IDE, I hope now to be aligned with you in terms of libraries.
I installed your latest library version with the "interrupt receive mode" feature and used "BresserWeatherSensorBasic.ino" for testing.

I tried both with RadioLib version 6.1.0 and 6.2.0, but the result is the same. It listens, but then reboot following a core panic.

I attach the outputs taken and the bulk, I hope they help you.

[20231125.zip](https://github.com/matthias-bs/BresserWeatherSensorReceiver/files/13464978/20231125.zip)

have you had the opportunity to test your library for the combination wemos d1 mini 8266 and the EByte E32-868t20d radio modulator, it should be a sx1276

matthias-bs, 2023-11-25:
Hi,

The compile output looks perfectly fine! I fed the .elf file and the exception dump into the Stack Trace Decoder https://maximeborges.github.io/esp-stacktrace-decoder/, but I did not get any output at all.

I'll try another exception decoder.

I never used the EByte E32-868t20d , but the E22 device has an interface which is not suitable for using the integrated radio modem in FSK mode (and also the LoRa mode can only be used in a pre-defined configuration).

matthias-bs, 2023-11-25:
With https://github.com/dankeboy36/esp-exception-decoder:
```
ESP Exception Decoder
Sketch: BresserWeatherSensorBasic FQBN: esp32:esp32:heltec_wifi_lora_32_V3

Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x42003515  PS      : 0x00060830  A0      : 0x82001eac  A1      : 0x3fcebf00  
A2      : 0x00000000  A3      : 0x3fc95834  A4      : 0x00000000  A5      : 0xc6cee261  
A6      : 0x3589148e  A7      : 0x02473389  A8      : 0x82003515  A9      : 0x3fcebed0  
A10     : 0xc2950000  A11     : 0x80000000  A12     : 0x000000ff  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x3c030783  SAR     : 0x0000001d  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xc6cee2a1  LBEG    : 0x40056f08  LEND    : 0x40056f12  LCOUNT  : 0xffffffff  

Backtrace: 0x42003512:0x3fcebf00 0x42001ea9:0x3fcebf60 0x42009395:0x764a4881 |<-CORRUPTED

PC: 0x42003515:  is in Module::SPIreadStream(unsigned char*, unsigned char, unsigned char*, unsigned int, bool, bool) (/home/mp/Arduino/libraries/RadioLib/src/Module.cpp:221).
EXCVADDR: 0xc6cee2a1

Decoding stack results
0x42003512:  is in Module::SPIreadStream(unsigned char*, unsigned char, unsigned char*, unsigned int, bool, bool) (/home/mp/Arduino/libraries/RadioLib/src/Module.cpp:221).
0x42001ea9:  is in WeatherSensor::findSlot(unsigned int, DecodeStatus*) (/home/mp/Arduino/libraries/BresserWeatherSensorReceiver/src/WeatherSensor.cpp:438).
0x42009395: ledc_timer_rst at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/driver/ledc.c:249
```

evon800c, 2023-11-25:
HI,

a little while ago I tried to change the antenna and I saw some different messages.
They don't always come out but every now and then they do.
I don't know if this can help you.

22:48:31.576 -> Rebooting...
22:48:31.576 -> �ESP-ROM:esp32s3-20210327
22:48:31.616 -> Build:Mar 27 2021
22:48:31.616 -> rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
22:48:31.616 -> Saved PC:0x4202c832
22:48:31.616 -> SPIWP:0xee
22:48:31.616 -> mode:DIO, clock div:1
22:48:31.616 -> load:0x3fce3808,len:0x44c
22:48:31.616 -> load:0x403c9700,len:0xbd8
22:48:31.616 -> load:0x403cc700,len:0x2a80
22:48:31.616 -> entry 0x403c98d0
22:48:31.735 -> Starting execution...
22:48:31.735 -> [   127][D][WeatherSensor.cpp:124] begin(): [SX1262] Initializing ... 
22:48:31.933 -> [   359][D][WeatherSensor.cpp:139] begin(): success!
22:48:31.933 -> [   360][D][WeatherSensor.cpp:178] begin(): [SX1262] Setup complete - awaiting incoming messages...
22:48:43.370 -> [ 11770][D][WeatherSensor.cpp:276] getMessage(): [SX1262] R [D4] RSSI: 0.0
22:48:43.370 -> [ 11770][D][WeatherSensor.cpp:988] decodeBresser7In1Payload(): Data sanity check failed
22:48:43.370 -> [ 11773][D][WeatherSensor.cpp:1001] decodeBresser7In1Payload(): Digest check failed - [75AF] vs [0B1E] (7EB1)
22:48:43.370 -> [ 11783][D][WeatherSensor.cpp:435] findSlot(): sensor[0]: v=0 id=0x00000000 t=0 c=0
22:48:43.420 -> [ 11790][D][WeatherSensor.cpp:934] decodeBresser6In1Payload(): Flags: Temp=0  Hum=0  Wind=1  Rain=1  UV=0
22:48:43.420 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
22:48:43.439 -> 
22:48:43.439 -> Core  1 register dump:
22:48:43.439 -> PC      : 0x42003615  PS      : 0x00060830  A0      : 0x82001eac  A1      : 0x3fcebf00  
22:48:43.439 -> A2      : 0x00000001  A3      : 0x00000000  A4      : 0x8e721d00  A5      : 0x5b152639  
22:48:43.473 -> A6      : 0xbac660cd  A7      : 0x495e3861  A8      : 0x82003587  A9      : 0x3fcebee0  
22:48:43.473 -> A10     : 0x00000000  A11     : 0x3fcebf22  A12     : 0x0000001a  A13     : 0x00000001  
22:48:43.473 -> A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
22:48:43.473 -> EXCVADDR: 0xbac660cd  LBEG    : 0x4037dbf0  LEND    : 0x4037dbfe  LCOUNT  : 0x00000001  
22:48:43.509 -> 
22:48:43.509 -> 
22:48:43.509 -> Backtrace: 0x42003612:0x3fcebf00 0x42001ea9:0x3fcebf60 0x42009395:0x6a635b09 |<-CORRUPTED
22:48:43.509 -> 
22:48:43.509 -> 
22:48:43.509 -> 
22:48:43.509 -> 
22:48:43.509 -> ELF file SHA256: 8ec4fcf7f2c7023b
22:48:43.509 -> 
22:48:43.620 -> Rebooting...

Davide

matthias-bs, 2023-11-25:
Yes, this helps! You got a valid message which has successfully been decoded by decodeBresser6In1Payload().
That means, the wiring seems to be correct!

matthias-bs, 2023-11-25:
```
ESP Exception Decoder
Sketch: BresserWeatherSensorBasic FQBN: esp32:esp32:heltec_wifi_lora_32_V3

22:48:43.439 -> Core 1 register dump:
22:48:43.439 -> PC : 0x42003615 PS : 0x00060830 A0 : 0x82001eac A1 : 0x3fcebf00
22:48:43.439 -> A2 : 0x00000001 A3 : 0x00000000 A4 : 0x8e721d00 A5 : 0x5b152639
22:48:43.473 -> A6 : 0xbac660cd A7 : 0x495e3861 A8 : 0x82003587 A9 : 0x3fcebee0
22:48:43.473 -> A10 : 0x00000000 A11 : 0x3fcebf22 A12 : 0x0000001a A13 : 0x00000001
22:48:43.473 -> A14 : 0x00000001 A15 : 0x00000001 SAR : 0x00000020 EXCCAUSE: 0x0000001c
22:48:43.473 -> EXCVADDR: 0xbac660cd LBEG : 0x4037dbf0 LEND : 0x4037dbfe LCOUNT : 0x00000001
22:48:43.509 ->
22:48:43.509 ->
22:48:43.509 -> Backtrace: 0x42003612:0x3fcebf00 0x42001ea9:0x3fcebf60 0x42009395:0x6a635b09 |<-CORRUPTED

PC: 0x42003615: SX1262::SX1262(Module*) at /home/mp/Arduino/libraries/RadioLib/src/modules/SX126x/SX1262.cpp:4
EXCVADDR: 0xbac660cd

Decoding stack results
0x42003612:  is in SX1262::setOutputPower(signed char) (/home/mp/Arduino/libraries/RadioLib/src/modules/SX126x/SX1262.cpp:103).
0x42001ea9:  is in WeatherSensor::findSlot(unsigned int, DecodeStatus*) (/home/mp/Arduino/libraries/BresserWeatherSensorReceiver/src/WeatherSensor.cpp:438).
0x42009395: ledc_timer_rst at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/driver/ledc.c:249
```

matthias-bs, 2023-11-26:
I have modified the RadioLib example with the same settings used in my code:
https://github.com/matthias-bs/RadioLib/blob/master/examples/SX126x/SX126x_Receive_Interrupt/SX126x_Receive_Interrupt.ino

Could you please try it with the board setting "Heltec WiFi LoRa 32(V3)" and see if an exception occurs?

evon800c, 2023-11-26:
Hi,
inside the files some result. 
[20231126.zip](https://github.com/matthias-bs/BresserWeatherSensorReceiver/files/13467032/20231126.zip)

I also tried to do the same thing, not with the "SX126x_Receive_Interrupt" example but with the "SX126x_FSK_Modem" example, 
setting the radio module initialization parameters as in your library.
I had modified the code a little to see if something could be seen.
in fact I could see "data" every 12 seconds or so, which is basically the transmission interval of the weather sensor

matthias-bs, 2023-11-26:
You mean no exception and repeated transmissions with your test code?

evon800c, 2023-11-26:
Hi,
my aim was to simply understand if setting the radio module with your parameters, I could capture something (anything).
I took the example of the RadioLib -> SX126x -> SX126x_FSK_Modem library, I deleted everything I didn't think was useful and changed the output slightly.
I repeat, I only needed to understand if something would was intercepted.
[test.zip](https://github.com/matthias-bs/BresserWeatherSensorReceiver/files/13467979/test.zip)

matthias-bs, 2023-11-26:
Hi,
I think we are getting closer...
Could you please test the updated [feature-heltec-wifi-lora-32-v3](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/feature-heltec-wifi-lora-32-v3/) branch again?
My assumption was that a package could not be larger than set with `radio.fixedPacketLengthMode(MSG_BUF_SIZE)`, but that was obviously wrong. So I read `numBytes = radio.getPacketLength()` bytes into recvData[] - sometimes exceeding its size (27 bytes), which might have caused the exceptions.

evon800c, 2023-11-26:
near perfect
give me some minute and i'll share you some information

matthias-bs, 2023-11-26:
Hi,

Nice to see a useful result!

Please have a look at the README (https://github.com/matthias-bs/BresserWeatherSensorReceiver/tree/main#readme) in the last line above the table...
Even if your sensor provides 5 different values and the station is sold as 5-in-1, it is well possible that the sensor transmits the 6-in-1 protocol. Actually the part number of the sensor is the interesting one (not the one of the station).

If the values in no_5in1_fixed.zip make sense (please bear in mind that the 6-in-1 protocol used two different messages in alternating order to send all data; this is handled by the SW), don't worry about the protocol name.

Next week I'd like to ask you to check if the blocking mode also works. The interrupt mode would be a little better indeed, but a common implementation for all three radio chips is not trivial.

evon800c, 2023-11-26:
Hi, 

yes, I have read your entire document and the note regarding 5 in 1 and 6 in 1. in the next few days I will try to check if the data I receive are coerent. for what I see, I think that they will be.

I see that when the sensor is coded as 7in1 there is a "Data integrity check failed", instead when code as 6in1 not. I think that this is a good point.

in debug mode I see also that you write a flags information for values. If I well understand is 1 when value is present and 0 where not. I think that this approc is very usefull to better manipulate sensor info in a chain of work.

I will study the others yours example to understend how send sensors data  in a lorawan network.

at the moment thanks a lot for you time and your work. it's great.

feel free to contact me to do other test of you code

thanks.

matthias-bs, 2023-11-27:
Hi,

The available decoders are tried sequentially until the matching one is found. If the decoder does not match, the integrity check (CRC, digest or whatever) will fail.

Yes, the flags indicate which values are present in a packet using the 6-in-1 protocol.

I merged the not-interrupt version for Heltec WiFi LoRa32 V3 into the main branch. Could you please check if it works?

BTW: [BresserWeatherSensorTTN](https://github.com/matthias-bs/BresserWeatherSensorTTN) does not support the SX1262, because the underlying LMIC libary doesn't.

evon800c, 2023-11-27:
Hi,

I hope to have done in the right way, but the result is not ok.
radio module initialize but remain in waiting and no data are captured.
[20231127.zip](https://github.com/matthias-bs/BresserWeatherSensorReceiver/files/13474284/20231127.zip)

Do you have any suggestions for me regarding LoRaWAN libraries?
With the heltec library I was able to connect my card to both TTN and Chirpstack network, but after what I experiment with their library, i prefer to go in a more agostic direction.

evon800c, 2023-12-01:
Hi Mattias,

I'm sorry to bring no good news ...
I see that you have published a new library version with change that we have tested.
I have changed my local library with your new version 19.6... unfortunatly something gone wrong.
With you basic example, that I use for testing, radio module start but don't get anythings. The only msg that I can see is "T" in verbose mode.

I add some data for you, I hope theu can be usefull for you. 
[20231201.zip](https://github.com/matthias-bs/BresserWeatherSensorReceiver/files/13532124/20231201.zip)

Davide

matthias-bs, 2023-12-02:
Hi Davide,

Thanks for testing, but the SX1262 is not supported yet in the latest release (see [Release Notes](https://github.com/matthias-bs/BresserWeatherSensorReceiver/releases/tag/v0.19.6)). I'm still struggling to get a common working solution for all three transceivers.

In the meantime, the [feature-heltec-wifi-lora-32-v3](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/feature-heltec-wifi-lora-32-v3/) branch should hopefully still work for you.

Cheers,
Matthias

matthias-bs, 2024-01-09:
The code has been proven to work with Heltec Wireless Stick V3 as can be seen in https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/121.

mothra78, 2024-01-13:
I finally got time to test the code on Heltec Wireless Stick V3 and got it to work with an update of the pins and adding SPI.begin(LoRa_SCK, LoRa_MISO, LoRa_MOSI, LoRa_nss);  before radio.beginFSK(); in WeatherSensor::begin(void).
Correct pins for me are : 
#define LoRa_MOSI 10
#define LoRa_MISO 11
#define LoRa_SCK 9
#define LoRa_nss 8

matthias-bs, 2024-01-15:
@mothra78 Thank you! I'd rather like to change the board definition in [espressif/arduino-esp32](https://github.com/espressif/arduino-esp32) than to implement a workaround in this project.

matthias-bs, 2024-02-02:
See "feat: Add board revisions V2/V3 to Heltec Wireless Stick" https://github.com/espressif/arduino-esp32/pull/9205

matthias-bs, 2024-05-03:
See https://github.com/matthias-bs/BresserWeatherSensorLW/issues/4

closed 2024-05-03 by matthias-bs