matthias-bs/BresserWeatherSensorReceiver #81: WeatherSensor::getMessage(void) does not seem to function as planned

mothra78, 2023-08-13:
When using the basic example I am looking to debug sx1262 and radio and noticed something in the code that seems off.

in WeatherSensor::getMessage(void) when (state != RADIOLIB_ERR_NONE) I would expect to be able to use the code to following know if I reached RADIOLIB_ERR_RX_TIMEOUT or another error.

It seems this code is within if (state == RADIOLIB_ERR_NONE) so will never execute.

I would suggest placing it after 
    } // if (state == RADIOLIB_ERR_NONE)

so would look more like : 

if (state == RADIOLIB_ERR_NONE) {
        // Verify last syncword is 1st byte of payload (see setSyncWord() above)
        if (recvData[0] == 0xD4) {
            #if CORE_DEBUG_LEVEL == ARDUHAL_LOG_LEVEL_VERBOSE
                char buf[128];
                *buf = '\0';
                for(size_t i = 0 ; i < sizeof(recvData) ; i++) {
                    sprintf(&buf[strlen(buf)], "%02X ", recvData[i]);
                }
                log_v("%s Data: %s", RECEIVER_CHIP, buf);
            #endif
            log_d("%s R [%02X] RSSI: %0.1f", RECEIVER_CHIP, recvData[0], rssi);

            decode_res = decodeMessage(&recvData[1], sizeof(recvData) - 1);
        } // if (recvData[0] == 0xD4)
     
    } // if (state == RADIOLIB_ERR_NONE)
   else if (state == RADIOLIB_ERR_RX_TIMEOUT) {
            log_v("T");
        } // if (state == RADIOLIB_ERR_RX_TIMEOUT)
        else {
            // some other error occurred
            log_d("%s Receive failed: [%d]", RECEIVER_CHIP, state);
        }

matthias-bs, 2023-08-14:
Could you please re-format your posting regarding the suggested code-change?

matthias-bs, 2023-08-14:
You are right! Thanks a lot for reporting!

This bug has been introduced ages ago. Fortunately, only debug output is affected by this.