matthias-bs/BresserWeatherSensorReceiver #84: Possible incorrect reading of the rain value on Bresser 7002585

array81, 2023-08-26:
Hi,

I'm testing your library with a NodeMCU (esp8266) and a Bresser 7002585 weather station (5in1). I would like to use the MQTT so I uploaded the BresserWeatherSensorMQTT example.
Everything seems to work fine but I noticed that the rain value is incorrect:

`{
  "id": 318835650,
  "ch": 0,
  "battery_ok": 1,
  "temp_c": 29.6,
  "temp_c": 29.7,
  "humidity": 45,
  "wind_gust": 0,
  "wind_avg": 0,
  "wind_gust": 1.1,
  "wind_avg": 1.1,
  "wind_dir": 202,
  "uv": 0,
  "rain": 10,
  "rain_d": 0,
  "rain_w": 0,
  "rain_m": 0
}`

As you can see it indicates 10 while it's been days that it hasn't rained here and it's sunny. In fact, the internal unit of my station does not indicate mm of rain while on my esp8266 I constantly get 10.

I compiled the example with the following configuration:
#define SLEEP_EN        false
#define USE_WIFI

matthias-bs, 2023-08-26:
Hi,

the `rain` value is the value provided directly by the sensor (tipping bucket rain gauge). The sensor accumulates the rain over time. Thus, this is the expected, correct behavior.

Regards
Matthias

array81, 2023-08-26:
Thanks for your replay.
Shouldn't the "rain" field give the current mm/h of rain? While and fields "rain_d", "rain_w" and "rain_d" the statistical data for day, week and month? Alternatively, where can I find the current pisggia values? I would like to use them to understand if it rains at a given moment or not.

matthias-bs, 2023-08-26:
`rain_h` is the hourly (last 60 minutes) rain data. This is currently missing in MQTT example, but it can be added easily.
see
https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/48e6341718a4253acc44479320930ef41df79116/src/RainGauge.h#L134

matthias-bs, 2023-08-26:
You can change
```
      if (weatherSensor.sensor[i].rain_ok || complete) {
          mqtt_payload += String(",\"rain\":")   + JSON_FLOAT(String(weatherSensor.sensor[i].rain_mm, 1));
          mqtt_payload += String(",\"rain_d\":") + JSON_FLOAT(String(rainGauge.currentDay(), 1));
          mqtt_payload += String(",\"rain_w\":") + JSON_FLOAT(String(rainGauge.currentWeek(), 1));
          mqtt_payload += String(",\"rain_m\":") + JSON_FLOAT(String(rainGauge.currentMonth(), 1));
      }
```
to
```
      if (weatherSensor.sensor[i].rain_ok || complete) {
          mqtt_payload += String(",\"rain\":")   + JSON_FLOAT(String(weatherSensor.sensor[i].rain_mm, 1));
          mqtt_payload += String(",\"rain_h\":") + JSON_FLOAT(String(rainGauge.pastHour(), 1));
          mqtt_payload += String(",\"rain_d\":") + JSON_FLOAT(String(rainGauge.currentDay(), 1));
          mqtt_payload += String(",\"rain_w\":") + JSON_FLOAT(String(rainGauge.currentWeek(), 1));
          mqtt_payload += String(",\"rain_m\":") + JSON_FLOAT(String(rainGauge.currentMonth(), 1));
      }
```

array81, 2023-08-26:
This works but show me mm in the last hour.
So there is not a punctual value for the rain like for the wind, right?
For the wind the speed which allows me to know if there is wind at this instant.
I guess "rain_h" takes time (perhaps an hour?) to eventually reset so it can't provide me with instant information (or in any case on a short-term basis, like 5/10 minutes) on the presence of rain.

matthias-bs, 2023-08-27:
Yes, there is no "instant value" for rainfall.

The hourly value starts counting immediately after power on - you will not have to wait for an hour. It is currently implemented as a circular buffer holding the values of the past 60 minutes.

Of course it would be possible to calculate a short-term value using the same principle as for the daily/weekly/monthly values. I just did not implemented it, because [BresserWeatherSensorTTN](https://github.com/matthias-bs/BresserWeatherSensorTTN) is the main application I have in mind. With LoRaWAN, air-time is restricted by a small percentage of time, requiring the device to stay in deep-sleep mode most of the time. Available energy is restricted, because the node is powered by a small PV module and a battery. And last but not least, it is difficult to guarantee a fixed, relatively short transmit cycle time with LoRaWAN.

array81, 2023-08-28:
I understand I thought this information was sent directly from the sensor instead it is calculated via software and I should edit the RainGuage file, right?

matthias-bs, 2023-08-28:
Yes, that's right.

array81, 2023-08-30:
I'm not sure if the code works.
Let's imagine turning on the exp. The esp will obtain as its first value that of "rain_mm" which is not necessarily zero, in fact it most likely will not be and therefore a (probably) high value will be obtained.

Also Having test Rain Gauge is updated before rainShortterm which will only be available after time has elapsed at RAIN_INTERVAL_SHORTTERM.
I err?

matthias-bs, 2023-08-31:
Well, that was intended as a draft, not as a final solution.

You are right, the first value after startup will not be correct. You can suppress the output of `rainShortterm` (or set it to zero) until `rainPrev` was assigned for the first time. This can be achieved by setting the initial value of `rainPrev` to -1 and comparing against this value.

What do you mean by "Also having test Rain Gauge is updated before rainShortterm"?

The code above works independent of the `rainGauge` object. Only the value `weatherSensor.sensor[i].rain_mm` is used, which is provided by `weatherSensor.getData()`.

matthias-bs, 2023-10-30:
Closed due to inactivity.

closed 2023-10-30 by matthias-bs