matthias-bs/BresserWeatherSensorReceiver #85: Support for building with PlatformIO?

amotl, 2023-08-26:
Dear Matthias,

we are huge fans of both your project, and [PlatformIO](https://platformio.org/).

Recently, we added corresponding PlatformIO build support to all our projects at https://github.com/hiveeyes/arduino, and at [bresser-to-mqtt](https://github.com/daq-tools/bresser-to-mqtt), we made an attempt to converge the directory layout of your BresserWeatherSensorReceiver project into a shape which is usable by PlatformIO.

Now, and because our "fork" lags behind very much [^1] we would like to take the chance to ask if you would accept a corresponding pull request, to make the project compatible with PlatformIO?

On the other hand, when looking at the CI configuration here, it seems like you are heavily using `arduino-cli` already. We [only recently discovered it](https://community.hiveeyes.org/t/probleme-mit-servo-lib-version-0-13/4866/4), and did not give it a spin yet.

If you think using PlatformIO is the wrong choice, and would like users like us to level up and use Arduino CLI instead, may we ask if there is some documentation about the sandbox setup process? Or would it roughly be to replicate what has been written down at the CI configuration file [^2]?

With kind regards,
Andreas.

/cc @einsiedlerkrebs

[^1]: Thank you so much for the excellent work you are doing here.
[^2]: https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/.github/workflows/CI.yml

matthias-bs, 2023-08-27:
Dear Andreas,

I am open to add support for PlatformIO, as long as compatibility with Arduino IDE is preserved. I do not want to raise the hurdles of using this project in any way.

I already tried to support PlatformIO, but gave up when it did not work immediately...

`arduino-cli` works pretty well in the CI setup. I also got the [VSCode extension for Arduino](https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.vscode-arduino) up and running recently (which is also based on `arduino-cli`) - this gives a pretty good development environment in combination with the `GitHub` extension and source control while keeping full compatibility with `Arduino IDE`.

I initially copied the CI setup from [jgromes/RadioLib](https://github.com/jgromes/RadioLib) and tailored it for my projects. The `arduino-cli` part is basically copy-paste, only adding options or board definitions requires some effort.

What is still missing is a working unit testing environment (see https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/27). Next I am planning to try [googletest](https://github.com/google/googletest) and [Arduino-CI](https://github.com/Arduino-CI) for this.

Best regards
Matthias

matthias-bs, 2023-10-30:
Closed due to inactivity.

closed 2023-10-30 by matthias-bs