matthias-bs/BresserWeatherSensorReceiver #88: Wemos D1 mini + CC1101+ 7003210

CedjaY, 2023-10-06:
Hello,
Your project looks extraordinary.
However I'm having some difficulty getting it to work...

After several unsuccessful tests on NodeMCU ESP32 and Node MCU ESP8826, I returned to your base with a Wemos ESP8826D1 mini.

I followed your "diagram", I just have a doubt about "To wake up the ESP8266 from deep sleep, GPIO16 should be connected to the RST pin.". Is this connection permanent and mandatory to function?
It didn't seem to change much with it.

I specify that the antenna is a CC1101 (868Mhz version).

I compiled and uploaded the program "BresserWeatherSensorBasic.ino".
Here is what I get:

```
18:02:48.207 -> SDK:2.2.2-dev(38a443e)/Core:2.7.3-3-g2843a5ac=20703003/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-30-g92add50/BearSSL:5c771be
18:02:48.254 -> Starting execution...
18:02:48.254 -> [CC1101] Initializing ... 
18:02:48.394 -> success!
18:02:48.394 -> [CC1101] Setup complete - awaiting incoming messages...
```
My weather station is 7003210: https://www.bresser.de/fr/Temps/Centrale-Station-meteo-WIFI/Station-meteo-6-jours-BRESSER-capteur-7-en-1-solaire.html
Maybe it's not supported yet?

What could be the steps to achieve a functional result?
Thank you all for reading

matthias-bs, 2023-10-06:
Hi,

yes, the GPIO/RST connection must be present, otherwise the MCU will never wake up from sleep mode. On the other hand, the connection must be removed while flashing.

To get more information, please enable debug output as described in https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/DEBUG_OUTPUT.md.

I cannot promise that your sensor is supported, but I am quite confident that it is...

Let's see what the debug output reveals... It's probably just (well...) a bad radio link.

Regards,
Matthias

matthias-bs, 2023-10-07:
Hi,
That's strange, I am using Radiolib 6.2.0 without any problems.

Could you please provide some debug output to allow me to check if the decoding works for your sensor?

Regards
Matthias

CedjaY, 2023-10-07:
No problem, how can I do this to you? Because in the console it waits for a message and that's it.

I managed to recover the data in MQTT with BresserWeatherSensorMQTTWifiMgr (I modified it because the password provided by my broker is longer than 20 characters!)

On the other hand, I didn't really understand rain_d, rain_w, rain_m, they return the same value to me and it seems to increment with each reception (1500, 2501 ...). rain_avg is interesting but it's not instant rain...

matthias-bs, 2023-10-07:
If you enable debug output, you should see a lot of text in the console. Did you select the right port and baudrate (115200 Baud)?

Did you read https://github.com/matthias-bs/BresserWeatherSensorReceiver/wiki/03.-Implementing-Rain-Gauge-Statistics? Rain data statistics is currently not supported on the ESP8266, that's why the values are meaningless. You should have seen a compiler warning which indicates this.

I'd be happy to integrate the air quality sensor. If you could provide some raw data with many different values, normal/low battery etc. and ideally the matching readings on the original display, this might be feasible.

matthias-bs, 2023-10-07:
Wow, the price for the air quality sensor dropped from 109€ to 39€!
:smile:

CedjaY, 2023-10-08:
> If you enable debug output, you should see a lot of text in the console. Did you select the right port and baudrate (115200 Baud)?

I don't have any more information with debugging in verbose or debug mode.
I'm upgrading to an ESP32 (AZ Delivery ESP32-WROOM-32). This time, I had compilation problems. There is a problem with EEPROM when I use RadioLib 6.2.0. I have no problems with RadioLib 6.1.0.

> Did you read https://github.com/matthias-bs/BresserWeatherSensorReceiver/wiki/03.-Implementing-Rain-Gauge-Statistics? Rain data statistics is currently not supported on the ESP8266, that's why the values are meaningless. You should have seen a compiler warning which indicates this.

Oh yes indeed! I didn't think it was so complex! What a crazy job!

> I'd be happy to integrate the air quality sensor. If you could provide some raw data with many different values, normal/low battery etc. and ideally the matching readings on the original display, this might be feasible.

Um... how can I do that? with what script? I really wish I could provide that.

> Wow, the price for the air quality sensor dropped from 109€ to 39€!
😄

Yes it has gone down a lot! 😀

matthias-bs, 2023-10-08:
> I don't have any more information with debugging in verbose or debug mode. I'm upgrading to an ESP32 (AZ Delivery ESP32-WROOM-32). This time, I had compilation problems. There is a problem with [...]

BTW: Did you re-compile after changing the debug level? (Setting the debug level inserts a certain set of printf-statements source code, which has to be compiled and uploaded to take effect!) 

I cannot reproduce both issues. Which Arduino IDE version and ESP32 board package are you using?

I just compiled with Arduini IDE 2.2.1 and ESP board package 2.0.14 - no issues regarding RadioLib 6.2.0. Could you provide some compiler output?

> Oh yes indeed! I didn't think it was so complex! What a crazy job!
>
>
> Um... how can I do that? with what script? I really wish I could provide that.

This can be done by running [BresserWeatherSensorBasic.ino](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/examples/BresserWeatherSensorBasic/BresserWeatherSensorBasic.ino).
If you set `Core Debug Level: Verbose` before compiling, you should be able to see debug output in your serial console. This output should contain message data as hex numbers also for unknown sensors. Then you can identify messages belonging to your air quality sensor, e.g. by removing/inserting its batteries.
Next, you can document the current sensor state e.g. by taking photos of your station's display or just taking note of its readings. With this information (and some knowledge about other sensors), it should be possible to find the positions and encoding of the data in the messages.
By manipulating the environment, you can get a little variation in data.
* Insert new batteries
* Insert weak batteries or set a power supply to a low battery voltage level
* Generate some dust/smoke to get different PM readings
* ...

> 
> > Wow, the price for the air quality sensor dropped from 109€ to 39€!
> > 😄
> 
> Yes it has gone down a lot! 😀

CedjaY, 2023-10-08:
This is all driving me crazy...!

> > > If you enable debug output, you should see a lot of text in the console. Did you select the right port and baudrate (115200 Baud)?
> > 
> > 
> > I don't have any more information with debugging in verbose or debug mode. I'm upgrading to an ESP32 (AZ Delivery ESP32-WROOM-32). This time, I had compilation problems. There is a problem with EEPROM when I use RadioLib 6.2.0. I have no problems with RadioLib 6.1.0.
> 
> BTW: Did you re-compile after changing the debug level? (Setting the debug level inserts a certain set of printf-statements into the source code, which has to be compiled and uploaded to take effect!)
> 
> I cannot reproduce both issues. Which Arduino IDE version and ESP32 board package are you using?
> 
> I just compiled with Arduino IDE 2.2.1 and ESP32 board package 2.0.14 - no issues regarding RadioLib 6.2.0. Could you provide some compiler output?
> 

This is probably the problem, I understand now. I'm navigating between two machines, it's a bit of anarchy.
IDE 2.2.1 et ESP32 by Espressif Systems board package 2.0.14
`C:\Users\CJ\Documents\Arduino\libraries\ESP_EEPROM\src\ESP_EEPROM.cpp:90:10: fatal error: c_types.h: No such file or directory
 #include "c_types.h"
          ^~~~~~~~~~~`

> This can be done by running [BresserWeatherSensorBasic.ino](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/examples/BresserWeatherSensorBasic/BresserWeatherSensorBasic.ino). [...]

OK ! I will do this as soon as possible, I think I will buy more antennas, since I only have one.
I excluded a sensor (because I think it's a neighbor). No chance that it's the Air quality probe?

On the other hand, I have a problem with BresserWeatherSensorMQTTWifiMgr, as soon as the esp32 restarts it loses the MQTT information. I also ended up deactivating sleep mode, because I don't find it of any interest in my case!? There is a lot of data I would like to recover it all.

`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
[     4][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[    29][I][BresserWeatherSensorMQTTWifiMgr.ino:782] setup(): 

BresserWeatherSensorMQTTWifiMgr

E (28) SPIFFS: mount failed, -10025
[    31][E][SPIFFS.cpp:89] begin(): Mounting SPIFFS failed! Error: -1
[    36][E][vfs_api.cpp:24] open(): File system is not mounted
[    42][E][vfs_api.cpp:24] open(): File system is not mounted
mounting FS...
E (55) SPIFFS: mount failed, -10025
[    58][E][SPIFFS.cpp:89] begin(): Mounting SPIFFS failed! Error: -1
[    58][E][BresserWeatherSensorMQTTWifiMgr.ino:421] wifimgr_setup(): failed to mount FS
[    83][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   150][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
[   150][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
*wm:AutoConnect 
*wm:Connecting to SAVED AP: WiFi IoT
[   662][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
*wm:connectTimeout not set, ESP waitForConnectResult... 
[   789][V][WiFiGeneric.cpp:355] _arduino_event_cb(): STA Connected: SSID: WiFi IoT, BSSID: d2:21:f9:98:56:c2, Channel: 1, Auth: WPA2_PSK
[   791][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[   825][V][WiFiGeneric.cpp:369] _arduino_event_cb(): STA Got New IP:192.168.20.201
[   825][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[   828][D][WiFiGeneric.cpp:1102] _eventCallback(): STA IP: 192.168.20.201, MASK: 255.255.255.0, GW: 192.168.20.1
*wm:AutoConnect: SUCCESS 
*wm:STA IP Address: 192.168.20.201
[   870][I][BresserWeatherSensorMQTTWifiMgr.ino:483] wifimgr_setup(): connected...yeey :)
[   872][I][BresserWeatherSensorMQTTWifiMgr.ino:490] wifimgr_setup(): The values in the file are: 
[   881][I][BresserWeatherSensorMQTTWifiMgr.ino:491] wifimgr_setup(): .mqtt_server : 
[   888][I][BresserWeatherSensorMQTTWifiMgr.ino:492] wifimgr_setup(): .mqtt_port : 1883
[   896][I][BresserWeatherSensorMQTTWifiMgr.ino:493] wifimgr_setup(): .mqtt_user : 
[   903][I][BresserWeatherSensorMQTTWifiMgr.ino:494] wifimgr_setup(): .mqtt_pass : ***
[   911][I][BresserWeatherSensorMQTTWifiMgr.ino:518] wifimgr_setup(): local ip: 192.168.20.201
Checking wifi...
MQTT connecting... [   930][E][WiFiGeneric.cpp:1582] hostByName(): DNS Failed for 
.[  1930][E][WiFiGeneric.cpp:1582] hostByName(): DNS Failed for 
.[  2930][E][WiFiGeneric.cpp:1582] hostByName(): DNS Failed for 
.[  3930][E][WiFiGeneric.cpp:1582] hostByName(): DNS Failed for 
.[  4930][E][WiFiGeneric.cpp:1582] hostByName(): DNS Failed for 
.[  5930][E][WiFiGeneric.cpp:1582] hostByName(): DNS Failed for 
.[  6930][E][WiFiGeneric.cpp:1582] hostByName(): DNS Failed for 
.[  7930][E][WiFiGeneric.cpp:1582] hostByName(): DNS Failed for 
.[  8930][E][WiFiGeneric.cpp:1582] hostByName(): DNS Failed for 
.[  9930][E][WiFiGeneric.cpp:1582] hostByName(): DNS Failed for 
.[ 10930][E][WiFiGeneric.cpp:1582] hostByName(): DNS Failed for 
.[ 11930][E][WiFiGeneric.cpp:1582] hostByName(): DNS Failed for 
.ets Jun  8 2016 00:22:57`

matthias-bs, 2023-10-08:
Ah, it's getting better! I don't think you need this `ESP_EEPROM` library. I don't have it installed. You can uninstall it or move it from the `libraries` directory temporarily. It's a disadvantage of the Arduino IDE that all libraries are installed in global directory.

Maybe you have seen the air quality probe. You could check if any message output coincides with the probe's transmission indicator.

CedjaY, 2023-10-08:
> Ah, it's getting better! I don't think you need this `ESP_EEPROM` library. I don't have it installed. You can uninstall it or move it from the `libraries` directory temporarily. It's a disadvantage [...]

No, it gives me information like temperature, wind, UV etc.

BresserWeatherSensorMQTT requires SSL for wifi? I'm never going to get out of this 😶

matthias-bs, 2023-10-08:
No, it's your choice: 
`USE_SECUREWIFI` - needs SSL
`USE_WIFI` - no need for SSL

This is a little bit misleading, because it's actually not directly related to WiFi but provided in the WiFi library (or not).

CedjaY, 2023-10-08:
> No, it's your choice: `USE_SECUREWIFI` - needs SSL `USE_WIFI` - no need for SSL
> 
> This is a little bit misleading, because it's actually not directly related to WiFi but provided in the WiFi library (or not).
> 
> No worries - some things are a little tricky...

This compiled well with RadioLib 6.2.0 but I definitely don't get any messages when it is used.

`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
[     4][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[    24][I][BresserWeatherSensorMQTT.ino:629] setup(): 

BresserWeatherSensorMQTT 20230711

[    24][I][BresserWeatherSensorMQTT.ino:365] mqtt_setup(): Attempting to connect to SSID: WiFi IoT
[    43][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   110][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
[   112][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[   112][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
.[   217][V][WiFiGeneric.cpp:355] _arduino_event_cb(): STA Connected: SSID: WiFi IoT, BSSID: d2:21:f9:98:56:c2, Channel: 1, Auth: WPA2_PSK
[   219][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[   750][V][WiFiGeneric.cpp:369] _arduino_event_cb(): STA Got New IP:192.168.20.201
[   751][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[   754][D][WiFiGeneric.cpp:1102] _eventCallback(): STA IP: 192.168.20.201, MASK: 255.255.255.0, GW: 192.168.20.1
[  1132][I][BresserWeatherSensorMQTT.ino:370] mqtt_setup(): connected!
Checking wifi...
MQTT connecting... [  1157][I][BresserWeatherSensorMQTT.ino:441] mqtt_connect(): 
connected!
[  1161][I][BresserWeatherSensorMQTT.ino:443] mqtt_connect(): ESPWeather-CAE710/status: online

[  1162][D][WeatherSensor.cpp:100] begin(): [CC1101] Initializing ... 
[  1334][D][WeatherSensor.cpp:113] begin(): success!
[  1336][D][WeatherSensor.cpp:148] begin(): [CC1101] Setup complete - awaiting incoming messages...
`

With RadioLib 6.1.0 everything is ok

`[1363121][V][WeatherSensor.cpp:243] getMessage(): T
[1363621][V][WeatherSensor.cpp:243] getMessage(): T
[1364122][V][WeatherSensor.cpp:243] getMessage(): T
[1364623][V][WeatherSensor.cpp:243] getMessage(): T
[1365124][V][WeatherSensor.cpp:243] getMessage(): T
[1365624][V][WeatherSensor.cpp:243] getMessage(): T
[1366125][V][WeatherSensor.cpp:243] getMessage(): T
[1366517][V][WeatherSensor.cpp:235] getMessage(): [CC1101] Data: D4 04 9A 00 92 BA 8A 18 AA AA AA AA AA D9 CC 8E DA FD AA 98 39 AA AA AA AA AA 00 
[1366519][D][WeatherSensor.cpp:237] getMessage(): [CC1101] R [D4] RSSI: -67.5
[1366526][V][WeatherSensor.h:458] log_message():           Byte #: 00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 
[1366538][V][WeatherSensor.h:468] log_message(): De-whitened Data: AE 30 AA 38 10 20 B2 00 00 00 00 00 73 66 24 70 57 00 32 93 00 00 00 00 00 AA 
[1366551][V][WeatherSensor.cpp:334] findSlot(): find_slot(): ID=0000AA38
[1366557][V][WeatherSensor.cpp:388] findSlot(): find_slot(): Storing into slot #0
[1366567][I][BresserWeatherSensorMQTT.ino:584] publishWeatherdata(): ESPWeather-CAE710/WeatherSensor/data: {"id":43576,"battery_ok":1,"temp_c":24.70,"humidity":57,"wind_gust":0.00,"wind_avg":0.00,"wind_dir":102.00,"uv":0.00,"light_klx":3.29,"rain":7.30,"rain_h":1.50,"rain_d":1.50,"rain_w":1.50,"rain_m":1.50}

[1366595][I][BresserWeatherSensorMQTT.ino:599] publishWeatherdata(): ESPWeather-CAE710/extra: {"wind_dir_txt":"ESE","wind_gust_bft":0,"wind_avg_bft":0,"dewpoint_c":15.60,"perceived_temp_c":25.76}

[1366610][I][BresserWeatherSensorMQTT.ino:697] loop(): ESPWeather-CAE710/status: online

[1366617][I][BresserWeatherSensorMQTT.ino:617] publishRadio(): ESPWeather-CAE710/radio: {"rssi":-67.5}

[1367127][V][WeatherSensor.cpp:243] getMessage(): T
[1367628][V][WeatherSensor.cpp:243] getMessage(): T
[1368129][V][WeatherSensor.cpp:243] getMessage(): T
[1368629][V][WeatherSensor.cpp:243] getMessage(): T
[1369130][V][WeatherSensor.cpp:243] getMessage(): T
[1369341][V][WeatherSensor.cpp:235] getMessage(): [CC1101] Data: D4 99 9B 4E D9 AA EA 89 AA EA AA 8A AA 8A AA 9A AA 3A AB BA AB 8A EA FA AA AA 00 
[1369343][D][WeatherSensor.cpp:237] getMessage(): [CC1101] R [D4] RSSI: -41.0
[1369350][V][WeatherSensor.h:458] log_message():           Byte #: 00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 
[1369362][V][WeatherSensor.h:468] log_message(): De-whitened Data: 33 31 E4 73 00 40 23 00 40 00 20 00 20 00 30 00 90 01 10 01 20 40 50 00 00 AA 
[1369375][V][WeatherSensor.cpp:334] findSlot(): find_slot(): ID=0000E473
[1369381][V][WeatherSensor.cpp:341] findSlot(): In Exclude-List, skipping!
[1369889][V][WeatherSensor.cpp:243] getMessage(): T
[1370389][V][WeatherSensor.cpp:243] getMessage(): T
[1370890][V][WeatherSensor.cpp:243] getMessage(): T
[1371391][V][WeatherSensor.cpp:243] getMessage(): T`

matthias-bs, 2023-10-08:
Thanks for the log!
I didn't consider in my previous postings regarding Radiolib that I didn't do any test with a CC1101 for quite a while!
I will keep it in mind...

CedjaY, 2023-10-14:
Thank you for your incredible responsiveness.
My NodeMCU ESP32 + CC1101 has been running for a week now without any problems.

I'm waiting for a new antenna (this should take 15 days) in order to do more testing (mainly for Air Quality)
And so as not to break what works well. 😊

matthias-bs, 2023-10-14:
I ordered an air quality sensor, too. Stay tuned! 😉

matthias-bs, 2023-10-24:
Air quality sensor has arrived and implementation of a decoder is in progress! (Just a variant of the existing 7-in-1 decoder - easy!)

CedjaY, 2023-10-25:
Great ! I can't wait to see this!
I still haven't received my antenna...

matthias-bs, 2023-10-25:
You can give it a try: [Release v0.19.0](https://github.com/matthias-bs/BresserWeatherSensorReceiver/releases/tag/v0.19.0)
See https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/97 for more information.

CedjaY, 2023-10-27:
Good morning !
Sorry if this has already been explained, but on my ESP32 no rain gauge resets automatically.
Is this normal? Is there anything to activate?

And otherwise I send MQTT messages for reset, by sending 2 this resets me "day", by sending 6 this resets me "week", by sending 8 this resets me "month".

I can't reset "hour" or "total".

I am currently still in 0.18.0.

matthias-bs, 2023-10-30:
Hi,

so the good news is that the Air Quality Sensor is already working! ;-)

**Automatic Reset**
The automatic reset works as follows:
* `rain` is the raw sensor output which is never reset by the sketch
* `rain_m` resets when a change of the calendar month occurs
*  `rain_w` resets when a change of the calendar month occurs (Monday 00:00h)
* `rain_d` resets when a change of the day occurs
* `rain_h` is a the sum of the rainfall in the sliding time frame of the past 60 minutes

**Manual Reset**
https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/40bec466c2c78da42a6a7889712bde1155d16271/src/RainGauge.h#L70
```
 #define RESET_RAIN_H 1
 #define RESET_RAIN_D 2
 #define RESET_RAIN_W 4
 #define RESET_RAIN_M 8
```
You can add the values to reset multiple counters in one go.

**Sensor name issue:**
I did not find anything suspicious in the code.
Did you also update the sketch from the new library version? Have you tried with a shorter name for "AirQualitySensor"?(There might be limitations in previous versions of the sketch.)
Besides, you have spaces in the JSON string (e.g: "battery_ok ") which do not originate from the sketch.

Cheers,
Matthias

matthias-bs, 2023-10-30:
See https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/100

CedjaY, 2023-11-03:
>so the good news is that the Air Quality Sensor is already working! ;-)

Yes it works wonderfully!

However, the automatic reset doesn't seem to work for me (And it didn't rain non-stop for several days 😁) :

[image]
[image]

I shouldn't need to reset manually but it is not functional for rain_h. Even if I send 1 nothing happens.

As stated before, I am forced to send 2,6,8 instead of 2,4,8.
Maybe I need to send the message in another format?

>Sensor name issue
>Edit: That's a bug!

Phew! thank you for the correction. Phew! thank you for the correction. So I'm not crazy. 

>Besides, you have spaces in the JSON string (e.g: "battery_ok ") which do not originate from the sketch.

oops, probably just a problem translating my post. I have no space in the MQTT logs. No problems reported here.

matthias-bs, 2023-11-03:
Which sketch are you running? Could you please have a look at the debug output if the time and date is set up properly after each wake-up?

Currently, this is not fully implemented in the MQTT example. Time and date is only set from NTP if USE_SECUREWIFI is defined.

The required changes are:
* set time/date independent of USE_SECUREWIFI  or USE_WIFI
* set HW RTC from NTP and restore time from HW RTC after wake-up (improves robustness in case of failing NTP access)

I tend to have mainly [BresserWeatherSensorTTN](https://github.com/matthias-bs/BresserWeatherSensorTTN) in focus, that's why I did't realize that something might by missing...

In case time and date are not set up properly, it will always start at 1970-01-01 01:00:00 after wake-up, a change of day/month/week will never occur and the hourly rainfall will behave in an odd way, too.

matthias-bs, 2023-11-03:
https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/103

matthias-bs, 2023-11-03:
The fix is ready for testing...

matthias-bs, 2023-11-07:
Fixed in Release [v0.19.4](https://github.com/matthias-bs/BresserWeatherSensorReceiver/releases/tag/v0.19.4)

closed 2023-11-07 by matthias-bs