matthias-bs/BresserWeatherSensorReceiver #91: Reception with CC1101 no longer working after update to RadioLib v6.2.0

matthias-bs, 2023-10-08:
**Context**
- Which example sketch are you using?
   [BresserWeatherSensorBasic.ino](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/examples/BresserWeatherSensorBasic/BresserWeatherSensorBasic.ino)
- Which changes to the code did you make (if any)?
  none
- Which sensor are you using?
   various

**Describe the bug**
Data reception with CC1101 fails after switching from RadioLib v6.1.0 to v6.2.0.
This occurs with both ESP32 and ESP8266.

**To Reproduce**
Steps to reproduce the behavior:
Compile [BresserWeatherSensorBasic.ino](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/examples/BresserWeatherSensorBasic/BresserWeatherSensorBasic.ino) for CC1101 with RadioLib v6.2.1 -> no messages received
Compile [BresserWeatherSensorBasic.ino](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/examples/BresserWeatherSensorBasic/BresserWeatherSensorBasic.ino) for CC1101 with RadioLib v6.2.1 -> messages received

**Expected behavior**
No functional change expected between RadioLib v6.1.0 and v6.2.0

**Debug Log**
```
21:03:48.550 -> Starting execution...
21:03:48.550 -> begin(), l.100: [CC1101] Initializing ... 
21:03:48.713 -> begin(), l.113: success!
21:03:48.713 -> begin(), l.148: [CC1101] Setup complete - awaiting incoming messages...
```

**Embedded Device (please complete the following information):**
 - Arduino IDE Version: 2.2.1
 - ESP32/ESP8266 Board Support Package Version: ESP8266 v3.1.2
 - Library Version: RadioLib v6.2.0
 - Your Board: ESP8266 D1 Mini
 - Selected Board in the Arduino IDE: LOLIN(Wemos) D1 R2 & mini
 - Your Radio Transceiver Module (if external): CC1101 868 MHz
 

**Additional context**
-

matthias-bs, 2023-10-08:
As a workaround, use RadioLib v6.2.1

matthias-bs, 2023-10-08:
@jgromes For your kind attention. I'll create an issue in https://github.com/jgromes/RadioLib as soon as I find the time.

jgromes, 2023-10-09:
@matthias-bs thank you for the notification, this is most likely related to https://github.com/jgromes/RadioLib/issues/839, it seems like blocking reception on CC1101 is indeed broken in RadioLib 6.2.0.

jgromes, 2023-11-18:
@matthias-bs FYI pushed an update to RadioLib master that should address this.

matthias-bs, 2023-11-18:
@jgromes Thank you for the update! I installed a ZIP archive from the master branch. Unfortunately, reception still hangs while it works with v6.1.0.

jgromes, 2023-11-19:
Tried again, unfortunately I was not able to replicate hte issue this time. I'm using a CC1101 transmitter and CC1101 receiver configured as close to your setup as I could, however, the receiver is able to receive the messages:

Transmitter:
```c++
int state = radio.begin(868.3, 8.21, 57.136417, 270, 10, 32);
state |= radio.setCrcFiltering(false);
state |= radio.fixedPacketLengthMode(12);
state |= radio.setSyncWord(0xAA, 0x2D, 0, false);

(...)

radio.transmit("Hello World!");
```

Receiver:
```c++
int state = radio.begin(868.3, 8.21, 57.136417, 270, 10, 32);
state |= radio.setCrcFiltering(false);
state |= radio.fixedPacketLengthMode(12);
state |= radio.setSyncWord(0xAA, 0x2D, 0, false);

(...)

String str;
int state = radio.receive(str);
```

jgromes, 2023-11-21:
You can try running both versions in debug mode, maybe that could provide extra insight. Though I have to say that using the blocking `receive` method is not something I would encourage, interrupt-driven reception is usually the better pick.

matthias-bs, 2023-11-25:
Hi,
I just tried the interrupt-driven reception with CC1101 and SX1276 - both are working with RadioLib v6.2.0. When I followed your example, I was only able to receive a single message. Only after calling `radio.startReceive()` again, I was able to receive the next message. Do you have any idea why?

Laserzicke, 2023-11-26:
Hello,
I have the same problem. (ESP32 Dev Kit C V2 (38p) with CC1101).
Because of the problem with negative temperature display (-0.5 °C -> -99.5°C) I tried an update.
After updating the WeatherSensor(Cfg, h, cpp) files, my sketch no longer works. (November 2023). Not even with RadioLib version 6.1.0.
In debug mode appears: [ 127][E][WeatherSensor.cpp:158] begin(): [CC1101] Error initializing: [-2].
After changing the pin assignment to G27 -> CSN, G5 -> GDO0, G4 -> GDO2, G18 -> SCLK, the following appears 
[27][D][WeatherSensor.cpp:108] begin(): [CC1101] Initializing ... 
[ 191][D][WeatherSensor.cpp:123] begin(): success!
[ 192][D][WeatherSensor.cpp:162] begin(): [CC1101] Setup complete - awaiting incoming messages... but no data is received.

I had uploaded the old sketch to the ESP32 Dev Kit in June 2023. However, the old WeatherSensor files are in the same directory as the sketch. The pin assignment must be G5 -> CSN, G2 -> GDO0, G4 -> GDO2, G18 -> SCLK. This is the only way it works. (Data received every 10 seconds).
Why does the sketch no longer work after the update? What is the correct pin assignment for the ESP32 Dev Kit C V2 (38p)?
I use Bresser Weatherstation 7002531 and Arduino-IDE 2.2.1 (iMac and Linux Mint).

matthias-bs, 2023-11-27:
@Laserzicke I do not quite understand your explanations of "old" vs. "new" sketch and where you had to change the pin assignments.

Since ESP32 Dev Kit C V2 is not a target which is supported out-of-the-box, you have to do [Manual Configuration](https://github.com/matthias-bs/BresserWeatherSensorReceiver#manual-configuration).

* Please check that you select GPIO pins which are not connected to anything else (board's schematic)
* Check the SPI pins (board documentation, schematic)
* Check your own wiring (at least twice!)
* *Keep in mind that the ArduinoIDE might chose to use .cpp/.h files from `Arduino/libraries/BresserWeatherSensorReceiver` rather than from your sketch directory!*
* Modify WeatherSensorCfg.h
   * Set the correct radio transceiver in the section "Radio Transceiver"
   * Define the pins according to your setup in the section starting with
     ```
     #elif defined(ESP32)
        // Generic pinning for ESP32 development boards
     ```
* Check the files actually used and the pin definitions actually used in the compiler output - 
   enable  "Show verbose output during compile" in Preferences for this

If you need more help, please create a new issue and provide as much information on your setup (schematic, wiring diargram, `WeatherSensorCfg.h`, ArduinoIDE settings, folder structure, ...) as possible.

Laserzicke, 2023-11-28:
Hello matthias-bs,
Thank you for your reply.
I did not describe my problem well enough. Sorry.
By old files I meant the WeatherSensor Lib and Example files with history date up to March and April 2023 respectively. My own sketch was also created at that time. 
Data packets from the weather sensor are received every 10 seconds, even with "BresserWeatherSensorBasic.ino".

After updating the library, they no longer work properly on this board.
Then I also had to change the pin wiring as described to get the C1101 initialisation working again. (only CSN and GDO0).
After each board reset, the sketch e.g. "BresserWeatherSensorBasic.ino" sometimes only receives a single data packet from the weather sensor. 
Nothing else.
No matter with which RadioLib version.

I will try this again later with another board (Heltec WiFi Kit or ESP8266).
(I have now found a simple solution for the minus °C temperature problem)

matthias-bs, 2023-11-28:
Hi,

Which sensor are you using? Which decoder works for it?

Laserzicke, 2023-11-28:
Hello,
I use
- TECNOIOT CC1101 868MHz
- ESP32 Dev Kit C V2
- NodeMCU 1.0 (ESP-12E)
- Bresser Weatherstation 7002531 with outdoor sensor (only temperature, humidity, wind speed)
- RadioLib 6.1.0
- Arduino IDE 2.2.1

The wiring specifications of the WeatherSensorCfg.h do not match my boards or I am too stupid to understand this file.
They all don't work -->  [ 127][E][WeatherSensor.cpp:158] begin(): [CC1101] Error initializing: [-2].
I got working wiring examples from google. I am not an expert :-/

ESP32 debug output: (BresserWeatherSensorBasic.ino)
[ 5][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 HStarting execution...
[ 13][D][WeatherSensor.cpp:108] begin(): [CC1101] Initialising ... 
[ 178][D][WeatherSensor.cpp:123] begin(): success!
[ 179][D][WeatherSensor.cpp:162] begin(): [CC1101] Setup complete - awaiting incoming messages...
[ 681][V][WeatherSensor.cpp:257] getMessage(): T
[ 1281][V][WeatherSensor.cpp:257] getMessage(): T
[ 1881][V][WeatherSensor.cpp:257] getMessage(): T
etc.... nothing more

Since the update, no or rarely only one data is received with ESP32. I don't understand it.
But with the backup files from March, data is received every 12 seconds, the hardware is not defective.

Now I have tried the same example with NodeMCU and the same CC1101.
It receives the data every 12 seconds (RadioLib 6.1.0). (with RadioLib 6.2.0. no data)
...
17:35:45.287 -> Id: [29700512] Typ: [1] Ch: [0] St: [0] Bat: [OK ] RSSI: [ -61.5dBm] Temp: [-95.7C] Hum: [ 96%] Wmax: [ 0.0m/s] Wavg: [ 0.0m/s] Wdir: [  0.0deg] Rain: [-----.-mm] UVidx: [0.0] Light: [--.-klx] 
17:35:57.265 -> Id: [29700512] Typ: [1] Ch: [0] St: [0] Bat: [OK ] RSSI: [ -60.5dBm] Temp: [-95.7C] Hum: [ 96%] Wmax: [ 0.0m/s] Wavg: [ 0.0m/s] Wdir: [  0.0deg] Rain: [-----.-mm] UVidx: [0.0] Light: [--.-klx] 
...
By the way, with BresserWeatherSensorBasic.ino (your last history entry 20231101) -4.3°C is still displayed as -95.7°C.

matthias-bs, 2023-11-28:
Hello,

let's start with your second case, NodeMCU and CC1101. A solution (or rather workaround) for RadioLib v6.2.0 is in progress. Your sensor 7002531 is "special": it uses the same decoder and transmits the same type as many other sensors, but the temperature is encoded differently. I did not find a way to distinguish the two temperature encodings automatically.

As a workaround, specify your sensor ID in `WeatherSensorCfg.h` as follows:
```
// List of sensor IDs of the model "BRESSER 3-in-1 Professional Wind Gauge / Anemometer"
// P/N 7002531 - requiring special handling in decodeBresser5In1Payload()
//#define SENSOR_IDS_DECODE3IN1 {}
#define SENSOR_IDS_DECODE3IN1 { 0x29700512 }
```

matthias-bs, 2023-11-28:
Now the ESP32 DecKitC V2. 

The connection to CC1101 needs 3 SPI pins and 3 GPIO pins.

Let's start with the SPI pins. You did not provide additional information about the board, so I have to make assumptions. I hope https://github.com/espressif/arduino-esp32/blob/master/variants/doitESP32devkitV1/pins_arduino.h matches your board. According to this, we have 
```
static const uint8_t SS    = 5;
static const uint8_t MOSI  = 23;
static const uint8_t MISO  = 19;
static const uint8_t SCK   = 18;
```
This seems to be the default for most ESP32 boards as mentioned in https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/0d37102ed6d41bf5c6523b0a990f821a835f9253/src/WeatherSensorCfg.h#L359
MOSI, MISO and SCK should be labelled on your CC1101 module. Connect to the GPIO pins according to the labels on the  DevKit board.

SS (Slave Select) is also called CS (Chip Select). We'll take care of it later.

Now the GPIO pins.
I am referring to https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-devkitc-v2.html. On that page, you find a schematic. 

Lets see if the default pins for ESP32 in `WeatherSensorCfg.h` are available, i.e. not connected to anything else:

```
#elif defined(ESP32)
    // Generic pinning for ESP32 development boards
    #define PIN_RECEIVER_CS   27

    // CC1101: GDO0 / RFM95W/SX127x: G0
    #define PIN_RECEIVER_IRQ  21

    // CC1101: GDO2 / RFM95W/SX127x: G1
    #define PIN_RECEIVER_GPIO 33

    // RFM95W/SX127x - GPIOxx / CC1101 - RADIOLIB_NC
    #define PIN_RECEIVER_RST  32
```
IO27: yes
IO21: yes
IO33: yes
IO32: yes

(Instead of IO27, wie could also use IO5 for CS as indicated above, but we leave it as it is.

Connect the pins as indicated above.

Laserzicke, 2023-11-29:
Hello,
I have gone through the steps again exactly according to your instructions:
// List of sensor IDs of the model "BRESSER 3-in-1 Professional Wind Gauge / Anemometer"
// P/N 7002531 - requiring special heandling in decodeBresser5In1Payload()
//#define SENSOR_IDS_DECODE3IN1 {}
#define SENSOR_IDS_DECODE3IN1 { 0x2C100512 } - this is correct

Board select in IDE: DOIT ESP32 DEVKIT V1 

Wiring:
CC1101 	ESP32(GPIO)	
MOSI	MOSI = 23	
MISO	MISO = 19	
SCLK	SCK = 18		
CSN		SS = 27		
GDO0	SDA = 21		
GD02	A5 = 33		

I receive a successful initialisation (CC1101), but still no data.
If I connect CSN (CC1101) to GPIO5 (SS), I get [CC1101] Error initialising: [-2].
I have also tried the bord firebeatle ESP32 and the wiring in the IDE.

The following circuit diagram was supplied with my ESP32. (see Schematic_ESP_DevKitC_V2.png)
[image]

See also the two files with the IDE output. 
[issue_ide_board_select_firebeatle_ESP32.txt](https://github.com/matthias-bs/BresserWeatherSensorReceiver/files/13502719/issue_ide_board_select_firebeatle_ESP32.txt)
[issue_ide_board_select_Doit_ESP32_devkit_v1.txt](https://github.com/matthias-bs/BresserWeatherSensorReceiver/files/13502726/issue_ide_board_select_Doit_ESP32_devkit_v1.txt)

Perhaps you can recognise something revealing in it.

Maybe I have a special variant of ESP32 that just happens to work with the backup files from March.
Unfortunately I have no other ESP32 variant for comparison.
I will have to postpone my analysis until a later date. I don't have much time left.

Thank you very much for your efforts.

matthias-bs, 2023-11-29:
Hi,
Both logfiles show that the compiler assumes that a FireBeetle is used!

Where is the configuration file located? Did you try to remove files from a previous build?

matthias-bs, 2023-11-29:
BTW: I think I can adjust the temperature decoding automatically instead of checking the sensor ID:

```
temp = ((sign) ? (temp_raw - 1000) : temp_raw) * 0.1f;

if (temp < -50.0) {
    temp = -(100.0 + temp);
}
```
The sensor range (as far as provided in the manuals) is -40...+60°C, so there is no risk that the intermediate values overlap.

matthias-bs, 2023-11-29:
See https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/114

Laserzicke, 2023-11-30:
Hello.
So, I have deleted, reinstalled and reconfigured everything.
BresserWeatherSensorReceiver 0.19.5
RadioLib 6.1.0
Something must have gone wrong with an update in the past or I'm too old now(60).

Now, i have uncommented „//define FIREBEETLE_ESP32_COVER_LORA“. — but is located in line 198.
 —> .../src/WeatherSensorCfg.h:210:121: note: #pragma message: ARDUINO_ESP32_DEV defined; select either LORAWAN_NODE or FIREBEETLE_ESP32_COVER_LORA manually! 
(The chip imprint on my ESP32-DevKit-board is: ESP32-WROOM-32)
 .../src/WeatherSensorCfg.h:504:170: #pragma message: Pin config: RST->32, CS->27, GD0/G0/IRQ->21, GDO2/G1/GPIO->33 - not work!

If I change to —> #pragma message: Pin config: RST->18, CS->5, GD0/G0/IRQ->2, GDO2/G1/GPIO->4  - It works now !
This has the disadvantage that I have to disconnect the cable (GPIO2) when uploading with USB.  Otherwise it blocks. This is not a problem with OTA.
Maybe I can find another combination that works better.
...
I had temporarily solved the display problem with the negativ temperature in weather.ino: (older modified version)
 if (weatherSensor.sensor[i].temp_c < 0.0){
        Temp_out = String((weatherSensor.sensor[i].temp_c + 100) * -1);
      } else {
      Temp_out = String(weatherSensor.sensor[i].temp_c);

matthias-bs, 2023-11-30:
Hi,

well, the remaining message from line 210 is confusing, I'll change this. Just set "DOIT ESP32 DEVKIT V1" as proposed.

Now  I don't understand how the config RST->18, CS->5, GD0/G0/IRQ->2, GDO2/G1/GPIO->4  was defined. Did you place it in the `#elif defined(ESP32)` section yourself or was it selected automatically.

Laserzicke, 2023-11-30:
Hello.
I have entered the definition there myself: Pin config: RST->18, CS->5, GD0/G0/IRQ->2, GDO2/G1/GPIO->4 - This is the only way it works.
I have already tried all other possibilities - without success.
The wiring suggestion to my ESP (with CC1101) via google does not show anything else.

matthias-bs, 2023-11-30:
Thanks for clarification. You can use any available GPIO pins, i.e. those which are not connected to anything else on the board.

closed 2023-12-02 by matthias-bs