matthias-bs/BresserWeatherSensorReceiver #97: Add additional data in Air Quality Sensor PM 2.5 / PM 10 decoder

matthias-bs, 2023-10-25:
Currently only PM 2.5 / PM 10 mass concentration values are provided, but the message payload seems to contain more, currently unknown data.

**Potential candidates**
- Battery status
- Presence of external power supply connection
- Update rate
- PM 0.5 / PM 4 mass concentration values
- PM number concentration values

MacH-21, 2023-11-18:
Hi Matthias,

I have just received a Bresser PM2.5/PM10 + dedicated Display!
Bresser part 7110300 £59.84 via Amazon dispatches Bresser-UK sold by Bresser-UK!

Bresser-UK online price £130 + delivery!!!

Bresser 7009970 PM2.5/PM10 NO display - Amazon £40.85 ... Bresser-UK £109!!!

Anyway... Initial testing... CRC 2bytes, ID (I suspect still 4bytes) changes on reset? aaaagh.
Also changed after ~ 12~24 Hrs (000000xx).

further tests to do ... any thoughts?

MacH

matthias-bs, 2023-11-18:
Hi MacH,

The ID is two bytes! I hope I didn't make any mistakes when I back-ported it from [WeatherSensor.cpp](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/src/WeatherSensor.cpp)... You might want to cross check it. I couldn't test the rtl_433 version.

Matthias

MacH-21, 2024-02-11:
Hi Matthias,

Did you ever monitor the Comms bus on the PM10/SPS30.

I have connected with a logic analyser and found some weird results,
some of which makes sense but does not follow the datasheet!

My PM10 does not have SEL connected to GND therfore Communication is via uart.
I have include my Capture (it is saleae)in zipfile.
I Reset the PM10 and wait for trigger then capture for 10 minutes.
IC communicates with sensor via TX and RX data, sort of makes sense as per datasheet.
IC to RF chip (Transmits 10 times)??? (No request for info no comms bus)?? ... then again 
IC communicates with sensor via TX and RX data sort of makes sense as per datasheet.

I also receive RF via SI4461 and monitor Clk/RxData.

What do you think.

MaCH
[PM10.zip](https://github.com/matthias-bs/BresserWeatherSensorReceiver/files/14234022/PM10.zip)

matthias-bs, 2024-02-13:
Hi @MacH-21,

Yes, the SEL is floating, i.e. communication is done via uart. I also checked that data is encoded as floating point by checking the specific configuration command/response.

Here is an example generated with the Python script mentoned above:
```
* * * *

2024/2/13
Experiment ID

* * * *

Date,Time,Mass,,,,Number
yyyy/m/d,h:m:s,PM1,PM2.5,PM4,PM10,0.3÷0.5,0.3÷1,0.3÷2.5,0.3÷4,0.3÷10,typical size
,,ug/m^3,ug/m^3,ug/m^3,ug/m^3,#/cm^3,#/cm^3,#/cm^3,#/cm^3,#/cm^3,um

CMD: 1, STATE: 67
CMD: 0, STATE: 0
CMD: 128, STATE: 0
CMD: 3, STATE: 0
2024/2/13,17:54:59 - 1.59,1.94,2.15,2.26,10.31,12.38,12.68,12.73,12.74,0.52
CMD: 3, STATE: 0
2024/2/13,17:55:0 - 1.63,1.99,2.20,2.31,10.55,12.68,12.97,13.03,13.04,0.52
CMD: 3, STATE: 0
2024/2/13,17:55:2 - 1.77,1.87,1.87,1.87,12.20,14.10,14.13,14.14,14.14,0.49
CMD: 1, STATE: 67
CMD: 0, STATE: 0
CMD: 128, STATE: 0
CMD: 3, STATE: 0
               AVG - 2,2,2,2,11.31,13.31,13.48,13.51,13.52,0.50
2024/2/13,17:58:49 - 7.87,8.32,8.32,8.32,54.21,62.64,62.80,62.81,62.82,0.49
CMD: 3, STATE: 0
2024/2/13,17:58:51 - 7.74,9.11,9.87,10.25,50.96,60.58,61.67,61.87,61.92,0.50
CMD: 3, STATE: 0
2024/2/13,17:58:52 - 7.53,7.97,7.97,7.97,51.91,59.98,60.13,60.15,60.16,0.47
CMD: 1, STATE: 67
CMD: 0, STATE: 0
CMD: 128, STATE: 0
CMD: 3, STATE: 0
               AVG - 8,8,9,9,52.24,60.79,61.18,61.24,61.26,0.48
2024/2/13,17:59:46 - 3.20,4.20,4.87,5.20,19.97,24.58,25.46,25.63,25.67,0.67
CMD: 3, STATE: 0
2024/2/13,17:59:47 - 3.19,4.04,4.59,4.86,20.22,24.60,25.33,25.47,25.51,0.67
CMD: 3, STATE: 0
2024/2/13,17:59:49 - 3.16,3.99,4.52,4.78,20.13,24.44,25.16,25.29,25.32,0.67
```

"STATE: 67" -> Command not allowed in current state (same as you observed)

Regards,
Matthias

matthias-bs, 2024-02-13:
Some data for racking your brains.

**SPS30 output (UART)**
```
* * * *

2024/2/13
Experiment ID

* * * *

Date,Time,Mass,,,,Number
yyyy/m/d,h:m:s,PM1,PM2.5,PM4,PM10,0.3÷0.5,0.3÷1,0.3÷2.5,0.3÷4,0.3÷10,typical size
,,ug/m^3,ug/m^3,ug/m^3,ug/m^3,#/cm^3,#/cm^3,#/cm^3,#/cm^3,#/cm^3,um

CMD: 1, STATE: 67
CMD: 0, STATE: 0
CMD: 128, STATE: 0
CMD: 3, STATE: 0
2024/2/13,19:53:46 - 8.01,8.47,8.47,8.47,55.19,63.77,63.94,63.95,63.96,0.43
CMD: 3, STATE: 0
2024/2/13,19:53:47 - 8.05,8.51,8.51,8.51,55.44,64.06,64.23,64.24,64.26,0.42
CMD: 3, STATE: 0
2024/2/13,19:53:48 - 8.10,8.57,8.57,8.57,55.81,64.49,64.66,64.67,64.68,0.43
CMD: 1, STATE: 67
CMD: 0, STATE: 0
CMD: 128, STATE: 0
CMD: 3, STATE: 0
               AVG - 8,9,9,9,55.57,64.20,64.37,64.39,64.40,0.42
2024/2/13,19:54:43 - 7.29,7.71,7.71,7.71,50.25,58.06,58.21,58.22,58.23,0.36
CMD: 3, STATE: 0
2024/2/13,19:54:44 - 7.33,7.76,7.76,7.76,50.52,58.38,58.53,58.54,58.55,0.37
CMD: 3, STATE: 0
2024/2/13,19:54:45 - 7.38,7.81,7.81,7.81,50.87,58.78,58.94,58.95,58.96,0.38
CMD: 1, STATE: 67
CMD: 0, STATE: 0
CMD: 128, STATE: 0
CMD: 3, STATE: 0
               AVG - 7,8,8,8,50.63,58.50,58.65,58.67,58.68,0.37
2024/2/13,19:55:40 - 5.70,6.03,6.03,6.03,39.28,45.39,45.51,45.52,45.53,0.43
CMD: 3, STATE: 0
2024/2/13,19:55:41 - 6.04,6.38,6.38,6.38,41.59,48.05,48.18,48.19,48.20,0.42
CMD: 3, STATE: 0
2024/2/13,19:55:42 - 6.34,6.70,6.70,6.70,43.66,50.45,50.58,50.59,50.60,0.43
CMD: 1, STATE: 67
CMD: 0, STATE: 0
CMD: 128, STATE: 0
CMD: 3, STATE: 0
               AVG - 6,6,6,6,42.05,48.58,48.71,48.72,48.73,0.43
2024/2/13,19:56:37 - 9.66,10.21,10.21,10.21,66.53,76.87,77.08,77.09,77.10,0.35
CMD: 3, STATE: 0
2024/2/13,19:56:38 - 9.55,10.10,10.10,10.10,65.81,76.05,76.25,76.26,76.27,0.35
CMD: 3, STATE: 0
2024/2/13,19:56:39 - 9.43,9.97,9.97,9.97,64.98,75.08,75.28,75.29,75.31,0.35
CMD: 1, STATE: 67
CMD: 0, STATE: 0
CMD: 128, STATE: 0
CMD: 3, STATE: 0
               AVG - 10,10,10,10,65.58,75.77,75.97,75.98,76.00,0.35
2024/2/13,19:57:34 - 10.79,11.41,11.41,11.41,74.35,85.91,86.14,86.15,86.17,0.45
CMD: 3, STATE: 0
2024/2/13,19:57:35 - 10.68,11.30,11.30,11.30,73.59,85.03,85.25,85.27,85.28,0.46
CMD: 3, STATE: 0
2024/2/13,19:57:36 - 10.58,11.18,11.18,11.18,72.86,84.19,84.41,84.43,84.44,0.47
```

**Air Quality Sensor Radio Message Decoding**

```
[479745][I][WeatherSensor.cpp:1180] decodeBresser7In1Payload(): PM1.0: 0010 PM2.5: 0011 PM10: 0011 PN1: 0067 PN2: 0078 PN3: 0078
[536691][I][WeatherSensor.cpp:1180] decodeBresser7In1Payload(): PM1.0: 0008 PM2.5: 0009 PM10: 0009 PN1: 0059 PN2: 0068 PN3: 0068
[593739][I][WeatherSensor.cpp:1180] decodeBresser7In1Payload(): PM1.0: 0007 PM2.5: 0008 PM10: 0008 PN1: 0049 PN2: 0057 PN3: 0057
[650687][I][WeatherSensor.cpp:1180] decodeBresser7In1Payload(): PM1.0: 0008 PM2.5: 0008 PM10: 0008 PN1: 0053 PN2: 0061 PN3: 0061
[707729][I][WeatherSensor.cpp:1180] decodeBresser7In1Payload(): PM1.0: 0009 PM2.5: 0009 PM10: 0009 PN1: 0061 PN2: 0070 PN3: 0070
```

**Observations**

* In the UART output, PM4 and PM10 are always very close
* Decoding of PM1.0, PM2.5 and PM10 seems to be correct
* Assumed particle number concentration values are slightly off w.r.t. UART output