matthias-bs/BresserWeatherSensorReceiver #105: rain per h/d/w and Month provide same values

mothra78, 2023-11-03:
**Context**
I am using MQTTCustom Sketch on ESP32 + CC1101 and I am super happy about your sketch, thanks a lot!
Am I missing something to have these values updated?

**Describe the bug**
MQTT data that is being sent is showing same date for rain_h, rain_d, rain_w and rain_m and does not seem to reset
Am I missing something?

{
  "id": 339744019,
  "ch": 0,
  "battery_ok": 1,
  "temp_c": 8.6,
  "humidity": 92,
  "wind_gust": 5.6,
  "wind_avg": 4.9,
  "wind_dir": 248,
  "uv": 0,
  "rain": 128,
  "rain_h": 35.6,
  "rain_d": 35.6,
  "rain_w": 35.6,
  "rain_m": 35.6
}

**To Reproduce**
Each time I get MQTT data (every 5min) I recieve the same values for all 4 rain values

**Expected behavior**
I would expect h to reset every Hour, D every day and so on.

**Debug Log**
{
  "id": 339744019,
  "ch": 0,
  "battery_ok": 1,
  "temp_c": 8.6,
  "humidity": 92,
  "wind_gust": 5.6,
  "wind_avg": 4.9,
  "wind_dir": 248,
  "uv": 0,
  "rain": 128,
  "rain_h": 35.6,
  "rain_d": 35.6,
  "rain_w": 35.6,
  "rain_m": 35.6
}

**Embedded Device (please complete the following information):**
 - Arduino IDE Version: 2.2.0
 - ESP32/ESP8266 Board Support Package Version: [e.g. ESP32 2.0.7]
 - Library Version 1.9
 - MQTT CUstom Sketch with 
 -  #define USE_WIFI              // use non-secure WIFI
 - 
 - Your Board: ESP32 Module
 - Selected Board in the Arduino IDE: [brand/model/version]
 - Your Radio Transceiver Module CC1101
 - **Pinning/Wiring**

**Additional context**
Add any other context about the problem here.

matthias-bs, 2023-11-03:
Hi,

Thanks for the detailed report! I'm currently fixing this: https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/103

See also https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/88#issuecomment-1792295535

Regards
Matthias

mothra78, 2023-11-03:
Oh thanks!
I will try to move the #ifdef USE_SECUREWIFI before the NTP code to see if it fixes the issue

matthias-bs, 2023-11-03:
The fix should be in the main branch in a few minutes...

mothra78, 2023-11-03:
Great

mothra78, 2023-11-03:
18:53:24.373 -> [ 10196][I][BresserWeatherSensorMQTTCustom1.9.ino:390] mqtt_setup(): Current time: Fri Nov  3 17:53:23 2023
18:53:24.373 -> 
18:53:24.373 -> Checking wifi...

matthias-bs, 2023-11-03:
Ready for testing...

mothra78, 2023-11-04:
Seems to change something but not yet what I was expecting.
Hourly values seem to reset correctly but _d, _w and _m seem to be resetting too soon.

See MQTT data showing a change compared to pre mod : 
{"id":339744019,"ch":0,"battery_ok":1,"temp_c":8.6,"humidity":96,"wind_gust":2.8,"wind_avg":2.5,"wind_dir":248.0,"uv":0.0,"rain":140.0,"rain_h":1.2,"rain_d":2.4,"rain_w":2.4,"rain_m":2.4}

And enclosed Graph (last 15 hours) which shows Hourly resetting regularly, Day resetting twice and Week and Month identicall and also having reset during the 15h window
[image]

matthias-bs, 2023-11-04:
Could you please provide a log file with 'debug' set?

mothra78, 2023-11-04:
With _DEBUG_CIRCULAR_BUFFER_ set?

mothra78, 2023-11-04:
Yes I am uploading a new version with Debug set

mothra78, 2023-11-04:
`10:59:11.338 -> [ 11418][D][WeatherSensor.cpp:118] begin(): success!
10:59:11.338 -> [ 11419][D][WeatherSensor.cpp:153] begin(): [CC1101] Setup complete - awaiting incoming messages...
10:59:17.419 -> [ 17515][D][WeatherSensor.cpp:242] getMessage(): [CC1101] R [D4] RSSI: -63.5
10:59:17.419 -> [ 17515][D][WeatherSensor.cpp:947] decodeBresser7In1Payload(): Data sanity check failed
10:59:17.494 -> [ 17519][D][WeatherSensor.cpp:960] decodeBresser7In1Payload(): Digest check failed - [3902] vs [EFC7] (D6C5)
10:59:17.494 -> [ 17528][D][WeatherSensor.cpp:394] findSlot(): sensor[0]: v=0 id=0x00000000 t=0 c=0
10:59:17.494 -> [ 17535][D][WeatherSensor.cpp:893] decodeBresser6In1Payload(): Flags: Temp=1  Hum=1  Wind=1  Rain=0  UV=1
10:59:29.419 -> [ 29517][D][WeatherSensor.cpp:242] getMessage(): [CC1101] R [D4] RSSI: -61.0
10:59:29.419 -> [ 29517][D][WeatherSensor.cpp:947] decodeBresser7In1Payload(): Data sanity check failed
10:59:29.419 -> [ 29520][D][WeatherSensor.cpp:960] decodeBresser7In1Payload(): Digest check failed - [D8BF] vs [894C] (51F3)
10:59:29.494 -> [ 29530][D][WeatherSensor.cpp:394] findSlot(): sensor[0]: v=1 id=0x14401513 t=1 c=0
10:59:29.494 -> [ 29537][D][WeatherSensor.cpp:893] decodeBresser6In1Payload(): Flags: Temp=0  Hum=0  Wind=1  Rain=1  UV=0
10:59:29.494 -> [  0 ]	[  1 ]	[  2 ]	[  3 ]	[  4 ]	[  5 ]	[  6 ]	[  7 ]	[  8 ]	[  9 ]	[ 10 ]	
10:59:29.512 ->  35940	     0	     0	     0	     0	     0	     0	     0	     0	     0	     0	
10:59:29.512 ->  140.4	   0.0	   0.0	   0.0	   0.0	   0.0	   0.0	   0.0	   0.0	   0.0	   0.0	
10:59:29.512 ->   T	   	   	   	   	   	   	   	   	   	   	
10:59:29.512 ->   H	   	   	   	   	   	   	   	   	   	   	
10:59:29.512 -> 
10:59:29.512 -> [ 29577][I][BresserWeatherSensorMQTTCustom1.9.ino:606] publishWeatherdata(): Ext/Bresser5in1/data: {"id":339744019,"ch":0,"battery_ok":1,"temp_c":9.1,"humidity":95,"wind_gust":2.2,"wind_avg":1.9,"wind_dir":202.0,"uv":0.0,"rain":140.4,"rain_h":0.0,"rain_d":0.0,"rain_w":0.0,"rain_m":0.0}
10:59:29.523 -> 
10:59:29.523 -> [ 29604][I][BresserWeatherSensorMQTTCustom1.9.ino:622] publishWeatherdata(): Ext/extra: {"wind_dir_txt":"SSW","wind_gust_bft":2,"wind_avg_bft":2,"dewpoint_c":8.3,"perceived_temp_c":8.2}
10:59:29.523 -> 
10:59:29.523 -> [ 29617][D][BresserWeatherSensorMQTTCustom1.9.ino:770] loop(): Data forwarding completed.
10:59:29.553 -> [ 29625][I][BresserWeatherSensorMQTTCustom1.9.ino:772] loop(): Sleeping for 300000 ms
10:59:29.553 -> 
10:59:29.553 -> [ 29632][I][BresserWeatherSensorMQTTCustom1.9.ino:773] loop(): Ext/status: offline
10:59:29.553 -> 
`

mothra78, 2023-11-04:
[20231104 - Bresser5in1 Log1.txt](https://github.com/matthias-bs/BresserWeatherSensorReceiver/files/13256310/20231104.-.Bresser5in1.Log1.txt)

mothra78, 2023-11-04:
After wakeup 5min later : 
11:07:41.522 -> [ 25978][D][WeatherSensor.cpp:893] decodeBresser6In1Payload(): Flags: Temp=1  Hum=1  Wind=1  Rain=0  UV=1
11:07:41.522 -> [  0 ]	[  1 ]	[  2 ]	[  3 ]	[  4 ]	[  5 ]	[  6 ]	[  7 ]	[  8 ]	[  9 ]	[ 10 ]	
11:07:41.522 ->  36120	 36120	     0	     0	     0	     0	     0	     0	     0	     0	     0	
11:07:41.522 ->  140.4	 140.4	   0.0	   0.0	   0.0	   0.0	   0.0	   0.0	   0.0	   0.0	   0.0	
11:07:41.522 ->   T	   	   	   	   	   	   	   	   	   	   	
11:07:41.523 ->    	  H	   	   	   	   	   	   	   	   	   	
11:07:41.523 -> 
11:07:41.523 -> [ 26018][I][BresserWeatherSensorMQTTCustom1.9.ino:606] publishWeatherdata(): Ext/Bresser5in1/data: {"id":339744019,"ch":0,"battery_ok":1,"temp_c":9.3,"humidity":94,"wind_gust":2.5,"wind_avg":2.3,"wind_dir":202.0,"uv":0.0,"rain":140.4,"rain_h":0.0,"rain_d":0.0,"rain_w":0.0,"rain_m":0.0}

mothra78, 2023-11-04:
And 5min later : 
11:13:41.516 -> [ 61088][D][WeatherSensor.cpp:893] decodeBresser6In1Payload(): Flags: Temp=1  Hum=1  Wind=1  Rain=0  UV=1
11:13:41.519 -> [  0 ]	[  1 ]	[  2 ]	[  3 ]	[  4 ]	[  5 ]	[  6 ]	[  7 ]	[  8 ]	[  9 ]	[ 10 ]	
11:13:41.519 ->  36120	 36120	 36420	     0	     0	     0	     0	     0	     0	     0	     0	
11:13:41.519 ->  140.4	 140.4	 140.4	   0.0	   0.0	   0.0	   0.0	   0.0	   0.0	   0.0	   0.0	
11:13:41.519 ->   T	   	   	   	   	   	   	   	   	   	   	
11:13:41.519 ->    	   	  H	   	   	   	   	   	   	   	   	
11:13:41.519 -> 
11:13:41.519 -> [ 61128][I][BresserWeatherSensorMQTTCustom1.9.ino:606] publishWeatherdata(): Ext/Bresser5in1/data: {"id":339744019,"ch":0,"battery_ok":1,"temp_c":9.4,"humidity":94,"wind_gust":0.9,"wind_avg":0.9,"wind_dir":158.0,"uv":0.0,"rain":140.4,"rain_h":0.0,"rain_d":0.0,"rain_w":0.0,"rain_m":0.0}
11:13:41.528 ->

mothra78, 2023-11-04:
And 5min later : 
11:19:17.465 -> [ 36898][D][WeatherSensor.cpp:394] findSlot(): sensor[0]: v=1 id=0x14401513 t=1 c=0
11:19:17.465 -> [ 36905][D][WeatherSensor.cpp:893] decodeBresser6In1Payload(): Flags: Temp=1  Hum=1  Wind=1  Rain=0  UV=1
11:19:17.495 -> [  0 ]	[  1 ]	[  2 ]	[  3 ]	[  4 ]	[  5 ]	[  6 ]	[  7 ]	[  8 ]	[  9 ]	[ 10 ]	
11:19:17.495 ->  36120	 36120	 36420	 36720	     0	     0	     0	     0	     0	     0	     0	
11:19:17.495 ->  140.4	 140.4	 140.4	 140.4	   0.0	   0.0	   0.0	   0.0	   0.0	   0.0	   0.0	
11:19:17.495 ->   T	   	   	   	   	   	   	   	   	   	   	
11:19:17.495 ->    	   	   	  H	   	   	   	   	   	   	   	
11:19:17.495 -> 
11:19:17.495 -> [ 36945][I][BresserWeatherSensorMQTTCustom1.9.ino:606] publishWeatherdata(): Ext/Bresser5in1/data: {"id":339744019,"ch":0,"battery_ok":1,"temp_c":9.5,"humidity":94,"wind_gust":1.6,"wind_avg":1.4,"wind_dir":202.0,"uv":0.0,"rain":140.4,"rain_h":0.0,"rain_d":0.0,"rain_w":0.0,"rain_m":0.0}

mothra78, 2023-11-04:
`11:24:41.487 -> [ 25039][D][WeatherSensor.cpp:893] decodeBresser6In1Payload(): Flags: Temp=0  Hum=0  Wind=1  Rain=1  UV=0
11:24:41.487 -> [  0 ]	[  1 ]	[  2 ]	[  3 ]	[  4 ]	[  5 ]	[  6 ]	[  7 ]	[  8 ]	[  9 ]	[ 10 ]	
11:24:41.487 ->  36120	 36120	 36420	 36720	 37080	     0	     0	     0	     0	     0	     0	
11:24:41.487 ->  140.4	 140.4	 140.4	 140.4	 140.4	   0.0	   0.0	   0.0	   0.0	   0.0	   0.0	
11:24:41.518 ->   T	   	   	   	   	   	   	   	   	   	   	
11:24:41.518 ->    	   	   	   	  H	   	   	   	   	   	   	
11:24:41.518 -> 
11:24:41.518 -> [ 25079][I][BresserWeatherSensorMQTTCustom1.9.ino:606] publishWeatherdata(): Ext/Bresser5in1/data: {"id":339744019,"ch":0,"battery_ok":1,"temp_c":9.5,"humidity":94,"wind_gust":2.6,"wind_avg":2.3,"wind_dir":135.0,"uv":0.0,"rain":140.4,"rain_h":0.0,"rain_d":0.0,"rain_w":0.0,"rain_m":0.0}
1`

matthias-bs, 2023-11-04:
Hi,
what I actually need is the information which time and date is set after each restart. Next it is important to know the rain values, of course.

mothra78, 2023-11-04:
Tail is still on i=0 and Head is moving to the end of the buffer (not sure you can see it well in the log.)

matthias-bs, 2023-11-04:
That's as expected. It is implemented as a ring buffer/FIFO.

matthias-bs, 2023-11-04:
BTW: If your cycle time is 5 minutes, you have to set RAINGAUGE_BUF_SIZE to 13 (60 minutes / 5 minutes) plus one extra entry.

mothra78, 2023-11-04:
OK I'll change the BUF_SIZE
Here the latest Log which shows NTP seems to work correctly
[20231104 - Bresser5in1 Log2.txt](https://github.com/matthias-bs/BresserWeatherSensorReceiver/files/13256363/20231104.-.Bresser5in1.Log2.txt)

mothra78, 2023-11-04:
I also added DAYLIGHT Saving (as we still have it in FR) and thus now print GMT and Local time

mothra78, 2023-11-04:
[20231104 - Bresser5in1 Log3.txt](https://github.com/matthias-bs/BresserWeatherSensorReceiver/files/13256420/20231104.-.Bresser5in1.Log3.txt)

mothra78, 2023-11-04:
Will let it run for a while

mothra78, 2023-11-04:
Added some additionnal traces to look into nvData.

mothra78, 2023-11-04:
[  8128][I][BresserWeatherSensorMQTTCustom1.9.ino:391] mqtt_setup(): Current GMT time: Sat Nov  4 12:43:02 2023

[  8133][I][BresserWeatherSensorMQTTCustom1.9.ino:393] mqtt_setup(): Current Local time: Sat Nov  4 13:43:02 2023

Checking wifi...
                MQTT connecting... ..[ 12205][I][BresserWeatherSensorMQTTCustom1.9.ino:447] mqtt_connect():
                                                                                                            connected!
[ 12213][I][BresserWeatherSensorMQTTCustom1.9.ino:449] mqtt_connect(): Ext/status: online

[ 12214][D][WeatherSensor.cpp:105] begin(): [CC1101] Initializing ...
[ 12383][D][WeatherSensor.cpp:118] begin(): success!
[ 12384][D][WeatherSensor.cpp:153] begin(): [CC1101] Setup complete - awaiting incoming messages...
[ 22051][D][WeatherSensor.cpp:242] getMessage(): [CC1101] R [D4] RSSI: -69.0
[ 22051][D][WeatherSensor.cpp:947] decodeBresser7In1Payload(): Data sanity check failed
[ 22055][D][WeatherSensor.cpp:960] decodeBresser7In1Payload(): Digest check failed - [C4EE] vs [8F57] (4BB9)
[ 22064][D][WeatherSensor.cpp:394] findSlot(): sensor[0]: v=0 id=0x00000000 t=0 c=0
[ 22071][D][WeatherSensor.cpp:893] decodeBresser6In1Payload(): Flags: Temp=1  Hum=1  Wind=1  Rain=0  UV=1
[ 34053][D][WeatherSensor.cpp:242] getMessage(): [CC1101] R [D4] RSSI: -68.5
[ 34053][D][WeatherSensor.cpp:947] decodeBresser7In1Payload(): Data sanity check failed
[ 34056][D][WeatherSensor.cpp:960] decodeBresser7In1Payload(): Digest check failed - [D873] vs [328A] (EAF9)
[ 34066][D][WeatherSensor.cpp:394] findSlot(): sensor[0]: v=1 id=0x14401513 t=1 c=0
[ 34073][D][WeatherSensor.cpp:893] decodeBresser6In1Payload(): Flags: Temp=0  Hum=0  Wind=1  Rain=1  UV=0
[  0 ]  [  1 ]  [  2 ]  [  3 ]  [  4 ]  [  5 ]  [  6 ]  [  7 ]  [  8 ]  [  9 ]  [ 10 ]  [ 11 ]  [ 12 ]
 45780       0       0       0       0       0       0       0       0       0       0       0       0
 140.4     0.0     0.0     0.0     0.0     0.0     0.0     0.0     0.0     0.0     0.0     0.0     0.0
  T
  H

nvData.tsDayBegin :    255
nvData.rainDayBegin :    0.0
nvData.tsWeekBegin :    255
nvData.wdayPrev :      6
nvData.rainWeekBegin :    0.0
nvData.tsMonthBegin :    255
nvData.rainMonthBegin :    0.0
nvData.startupPrev :      0
nvData.rainStartup :    0.0
nvData.rainPrev :  140.4
nvData.rainOvf :  140.4
[ 34147][I][BresserWeatherSensorMQTTCustom1.9.ino:609] publishWeatherdata(): Ext/Bresser5in1/data: {"id":339744019,"ch":0,"battery_ok":1,"temp_c":12.7,"humidity":85,"wind_gust":3.0,"wind_avg":2.9,"wind_dir":202.0,"uv":0.0,"rain":140.4,"rain_h":0.0,"rain_d":0.0,"rain_w":0.0,"rain_m":0.0}

mothra78, 2023-11-04:
[20231104 - Bresser5in1 Log5.txt](https://github.com/matthias-bs/BresserWeatherSensorReceiver/files/13256691/20231104.-.Bresser5in1.Log5.txt)

mothra78, 2023-11-04:
> Hi, what I actually need is the information which time and date is set after each restart. Next it is important to know the rain values, of course. First the time read from the RTC, second the time set from NTP (if possible).

How can I print RTC time?

matthias-bs, 2023-11-04:
`printDateTime();`

mothra78, 2023-11-04:
Do I need to add an additional library?

matthias-bs, 2023-11-04:
```
nvData.tsDayBegin :    255
nvData.rainDayBegin :    0.0
nvData.tsWeekBegin :    255
nvData.wdayPrev :      6
nvData.rainWeekBegin :    0.0
nvData.tsMonthBegin :    255
nvData.rainMonthBegin :    0.0
nvData.startupPrev :      0
nvData.rainStartup :    0.0
nvData.rainPrev :  140.4
nvData.rainOvf :  140.4
```

A value of 255 in the structure members shows that the data has just been initialized, either after a HW reset, power-on reset, spurious reset (glitch, power failure) or by calling `RainGauge::reset()`.

matthias-bs, 2023-11-04:
> Do I need to add an additional library?

No, you just have to `#include <time.h>`.
See https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/examples/BresserWeatherSensorMQTT/BresserWeatherSensorMQTT.ino

mothra78, 2023-11-04:
the 255 is replaced by a value that seems correct after reset : 
[  5077][I][BresserWeatherSensorMQTTCustom1.9.ino:391] mqtt_setup(): Current GMT time: Sat Nov  4 13:26:22 2023

[  5087][I][BresserWeatherSensorMQTTCustom1.9.ino:393] mqtt_setup(): Current Local time: Sat Nov  4 14:26:22 2023

Checking wifi...
                MQTT connecting... ..[  9179][I][BresserWeatherSensorMQTTCustom1.9.ino:447] mqtt_connect():
                                                                                                            connected!
[  9188][I][BresserWeatherSensorMQTTCustom1.9.ino:449] mqtt_connect(): Ext/status: online

[  9189][D][WeatherSensor.cpp:105] begin(): [CC1101] Initializing ...
[  9358][D][WeatherSensor.cpp:118] begin(): success!
[  9359][D][WeatherSensor.cpp:153] begin(): [CC1101] Setup complete - awaiting incoming messages...
[ 13209][D][WeatherSensor.cpp:242] getMessage(): [CC1101] R [D4] RSSI: -62.5
[ 13209][D][WeatherSensor.cpp:947] decodeBresser7In1Payload(): Data sanity check failed
[ 13213][D][WeatherSensor.cpp:960] decodeBresser7In1Payload(): Digest check failed - [DE19] vs [F249] (2C50)
[ 13222][D][WeatherSensor.cpp:394] findSlot(): sensor[0]: v=0 id=0x00000000 t=0 c=0
[ 13229][D][WeatherSensor.cpp:893] decodeBresser6In1Payload(): Flags: Temp=1  Hum=1  Wind=1  Rain=0  UV=1
[ 25211][D][WeatherSensor.cpp:242] getMessage(): [CC1101] R [D4] RSSI: -62.5
[ 25211][D][WeatherSensor.cpp:947] decodeBresser7In1Payload(): Data sanity check failed
[ 25214][D][WeatherSensor.cpp:960] decodeBresser7In1Payload(): Digest check failed - [279C] vs [2406] (039A)
[ 25224][D][WeatherSensor.cpp:394] findSlot(): sensor[0]: v=1 id=0x14401513 t=1 c=0
[ 25231][D][WeatherSensor.cpp:893] decodeBresser6In1Payload(): Flags: Temp=0  Hum=0  Wind=1  Rain=1  UV=0
[  0 ]  [  1 ]  [  2 ]  [  3 ]  [  4 ]  [  5 ]  [  6 ]  [  7 ]  [  8 ]  [  9 ]  [ 10 ]  [ 11 ]  [ 12 ]
 45780   45780   46080   46380   46740   47040   47400   47700   48000       0       0       0       0
 140.4   140.4   140.4   140.4   140.4   140.4   140.4   140.4   140.4     0.0     0.0     0.0     0.0
  T
                                                                  H

nvData.tsDayBegin :      6
nvData.rainDayBegin :  140.4
nvData.tsWeekBegin :      6
nvData.wdayPrev :      6
nvData.rainWeekBegin :  140.4
nvData.tsMonthBegin :     10
nvData.rainMonthBegin :  140.4
nvData.startupPrev :      0
nvData.rainStartup :    0.0
nvData.rainPrev :  140.4
nvData.rainOvf :  140.4
[ 25305][I][BresserWeatherSensorMQTTCustom1.9.ino:609] publishWeatherdata(): Ext/Bresser5in1/data: {"id":339744019,"ch":0,"battery_ok":1,"temp_c":12.8,"humidity":89,"wind_gust":1.0,"wind_avg":1.0,"wind_dir":292.0,"uv":0.0,"rain":140.4,"rain_h":0.0,"rain_d":0.0,"rain_w":0.0,"rain_m":0.0}

mothra78, 2023-11-04:
OK added code from https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/examples/BresserWeatherSensorMQTT/BresserWeatherSensorMQTT.ino

matthias-bs, 2023-11-04:
```
nvData.tsDayBegin : 6 // start day of week (for daily counter)
nvData.rainDayBegin : 140.4
nvData.tsWeekBegin : 6 // start day of week (for weekly counter)
nvData.wdayPrev : 6 // day of week at previous update
nvData.rainWeekBegin : 140.4 // rain gauge value at begin of week or start, respectively
nvData.tsMonthBegin : 10 // month  (for monthly counter, actual month-1 as defined by struct tm)
nvData.rainMonthBegin : 140.4 // rain gauge value at begin of month or start, respectively
nvData.startupPrev : 0 // previous startup flag value
nvData.rainStartup : 0.0 // last rain value before rising edge of startup flag
nvData.rainPrev : 140.4 // previous rain gauge value (to calculate difference)
nvData.rainOvf : 140.4 // this is odd - should be number of rain gauge overflows. Maybe a glitch in your print statement? 
```

mothra78, 2023-11-04:
I will check last printf

matthias-bs, 2023-11-04:
[BresserWeatherSensorMQTT.ino](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/examples/BresserWeatherSensorMQTT/BresserWeatherSensorMQTT.ino), [BresserWeatherSensorMQTTCustom.ino](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/examples/BresserWeatherSensorMQTTCustom/BresserWeatherSensorMQTTCustom.ino) and [BresserWeatherSensorMQTTWifiMgr.ino](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/examples/BresserWeatherSensorMQTTWifiMgr/BresserWeatherSensorMQTTWifiMgr.ino) should all provide the same changes I made yesterday.

mothra78, 2023-11-04:
Corrected the bug on the printf : 

14:42:52.147 -> [  5134][I][BresserWeatherSensorMQTTCustom1.9.ino:408] mqtt_setup(): Setting time using SNTP
14:42:52.615 -> .[  5638][I][BresserWeatherSensorMQTTCustom1.9.ino:423] mqtt_setup(): 
14:42:52.646 -> done!
14:42:52.646 -> [  5638][I][BresserWeatherSensorMQTTCustom1.9.ino:428] mqtt_setup(): Current time (GMT): Sat Nov  4 13:42:51 2023
14:42:52.646 -> 
14:42:52.646 -> [  5643][I][BresserWeatherSensorMQTTCustom1.9.ino:430] mqtt_setup(): Current time (Local): Sat Nov  4 14:42:51 2023
14:42:52.646 -> 
14:42:52.646 -> [  5653][I][BresserWeatherSensorMQTTCustom1.9.ino:431] mqtt_setup(): RTC time:
14:42:52.646 -> [  5660][I][BresserWeatherSensorMQTTCustom1.9.ino:366] printDateTime(): 2023-11-04 14:42:51
14:42:52.679 -> Checking wifi...
14:42:52.679 -> MQTT connecting... ..[  9734][I][BresserWeatherSensorMQTTCustom1.9.ino:486] mqtt_connect(): 
14:42:56.751 -> connected!
14:42:56.751 -> [  9747][I][BresserWeatherSensorMQTTCustom1.9.ino:488] mqtt_connect(): Ext/status: online
14:42:56.751 -> 
14:42:56.751 -> [  9748][D][WeatherSensor.cpp:105] begin(): [CC1101] Initializing ... 
14:42:56.922 -> [  9917][D][WeatherSensor.cpp:118] begin(): success!
14:42:56.922 -> [  9919][D][WeatherSensor.cpp:153] begin(): [CC1101] Setup complete - awaiting incoming messages...
14:43:06.055 -> [ 19061][D][WeatherSensor.cpp:242] getMessage(): [CC1101] R [D4] RSSI: -60.0
14:43:06.055 -> [ 19061][D][WeatherSensor.cpp:947] decodeBresser7In1Payload(): Data sanity check failed
14:43:06.055 -> [ 19065][D][WeatherSensor.cpp:960] decodeBresser7In1Payload(): Digest check failed - [53B3] vs [15CB] (4678)
14:43:06.055 -> [ 19074][D][WeatherSensor.cpp:394] findSlot(): sensor[0]: v=0 id=0x00000000 t=0 c=0
14:43:06.090 -> [ 19081][D][WeatherSensor.cpp:893] decodeBresser6In1Payload(): Flags: Temp=0  Hum=0  Wind=1  Rain=1  UV=0
14:43:18.062 -> [ 31063][D][WeatherSensor.cpp:242] getMessage(): [CC1101] R [D4] RSSI: -60.0
14:43:18.062 -> [ 31063][D][WeatherSensor.cpp:947] decodeBresser7In1Payload(): Data sanity check failed
14:43:18.062 -> [ 31066][D][WeatherSensor.cpp:960] decodeBresser7In1Payload(): Digest check failed - [F388] vs [DD9B] (2E13)
14:43:18.062 -> [ 31076][D][WeatherSensor.cpp:394] findSlot(): sensor[0]: v=1 id=0x14401513 t=1 c=0
14:43:18.096 -> [ 31083][D][WeatherSensor.cpp:893] decodeBresser6In1Payload(): Flags: Temp=1  Hum=1  Wind=1  Rain=0  UV=1
14:43:18.096 -> [  0 ]	[  1 ]	[  2 ]	[  3 ]	[  4 ]	[  5 ]	[  6 ]	[  7 ]	[  8 ]	[  9 ]	[ 10 ]	[ 11 ]	[ 12 ]	
14:43:18.096 ->      0	     0	     0	     0	     0	     0	     0	     0	     0	     0	     0	     0	     0

mothra78, 2023-11-04:
14:43:18.096 -> [  0 ]	[  1 ]	[  2 ]	[  3 ]	[  4 ]	[  5 ]	[  6 ]	[  7 ]	[  8 ]	[  9 ]	[ 10 ]	[ 11 ]	[ 12 ]	
14:43:18.096 ->      0	     0	     0	     0	     0	     0	     0	     0	     0	     0	     0	     0	     0	
14:43:18.096 ->  140.8	   0.0	   0.0	   0.0	   0.0	   0.0	   0.0	   0.0	   0.0	   0.0	   0.0	   0.0	   0.0	
14:43:18.126 ->   T	   	   	   	   	   	   	   	   	   	   	   	   	
14:43:18.126 ->   H	   	   	   	   	   	   	   	   	   	   	   	   	
14:43:18.126 -> 
14:43:18.126 -> nvData.tsDayBegin :    255	
14:43:18.126 -> nvData.rainDayBegin :    0.0	
14:43:18.126 -> nvData.tsWeekBegin :    255	
14:43:18.126 -> nvData.wdayPrev :      4	
14:43:18.126 -> nvData.rainWeekBegin :    0.0	
14:43:18.126 -> nvData.tsMonthBegin :    255	
14:43:18.126 -> nvData.rainMonthBegin :    0.0	
14:43:18.126 -> nvData.startupPrev :      0	
14:43:18.158 -> nvData.rainStartup :    0.0	
14:43:18.158 -> nvData.rainPrev :  140.8	
14:43:18.158 -> nvData.rainOvf :      0	
14:43:18.158 -> [ 31157][I][BresserWeatherSensorMQTTCustom1.9.ino:648] publishWeatherdata(): Ext/Bresser5in1/data: {"id":339744019,"ch":0,"battery_ok":1,"temp_c":12.7,"humidity":92,"wind_gust":2.4,"wind_avg":2.2,"wind_dir":315.0,"uv":0.0,"rain":140.8,"rain_h":0.0,"rain_d":0.0,"rain_w":0.0,"rain_m":0.0}
14:43:18.190 ->

matthias-bs, 2023-11-04:
If the struct falls back to the initial values without power-on or HW reset (or explicit invocation of the reset function), you have a problem with your RTC RAM.

matthias-bs, 2023-11-04:
It's raining at your place... ;-)

mothra78, 2023-11-04:
Yes it is :-) I have now taken all the changes you made and will log for a day.

mothra78, 2023-11-05:
Issue seems to be solved as rain_h is being updated and rain_d reset at midnight GMT while rain_w and rain_m are still increasing. 

[ 25159][I][BresserWeatherSensorMQTTCustom1.9.ino:538] publishWeatherdata(): Current time (GMT): Sun Nov  5 07:01:08 2023

[  0 ]  [  1 ]  [  2 ]  [  3 ]  [  4 ]  [  5 ]  [  6 ]  [  7 ]  [  8 ]  [  9 ]  [ 10 ]  [ 11 ]  [ 12 ]
 22980   23280   23640   23940   24240   24600   24900   21000   21360   21660   22020   22320   22620
 146.4   146.4   146.4   146.4   146.4   146.4   146.4   146.4   146.4   146.4   146.4   146.4   146.4
                                                                          T
                                                  H

nvData.tsDayBegin :      0
nvData.rainDayBegin :  146.0
nvData.tsWeekBegin :      6
nvData.wdayPrev :      0
nvData.rainWeekBegin :  141.6
nvData.tsMonthBegin :     10
nvData.rainMonthBegin :  141.6
nvData.startupPrev :      0
nvData.rainStartup :    0.0
nvData.rainPrev :  146.4
nvData.rainOvf :      0
[ 25233][I][BresserWeatherSensorMQTTCustom1.9.ino:653] publishWeatherdata(): Ext/Bresser5in1/data: {"id":339744019,"ch":0,"battery_ok":1,"temp_c":12.3,"humidity":67,"wind_gust":8.2,"wind_avg":7.4,"wind_dir":202.0,"uv":0.0,"rain":146.4,"rain_h":0.0,"rain_d":0.4,"rain_w":4.8,"rain_m":4.8}
[image]

matthias-bs, 2023-11-05:
Reset should actually occur at midnight local time (not GMT).

matthias-bs, 2023-11-07:
Fixed in Release [v0.19.4](https://github.com/matthias-bs/BresserWeatherSensorReceiver/releases/tag/v0.19.4)

closed 2023-11-07 by matthias-bs