matthias-bs/BresserWeatherSensorReceiver #107: Modify Rain Gauge Data Post-Processing

matthias-bs, 2023-11-08:
Currently the post-processing stores persistent (i.e. during sleep mode) data in RTC RAM.

This means
* The data is lost after a power-cycle or HW reset
* The implementation does not work for architectures without RTC RAM (e.g. RP2040) 

The implementation will be modified to use the Preferences library (i.e. Flash) for storing persistent data instead. To minimize wearing of the Flash, the algorithm and data structures will be modified to reduce the number of write cycles with changed data.

bakkerv, 2023-12-09:
This is a very good enhancement! 

Once finished, adding support for Home Assistant using a total_increasing sensor for a PRECIPITATION sensor can be build

matthias-bs, 2023-12-10:
Hi Vincent,

what exactly do you mean by "total_increasing sensor for precipitation"? I don't think it is related to this planned modification.

The sensor's raw rain gauge value already accumulates precipitation values unless a reset/battery change or an overflow occurs. There is also the internal value `rainCurr`  in `RainGauge.cpp` which is meant to keep accumulating even if a reset/battery change occurs by taking the "startup" flag into account. Additionally, the overflow condition is considered.

Maybe the latter features need some more testing, though.

Cheers,
Matthias

bakkerv, 2023-12-10:
If I understand this enhancement correctly, this will maintain a persistent state even after a power loss of the ESP board, i.e ‘fix’ the derived day/week/year values after power loss. 

One of the enhancements I would like to add is the support for home assistant via MQTT, basically following a convention in the MQTT topics and structure. The sensor type relevant for the rain gauge needs an increasing value. After reading the documentation on home assistant, I guess the current behavior is sufficient to easily integrate the rain gauge in Home Assistant.

matthias-bs, 2023-12-10:
> If I understand this enhancement correctly, this will maintain a persistent state even after a power loss of the ESP board, i.e ‘fix’ the derived day/week/year values after power loss.

Yes, that's correct, but currently there is no "year" value.

> One of the enhancements I would like to add is the support for home assistant via MQTT, basically following a convention in the MQTT topics and structure. The sensor type relevant for the rain gauge [...]

Please feel free to send the HA requirements, either as specification, example data or code changes with pull request.

closed 2024-01-19 by matthias-bs