matthias-bs/BresserWeatherSensorReceiver #108: [Help] How do I reset the MQTT credentials?

Calimerorulez, 2023-11-09:
Hi,

I just flashed BresserWeatherSensorMQTTWifiMgr in my LilyGo LoRa32 V2.1_1.6.1. I used the access point to configure the software, but I've made a typo in my MQTT credentials.

How can I change them, because the access point isn't available anymore, because the LilyGo does connect to my Wifi network, but it hasn't an config interface available.

Thank you for the help.

Calimerorulez, 2023-11-09:
I tried persing the reset button twice, but then my Lily LoRa32 Just reboots twice.

I erased the flash now, I see that the device connects to Wifi and MQTT. It publishes online, then offline, and then dead.

What am I missing?

matthias-bs, 2023-11-10:
Hi,

a double reset within 10 seconds should reset the configuration.
The MQTT connection is stored persistently, I just tried it with the same type of board.

I recommend to do the following steps:
1. Try [BresserWeatherSensorBasic](https://github.com/matthias-bs/BresserWeatherSensorReceiver/tree/main/examples/BresserWeatherSensorBasic) to see if the sensor data can be received
   * Enable "Show verbose output during compile" in the Arduino IDE preferences
   * Set "Core Debug Level: Debug" in the Arduino IDE (see [Debug Output Configuration in Arduino IDE](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/DEBUG_OUTPUT.md))
   * Compile
   * Look at the compiler output if there are any warnings and paste them here
   * Check the console output for sensor messages

If this works, you can go to the next step: 

2.  Try [BresserWeatherSensorMQTTWifiMgr.ino](https://github.com/matthias-bs/BresserWeatherSensorReceiver/blob/main/examples/BresserWeatherSensorMQTTWifiMgr/BresserWeatherSensorMQTTWifiMgr.ino)
   * Enable "Show verbose output during compile" in the Arduino IDE preferences
   * Set "Core Debug Level: Debug" in the Arduino IDE
   * Compile
   * Look at the compiler output if there are any warnings and paste them here
   * Check the console output to see what happens

Which sensor are you using?

Regards
Matthias

matthias-bs, 2023-11-10:
In my test run, I just saw
`Forcing config mode as there was a Double reset detected` 
in the console output, but in fact I did not press reset.

If this also occurs in your case, please comment out the line as shown below:
```
    drd = new DoubleResetDetector(DRD_TIMEOUT, DRD_ADDRESS);
    if (drd->detectDoubleReset())
    {
        Serial.println(F("Forcing config mode as there was a Double reset detected"));
        //forceConfig = true;
    }
```

Calimerorulez, 2023-11-10:
Hi Matthias,

I will test the reset this weekend. 

I'm using the 7003100 sensor (also a 7-in-1). I've disabled deep sleep, and now the Lily stays 'online' in MQTT. I see the data from the sensor coming in fine.

I will try you suggestions too, because I couldn't find my way in the debugging options in de Arduino IDE, but now with your directions I can :). Maybe I can use the display on de Lily now too ;)

I saw`log_v `and `log_i `in your code, but I couldn't find the output of it anywhere, but now I understand why.

Thanks for the help and regards,
Patrick

matthias-bs, 2023-11-10:
I'm afraid you have found a bug: https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/109

If you need a way to trigger re-configuration without erasing the flash, I could add querying an input pin. Unfortunately, the LilyGo does not have a user button. You would have to wire a button (or a jumper wire) externally, possibly with an additional pull-up resistor.

matthias-bs, 2023-11-10:
Fixed double reset detection in [Release v0.19.5](https://github.com/matthias-bs/BresserWeatherSensorReceiver/releases/tag/v0.19.5)

closed 2023-11-19 by Calimerorulez